evcc-io/evcc #14100: Skoda Enyaq 2024 Verbindung zum Connect Portal funktioniert nicht

eny34, 2024-05-28:
### Describe the bug

Hallo, 

das Skoda Connect Portal funktioniert nicht mit 2024er Skoda Enyaq Modellen. 
Laut der Info in diesem Thread wird die Funktion im Juli 2024 komplett abgeschaltet (also auch für ältere Modelle). 
https://github.com/skodaconnect/skodaconnect/issues/126
Eine Verbindung über EVCC ist daher nicht möglich. 

Es funktioniert nur noch die "MySkoda" App. 

### Steps to reproduce

1. 
2.
3.
...

### Configuration details

```yaml
vehicles:
- type: template
  template: enyaq
  title: Skoda Enyaq
  user: user@t-online.de
  password: pw
  vin: vin
  capacity: 62
  name: my_enyaq
```

### Log details

```text
[main ] ERROR 2024/05/28 23:25:58 creating vehicle my_enyaq failed: cannot create vehicle type 'template': cannot create vehicle type 'enyaq': cannot find vehicle, got: []
[site ] INFO 2024/05/28 23:25:58 vehicles:
[site ] INFO 2024/05/28 23:25:58 vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[lp-1 ] INFO 2024/05/28 23:25:58 vehicle updated: unknown -> Skoda Enyaq
```

### What type of operating system are you running?

Windows

### Version

v0.126.4

astrakid, 2024-05-29:
ich glaube, dass es nicht mit der auto-version zusammenhängt, sondern mit der zeit der registrierung. ich habe einen enyaq iv80x vor 2 wochen bekommen, dort habe ich das gleiche problem.

StefanSchoof, 2024-05-29:
Gibt es noch irgendeine Software wie HA oder Iobroker wo es geht?

astrakid, 2024-05-29:
ich habe es in iobroker getestet, da klappt es auch nicht.

eny34, 2024-05-29:
Das Thema wird u.a. auch in diesem Thread diskutiert, demnach funktioniert es auch mit HA nicht. 
https://github.com/skodaconnect/skodaconnect/issues/126

Ikarusit, 2024-05-29:
Im Thread den eny34 verlinkt hat ist auch eine Mail von Skoda Frankreich gepostet worden, dass die App Skoda Essentials und das Skoda Connect Portal ab Ende Juli gänzlich nicht mehr unterstützt werden und zur neuen App "MySkoda" umgezogen werden.
Daher ist stark davon auszugehen, dass sämtliche Neu-Registrierungen bereits nur noch über die neue App auf neuer Infrastruktur in neuer API angelegt werden.

andig, 2024-05-29:
`skoda` geht auch nicht? Dann müssen wir erstmal auf eine externe Lösung warten :(

astrakid, 2024-05-29:
wenn ein account zum testen gebraucht wird, könnte ich meinen anbieten. info dann bitte per mail an andregronwald78 at gmail dot com

eny34, 2024-05-29:
> `skoda` geht auch nicht? Dann müssen wir erstmal auf eine externe Lösung warten :( Funktioniert denn https://github.com/TA2k/ioBroker.vw-connect noch?

Ist damit das Skoda template gemeint? Das funktioniert auch nicht. 

Interessant ist auch der Beitrag von [Raducanu] im Enyaq Forum, demnach funktioniert die Verbindung zu TRONITY. Also gibt es scheinbar eine funktionierende Verbindung um die Daten abzurufen. 
Hoffe es ist ok den Beitrag hier rüber zu kopieren, evtl. helfen die Infos weiter:

_Zitat Raducanu, Enyaq Forum: 
Seit einigen Monaten gibt es aber das Problem, dass einige neue (und Teilweise auch alte auf Werkseinstellung resettet) Enyaqs sich nicht über die SkodaConnect Python Library ansprechen lassen.
Meine Vermutung:
Es gibt eine "alte" Welt (skoda-connect.com / Essentials App) -> Zugriff über klassische RestAPI / SkodaConnect Library
Neue Welt (MySkoda App) -> Zugriff über neue "MySkoda" App. Aktuell kein bekannter ReSTAPI Zugriff / Library.
Bis vor einigen Wochen wurden neue Enyaqs sowohl in der "alten" als auch in der "neuen" Welt verfügbar gemacht. Hier gibt es seit einigen Monaten aber wohl Probleme. Ob dieses nun ein "Bug" ist oder ob dieses "Gewollt" (erster Schritt der Abschaltung alter Welt) ist -> unbekannt.
Meine Diskussionen mit dem Skoda Support lief immer auf die Aussage: Altes Zeug wird abgeschaltet, was willst du eigentlich damit? Nutz doch die neue MySkoda App.
Neben der SkodaConnect Python Lib (die einfache reverse engineered ist) gibt es aber wohl "offizielle" 3rd Party APIs für Business Partner. Wie komme ich darauf: Mein Enyaq ist selber davon betroffen dass er nicht im skoda-connect Portal noch via EVCC noch im HomeAssistant verfügbar ist.
Allerdings kann ich das Auto aber Problemlos bei TRONITY einbinden. Heißt es gibt irgendwo eine Schnittstelle für 3rd Party Partner über die auch neue Autos verfübar sind.
Nachtrag zur "neuen Welt":
- API Endpunkt bekannt
- Reverse Engineering deutlich aufwendiger (SSL Pinning, vermutlich WebSocket statt ReST)_

andig, 2024-05-29:
> gibt es aber wohl "offizielle" 3rd Party APIs für Business Partner

Wo?

eny34, 2024-05-29:
So wie ich den Beitrag verstanden habe ist das nicht bekannt und war eine "Vermutung" des Users weil Tronity funktioniert.

andig, 2024-05-30:
> und würde gerne helfen

Siehe https://github.com/evcc-io/evcc/issues/14100#issuecomment-2137777710

GrimmiMeloni, 2024-05-30:
> * API Endpunkt bekannt
> * Reverse Engineering deutlich aufwendiger (SSL Pinning, vermutlich WebSocket statt ReST)_

@eny34 wo kann ich mehr Details finden?

MarcelMertens, 2024-05-30:
> > * API Endpunkt bekannt
> > * Reverse Engineering deutlich aufwendiger (SSL Pinning, vermutlich WebSocket statt ReST)_
> 
> @eny34 wo kann ich mehr Details finden?

Hi, ich bin Raducanu aus dem Enyaq Forum. Mein Enyaq geht nicht über die SkodaConnect Python Lib und somit über keine der damit assozierten Apps (Ausprobiert: HA und EVCC). Aber mein Enyaq geht z.B. Problemlos über TRONITY. Daher meine Vermutung, dass es irgendwo eine weitere API geben muß. Ich kann mir auch nicht vorstellen, dass TRONITY ihr Geschäftsmodell auf eine reverse engineered API ohne jeglichen Herstellersupport aufbaut.

Bzgl.  der neuen API und deren Informationen (Die Infos mit SSL Pinning kommen von mir) im Discord vom Skodaconnect schauen.
https://discord.gg/826X9jEtCh

GrimmiMeloni, 2024-06-03:
Jemand hat das SSL pinning aus der Android Version heraus gepatched und dann mit MITM einen Mitschnitt gemacht. 

https://gitlab.com/prior99/skoda

Ich schaue mir das in den kommenden Tagen mal an, und schaue ob ich die Daten für mein Fahrzeug über die neue API lesen kann.

MarcelMertens, 2024-06-03:
Mein Enyaq funktioniert nicht mit der SkodaPythonLib / Skoda-Connect / Skoda MyEssentials App.
Nur in der neuen MySkoda App.
Mit den Python Script kann ich meinen Enyaq Abfragen:

```
(venv) marcel@ubuntu01:~/skoda$ python ./skoda.py SkodaID SkodaIDPW
User is xxxxxxxxx
Vehicle Škoda Enyaq with 58kWh capacity and 132kW engine
Vehicle doors locked: NO, windows open: CLOSED
Battery at 72% with a range of 268000m
Air conditioning is False, target temperature is 22.0°C
Charger state: CONNECTED and locked: LOCKED
Vehicle is located at xxxxxxxxxxxxxxx
GPS xxx.999861/xxx.038604
Mileage: 2879km
(venv) marcel@ubuntu01:~/skoda$
```

andig, 2024-06-03:
Top. Hast du das als MITM Trace? Wo kommen die Tokens her?

MarcelMertens, 2024-06-03:
> Top. Hast du das als MITM Trace? Wo kommen die Tokens her?

Nein. Ich hab mir einfach das [skoda](https://gitlab.com/prior99/skoda) skoda.py script runtergeladen und ausgeführt...
Es gibt im SkodaConnect Python Discord eine Diskussion mit dem Urheber des Scripts. [Gerade die Anmeldung ist wohl ungewöhnlich. 
](https://discord.gg/SgHVDJrp)

motze92, 2024-06-03:
Hab die aus dem Code der App geholt. 

Zuerst normales OAuth über VW-Identity und dann kann man sich von `/api/v1/authentication/exchange-authorization-code` den Bearer Token holen. 

Sollte also nicht so großer aufwand sein, dass in evcc zu implementieren.

Vlt komme ich später noch dazu.

MarcelMertens, 2024-06-03:
Mit dem Token funktionieren zumindest die wichtigsten GET Endpunkte

https://mysmob.api.connect.skoda-auto.cz/api/v2/garage/initial-vehicle?connectivityGenerations=MOD3&connectivityGenerations=MOD4

Liefert alle registrierten Autos mit Infos

```
{
    "deliveredVehicle": {
        "vin": "TMBJB9NY0RF024245",
        "name": "Enyaq",
        "workshopModeEnabled": false,
        "licensePlate": "BE-----",
        "state": "ACTIVATED",
        "devicePlatform": "WCAR",
        "softwareVersion": "3.7",
        "specification": {
            "title": "Škoda Enyaq",
            "manufacturingDate": "2024-04-02",
            "model": "Enyaq",
            "modelYear": "2024",
            "body": "SUV",
            "trimLevel": "60 Sportline",
            "systemCode": "UNKNOWN",
            "systemModelId": "5AZPF2",
            "maxChargingPowerInKW": 100,
            "battery": {
                "capacityInKWh": 58
            },
            "engine": {
                "type": "iV",
                "powerInKW": 132
            },
            "gearbox": {
                "type": "E1H"
            }
        },
```

https://mysmob.api.connect.skoda-auto.cz/api/v2/garage/initial-vehicle?connectivityGenerations=MOD3&connectivityGenerations=MOD4

https://mysmob.api.connect.skoda-auto.cz/api/v1/charging/{{vin}}
`{
    "isVehicleInSavedLocation": false,
    "status": {
        "chargingRateInKilometersPerHour": 25.0,
        "chargePowerInKw": 4.0,
        "remainingTimeToFullyChargedInMinutes": 40,
        "state": "CHARGING",
        "chargeType": "AC",
        "battery": {
            "remainingCruisingRangeInMeters": 278000,
            "stateOfChargeInPercent": 75
        }
    },
    "settings": {
        "targetStateOfChargeInPercent": 80,
        "preferredChargeMode": "MANUAL",
        "availableChargeModes": [
            "MANUAL"
        ],
        "chargingCareMode": "ACTIVATED",
        "autoUnlockPlugWhenCharged": "OFF",
        "maxChargeCurrentAc": "MAXIMUM",
        "batterySupport": "DISABLED"
    },
    "carCapturedTimestamp": "2024-06-03T12:20:12Z",
    "errors": []
}`

https://mysmob.api.connect.skoda-auto.cz/api/v1/vehicle-information/{{vin}}

`{
    "devicePlatform": "WCAR",
    "vehicleSpecification": {
        "title": "Škoda Enyaq",
        "manufacturingDate": "2024-04-02",
        "model": "Enyaq",
        "modelYear": "2024",
        "body": "SUV",
        "trimLevel": "60 Sportline",
        "systemCode": "UNKNOWN",
        "systemModelId": "5AZPF2",
        "maxChargingPowerInKW": 100,
        "battery": {
            "capacityInKWh": 58
        },
        "engine": {
            "type": "iV",
            "powerInKW": 132
        },
        "gearbox": {
            "type": "E1H"
        }
    },`

andig, 2024-06-03:
Das ist alles prima 😊. Aber: wenn das implementiert werden soll brauchts bitte vollständigen MITM Trace und Zugangsdaten für Test an info@evcc.io. Oder alternativ gleich einen PR- Vorlagen dafür gibt es in Hülle und Fülle 👍🏻.

GrimmiMeloni, 2024-06-03:
Ich habe heute Abend angefangen das mal reinzuhacken. Die Authentifizierung an sich ist unverändert. Der Garage Endpoint scheint aber nicht mehr der gleiche zu sein. Ich frage mal im Discord noch einem weiteren MITM dump von einer Fahrzeugauswahl bzw. Übersicht.
Wenn ich meine VIN hardcode, bekomme ich schonmal per API erste Daten. Das ganze mappt aber noch nicht, weil die Responses jetzt anders aussehen. Ist jetzt aber defintiv kein Hexenwerk. Ich bleib dran und denke in 1-2 Tagen haben wir etwas lauffähiges.

ruffieuxh, 2024-06-05:
Ich habe genau das gleiche Problem bei meinem Ford Mustang Mach-e auf der OpenWB mit SW V1.9 seit ca 2 Wochen:

Auszug aus dem SoC.log:

2024-06-05 13:13:25: PID: 25937: Lp1: Requesting SoC (LV0)
2024-06-05 13:13:31: PID: 25937: Lp1: Error from EVCC: 2024/06/05 13:13:31 rpc error: code = Unknown desc = cannot create vehicle type 'ford': login failed: unexpected status: 403 (Forbidden) (LV0)

Mit der App und dem Portal kann ich mich problemlos einloggen und die Credentials sind ok. Das Token hatte ich ebenfalls schon mal erneuert.

andig, 2024-06-05:
Ford != Skoda

ruffieuxh, 2024-06-05:
Ist mir schon klar. Ich wollte nur sicherstellen, dass klar ist, dass das nicht ein Skoda alleine Problem ist. Sondern m.E. ein generelles Problem. Im OpenWB forum wurde dieselbe Problematik mit demselben Fehler von verschiedenen Usern angebracht.

https://forum.openwb.de/viewtopic.php?t=3134&start=920

andig, 2024-06-08:
Wissen wir, ob *alle* Enyaqs das neue Protokoll unterstützen?

GrimmiMeloni, 2024-06-08:
> Wissen wir, ob _alle_ Enyaqs das neue Protokoll unterstützen?

100% sicher wissen wir es noch nicht. 

Mein „junger Gebrauchter“ aus 2022 unterstützt es. Umgekehrt wissen wir das alle neueren Autos nicht mehr in Skoda Connect registriert werden können, und diese somit bei der alten API raus sind.

PR #14186 (2024-06-10): add MySkoda support
fix #14100 

- [x] update wakeup
- [x] update start/stop charge
closed 2024-06-10 by andig