evcc-io/evcc #14102: Wechselrichter in der UI labeln

floqqi, 2024-05-29:
**Is your feature request related to a problem? Please describe.**

Ich habe (historisch bedingt) drei Wechselrichter von denen Erträge in evcc einfließen und kann in der UI nicht erkennen von welchem WR der jeweilige Ertrag stammt.

**Describe the solution you'd like**

Ich würde mir wünschen, dass ich beim Hovern den jeweiligen Wechselrichter erkennen kann. Lösen könnte man das zum Beispiel über ein zusätzlicher Feld `title` bei den `meters`. Wenn das nicht gesetzt ist, könnte man auf `name` fallbacken.

[image]

Würde dann bei mir resultieren in:

> pv1: 5,8 kW
> pv2: 6,0 kW
> pv3: 2,7 kW

Oder halt alternativ, wenn ich einen `title` vergeben hätte:

> Hauptgebäude: 5,8 kW
> Nebengebäude: 6,0 kW
> Garagendach: 2,7 kW

panzerdev, 2024-05-29:
https://github.com/evcc-io/evcc/issues/6781

floqqi, 2024-05-29:
@panzerdev ist damit ein Duplikat. Danke.

closed 2024-05-29 by floqqi