evcc-io/evcc #14106: Charger charges too fast in solar mode

mansurs, 2024-05-29:
### Describe the bug

When I plug in my car, it starts to load with around 10kw and normally it should go down to around 4kw, but this does not happen most of the time. It works, when I plug out from one car and plug in into another car. But if the charger was plugged out for too long and then I plug in it fails to get to 4kw.

What also helps is to switch from Solar to Fast mode and then back to Solar. Then it easily changes the charging speed.

I had a discussion https://github.com/evcc-io/evcc/discussions/13888 where I got the hint to write an issue.

[evcc-20240516-161251-debug (1).log](https://github.com/evcc-io/evcc/files/15483798/evcc-20240516-161251-debug.1.log)

### Steps to reproduce

1. Leave wallbox unplugged for a while, have enough SoC in battery for battery-supported charging
2. Plug-in cable
3. Charging starts with over 9kw and does not go down

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: manpi.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  etrel: trace
#  cache: error

# unique installation id
plant: *****

interval: 60s # control cycle interval

sponsortoken: *****

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
  - type: template
    template: sonnenbatterie
    usage: grid
    host: 172.16.5.51
    port: 8080
    name: grid1
  - type: template
    template: sonnenbatterie
    usage: pv
    host: 172.16.5.51
    port: 8080
    name: pv2
  - type: template
    template: sonnenbatterie
    usage: battery
    token: *****
    host: 172.16.5.51
    port: 8080
    name: battery3

vehicles:
  - name: ev4
    type: template
    template: hyundai
    title: IONIQ5
    user: *****
    password: *****
    vin: *****
    capacity: 70
    language: de
    mode: pv
  - name: ev5
    type: template
    template: bmw
    title: BMW iX3
    user: *****
    password: *****
    vin: *****
    capacity: 74
    mode: pv

chargers:
  - type: template
    template: etrel 
    host: 172.16.5.39
    port: 502  
    name: wallbox6

loadpoints:
  - title: Carport
    charger: wallbox6
    mode: now

site:
  title: Zuhause
  residualPower: 100
  meters:
    grid: grid1
    pv:
      - pv2
    battery:
      - battery3

tariffs:
  currency: EUR # (default EUR)
  grid:
    type: tibber
    token: *****
  co2:
    type: grünstromindex
    zip: 50354

influx:
  url: http://localhost:8086
  database: evcc_log
  token: *****
  org: home
```

### Log details

```text
Log is attached in description
```

### What type of operating system are you running?

Linux

### Version

0.126.4

VolkerK62, 2024-05-29:
Ich war in die Diskussion involviert.

Die Box fängt nach dem Anstecken eigenmächtig an mit voller Leistung zu laden.
Dann kommt `charger enable` und weil `bufferSoc` überschritten ist, wird weitergeladen.
evcc denkt es wird mit 6A geladen und greift nicht weiter ein. 
Wenn der Hausspeicher noch ziemlich leer ist, wird direkt wieder ausgeschaltet. Die Steuerung an sich funktioniert also.
MMn fehlt in Zusammenhang mit dem eigenmächtigen Start und `charger enable` noch ein `max charge current: 6A`.

premultiply, 2024-05-29:
```yaml
loadpoints:
  - title: Carport
    charger: wallbox6
    mode: now
```

Die Vorgabe ist ja auch hierdurch sofort mit voller Leistung zu laden.

mansurs, 2024-05-29:
Ja, das war ein Versuch. Initial war die loadpoint config ohne mode. Dann hatte ich das mit off getestet und dann noch mit now. Siehe https://github.com/evcc-io/evcc/discussions/13888

mansurs, 2024-05-29:
Hier ist ein aktuelles Beispiel mit Trace:

```
[site ] DEBUG 2024/05/29 15:22:10 ----
[etrel ] TRACE 2024/05/29 15:22:10 modbus: send 1c c0 00 00 00 06 01 04 00 1a 00 02
[etrel ] TRACE 2024/05/29 15:22:10 modbus: recv 1c c0 00 00 00 07 01 04 04 41 1c ac 08
[lp-1 ] DEBUG 2024/05/29 15:22:10 charge power: 9792W
[etrel ] TRACE 2024/05/29 15:22:10 modbus: send 1c c1 00 00 00 06 01 04 00 0e 00 06
[influx] TRACE 2024/05/29 15:22:10 write chargePower=map[value:9792] (map[loadpoint:Carport])
[etrel ] TRACE 2024/05/29 15:22:10 modbus: recv 1c c1 00 00 00 0f 01 04 0c 41 71 26 e9 41 6c 8b 44 41 6e 14 7b
[lp-1 ] DEBUG 2024/05/29 15:22:10 charge currents: [15.1 14.8 14.9]A
[influx] TRACE 2024/05/29 15:22:10 write chargeCurrents=map[l1:15.071999549865723 l2:14.784000396728516 l3:14.880000114440918] (map[loadpoint:Carport])
[http ] TRACE 2024/05/29 15:22:10 GET http://172.16.5.51:8080/api/v1/status
[http ] TRACE 2024/05/29 15:22:10 {"Apparent_output":4535,"BackupBuffer":"0","BatteryCharging":false,"BatteryDischarging":true,"Consumption_Avg":7703,"Consumption_W":10506,"Fac":49.99599838256836,"FlowConsumptionBattery":true,"FlowConsumptionGrid":true,"FlowConsumptionProduction":true,"FlowGridBattery":false,"FlowProductionBattery":false,"FlowProductionGrid":false,"GridFeedIn_W":-4232,"IsSystemInstalled":1,"OperatingMode":"2","Pac_total_W":4540,"Production_W":1732,"RSOC":65,"RemainingCapacity_Wh":13932,"Sac1":4535,"Sac2":null,"Sac3":null,"SystemStatus":"OnGrid","Timestamp":"2024-05-29 15:22:10","USOC":62,"Uac":228,"Ubat":205,"dischargeNotAllowed":false,"generator_autostart":false,"NVM_REINIT_STATUS":0}
[site ] DEBUG 2024/05/29 15:22:10 pv power: 1732W
[influx] TRACE 2024/05/29 15:22:10 write pvPower=map[value:1732] (map[])
[http ] TRACE 2024/05/29 15:22:10 GET http://172.16.5.51:8080/api/v1/status
[influx] TRACE 2024/05/29 15:22:10 write pvEnergy=map[value:0] (map[])
[influx] TRACE 2024/05/29 15:22:10 write pvPower=map[value:1732] (map[id:1])
[influx] TRACE 2024/05/29 15:22:10 write pvEnergy=map[value:0] (map[id:1])
[http ] TRACE 2024/05/29 15:22:10 {"Apparent_output":4535,"BackupBuffer":"0","BatteryCharging":false,"BatteryDischarging":true,"Consumption_Avg":7703,"Consumption_W":10506,"Fac":49.99599838256836,"FlowConsumptionBattery":true,"FlowConsumptionGrid":true,"FlowConsumptionProduction":true,"FlowGridBattery":false,"FlowProductionBattery":false,"FlowProductionGrid":false,"GridFeedIn_W":-4232,"IsSystemInstalled":1,"OperatingMode":"2","Pac_total_W":4540,"Production_W":1732,"RSOC":65,"RemainingCapacity_Wh":13932,"Sac1":4535,"Sac2":null,"Sac3":null,"SystemStatus":"OnGrid","Timestamp":"2024-05-29 15:22:10","USOC":62,"Uac":228,"Ubat":205,"dischargeNotAllowed":false,"generator_autostart":false,"NVM_REINIT_STATUS":0}
[http ] TRACE 2024/05/29 15:22:10 GET http://172.16.5.51:8080/api/v1/status
[http ] TRACE 2024/05/29 15:22:10 {"Apparent_output":4535,"BackupBuffer":"0","BatteryCharging":false,"BatteryDischarging":true,"Consumption_Avg":7703,"Consumption_W":10506,"Fac":49.99599838256836,"FlowConsumptionBattery":true,"FlowConsumptionGrid":true,"FlowConsumptionProduction":true,"FlowGridBattery":false,"FlowProductionBattery":false,"FlowProductionGrid":false,"GridFeedIn_W":-4232,"IsSystemInstalled":1,"OperatingMode":"2","Pac_total_W":4540,"Production_W":1732,"RSOC":65,"RemainingCapacity_Wh":13932,"Sac1":4535,"Sac2":null,"Sac3":null,"SystemStatus":"OnGrid","Timestamp":"2024-05-29 15:22:10","USOC":62,"Uac":228,"Ubat":205,"dischargeNotAllowed":false,"generator_autostart":false,"NVM_REINIT_STATUS":0}
[site ] DEBUG 2024/05/29 15:22:10 battery soc: 62%
[influx] TRACE 2024/05/29 15:22:10 write batteryCapacity=map[value:0] (map[])
[influx] TRACE 2024/05/29 15:22:10 write batterySoc=map[value:62] (map[])
[site ] DEBUG 2024/05/29 15:22:10 battery power: 4540W
[influx] TRACE 2024/05/29 15:22:10 write batteryPower=map[value:4540] (map[])
[influx] TRACE 2024/05/29 15:22:10 write batteryEnergy=map[value:0] (map[])
[http ] TRACE 2024/05/29 15:22:10 GET http://172.16.5.51:8080/api/v1/status
[influx] TRACE 2024/05/29 15:22:10 write batteryPower=map[value:4540] (map[id:1])
[influx] TRACE 2024/05/29 15:22:10 write batteryEnergy=map[value:0] (map[id:1])
[influx] TRACE 2024/05/29 15:22:10 write batterySoc=map[value:62] (map[id:1])
[influx] TRACE 2024/05/29 15:22:10 write batteryCapacity=map[value:0] (map[id:1])
[influx] TRACE 2024/05/29 15:22:10 write batteryControllable=map[value:true] (map[id:1])
[http ] TRACE 2024/05/29 15:22:10 {"Apparent_output":4535,"BackupBuffer":"0","BatteryCharging":false,"BatteryDischarging":true,"Consumption_Avg":7703,"Consumption_W":10506,"Fac":49.99599838256836,"FlowConsumptionBattery":true,"FlowConsumptionGrid":true,"FlowConsumptionProduction":true,"FlowGridBattery":false,"FlowProductionBattery":false,"FlowProductionGrid":false,"GridFeedIn_W":-4232,"IsSystemInstalled":1,"OperatingMode":"2","Pac_total_W":4540,"Production_W":1732,"RSOC":65,"RemainingCapacity_Wh":13932,"Sac1":4535,"Sac2":null,"Sac3":null,"SystemStatus":"OnGrid","Timestamp":"2024-05-29 15:22:10","USOC":62,"Uac":228,"Ubat":205,"dischargeNotAllowed":false,"generator_autostart":false,"NVM_REINIT_STATUS":0}
[site ] DEBUG 2024/05/29 15:22:10 grid meter: 4232W
[site ] DEBUG 2024/05/29 15:22:10 site power: 8872W
[influx] TRACE 2024/05/29 15:22:10 write gridPower=map[value:4232] (map[])
[influx] TRACE 2024/05/29 15:22:10 write homePower=map[value:712] (map[])
[lp-1 ] DEBUG 2024/05/29 15:22:10 !! session: chargeRater.chargedEnergy=0.1 - chargedAtStartup=0.0
[... cut ...] tery":true,"FlowConsumptionGrid":true,"FlowConsumptionProduction":true,"FlowGridBattery":false,"FlowProductionBattery":false,"FlowProductionGrid":false,"GridFeedIn_W":-3719,"IsSystemInstalled":1,"OperatingMode":"2","Pac_total_W":4533,"Production_W":2001,"RSOC":63,"RemainingCapacity_Wh":13542,"Sac1":4533,"Sac2":null,"Sac3":null,"SystemStatus":"OnGrid","Timestamp":"2024-05-29 15:24:20","USOC":60,"Uac":229,"Ubat":204,"dischargeNotAllowed":false,"generator_autostart":false,"NVM_REINIT_STATUS":0}
```

andig, 2024-05-29:
Warum taucht denn hier jede Nachricht 3x auf? Da stimmt doch schon was nicht???

mansurs, 2024-05-29:
Ja, sonnen… 😢 
Hab http und influx raus gegrep'd:
```
[site  ] DEBUG 2024/05/29 15:22:10 ----
[etrel ] TRACE 2024/05/29 15:22:10 modbus: send 1c c0 00 00 00 06 01 04 00 1a 00 02
[etrel ] TRACE 2024/05/29 15:22:10 modbus: recv 1c c0 00 00 00 07 01 04 04 41 1c ac 08
[lp-1  ] DEBUG 2024/05/29 15:22:10 charge power: 9792W
[etrel ] TRACE 2024/05/29 15:22:10 modbus: send 1c c1 00 00 00 06 01 04 00 0e 00 06
[etrel ] TRACE 2024/05/29 15:22:10 modbus: recv 1c c1 00 00 00 0f 01 04 0c 41 71 26 e9 41 6c 8b 44 41 6e 14 7b
[lp-1  ] DEBUG 2024/05/29 15:22:10 charge currents: [15.1 14.8 14.9]A
[site  ] DEBUG 2024/05/29 15:22:10 pv power: 1732W
[site  ] DEBUG 2024/05/29 15:22:10 battery soc: 62%
[site  ] DEBUG 2024/05/29 15:22:10 battery power: 4540W
[site  ] DEBUG 2024/05/29 15:22:10 grid meter: 4232W
[site  ] DEBUG 2024/05/29 15:22:10 site power: 8872W
[lp-1  ] DEBUG 2024/05/29 15:22:10 !! session: chargeRater.chargedEnergy=0.1 - chargedAtStartup=0.0
[etrel ] TRACE 2024/05/29 15:22:10 modbus: send 1c c2 00 00 00 06 01 04 00 00 00 01
[etrel ] TRACE 2024/05/29 15:22:10 modbus: recv 1c c2 00 00 00 05 01 04 02 00 04
[lp-1  ] DEBUG 2024/05/29 15:22:10 charger status: C
[lp-1  ] INFO 2024/05/29 15:22:10 car connected
[lp-1  ] DEBUG 2024/05/29 15:22:10 vehicle api refresh
[lp-1  ] DEBUG 2024/05/29 15:22:10 pv timer elapse
[lp-1  ] DEBUG 2024/05/29 15:22:10 pv timer inactive
[lp-1  ] INFO 2024/05/29 15:22:10 start charging ->
[lp-1  ] DEBUG 2024/05/29 15:22:10 wake-up timer: stop
[db    ] TRACE 2024/05/29 15:22:10 INSERT INTO `sessions` (`created`,`finished`,`loadpoint`,`identifier`,`vehicle`,`odometer`,`meter_start_kwh`,`meter_end_kwh`,`charged_kwh`,`charge_duration`,`solar_percentage`,`price`,`price_per_kwh`,`co2_per_kwh`) VALUES ("2024-05-29 15:22:10.445","0000-00-00 00:00:00","Carport","","",NULL,NULL,NULL,0,NULL,NULL,NULL,NULL,NULL) RETURNING `id` 1 <nil>
[hyundai] TRACE 2024/05/29 15:22:10 GET https://prd.eu-ccapi.hyundai.com:8080/api/v1/spa/vehicles/fc6c689d-a39d-4a83-a650-d2487ec297e0/status/latest
[hyundai] TRACE 2024/05/29 15:22:10 {"retCode":"S","resCode":"0000","resMsg":{"vehicleStatusInfo":{"vehicleLocation":{"coord":{"lat":50.888317,"lon":6.828597,"alt":0,"type":0},"head":125,"speed":{"value":0,"unit":0},"accuracy":{"hdop":0,"pdop":0},"time":"20240528124756"},"vehicleStatus":{"airCtrlOn":false,"engine":false,"doorLock":true,"doorOpen":{"frontLeft":0,"frontRight":0,"backLeft":0,"backRight":0},"trunkOpen":false,"airTemp":{"value":"02H","unit":0,"hvacTempType":1},"defrost":false,"acc":false,"evStatus":{"batteryCharge":false,"batteryStatus":73,"batteryPlugin":0,"remainTime2":{"etc1":{"value":32,"unit":1},"etc2":{"value":470,"unit":1},"etc3":{"value":115,"unit":1},"atc":{"value":115,"unit":1}},"drvDistance":[{"rangeByFuel":{"evModeRange":{"value":300,"unit":1},"totalAvailableRange":{"value":300,"unit":1}},"type":2}],"reservChargeInfos":{"reservChargeInfo":{"reservChargeInfoDetail":{"reservInfo":{"day":[9],"time":{"time":"1200","timeSection":0}},"reservChargeSet":false,"reservFatcSet":{"defrost":false,"airTemp":{"value":"00H","unit":0,"hvacTempType":1},"airCtrl":0,"heating1":0}}},"offpeakPowerInfo":{"offPeakPowerTime1":{"starttime":{"time":"1200","timeSection":0},"endtime":{"time":"1200","timeSection":0}},"offPeakPowerFlag":0},"reserveChargeInfo2":{"reservChargeInfoDetail":{"reservInfo":{"day":[9],"time":{"time":"1200","timeSection":0}},"reservChargeSet":false,"reservFatcSet":{"defrost":false,"airTemp":{"value":"00H","unit":0,"hvacTempType":1},"airCtrl":0,"heating1":0}}},"reservFlag":0,"ect":{"start":{"day":0,"time":{"time":"0000","timeSection":0}},"end":{"day":0,"time":{"time":"0000","timeSection":0}}},"targetSOClist":[{"targetSOClevel":100,"dte":{"rangeByFuel":{"evModeRange":{"value":421,"unit":1},"totalAvailableRange":{"value":421,"unit":1}},"type":2},"plugType":0},{"targetSOClevel":100,"dte":{"rangeByFuel":{"evModeRange":{"value":421,"unit":1},"totalAvailableRange":{"value":421,"unit":1}},"type":2},"plugType":1}]},"chargePortDoorOpenStatus":2,"batteryPreconditioning":false,"batterySoh":0},"ign3":true,"hoodOpen":false,"transCond":false,"steerWheelHeat":0,"sideBackWindowHeat":0,"tirePressureLamp":{"tirePressureLampAll":0,"tirePressureLampFL":0,"tirePressureLampFR":0,"tirePressureLampRL":0,"tirePressureLampRR":0},"battery":{"batSoc":81,"batState":0,"sjbDeliveryMode":0,"batSignalReferenceValue":{"batWarning":65},"powerAutoCutMode":2},"lampWireStatus":{"stopLamp":{"leftLamp":false,"rightLamp":false},"headLamp":{"headLampStatus":false,"leftLowLamp":false,"rightLowLamp":false,"leftHighLamp":false,"rightHighLamp":false,"leftBifuncLamp":false,"rightBifuncLamp":false},"turnSignalLamp":{"leftFrontLamp":false,"rightFrontLamp":false,"leftRearLamp":false,"rightRearLamp":false}},"smartKeyBatteryWarning":false,"washerFluidStatus":false,"breakOilStatus":false,"sleepModeCheck":false,"time":"20240529141521","remoteWaitingTimeAlert":{"remoteControlAvailable":1,"remoteControlWaitingTime":168,"elapsedTime":"44:24:49"},"systemCutOffAlert":0,"tailLampStatus":0,"hazardStatus":0},"odometer":{"value":27791.8,"unit":1}}},"msgId":"f40f2041-93cf-49a0-89d5-155e3fb89366"}
[hyundai] TRACE 2024/05/29 15:22:10 GET https://prd.eu-ccapi.hyundai.com:8080/api/v1/spa/vehicles/fc6c689d-a39d-4a83-a650-d2487ec297e0/status
[hyundai] TRACE 2024/05/29 15:22:15 {"retCode":"S","resCode":"0000","resMsg":{"airCtrlOn":false,"engine":false,"doorLock":true,"doorOpen":{"frontLeft":0,"frontRight":0,"backLeft":0,"backRight":0},"trunkOpen":false,"airTemp":{"value" [... cut ...]
[lp-1  ] DEBUG 2024/05/29 15:24:10 charger status: C
[lp-1  ] DEBUG 2024/05/29 15:24:10 soc estimated: 46.40% (vehicle: 46.00%)
[lp-1  ] DEBUG 2024/05/29 15:24:10 vehicle soc: 46%
[lp-1  ] DEBUG 2024/05/29 15:24:10 vehicle range: 149km
[etrel ] TRACE 2024/05/29 15:24:10 modbus: send 1c d0 00 00 00 06 01 04 00 04 00 02
[etrel ] TRACE 2024/05/29 15:24:10 modbus: recv 1c d0 00 00 00 07 01 04 04 41 80 00 00
[lp-1  ] DEBUG 2024/05/29 15:24:10 pv charge current: 0A = 6A + -12.1A (8328W @ 3p)
[site  ] DEBUG 2024/05/29 15:25:10 ----
```

andig, 2024-05-30:
Anscheinend fängt die Box selbst an zu laden:

> [lp-1  ] WARN 2024/05/29 15:22:15 charger out of sync: expected disabled, got enabled

Danach sollte eigentlich deaktiviert werden:

> [lp-1  ] DEBUG 2024/05/29 15:22:16 pv charge current: 0A = 6A + -12.9A (8914W @ 3p)

Das scheint entweder nicht zu passieren oder funktioniert nicht. Warum ist aus dem Log nicht klar- da müsste man wohl mit dem Debugger rein schauen.

Besteht das Problem denn immer oder nur manchmal?

VolkerK62, 2024-05-30:
bufferSoc ist überschritten.

Wenn der Speicher "leer" ist, funktioniert das abschalten.
https://github.com/evcc-io/evcc/discussions/13888#discussioncomment-9590025

andig, 2024-05-30:
@VolkerK62 Du bist ein Schatz! Siehe https://github.com/evcc-io/evcc/pull/14126 um das im Log klarer zu machen, zumal wir auch den bufferSoc sonst nicht mit loggen...

closed 2024-05-30 by andig

VolkerK62, 2024-05-30:
Löst aber das Problem nicht.

> Die Box fängt nach dem Anstecken eigenmächtig an mit voller Leistung zu laden. Dann kommt `charger enable` und weil `bufferSoc` überschritten ist, wird weitergeladen. evcc denkt es wird mit 6A [...]

andig, 2024-05-30:
Bin nicht sicher- aber kann doch einfach morgen im nightly ausprobiert werden!

mansurs, 2024-05-30:
Ok dann logge ich das mal mit dem Nightly

mansurs, 2024-05-31:
Na mega, jetzt funktioniert es! Lag das an der [Änderung in der if-condition](https://github.com/evcc-io/evcc/pull/14126/files#diff-c456526d4bd9022a488ce32de208b40d86016e2c40b0c81a2dffe557c186377dR1217)?
[evcc-nightly-filtered.log](https://github.com/user-attachments/files/15511027/evcc-nightly-filtered.log)

mansurs, 2024-06-04:
Sieht so aus als wäre es ein Zufall, dass es geklappt hat oder ich hab was Neues entdeckt [evcc-20240604-1527-trace-filtered.log](https://github.com/user-attachments/files/15552174/evcc-20240604-1527-trace-filtered.log)

Klappt wieder nicht :/

cvg-de, 2024-06-05:
Zusätzlicher Datenpunkt von mir:
[2024-06-05 Log-Auszug.txt](https://github.com/user-attachments/files/15585302/2024-06-05.Log-Auszug.txt)

evcc setzt den Ladestrom um 11:03:26 auf 9A hoch, um 11:03:46 ist das sichtbar.
Um 11:04:47 wird versucht, ihn wieder auf 6A herunterzusetzen, das klappt aber offenbar nicht. Es wird aber auch kein Fehler protokolliert.
Ich habe eine Vermutung: Die Funktion lp.setlimit setzt die Variable lp.chargeCurrent auf den ermittelten Sollwert, und die Funktion lp.effectiveCurrent gibt genau diesen Wert zurück (und nicht den gemessenen Ladestrom). Damit beziehen sich alle weiteren Berechnungen auf einen zu kleinen Wert, und wenn der schon das Minimum 6A ist, sieht nie die Notwendigkeit, noch weiter runterzugehen. Erst ein erneutes Hochschalten des Ladestrom wenn mehr Sonne da ist bringt die Logik wieder ans Laufen, weil lp.chargeCurrent dann wieder auf einen höheren Wert gesetzt wird.
Evtl. würde ein Debug-Logging auch in setLimit() weiter helfen? Problem scheint ja zu sein, dass der Charger auf den Befehl nicht reagiert. Bei mir ein Hardy Barth eCharge ePH1-2T22.

mansurs, 2024-06-12:
@andig Irgendwas stimmt da immer noch nicht. 
[evcc-20240611-182110-trace-filtered.log](https://github.com/user-attachments/files/15802877/evcc-20240611-182110-trace-filtered.log)

Um 17:46 ging's los, der Charger startet wie immer von alleine. Dann wurde das Auto erkannt und auf PV mode gewechselt, aber die Ladegeschwindigkeit bleibt bei 9k...

andig, 2024-06-12:
Damit wir es nicht doppelt machen: es gib schon eine Discussion dazu. 1x reicht.