evcc-io/evcc #14110: Loadpoint on 3phase 3*230 no neutral in delta config : bad power reporting

hcouplet, 2024-05-29:
### Describe the bug

Here in belgium I have 3 phase 230V no neutral (in delta config).
I have a wallbox pulsar (ocpp) and a tesla model 3.   
With my setup, the charging power for a 28A current is 11 Kw (3 * 230 * 28 / 1,73).  1,73= square_root (3).
With a normal 3 phase 400V, this would be 19,3 Kw (3 * 400 * 28 / 1,73). 

I tried to setup the loadpoint phases to 1,73 intstead of 3 but this doesn't work. 
This is causing issue as the system is trying to adapt charging power with bad input value.

### Steps to reproduce

1. connect to a 230V tri phase in delta.
2.Start charging at max (28A)
3. check the charging power is 19,3 instead of 11Kw
...

### Configuration details

```yaml
chargers:
  - name: WB_Home
    type: template
    template: pulsarplus
    timeout: 2m # optional 

vehicles:
  - name: Model3
    type: template
    template: tesla
    title: # Wird in der Benutzeroberfläche angezeigt (optional)
    accessToken: ...
    refreshToken: ...
    vin: LRW3E..
    capacity: 60

loadpoints:
  - title: House # display name for UI
    charger: WB_Home # charger
    #meter: charge # charge meter
    mode: "pv" # default charge mode to apply when vehicle is disconnected; use "off" to disable by default if charger is publicly available
    phases: 1.73
    minCurrent: 6
    maxCurrent: 28
#
```

### Log details

```text
[site ] DEBUG 2024/05/29 16:26:42 ----
[lp-1 ] DEBUG 2024/05/29 16:26:42 charge power: 14490W
[site ] DEBUG 2024/05/29 16:26:42 grid meter: -5520W
[site ] DEBUG 2024/05/29 16:26:42 grid currents: [-10.2 26.5 -19.4]A
[site ] DEBUG 2024/05/29 16:26:42 pv power: 20010W
[site ] DEBUG 2024/05/29 16:26:42 site power: -5520W
[lp-1 ] DEBUG 2024/05/29 16:26:42 !! session: chargeRater.chargedEnergy=14.5 - chargedAtStartup=0.0
[lp-1 ] DEBUG 2024/05/29 16:26:42 charger status: C
[lp-1 ] DEBUG 2024/05/29 16:26:42 soc estimated: 78.80% (vehicle: 74.00%)
[lp-1 ] DEBUG 2024/05/29 16:26:42 vehicle soc: 79%
[lp-1 ] DEBUG 2024/05/29 16:26:42 vehicle soc limit: 100%
[lp-1 ] DEBUG 2024/05/29 16:26:42 vehicle range: 325km
[lp-1 ] DEBUG 2024/05/29 16:26:42 pv charge current: 29A = 21A + 8A (-5520W @ 3p)
[lp-1 ] DEBUG 2024/05/29 16:26:42 max charge current: 28A
[site ] DEBUG 2024/05/29 16:27:12 ----
[lp-1 ] DEBUG 2024/05/29 16:27:12 charge power: 19320W
[site ] DEBUG 2024/05/29 16:27:12 grid meter: -4709W
[site ] DEBUG 2024/05/29 16:27:12 grid currents: [-8.49 24.3 -18.4]A
[site ] DEBUG 2024/05/29 16:27:12 pv power: 24029W
[site ] DEBUG 2024/05/29 16:27:12 site power: -4709W
[lp-1 ] DEBUG 2024/05/29 16:27:12 !! session: chargeRater.chargedEnergy=14.6 - chargedAtStartup=0.0
[lp-1 ] DEBUG 2024/05/29 16:27:12 charger status: C
[lp-1 ] DEBUG 2024/05/29 16:27:12 soc estimated: 79.05% (vehicle: 74.00%)
[lp-1 ] DEBUG 2024/05/29 16:27:12 vehicle soc: 79%
[lp-1 ] DEBUG 2024/05/29 16:27:12 vehicle soc limit: 100%
[lp-1 ] DEBUG 2024/05/29 16:27:12 vehicle range: 325km
[lp-1 ] DEBUG 2024/05/29 16:27:12 pv charge current: 34.8A = 28A + 6.82A (-4709W @ 3p)
```

### What type of operating system are you running?

Linux

### Version

0.126.4

hcouplet, 2024-05-29:
[image]
[image]

hcouplet, 2024-05-29:
also here are more details on tesla charger with 3x220 in delta : https://teslamotorsclub.com/tmc/threads/wiring-evse-on-230v-three-phase-without-neutral.78134/#post-7693629

andig, 2024-05-29:
@premultiply could you take a look?

VolkerK62, 2024-05-29:
Due to there is no measurement, `phases: 2` at vehicle could be a "near by solution".

premultiply, 2024-05-31:
Exactly. You should set the loadpoint to 3 phases and try to set the vehice config to 2 phases.

closed 2026-10-19 by premultiply