evcc-io/evcc #14111: API Point "POST /api/loadpoints/<id>/plan/energy/<energy>/<time>" funktioniert nicht

puzich, 2024-05-29:
### Describe the bug

Ich versuche über u. g. Post Request eine Ladeplanung zu setzen. und erhalte auch u. g. Ergebnis. Aber die Ladeplanung wird trotzdem nicht gesetzt und ist weder via API Call zu sehen noch via GUI. 

# curl -s -X POST http://localhost/api/loadpoints/1/plan/energy/11040/2024-05-30T04:00:00Z | jq .
{
  "result": {
    "energy": 11040,
    "time": "2024-05-30T04:00:00Z"
  }
}

### Steps to reproduce

# curl -s -X POST http://localhost/api/loadpoints/1/plan/energy/11040/2024-05-30T04:00:00Z | jq .

### Configuration details

```yaml
<details><summary>Konfiguration (/etc/evcc.yaml)</summary>

# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 80

#log: debug
log: info
#levels:
#  bmw: debug
#  cache: error
##  modbus: trace
##log.easee: debug

# unique installation id
plant: *****
sponsortoken: *****
#interval: 30s # control cycle interval
#interval: 10s # control cycle interval
interval: 15s # control cycle interval

site:
  title: Home
  meters:
    grid: my_grid
    pv: my_pv
    battery: my_battery
  ##prioritySoc: 50
#  residualPower: 350
  residualPower: 100
  ##bufferSoc: 80
  ##batteryDischargeControl: true

meters:
  - name: my_grid
    type: template
    template: sofarsolar-g3
    usage: grid
    # modbus: rs485serial
    modbus: tcpip
    id: 1
    # device: /dev/ttyUSB0
    # baudrate: 9600 # Pruefe Geraeteeinstellungen, typische Werte sind 9600, 19200, 38400, 57600, 115200
    # comset: "8N1" # Kommunikationsparameter fuer den Adapter
    # delay: 1s # Optional
    host: 192.168.166.12
    port: 8899
  - name: my_pv
    type: template
    template: sofarsolar-g3
    usage: pv
    # modbus: rs485serial
    modbus: tcpip
    id: 1
    # device: /dev/ttyUSB0
    # baudrate: 9600 # Pruefe Geraeteeinstellungen, typische Werte sind 9600, 19200, 38400, 57600, 115200
    # comset: "8N1" # Kommunikationsparameter fuer den Adapter
    # delay: 1s # Optional
    host: 192.168.166.12
    port: 8899
  - name: my_battery
    type: template
    template: sofarsolar-g3
    usage: battery
    # modbus: rs485serial
    modbus: tcpip
    id: 1
    # device: /dev/ttyUSB0
    # baudrate: 9600 # Pruefe Geraeteeinstellungen, typische Werte sind 9600, 19200, 38400, 57600, 115200
    # comset: "8N1" # Kommunikationsparameter fuer den Adapter
    # delay: 1s # Optional
    host: 192.168.166.12
    port: 8899

chargers:
- name: wallbox1
  type: template
  template: easee
  user: *****
  password: *****
  charger: EHPW8R46
- name: wallbox2
  type: template
  template: easee
  user: *****
  password: *****
  charger: EHLC2AVR

vehicles:
- type: template
  template: bmw
  title: BMW i3
  user: *****
  password: *****
  vin: *****
  priority: 100
  name: bmw_i3
  capacity: 28
- type: template
  template: tesla-command
  title: Tesla ECTO-1
  accessToken: *****
  refreshToken: *****
  priority: 10
  name: tesla_ecto1
  capacity: 78

loadpoints:
  - title: Carport links # display name for UI
    charger: wallbox1 # Wallbox Referenz
    vehicle: bmw_i3 # Referenz auf Standardfahrzeug
    mode: pv # charge mode (off, now, minpv, pv)
    #resetOnDisconnect: true
    priority: 100
  - title: Carport rechts # display name for UI
    charger: wallbox2 # Wallbox Referenz
    vehicle: tesla_ecto1 # Referenz auf Standardfahrzeug
    mode: pv # charge mode (off, now, minpv, pv)
    #resetOnDisconnect: true
    priority: 1
#    enable: # pv mode enable behavior
#      delay: 1m # threshold must be exceeded for this long
#      threshold: 0 # grid power threshold (in Watts, negative=export). If zero, export must exceed minimum charge power to enable
#    disable: # pv mode disable behavior
#      delay: 5m # threshold must be exceeded for this long
#      threshold: 0 # maximum import power (W)
#    guardDuration: 5m # switch charger contactor not more often than this (default 5m)

tariffs:
  currency: EUR # (default EUR)
  grid:
    # static grid price
    type: fixed
    price: 0.279 # [currency]/kWh
  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.082 # [currency]/kWh
#  co2:
#    type: grünstromindex
#    zip: 21244

# mqtt message broker
mqtt:
  broker: 192.168.166.8:1883
  topic: evcc # root topic for publishing, set empty to disable publishing
#  clientid: EVCC
  user: *****
  password: *****

#levels:
#  bmw: debug
#  lp-1: debug

#influx:
#  url: http://192.168.166.6:8086
#  database: evcc
#  token: *****
#  org: default

</details>
Version: `0.126.4`
```

### Log details

```text
kein Eintrag im Log hierzu. Loglevel war "debug"
```

### What type of operating system are you running?

Linux

### Version

0.126.4

naltatis, 2024-05-29:
Mit dem Command setzt du einen energiemengenbasierten Ladeplan. Der ist nur dann aktiv, wenn am Ladepunkt ein Fahrzeug ohne SoC steckt (Offline Vehicle oder Gast).

puzich, 2024-05-29:
Oh Keee? Und wie setze ich die Ladeplanung?

naltatis, 2024-05-29:
Bei Fahrzeugen mit bekanntem SoC und Kapazität kannst du das direkt am Fahrzeug machen:

Beispiel:
```
curl http://selsillendusche.local:7070/api/vehicles/[vehicle]/plan/soc/80/2024-05-31T05:00:00.000Z -X POST
```

naltatis, 2024-05-29:
`[vehicle]` ist hier der Name des Fahrzeugs aus der `evcc.yaml`. Also bspw. `bmw_i3`.

puzich, 2024-05-29:
Vielen Dank! Habe mir nur die Loadpoints APIs angeschaut und kam nicht auf die Idee bei den Vehicles zu schauen! Vielen Dank für den Support. Problem war mal wieder an der Tastatur!

closed 2024-05-29 by puzich

naltatis, 2024-05-29:
Gerne.