evcc-io/evcc #14115: Charger shows 0.0 kW while charging with 11 kW

mztiq, 2024-05-29:
### Describe the bug

Hi,

like the title says, whenever I'm charging my car it does not show the power used and therefor doesn't show any historical charging sessions either. (The car is charging at 11 kW on those screenshots).  

[image]  

[image]

**Inverter:**
Fronius Symo GEN24 10.0 Plus, connected via Solar API.  

**Charger:**
Everon EVBox Elvie, connected via template but without integrated meter.  

The documentation says, that even without an integrated meter at the wallbox it should show those charging values:

> This entry is only needed if the used charger doesn't perform its own current measurement or if the measurement values cannot be read from the charger by evcc. However, even in this case, this entry is optional, because evcc assumes that the set maximum current will also be used for charging.
(Source: https://docs.evcc.io/en/docs/reference/configuration/loadpoints#meter)  

Even the log is showing `charge power: 0W`.

Any help is appreciated.

### Steps to reproduce

1. Start charging the car
2. Check GUI
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: <ID>

interval: 30s # control cycle interval

meters:
- type: template
  template: fronius-solarapi-v1 
  usage: grid  
  host: <IPADDRESS>  
  user: customer  
  password: <PASSWORD> 
  name: grid1
- type: template
  template: fronius-solarapi-v1 
  usage: pv  
  host: <IPADDRESS>  
  user: customer  
  password: <PASSWORD> 
  name: pv2
- type: template
  template: fronius-solarapi-v1 
  usage: battery  
  host: <IPADDRESS>  
  user: customer  
  password: <PASSWORD> 
  name: battery3

chargers:
- type: template
  template: elvi 
  stationid: <STATION_ID>  
  idtag: <ID_TAG>
  connecttimeout: 5m
  timeout: 2m
  meter: false
  name: wallbox_garage

vehicles:
- type: template
  template: hyundai 
  title: Hyundai IONIQ 5  
  icon: car  
  user: <USER>  
  password: <PASSWORD>  
  vin: <VIN>  
  capacity: 77.4  
  phases: 3  
  cache: 15m
  language: de  
  mode: pv 
  minCurrent: 6  
  maxCurrent: 16
  name: ev4

loadpoints:
- title: Garage
  charger: wallbox_garage
  vehicle: ev4
  mode: pv
  enable:
    threshold: 0
    delay: 15m
  disable:
    threshold: 300
    delay: 15m

site:
  title: Mein Zuhause
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3
```

### Log details

```text
May 29 19:30:30 evcc evcc[184878]: [lp-1  ] INFO 2024/05/29 19:30:30 vehicle updated: unknown -> Hyundai IONIQ 5
May 29 19:30:30 evcc evcc[184878]: [lp-1  ] DEBUG 2024/05/29 19:30:30 set charge mode: pv
May 29 19:30:30 evcc evcc[184878]: [site  ] DEBUG 2024/05/29 19:30:30 ----
May 29 19:30:30 evcc evcc[184878]: [lp-1  ] DEBUG 2024/05/29 19:30:30 charge power: 0W
May 29 19:30:30 evcc evcc[184878]: [site  ] DEBUG 2024/05/29 19:30:30 pv power: 10W
May 29 19:30:30 evcc evcc[184878]: [site  ] DEBUG 2024/05/29 19:30:30 battery soc: 48%
May 29 19:30:30 evcc evcc[184878]: [site  ] DEBUG 2024/05/29 19:30:30 battery power: 5578W
May 29 19:30:30 evcc evcc[184878]: [site  ] DEBUG 2024/05/29 19:30:30 grid meter: 5640W
May 29 19:30:30 evcc evcc[184878]: [site  ] DEBUG 2024/05/29 19:30:30 site power: 11218W
May 29 19:30:30 evcc evcc[184878]: [lp-1  ] DEBUG 2024/05/29 19:30:30 vehicle odometer: 5132km
May 29 19:30:30 evcc evcc[184878]: [lp-1  ] DEBUG 2024/05/29 19:30:30 charge total import: 0.000kWh
May 29 19:30:30 evcc evcc[184878]: [lp-1  ] DEBUG 2024/05/29 19:30:30 charger status: C
May 29 19:30:30 evcc evcc[184878]: [lp-1  ] DEBUG 2024/05/29 19:30:30 charge start energy: 0kWh
May 29 19:30:30 evcc evcc[184878]: [lp-1  ] INFO 2024/05/29 19:30:30 car connected
May 29 19:30:30 evcc evcc[184878]: [lp-1  ] DEBUG 2024/05/29 19:30:30 pv timer elapse
May 29 19:30:30 evcc evcc[184878]: [lp-1  ] DEBUG 2024/05/29 19:30:30 pv timer inactive
May 29 19:30:30 evcc evcc[184878]: [lp-1  ] DEBUG 2024/05/29 19:30:30 charge total import: 0.000kWh
May 29 19:30:30 evcc evcc[184878]: [lp-1  ] DEBUG 2024/05/29 19:30:30 charge start energy: 0kWh
May 29 19:30:30 evcc evcc[184878]: [lp-1  ] INFO 2024/05/29 19:30:30 start charging ->
May 29 19:30:30 evcc evcc[184878]: [lp-1  ] DEBUG 2024/05/29 19:30:30 wake-up timer: stop
May 29 19:30:35 evcc evcc[184878]: [lp-1  ] DEBUG 2024/05/29 19:30:35 vehicle status: C
May 29 19:30:35 evcc evcc[184878]: [lp-1  ] DEBUG 2024/05/29 19:30:35 vehicle soc: 97%
May 29 19:30:35 evcc evcc[184878]: [lp-1  ] DEBUG 2024/05/29 19:30:35 vehicle soc limit: 100%
May 29 19:30:35 evcc evcc[184878]: [lp-1  ] DEBUG 2024/05/29 19:30:35 vehicle range: 459km
May 29 19:30:35 evcc evcc[184878]: [lp-1  ] WARN 2024/05/29 19:30:35 charger logic error: disabled but charging
May 29 19:30:35 evcc evcc[184878]: [lp-1  ] DEBUG 2024/05/29 19:30:35 charger enable
May 29 19:30:35 evcc evcc[184878]: [lp-1  ] DEBUG 2024/05/29 19:30:35 pv charge current: 0A = 0A + -16.3A (11218W @ 3p)
May 29 19:30:35 evcc evcc[184878]: [lp-1  ] DEBUG 2024/05/29 19:30:35 pv disable timer reset
May 29 19:30:35 evcc evcc[184878]: [lp-1  ] DEBUG 2024/05/29 19:30:35 pv timer inactive
May 29 19:30:35 evcc evcc[184878]: [lp-1  ] DEBUG 2024/05/29 19:30:35 max charge current: 6A
May 29 19:30:35 evcc evcc[184878]: [site  ] DEBUG 2024/05/29 19:30:35 ----
May 29 19:30:35 evcc evcc[184878]: [lp-1  ] DEBUG 2024/05/29 19:30:35 charge power: 0W
May 29 19:30:35 evcc evcc[184878]: [site  ] DEBUG 2024/05/29 19:30:35 pv power: 10W
May 29 19:30:35 evcc evcc[184878]: [site  ] DEBUG 2024/05/29 19:30:35 battery soc: 48%
May 29 19:30:35 evcc evcc[184878]: [site  ] DEBUG 2024/05/29 19:30:35 battery power: 5578W
May 29 19:30:35 evcc evcc[184878]: [site  ] DEBUG 2024/05/29 19:30:35 grid meter: 5645W
May 29 19:30:35 evcc evcc[184878]: [site  ] DEBUG 2024/05/29 19:30:35 site power: 11223W
May 29 19:30:35 evcc evcc[184878]: [lp-1  ] DEBUG 2024/05/29 19:30:35 vehicle odometer: 5132km
May 29 19:30:35 evcc evcc[184878]: [lp-1  ] DEBUG 2024/05/29 19:30:35 charge total import: 0.000kWh
May 29 19:30:35 evcc evcc[184878]: [lp-1  ] DEBUG 2024/05/29 19:30:35 charger status: C
May 29 19:30:35 evcc evcc[184878]: [lp-1  ] DEBUG 2024/05/29 19:30:35 soc estimated: 97.00% (vehicle: 97.00%)
May 29 19:30:35 evcc evcc[184878]: [lp-1  ] DEBUG 2024/05/29 19:30:35 vehicle soc: 97%
May 29 19:30:35 evcc evcc[184878]: [lp-1  ] DEBUG 2024/05/29 19:30:35 vehicle soc limit: 100%
May 29 19:30:35 evcc evcc[184878]: [lp-1  ] DEBUG 2024/05/29 19:30:35 vehicle range: 459km
May 29 19:30:35 evcc evcc[184878]: [lp-1  ] DEBUG 2024/05/29 19:30:35 pv charge current: 0A = 6A + -16.3A (11223W @ 3p)
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.126.5

andig, 2024-05-30:
Seems charger is starting by itself:

> May 29 19:30:35 evcc evcc[184878]: [lp-1  ] WARN 2024/05/29 19:30:35 charger logic error: disabled but charging

After that the disable timer starts:

> May 29 19:30:35 evcc evcc[184878]: [lp-1  ] DEBUG 2024/05/29 19:30:35 pv disable timer reset

But according to config it will take 15m to do anything. I think we're not reducing the current within these 15min since we kinda assume that disabling the loadpoint will sooner or later fix the problem.

mztiq, 2024-05-31:
And what exactly does that mean? Like why is the charger logic not working but charging anyways?  
Any suggestions?

PR #15876 (2024-09-06): Evbox Elvi: undeprecate meter-less configuration
Alternative approach to #15882

- Skip broken measurands decoration on Elvi chargers without internal meter if configured using `meter: false` template option.

May fix #14115 #14138