evcc-io/evcc #14124: Nulleinspeißung deaktivert durch EVCC aktive Batteriesteuerung

stadlaa, 2024-05-30:
Seit ich EVCC mit dem Fronius Gen24 inkl. Speicher via Modbus TCP verbunden habe speißt der WR unerlaubterweise ins Netz ein. 

Habe den Fehler temporär behoben indem ich den Parameter Conn im control modul auf 0 gesetzt habe. Allerdings deaktivert das alle Wechselrichteraktivitäten. Ich möchte ja nur das nichts eingespeißt wird. 

1. Ich habe in der UI unter Batterieeinstellungen keine Option aktive Batteriesteuerung. (Da hab ich wohl was nicht verstanden) 
2. Welche Register werden den von Evcc generell potentiell angesteuert damit ich den Urzustand meines WR wiederherstellen kann.

mfg
Stadlaa

_Originally posted by @stadlaa in https://github.com/evcc-io/evcc/discussions/14099#discussioncomment-9596787_

andig, 2024-05-30:
@premultiply es sei denn Du sagtest, dass es hier etwas zu tun gibt?

andig, 2024-05-30:
Und bitte: lasst uns Einspeisung richtig schreiben ;)

premultiply, 2024-05-30:
Nein, ich verstehe ehrlich gesagt noch nicht mal um was es hier geht und was evcc zu tun haben sollte.

stadlaa, 2024-05-30:
Okay Sorry die späte Antwort. Zurzeit ist es nicht leicht.

Ich möchte nicht claimen das das Problem bei evcc liegt, aber nunmal sichergehen das es das nicht tut. 

Ausgangslage:
Mein WR Gen24 wurde vom Installateur in Betrieb genommen und die Settings in der WR UI via Technician Passwort so konfiguriert, dass das Gerät Strom für unser Haus produziert und auch im Speicher speichert, allerdings unter keinen Umständen ins Netz einspeist. (Die Anlage ist seitens Netzbetreiber noch nicht abgenommen) 

Ich habe am Samstag den 25.Mai um 22:00 evcc gestartet und die 3 Meter pv, battery, grid via Modbus TCP in der yaml hinzugefügt. Aktive Batteriesteuerung ist somit auch gar nicht möglich, da diese nur mit Solar API verfügbar wäre oder?

Seitdem ich diese 3 Meter über den Wechselrichter konfiguriert habe fing der Wechselricher gleich am nächsten Tag an ins Netz einzuspeisen. Die Vermutung liegt daher sehr nahe das da ein Zusammenhang besteht.

Einen Logausschnitt vom 25.Mai als ich Evcc mit neuer Konfig hochgefahren hab und den Yaml ausschnitt mit den Gen24-Metern poste ich unten, falls das hilft.

Skuril ist das ich diese oben Beschriebene Option nicht in den Modbus-Registern von Fronius finden kann. Falls ihr ausschließen könnt das Evcc dieses Problem verursacht haben könnte wäre mir schon geholfen. Dann werde ich den Installateur kontaktieren, der das hoffentlich wieder in Ordnung bringt.

LOG:
May 25 22:17:00 raspberrypi systemd[1]: Started evcc.service - evcc.
May 25 22:17:01 raspberrypi evcc[662183]: [main  ] INFO 2024/05/25 22:17:01 evcc 0.124.9
May 25 22:17:01 raspberrypi evcc[662183]: [main  ] INFO 2024/05/25 22:17:01 using config file: /etc/evcc.yaml
May 25 22:17:01 raspberrypi evcc[662183]: [main  ] INFO 2024/05/25 22:17:01 starting ui and api at :7070
May 25 22:17:06 raspberrypi evcc[662183]: [lp-1  ] WARN 2024/05/25 22:17:06 poll mode '{always 30m0s}' may deplete your battery or lead to API misuse. USE AT YOUR OWN RISK.
May 25 22:17:06 raspberrypi evcc[662183]: [lp-1  ] WARN 2024/05/25 22:17:06 deprecated: minCurrent setting is ignored, please remove
May 25 22:17:06 raspberrypi evcc[662183]: [lp-1  ] WARN 2024/05/25 22:17:06 deprecated: maxcurrent setting is ignored, please remove
May 25 22:17:06 raspberrypi evcc[662183]: [site  ] WARN 2024/05/25 22:17:06 battery configured but residualPower is missing or <= 0 (add residualPower: 100 to site), see https://docs.evcc.io/en/docs/reference/configuration/site#residualpower
May 25 22:17:06 raspberrypi evcc[662183]: [site  ] INFO 2024/05/25 22:17:06 site config:
May 25 22:17:06 raspberrypi evcc[662183]: [site  ] INFO 2024/05/25 22:17:06   meters:      grid ✓ pv ✓ battery ✓
May 25 22:17:06 raspberrypi evcc[662183]: [site  ] INFO 2024/05/25 22:17:06     grid:      power ✓ energy ✓ currents ✓
May 25 22:17:06 raspberrypi evcc[662183]: [site  ] INFO 2024/05/25 22:17:06     pv 1:      power ✓ energy ✗ currents ✗
May 25 22:17:06 raspberrypi evcc[662183]: [site  ] INFO 2024/05/25 22:17:06     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✗
May 25 22:17:06 raspberrypi evcc[662183]: [site  ] INFO 2024/05/25 22:17:06   vehicles:
May 25 22:17:06 raspberrypi evcc[662183]: [site  ] INFO 2024/05/25 22:17:06     vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
May 25 22:17:06 raspberrypi evcc[662183]: [lp-1  ] INFO 2024/05/25 22:17:06 loadpoint 1:
May 25 22:17:06 raspberrypi evcc[662183]: [lp-1  ] INFO 2024/05/25 22:17:06   mode:        pv
May 25 22:17:06 raspberrypi evcc[662183]: [lp-1  ] INFO 2024/05/25 22:17:06   charger:     power ✓ energy ✗ currents ✓ phases ✓ wakeup ✗
May 25 22:17:06 raspberrypi evcc[662183]: [lp-1  ] INFO 2024/05/25 22:17:06   meters:      charge ✓
May 25 22:17:06 raspberrypi evcc[662183]: [lp-1  ] INFO 2024/05/25 22:17:06     charge:    power ✓ energy ✗ currents ✓
May 25 22:17:06 raspberrypi evcc[662183]: [lp-1  ] DEBUG 2024/05/25 22:17:06 phase timer inactive
May 25 22:17:06 raspberrypi evcc[662183]: [lp-1  ] DEBUG 2024/05/25 22:17:06 pv timer inactive
May 25 22:17:06 raspberrypi evcc[662183]: [lp-1  ] INFO 2024/05/25 22:17:06 vehicle updated: unknown -> ZOE
May 25 22:17:06 raspberrypi evcc[662183]: [site  ] WARN 2024/05/25 22:17:06 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
May 25 22:17:06 raspberrypi evcc[662183]: [site  ] DEBUG 2024/05/25 22:17:06 ----
May 25 22:17:06 raspberrypi evcc[662183]: [lp-1  ] DEBUG 2024/05/25 22:17:06 charge power: 0W
May 25 22:17:06 raspberrypi evcc[662183]: [site  ] DEBUG 2024/05/25 22:17:06 pv power: 4W
May 25 22:17:06 raspberrypi evcc[662183]: [site  ] DEBUG 2024/05/25 22:17:06 battery soc: 77%
May 25 22:17:06 raspberrypi evcc[662183]: [site  ] DEBUG 2024/05/25 22:17:06 battery power: 3373W
May 25 22:17:06 raspberrypi evcc[662183]: [site  ] DEBUG 2024/05/25 22:17:06 grid meter: -1186W
May 25 22:17:06 raspberrypi evcc[662183]: [site  ] DEBUG 2024/05/25 22:17:06 grid currents: [-6.44 3.68 -4.24]A
May 25 22:17:06 raspberrypi evcc[662183]: [site  ] DEBUG 2024/05/25 22:17:06 site power: 2187W
May 25 22:17:06 raspberrypi evcc[662183]: [lp-1  ] DEBUG 2024/05/25 22:17:06 vehicle odometer: 0km
May 25 22:17:06 raspberrypi evcc[662183]: [lp-1  ] DEBUG 2024/05/25 22:17:06 charge currents: [0 0 0]A
May 25 22:17:06 raspberrypi evcc[662183]: [lp-1  ] DEBUG 2024/05/25 22:17:06 charger status: A
May 25 22:17:06 raspberrypi evcc[662183]: [lp-1  ] INFO 2024/05/25 22:17:06 car disconnected
[... cut ...]
meters:
  - name: gen24_grid
    type: template
    template: fronius-gen24
    usage: grid
    host: 10.0.0.230 # IP-Adresse oder Hostname
    port: 502 # Port (optional) 

  - name: gen24_pv
    type: template
    template: fronius-gen24
    usage: pv
    host: 10.0.0.230 # IP-Adresse oder Hostname
    port: 502 # Port (optional) 

  - name: gen24_batt
    type: template
    template: fronius-gen24
    usage: battery
    host: 10.0.0.230 # IP-Adresse oder Hostname
    port: 502 # Port (optional)