evcc-io/evcc #14133: Funny error message when seting min.current to less than 6 Amps

edol1970, 2024-05-30:
### Describe the bug

If selecting less than 6Amps as min.current, the following error message appears in the logs and the charger stops charging

[lp-1  ] ERROR 2024/05/30 18:59:15 max charge current 5A: invalid current 5.0

Why is this error occuring? (is not clear from the message logs)
Same error messages appears with other current settings (below 6A)

The charger is a  Sungrow AC011E-01 (which is configured behind a Sungrow inverter)
According to the documentation this wallbox supports [mA Regelung] and [1P3P].

I'm using the latest nighty build.(v0.126.5), running on docker (in a Raspberry Pi).

If selecting 6Amp as min-current all works fine.

Switching from 3P to 1P seems also an issue.(most of the time it fails and the system goes back to 3P)

Many thanks for your help.

### Steps to reproduce

1.Select Auto-switch phases
2. Select 5Amps or less as min_current
3. Check the logs for the error
...

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: evcc.local
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 30s # control cycle interval. Interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval

# database configuration for persisting charge sessions and settings
# database:
#   type: sqlite
#   dsn: <path-to-db-file>

# sponsor token enables optional features (request at https://sponsor.evcc.io)
# sponsortoken:
# trial token, valid until 2024-06-18
sponsortoken: eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJldmNjLmlvIiwic3ViIjoiVHJpYWwgU3BvbnNvcnNoaXAg8J-SqyIsImV4cCI6MTcxODcxOTIwMCwiaWF0IjoxNzE2MTI3MjAwLCJzcGUiOnRydWUsInNyYyI6Im1hIn0.NUDQh6tj14OzJjFdpKSdiLewvL7pwApQERrCMJQcv5g

# telemetry enables aggregated statistics
#
# Telemetry allows collecting usage data (grid and green energy, charge power).
# Data is aggregated, no individual charging sessions are tracked. The collected,
# anonymous data can be retrieved using https://api.evcc.io.
#
# See https://github.com/evcc-io/evcc/pull/4343 or details.
#
# For time being, this is only available to sponsors, hence data is associated with
# the sponsor token's identity.
#
#telemetry: true

# log settings
log: info
levels:
[... 760 lines cut ...]
[site  ] DEBUG 2024/05/30 19:16:45 grid meter: 1W
[site  ] DEBUG 2024/05/30 19:16:45 site power: -194W
[lp-1  ] DEBUG 2024/05/30 19:16:45 charge total import: 1867.018kWh
[lp-1  ] DEBUG 2024/05/30 19:16:45 charger status: B
[lp-1  ] DEBUG 2024/05/30 19:16:45 pv charge current: 0.843A = 0A + 0.843A (-194W @ 1p)
[site  ] DEBUG 2024/05/30 19:17:15 ----
[lp-1  ] DEBUG 2024/05/30 19:17:15 charge power: 0W
[lp-1  ] DEBUG 2024/05/30 19:17:15 charge currents: [0 0 0]A
[site  ] DEBUG 2024/05/30 19:17:15 pv power: 594W
[site  ] DEBUG 2024/05/30 19:17:15 battery soc: 80%
[site  ] DEBUG 2024/05/30 19:17:15 battery power: -190W
[site  ] DEBUG 2024/05/30 19:17:15 grid meter: -0W
[site  ] DEBUG 2024/05/30 19:17:15 site power: -190W
[lp-1  ] DEBUG 2024/05/30 19:17:15 charge total import: 1867.018kWh
[lp-1  ] DEBUG 2024/05/30 19:17:15 charger status: B
[lp-1  ] DEBUG 2024/05/30 19:17:15 pv charge current: 0.826A = 0A + 0.826A (-190W @ 1p)
```

### What type of operating system are you running?

Docker container

### Version

0.126.5

andig, 2024-05-30:
Because its invalid…

closed 2024-05-30 by andig

kvray, 2024-06-01:
You change everything beyond recognition, except for the email and password @edol1970

edol1970, 2024-06-01:
> You change everything beyond recognition, except for the email and password @edol1970

oops…thanks for pointing…any way to remove it?

andig, 2024-06-01:
@edol1970 consider your PW burned and immediately create a new one. For issues we suggest 

    evcc discuss

which will mask any passwords.

edol1970, 2024-06-01:
Many thanks!