evcc-io/evcc #14134: UI - Viele Ladepunkte - Toggle Card/List view

michiproep, 2024-05-31:
Hallo zusammen,
bei mir häufen sich schön langsam die Ladepunkte. Aktuell hab ich 5, es sind aber noch weitere geplant.
Häufig möchte ich in evcc nichts verstellen, sondern eigentlich nur kontrollieren, ob alles läuft wie erwartet.
Die Anzahl der Ladepunkte passt mit der Card-Ansicht natürlich nicht auf einen Screen. (ja, Handy ist nochmal ne andere Baustelle).

Wäre es möglich, via ToggleButton (oben) zwischen 2 views zu wechseln?
Auf kleineren Devices (Handy, Raspberry 7'') könnten ja die einzelnen Zeilen auch nochmal umbrechen und/oder Icons statt Texte anzeigen.
Wäre sowas denkbar?

Hier mal ein Mockup:
[image]

naltatis, 2024-05-31:
@michiproep gehts bei dir um Fahrzeuge und Wallboxen oder sind das größtenteils auch "andere Geräte", die du da im Überblick haben möchtest?

Eine so kompakte Darstellung, wie du sie da zeichnest, kann ich mir in der Tat nicht gut vorstellen. Vor allem nicht als full-featured Ersatz für die aktuelle Darstellung, die auch noch responsiv funktioniert. Es gibt aber immer noch ein Proposal für eine Kompaktdarstellung von "Nicht Wallboxen". Szenarien wie Schaltsteckdosen, Heizgeräte und co. bei denen man kein Fahrzeugwechsel, Limit-Soc, Planung und co. braucht. In dem Zuge werden wir ohnehin noch mal an das Design ran gehen und mehr Übersichtlichkeit schaffen. 

Ich würd diesen Feature-Wunsch, da einfach mal verlinken. (duplicate of #7235)

closed 2024-05-31 by naltatis

michiproep, 2024-05-31:
@naltatis Wie bei den meisten Leuten sind eigentlich maximal, wenn überhaupt, zwei Wallboxen im Einsatz. evcc hat sich tatsächlich als Wunderwaffe für alle möglichen Szenarien erwiesen. Wenn man so durch die Foren schaut, scheint das Thema Wallbox/Auto bei allen perfekt zu funktionieren und machen sich jetzt mehr und mehr an Heizen und Kühlen ran, die vielleicht keine Planung brauchen, aber doch mit Soc und Hysteresen hantieren.

Schade, dass du Dir "das nicht gut vorstellen kannst". Ich hatte gehofft, dass das gar nicht mal viel Aufwand wäre.
Der Button schaltet zwischen zwei "div" um, wobei der neue "div" halt nur eine Zeilenansicht der selben "Datenpunkte" - ich denk via DataBinding - darstellt.

Ich hatte schon mal in einem anderen Ticket gefragt, ob man nicht eine custom-css injecten darf, deren Pfad man im yaml angeben kann? Die UI würde dann nur einen <link rel....> mit der Datei ausgeben und ich könnte mir den Rest selbst zusammenbauen. Wäre das denkbar?

michiproep, 2024-05-31:
Hab mal kurz mit den Browser-Dev-Tools ein wenig getüftelt...
Vielleicht lasst ihr euch ja doch noch überreden ;-)
[image]

naltatis, 2024-05-31:
Wie gesagt. Ich find eine kompaktere Darstellung gut. Gerade für einfachere Geräte werden wir das auch machen.

Nen Quick-Fix mit Design-Kompromissen würde ich jetzt aber nicht machen. Man sieht bei deinem Screenshot ja schon, das das unglaublich viel Info in eine Zeile gequetscht ist. Du musst hier auch noch an längere Titel (mehr als "Garage") und andere Sprachen denken. Zudem ist das Fahrzeug auch nicht da :D. Die Idee einer separaten Ansicht/Dashboard will ich auch gar nicht ausschließen.

Custom CSS: Ich hab das leider nur selten in stabil, geschmackvoll und gut gesehen. Wir verwenden Vue mit Scoped Styles. Da kann man natürlich probieren, mit ner custom CSS gegenzustylen, aber das wird immer brüchig sein, weil wir hier keinerlei Garantien auf stabile Klassen geben können. Wenn wir die Option zur Eingabe von Custom CSS einbauen (technisch ist das natürlich ganz einfach) wird, zu recht, aber die Erwartungshaltung entstehen, dass wir hier Stabilität bieten.

Über Browser Plugins oder Reverse-Proxy-Injection kann man sich natürlich heute schon Custom Styles selbst bauen.