evcc-io/evcc #14135: Stabile LoadpointIDs

michiproep, 2024-05-31:
Häufig wird evcc zusammen mit anderen Automatisierungs- und Monitoringsystemen verwendet, z.B. iobroker-Integration/Skriptsteuerung oder auch HEMS (SMA).
Sobald man nun neue Ladepunkte hinzufügt, muss man diese zwingend - um nicht alles andere kaputt zu machen - immer am Ende einfügen.
Alle API Zugriffe und auch MQTT Kommunikation beruhen auf dem Index der Loadpoints.

Es wäre zwar denkbar, dass man den "name" verwendet, aber auch der muss nicht zwingend stabil sein.

Was spräche dagegen, im yaml pro Loadpoint eine frei wählbare ID zu hinterlegen - sei es int oder guid oder string - 
und - falls notwendig wegen Rückwärtskompatibilität - unter den globalen Einstellungen wählen muss ob mit ID oder Index gearbeitet wird?

naltatis, 2024-05-31:
Mit Einführung der Config UI werden wir für alle Ladepunkte eine eindeutige Datenbank-ID bekommen. Diese sollten wir dann auch als Identifier in allen API verwenden.

Siehe auch https://github.com/evcc-io/evcc/issues/12903, https://github.com/evcc-io/evcc/pull/12958, https://github.com/evcc-io/evcc/issues/6029

Eine Übergangsphase mit frei wählbaren Identifiern aus der yaml, würde ich aktuell nicht sehen.
//cc @andig

andig, 2024-05-31:
Ich wollte das gleiche schreiben. Wird "automatisch" kommen!

closed 2024-05-31 by andig

michiproep, 2024-05-31:
Und das funktioniert dann auch wenn man die Loadpoints im yaml hin und her schiebt? Oder verschwinden die im yaml komplett?

naltatis, 2024-05-31:
Yaml verschwindet komplett. Zumindest als Datei. Für advanced Use-Cases wirds weiterhin yaml via Web-UI geben.