evcc-io/evcc #14142: Darstellungsmenü verschiebt Inhalt auf dem Desktop wenn Scrollbar da ist

StefanSchoof, 2024-05-31:
### Describe the bug

Wenn man auf dem Desktop evcc offen hat und die vertikale Scrollbar zu sehen ist (weil soviel angezeigt wird oder weil das Fenster so klein gezogen wurde) und man das Overlay Menü Darstellung öffnet, verschiebt sich evcc im Hintergund um ein paar Pixel. Beim schließen schiebt sich dann zurück.

https://github.com/evcc-io/evcc/assets/4662023/c8e290e8-4bd8-41de-8e43-940ce7abadd9

@naltatis Ich denke das ist deine Area.

### Steps to reproduce

1. Browserfenster so anpassen das Scrollbar sichtbar ist
2. Auf Darstellungsmenü klicken
3. Hintergrund beobachten.
...

### Configuration details

```yaml
na
```

### Log details

```text
na
```

### What type of operating system are you running?

Linux

### Version

0.126.5

andig, 2024-05-31:
Vmtl. kommt die Frage nach dem Browser?

StefanSchoof, 2024-05-31:
Stimmt. Ich habe das im Chrome, Firefox und Edge also vermutlich in allen.

naltatis, 2024-05-31:
Schau dir mal die Dialoge in https://github.com/evcc-io/evcc/pull/13319 an. Dort ist einiges umgestellt und dieser Bug auch gefixt.

closed 2024-05-31 by naltatis