evcc-io/evcc #14150: Twingo Electric Fehler: Climater: json: cannot unmarshal number into Go struct field

exkanzler, 2024-06-01:
### Describe the bug

Hallo zusammen,
es treten in letzter Zeit vermehrt Fehler auf.
`Climater: json: cannot unmarshal number into Go struct field attributes.Data.Attributes.hvacStatus of type string`
Config und Log habe ich unten angehängt. 

Falls noch etwas fehlen sollte bitte kurz melden.

Danke schon mal für die Hilfe und die gute Arbeit.

### Steps to reproduce

1. Habe alle Versionen bis runter zur 0.125 getestet, mit demselben Ergebnis
2. Fehler tritt nicht auf, wenn die Ladeeinstellung auf "Aus" steht

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

interval: 10s # control cycle interval

meters:
  - name: grid_st
    type: template
    template: fronius-solarapi-v1
    usage: grid
    host: xxx
  - name: pv_st1
    type: template
    template: fronius-solarapi-v1
    usage: pv
    host: xxx

chargers:
  - name: wb_st2
    type: template
    template: go-e-v3
    host: xxx

vehicles:
  - name: ev_as1
    type: template
    template: renault
    title: TwingoZE
    user: xxx
    password: xxx
    vin: xxx
    capacity: 22.7
    phases: 3
 #   mode: pv targetSoc: 85
 #   minSoC: 20
 #   targetSoC: 85

loadpoints:
- title: Mobile-Go-e2
  charger: wb_st2
  vehicle: ev_as1
  mode: pv
  phases: 0
#  mincurrent: 8
[... 482 lines cut ...]
  "verified": "2022-05-23T11:47:08.099Z",
  "verifiedTimestamp": 1653306428099
}
[renault] TRACE 2024/06/01 10:20:09 GET https://api-wired-prod-1-euw1.wrd-aws.com/commerce/v1/accounts/09b20fa0-3d6e-4bbe-82fb-f8dc740e40d5/kamereon/kca/car-adapter/v1/cars/***/hvac-status?country=DE
[renault] TRACE 2024/06/01 10:20:09 {"data":{"id":"***","attributes":{"hvacStatus":1,"lastUpdateTime":"2024-06-01T05:21:50Z"}}}
[renault] TRACE 2024/06/01 10:20:09 GET https://api-wired-prod-1-euw1.wrd-aws.com/commerce/v1/accounts/09b20fa0-3d6e-4bbe-82fb-f8dc740e40d5/kamereon/kca/car-adapter/v1/cars/***/location?country=DE
[renault] TRACE 2024/06/01 10:20:09 {"data":{"id":"***","attributes":{"lastUpdateTime":"2024-05-31T11:15:17Z","gpsLatitude":xx.xxxxxxxxxxxxx,"gpsLongitude":yy.yyyyyyyyyyyyy}}}
Soc:           68%
Capacity:      22.7kWh
Charge status: B
Range:         xxxkm
Odometer:      xxxxxkm
Finish time:   2024-05-31 23:45:00 +0200 CEST
Climater:      json: cannot unmarshal number into Go struct field attributes.Data.Attributes.hvacStatus of type string
Position:      xx.xxxxxxxxxxxxx,yy.yyyyyyyyyyyyy
Features:      []
```

### What type of operating system are you running?

Windows

### Version

0.125.6

andig, 2024-06-01:
API-Änderung. Wissen wir, welcher hvacStatus für "aktiv" steht?

VolkerK62, 2024-06-01:
see https://github.com/hacf-fr/renault-api/issues/1202

andig, 2024-06-01:
I don‘t see an answer to the question 😮

VolkerK62, 2024-06-01:
oh, sorry. Sollte als ergänzende Info gedacht sein.

andig, 2024-06-01:
Closed in 9eb918b3ac0798f821273b2ee639e3dfc5274572

closed 2024-06-01 by andig

eltoro0815, 2024-06-02:
Hi,

from my Dacia Spring I can tell you:
```
"hvacStatus": 1    =    off
"hvacStatus": 2    =    on
```

andig, 2024-06-02:
Danke, im Master behoben!

exkanzler, 2024-06-04:
Hi zusammen,
mit der 0.126.6 ist der Fehler behoben.
Danke an alle.
```
C:\Users\xxx\Downloads\evcc>evcc vehicle
[main  ] INFO 2024/06/04 08:46:44 evcc 0.126.6
[main  ] INFO 2024/06/04 08:46:44 using config file: C:\Users\xxx\Downloads\evcc\evcc.yaml
[db    ] INFO 2024/06/04 08:46:44 using sqlite database: C:\Users\xxx\.evcc\evcc.db
Soc:            51%
Capacity:       22.7kWh
Charge status:  A
Range:          90km
Odometer:       19013km
Finish time:    2024-06-04 08:40:00 +0200 CEST
Climate active: false
Position:       xx.xxxxx,y.yyyyy
Features:       []
C:\Users\xxx\Downloads\evcc>
```

foto-andreas, 2024-06-13:
Hm,

taucht jetzt in 0.127.0 wieder auf.

```
Zuhause Wallbox: climater: json: cannot unmarshal string into Go struct field attributes.Data.Attributes.hvacStatus of type int
```

Wurde da was reverted - ggf. aus Versehen? Oder neues Problem?

Andreas

andig, 2024-06-13:
Wenn man jetzt das json hätte…

foto-andreas, 2024-06-14:
kommt hier...

```
[renault] TRACE 2024/06/14 03:29:31 {"data":{"id":"***","attributes":{"hvacStatus":"off","lastUpdateTime":"2024-06-13T18:39:13Z"}}}
```

Gerade nur das iPhone zur Hand. Kann im Verlauf des Vormittags noch ganzes Log von Pass/Token befreit schicken.

Andreas

foto-andreas, 2024-06-14:
übrigens Zoe hier. Da hat dann ja wohl eher Renault was zurückgedreht.

exkanzler, 2024-06-14:
Hi zusammen, 
den Fehler kann ich für den Twingo Electric mit 0.126.6 ebenfalls bestätigen.

Ggfls. wieder zurück wie vorher oder beide Datentypen nehmen ?!

https://github.com/hacf-fr/renault-api/issues/1202#issuecomment-2165531462
> Today, the endpoint for DACIA SPRING return strings again. I get "hvacStatus": "off". To mitigate this, I changed in the code the type of hvacStatus from hvacStatus: Optional[int] to hvacStatus: Optional[int] | Optional[str]

premultiply, 2024-06-14:
Das ist momentan offensichtlich ein wildes Gebastel bei Renault. Selbst deren App kommt bei den ständigen Änderungen nicht mit.
Hin- und her.

foto-andreas, 2024-06-17:
Aktuell wieder Fehler mit 0.127.2 und 
```
[renault] TRACE 2024/06/17 13:34:24 {"data":{"id":"***","attributes":{"hvacStatus":"off","lastUpdateTime":"2024-06-17T08:51:38Z"}}}
```

andig, 2024-06-17:
Ah, sorry. Seems I did half a revert only. Finally fixed in e353083a3.