evcc-io/evcc #14157: Batteriekonfiguration einstellbar für Loadpoints

Captainbrille, 2024-06-01:
Hi,

Ich nutze EVCC um mein Fahrzeug zu laden und ebenfalls um meine Heizstäbe im WW und Heizungsboiler über einen Shelly zu schalten. 
Vielen Dank erstmal für dieses tolle und gelungene Programm. 
Nun zu meinem feature wunsch:

Ich Würde gerne die gelungene Batteriesteuerung nutzen um mein Fahrzeug aus der Hausbatterie zu laden, jedoch gilt diese Einstellung für alle Loadpoints wodurch es regelmäßig dazu kommt, dass der Strom aus der Hausbatterie für den ineffizienten Heizstab genutzt wird und wenig fürs Auto übrig bleibt.

Ich wünsche mir, dass man wählen kann für welche Loadpoints die Hausbatterie genutzt werden soll. Aus meiner Sicht kann das über die yaml laufen oder man könnte das Fenster in dem man die Stromstärke einstellen kann erweitern mit dieser Option.

andig, 2024-06-01:
Eine sehr einfache (wenn auch nicht super schöne) Lösung könnte sein, dem Ladepunkt ein „keine Batterie“ Feature zu verpassen.

premultiply, 2024-06-01:
Ev. Thema für die Mini-Loadpoints?

naltatis, 2024-06-02:
Ich wäre dafür, die Funktion "Batteriesperre bei Schnelladen" generell an den Loadpoint zu verschieben. Ähnlich wie wir es beim Preislimit ja gemacht haben. Das ist nachvollziehbarer und einfacher verstehbar als eine globale Configuration + Opt-Out Settings pro Loadpunkt.

Wenn man viele Ladepunkte mit unterschiedlichen Einstellwünschen hat, kann das Umstellen etwas mühselig sein. Aber das ist sicher nicht der Standardfall.

Bonus: Wenn wir dann mal das Battery-Boost-Feature bauen (PV-Laden mit max. Batterie Umterstützung), wäre das auch gleich an der richtigen Stelle.

Captainbrille, 2024-06-02:
Ich möchte aber ja den Heizstab ausschließlich mit PV Überschuss betreiben, da der im Vergleich zur Wärmepumpe natürlich deutlich ineffizienter ist, aber da meine WP nicht SG Ready ist, kann ich sie nicht so einfach regeln.

andig, 2024-06-02:
@naltatis in jedem Falle würde *ein* Ladepunkt mit aktiver Batteriesperre dazu führen, dass diese auslösen würde- was ja auch richtig ist. Die Umsetzung wäre sehr einfach.

> Ich möchte aber ja 

@Captainbrille das Argument verstehe ich ich. Für oder gegen was spricht das jetzt?

Captainbrille, 2024-06-02:
Naja, ich hatte das wo verstanden, dass die Batteriesperre dann nur bei schnellladen aktiv ist, die Loadpoints stehen aber immer auf PV.

premultiply, 2024-06-02:
Bei PV wird doch ohnehin nur Überschuss genutzt. Da spielt nie die Batterie mit rein.

Captainbrille, 2024-06-02:
Hmm... vielleicht habe ich mich missverständlich ausgedrückt. 

Ich meine die Einstellung der Nutzung der Hausbatterie. Wenn ich die einstelle wie auf dem Screenshot zu sehen, dann wird der Heizstab über den Shelly angesteuert, wenn genug PV Leistung da ist, wenn aber nicht mehr genug da ist wird meine Hausbatterie für den Heizstab genutzt, was sehr ineffizient ist und dieser sehr zügig die Batterie leer zieht. Somit ist diese Einstellung für mich gerade unbrauchbar. Wenn ich bestimmen könnte welcher loadpoint die Batterie nutzen darf, kann ich die Hausbatterie für das Laddn meines BEV nutzen.

[image]

naltatis, 2024-06-03:
Ok, das heißt, dir gehts hier um "Batterieunterstütztes Laden" also `bufferSoc`.

naltatis, 2024-06-04:
Ich fasse mal kurz zusammen. Wir haben hier mMn. zwei unterschiedliche Themen:

1. Ein Ladepunkt soll rein mit PV-Überschuss arbeiten, nicht den Buffer-SoC Bereich nutzen. Bsp.: Heizstab.
2. Schnell-Laden eines Kleingerätes (bspw. E-Bike) soll nicht zur Batteriesperre führen, das Schnell-Laden des Autos aber doch.

Um 1. zu lösen würde ich auch, wie am Anfang von @andig vorgeschlagen, ein Opt-out Setting für den Buffer SoC Bereich sehen. Also eine Option `[ ] kein batterieunterstütztes Laden` am Ladepunkt. Das Einstellen anderer Schwellwerte pro Ladepunkt würde hier mMn. erstmal etwas zu weit gehen.

Punkt 2. ist mir immer mal wieder in der Benuzung aufgefallen, sollten wir auch was dran machen, siehe Ideen von oben, würde ich aber hier im Issue erstmal ausklammern, um den Fokus zu behalten.

Passt das für alle?

andig, 2024-06-04:
Ich frage mich, ob das aus Sicht des Ladepunktes nicht das identische Setting ist. Der `bufferSoc` ist eine Eigenschaft der Batterie, also global.

1) sagt "lade nicht aus dem Akku", also "nutze die Batterie nicht". 
2) sagt "sperre die Batterie nicht", also "mach nix mit der Batterie".

Sind nicht 1+2 identisch im Sinne von "ignoriere Batterie" und damit ein Setting am Loadpoint?

naltatis, 2024-06-05:
@andig ne, das sind zumindest bei mir unterschiedliche Anwendungsfälle. Ich geb mal ein Beispiel:

- Mein Heizstab kann gerne (mit niedriger Prio) den am Ende des Tages verbliebenen Buffer-Bereich aus dem Speicher nutzen. Ich möchte aber nicht, dass dieser "kleine" Verbraucher die Hausbatterie für alle Nutzer sperrt, wenn er mal im Schnell-Modus (manuell, min, plan) ist.

Bei @mfuchs1984 ist der Use-Case scheinbar so, dass sein Luftentfeuchter nur direkt mit PV-Energie betrieben werden soll. Vermutlich ist aber auch da der Wunsch (Unterstellung), dass manuelles Schnell-Schalten nicht die Speicherentladung global blockieren soll.

andig, 2024-06-05:
Das macht die Config dann aber sehr feingranular. Richtig gut find ich das nicht 😫

Wisley7, 2024-06-08:
@Captainbrille 
Um jetzt eine ganz einfache Lösung zu haben, bis eine adäquate Programmierung stattgefunden hat, empfehle ich dir einen 2. evcc zu nutzen. 
hier kannst du dann unterschiedliche Batterienutzung einstellen.

Captainbrille, 2024-06-15:
Danke für den Tipp 👍

andig, 2024-06-22:
Seems this is not something we'd currently like to do- closing.

closed 2024-06-22 by andig

CDank, 2025-10-01:
Hallo zusammen,

gibt es aktuell Überlegungen, Punkt 1 aus den oben genannten Vorschlägen umzusetzen?

Ich hatte ebenfalls das Problem, dass ich bufferSOC nicht nutzen konnte, da sonst der Heizstab bzw. die Zusatzelektroheizung unnötig Strom aus dem Speicher verbraten würden.
Grundsätzlich finde ich die Option sehr sinnvoll, um kurzfristige Erzeugungsschwankungen beim Laden des Autos abzufangen.

Ich habe das Thema vorerst extern über Home Assistant gelöst, da ich dort sowieso schon die Stufenschaltung des Heizstabs mit den drei Phasen abhängig vom SoC des Speichers steuere. Für einfachere Ein-/Aus-Heizgeräte wäre es aber meiner Meinung nach eine nützliche Funktion, wenn man das direkt in EVCC abbilden könnte. Dadurch bliebe die Logik übersichtlicher, weil dann alles an einem Ort gesteuert würde.

JoernHei, 2025-12-25:
Mir ist noch etwas wichtiges Aufgefallen: Es darf nur die Hausakkuladung nicht berücksichtigt werden. Die Entladung hingegen muss in die Berechnung nach wie vor mit einfließen.

naltatis, 2025-12-26:
> @naltatis siehst du hier noch etwas zu tun?

Ja, ist weiterhin ein Thema. Wir können grundsätzlich mal überlegen ob wir BufferSoC und Endladesperre von globalen Einstellungen (Site) zu LP-Einstellungen migrieren. Aus Komplexitätsgründen würde ich ungern eine Vererbung (Global + Überschreibungsmöglichkeit vom LP) machen.

Ist ein gutes Thema was wir bei der Restrukturierung für kleinere Verbraucher (Miniladepunkte) noch mal mit betrachten sollten.