evcc-io/evcc #14161: Solarwatt Energymanager - keine Meter Daten verfügbar

hegauwolke, 2024-06-02:
### Describe the bug

Seit kurzem kann EVCC nicht mehr die Meter Daten vom Solarwatt Energymanager beziehen. Es kommen folgende Fehlermeldungen.

Client.Timeout exceeded while awaiting headers

### Steps to reproduce

it is just not working. restarted the server many times.

### Configuration details

```yaml
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

interval: 30s # control cycle interval

sponsortoken: 
xxxxx

meters:
- type: template
  template: solarwatt
  usage: grid
  host: 192.168.1.40
  name: grid1
- type: template
  template: solarwatt
  usage: pv
  host: 192.168.1.40
  name: pv2
- name: my_battery1
  type: template
  template: solarwatt
  usage: battery
  host: 192.168.1.58 # IP-Adresse oder Hostname

chargers:
- type: template
  template: webasto-next
  host: 192.168.1.61
  port: 502
  name: NEXT

loadpoints:
- title: CarPort
  charger: NEXT
  mode: pv
  phases: 0
  mincurrent: 6
  maxcurrent: 16

site:
  title: Libellenweg
  meters:
    grid: grid1
    pv:
      - pv2

mqtt:
  broker: 192.168.1.20
  topic: evcc

influx:
  url: http://192.168.1.20:18086
  database: evcc # InfluxDB v2.x uses term `bucket` but for compatibility still named `database` here
  token: __T96wzU5_XxFCYlk2Rv00XCM1IpUaOk7lhj_6Yal49jaGNxQBfgSnK0iXUlJPAZjRZRN1H-JkX_VmswowYVyQ==
  org: Hegausonne
```

### Log details

```text
[site ] ERROR 2024/06/02 15:02:53 pv 1 power: Get "http://192.168.1.40/rest/kiwigrid/wizard/devices": net/http: request canceled (Client.Timeout exceeded while awaiting headers)
[site ] DEBUG 2024/06/02 15:02:53 pv power: 0W
[site ] ERROR 2024/06/02 15:03:03 grid meter: Get "http://192.168.1.40/rest/kiwigrid/wizard/devices": context deadline exceeded (Client.Timeout exceeded while awaiting headers)
[site ] DEBUG 2024/06/02 15:03:13 ----
[lp-1 ] DEBUG 2024/06/02 15:03:13 charge power: 0W
[lp-1 ] DEBUG 2024/06/02 15:03:13 charge currents: [0 0 0]A
[site ] ERROR 2024/06/02 15:03:23 pv 1 power: Get "http://192.168.1.40/rest/kiwigrid/wizard/devices": context deadline exceeded (Client.Timeout exceeded while awaiting headers)
[site ] DEBUG 2024/06/02 15:03:23 pv power: 0W
[site ] ERROR 2024/06/02 15:03:33 grid meter: Get "http://192.168.1.40/rest/kiwigrid/wizard/devices": context deadline exceeded (Client.Timeout exceeded while awaiting headers)
```

### What type of operating system are you running?

Linux

### Version

0.126.2

VolkerK62, 2024-06-02:
Was wurde geändert?

Unabhängig davon hat deine yaml ein paar Fehler.
Du hast ein battery meter, welches aber in der Site nicht definiert ist.
In deiner MQTT Konfig fehlt die Port Angabe.
Am Loadpoint können `phases, mincurrent, maxcurrent` gelöscht werden. Wird im Web UI eingestellt.

hegauwolke, 2024-06-02:
Es wurde das update auf 0.126.2 gemacht. dann lief nicht mehr

hegauwolke, 2024-06-02:
aktuelles Fehlerbild
[image]

andig, 2024-06-02:
Das Gerät antwortet nicht. An evcc hat sich nichts geändert.

> Es wurde das update auf 0.126.2 gemacht. dann lief nicht mehr

Dann bitte Downgrade auf alte Version. Falls es dort geht bitte wieder melden. Falls nein Geräteproblem.

closed 2024-06-02 by andig

hegauwolke, 2024-06-02:
Downgrade hat nichts gebracht. Daraufhin System neu installiert, EVCC neu installiert und jetzt funktioniert es wieder

Danke für die schnelle Reaktion

andig, 2024-06-02:
> EVCC neu installiert 

Damit kann es ja nichts zu tun haben!