evcc-io/evcc #14165: Go-e charger API v2 detection not working reliably, getting stuck at v1 all the time

andi0b, 2024-06-02:
### Describe the bug

- The go-e API detection seems to be only done once during evcc startup.
- If the go-e is **not reachable** during this time it falls back to v1
- If v1 is disabled, evcc can't communicate with the go-e charger until evcc is restarted

Source reference `upgradeV2()`:
https://github.com/evcc-io/evcc/blob/9ea73de7f640cf2a31e7565e19037e6729d97aaa/charger/go-e/api.go#L56-L73

This is really not great, I'm going to switch now to v1 or OCPP, because v2 is just not reliable enough for me (the go-e is a mobile charger only with WIFI, so it is not possible to provide a 100% top notch reliable connection to evcc).

Suggestions how to fix it:
- **Quick'n'dirty:** Just default to v2 API instead of v1, if go-e is unreachable
- **Configurable:** Add a parameter to the config for v1/v2/auto (current behavior)
- **Overengineered suggestion:**
  - check both API versions in parallel
  - repeat the check until there is a response from both endpoints
  - check for 404 instead of any error, don't assume the API is disabled on network errors
  - only if both API endpoints (v1 & v2) return a response, decide between API versions
  - re-check on any 404 error

### Steps to reproduce

1. set up go-e charger with v2 API connection to evcc
2. turn off go-e charger
3. start evcc
4. turn on go-e charger
5. evcc will flood the logs with 404 errors until restarted

### Configuration details

```yaml
chargers:
  - name: go_e
    type: template
    template: go-e-v3
    host: 192.168.178.33
```

### Log details

```text
Logs look more or less like that, evcc startup while go-e is not reachable 

[main  ] INFO 2024/06/02 16:23:44 evcc 0.126.5
...
seems to try the v2 upgrade, but fails:
...
[go-e  ] TRACE 2024/06/02 16:23:54 GET http://192.168.178.33/api/status?filter=alw
[go-e  ] TRACE 2024/06/02 16:23:34 GET http://192.168.178.33/status
[lp-1  ] ERROR 2024/06/02 16:23:37 charge power: Get "http://192.168.178.33/status": dial tcp 192.168.178.33:80: connect: no route to host
```

after network connection to go-e is restored:
```
[go-e  ] TRACE 2024/06/02 16:24:38 GET http://192.168.178.33/status
[go-e  ] TRACE 2024/06/02 16:24:38 Nothing matches the given URI
[lp-1  ] ERROR 2024/06/02 16:24:39 charge power: unexpected status: 404 (Not Found)
```

after an evcc restart with go-e reachable:
```
[main  ] INFO 2024/06/02 21:48:02 evcc 0.126.5
...
here goes the v2 upgrade, and it succeeds:
...
[go-e  ] TRACE 2024/06/02 21:48:12 GET http://192.168.178.33/api/status?filter=alw
[go-e  ] TRACE 2024/06/02 21:48:12 {"alw":false}
...
now v2 is used for periodic checks/commands:
...
[go-e  ] TRACE 2024/06/02 21:48:45 GET http://192.168.178.33/api/status?filter=alw,car,eto,nrg,wh,trx,cards
[go-e  ] TRACE 2024/06/02 21:48:45 {"alw":false,"eto":1477060,"car":3, [... all good]
```
	

logs filtered to the relevant parts, I can provide a few million lines of logs if required.
```

### What type of operating system are you running?

Docker container

### Version

0.126.5

andig, 2024-06-03:
> If the go-e is not reachable during this time it falls back to v1

Sounds like a bug with Go-E. Why should this be the case? Did you open a ticket? We don't intend to provide a workaround for this- it is easily fixable upstream.

andi0b, 2024-06-03:
> Sounds like a bug with Go-E. Why should this be the case? Did you open a ticket? We don't intend to provide a workaround for this- it is easily fixable upstream.

@andig: Seriously? Git blame proves that you were the person that introduced this bug 2 years ago with https://github.com/evcc-io/evcc/commit/a90c69e8b38016df25630246d1a969cb69b9dfa2

@naltatis Can you maybe help out here?

andi0b, 2024-06-03:
btw: go-e is the only reason I "need" sponsorship, if the evcc team is smug about supporting go-e, there is no reason for me to keep sponsoring.

andi0b, 2024-06-03:
@andig Ich versuche das Problem nochmal auf Deutsch zu beschreiben, vielleicht ist es ja ein Sprachproblem (auch wenn OSS doch besser englisch sein sollte). 

Wobei hier nicht unerwähnt bleiben soll, dass die go-e Integration als "sponsor-only" markiert ist, und auch ich dafür bezahle.

## Scope dieses Bug reports

Dieses Ticket betrifft die Integration des go-e chargers von evcc. Dieser code befindet sich hier in diesem Repository unter:
- `charger/go-e.go`
- `charger/go-e/*`

**Dieses Ticket betrifft NICHT das Verhalten des go-e chargers**

## Konkret betrifft dieser bug report folgenden Code:

https://github.com/evcc-io/evcc/blob/9ea73de7f640cf2a31e7565e19037e6729d97aaa/charger/go-e/api.go#L62-L73

Dieser Code macht folgendes:
1. `upgradeV2()` versucht festzustellen ob die API v2 vom go-e charger aktiv ist
2. Wenn v2 aktiv ist, wird ab diesem Punkt v2 benutzt, ansonsten wird v1 benutzt
3. `upgradeV2()` wird einmalig beim startup von evcc aufgerufen und das Ergebnis davon bis zum nächsten Neustart benutzt

Hier wird ein v2 API request zum Test abgesendet
https://github.com/evcc-io/evcc/blob/9ea73de7f640cf2a31e7565e19037e6729d97aaa/charger/go-e/api.go#L66

Und in Folge geprüft ob dabei ein Fehler aufgetreten ist:
https://github.com/evcc-io/evcc/blob/9ea73de7f640cf2a31e7565e19037e6729d97aaa/charger/go-e/api.go#L68

## Was ist das Problem damit?

In Zeile 86 wird bei jedem Fehler einfach auf API v1 gewechselt. Diese Fehlerfälle können unter anderem sein:
- HTTP 404: v2 API Endpoint nicht eingeschaltet, also ist ein fallback auf v1 ist **sinnvoll**
- Timeout/no route to host: Go-e charger ist nicht erreichbar, ein automatischer fallback auf v1 ist **nicht sinnvoll!**

API v1 hat keinen automatischen Phasenwechsel und ist wohl bei den meisten Usern nicht eingeschaltet.

Vom go-e Charger gibt es mobile Versionen, die nicht immer angesteckt sind. Wird nun evcc gestartet während der go-e Charger nicht angeschlossen und erreichbar ist, benutzt evcc die API v1 mit allen Problemen.

## Lösungsvorschläge (automatich übersetzt aus dem ursprünglichen ticket):

- Quick'n'dirty: Einfach standardmäßig auf die API v2 wechseln, wenn go-e nicht erreichbar ist
- Konfigurierbar: Einen Parameter zur Konfiguration des go-e templates hinzufügen für die API-Version: `v1`, `v2` oder `auto` (aktuelles Verhalten)
- Überkomplizierter Vorschlag:
    - beide API-Versionen parallel überprüfen
    - die Überprüfung wiederholen, bis eine Antwort von beiden Endpunkten vorliegt
    - auf 404 prüfen statt auf jeden Fehler, nicht davon ausgehen, dass die API bei Netzwerkfehlern deaktiviert ist
    - nur wenn beide API-Endpunkte (v1 & v2) eine Antwort liefern, zwischen den API-Versionen entscheiden
    - bei jedem 404-Fehler erneut überprüfen

andig, 2024-06-03:
evcc erwartet einen funktionierenden Charger. Wenn der unter dem gewünschten API nicht erreichbar ist muss das Problem gelöst werden. Das Problem hier ist entweder die Go-E oder Deine Netzwerkverbindung, aber nicht evcc. Die Alternative wäre, in dem Fall _gar nicht_ zu starten mangels erreichbarem Charger.

> Wobei hier nicht unerwähnt bleiben soll, dass die go-e Integration als "sponsor-only" markiert ist, und auch ich dafür bezahle.

Auf dieses Niveau der Diskussion möchte ich mich nicht begeben.

andi0b, 2024-06-03:
Alles klar, von mir kommen keine Bug reports mehr. Ich habe den Fehler 2 Stunden lang isoliert und dokumentiert, wenn das von Euch einfach ignoriert wird, dann ist das eben zu akzeptieren.

closed 2024-06-03 by andi0b

andi0b, 2024-06-03:
Vielleicht noch zur Erklärung: Ich contribute nicht kostenlos zu kommerziellen Projekten. evcc ist zwar MIT licensed, aber da der Build nun mal ein Sponsor-Token erfordert, ist es in meinen Augen ein kommerzielles Projekt.

Falls hier jemand einen Fork von evcc machen möchte, bitte gerne Kontakt mit mir aufnehmen.