evcc-io/evcc #14167: kleine Schönheitsfehler im UI

CarG70, 2024-06-03:
### Describe the bug

in der Landscape-Ansicht auf dem iPad werden die Loadpoints nebeneinander dargestellt. Dadurch wird aufgrund zu wenig Platz der Name nach wenigen Buchstaben mit … abgeschnitten. Auf dem iPhone wir in der obersten Zeile die Name des Loadpoints links und der Punkt für die Einstellungen rechts angezeigt. Darunter sind dann die Lademodi angeordnet. So würde es auf dem iPpad auch passen, da genug Platz vorhanden ist. Außerdem nutzt der Zustandsbalken oben nicht die mögliche Bildschirmbreite aus. 
Ich habe die zwei Punkte in der Bildschirmcopy markiert. Die Punkte sind aber jammern auf ganz hohem Niveau und gehören sicherlich ganz weit nach hinten.
[image]
 

### Steps to reproduce

1. öffne evcc auf dem Tablet

### Configuration details

```yaml
nicht erforderlich
```

### Log details

```text
nicht erforderlich
```

### What type of operating system are you running?

Linux

### Version

0.126.5

naltatis, 2024-06-04:
Danke für dein Feedback.

> Außerdem nutzt der Zustandsbalken oben nicht die mögliche Bildschirmbreite aus.

Das war in der Tat eine bewusste Design-Entscheidung. Auf größeren Screens ist der Balken mit dem Content der Ladepunkt Kacheln aligned. Sieht man ja auch ganz gut, wenn man deine inneren Linien weiter nach unten ziehen würde. Im ausgeklappten Zustand mit den Werten macht das noch mehr Sinn. Gerade auf größeren Screens bringt ein breiter Balken auch weniger neue Information.

> Dadurch wird aufgrund von zu wenig Platz der Name nach wenigen Buchstaben mit … abgeschnitten.

Ja, das ist ein guter Punkt. Durch die generelle Einführung des Settings-Buttons am Ladepunkt hat sich das Thema auf dieser Auflösung noch mal verschärft. Mittelfristig würde ich gerne die Modus-Auswahl verschlanken. Aber das ist ein etwas größeres Thema für später. Kurzfristig ist die von dir vorgeschlagene Lösung (Zeilenumbruch) sicher eine gute Idee. Ich schau mir das mal an.

CarG70, 2024-06-04:
auf dem iPhone ist die Kachel besser aufgeteilt. Kleinerer Bildschirm und nichts wird abgeschnitten. 
[image]

Jakob-Teske, 2024-06-04:
Hallo zusammen, wir haben uns das Issue im Rahmen des Hackergarten Stuttgart (https://www.hackergarten.net/) angeschaut und einen Pull Request #14185  mit der angesprochenen kurzfristigen Lösung erstellt.

Jakob-Teske, 2024-06-04:
[image]

closed 2024-07-12 by github-actions[bot]