evcc-io/evcc #14175: Http Request URL Encoding fails after update to 0.126.6

funnyfrish, 2024-06-04:
### Describe the bug

0.126.5 is working fine with the same configuration. But in the latest release the
URL Encoding does not work.
Configured is http://192.168.0.107/cm?cmnd=Status%208

But EVCC does not add the Space. It requests only http://192.168.0.107/cm?cmnd=Status 
without Space and 8 at the end

The Answer should look like this when requesting  http://192.168.0.107/cm?cmnd=Status%208
{
   "StatusSNS":{
      "Time":"2024-06-04T09:12:14",
      "Total":{
         "consumed":12791.299,
         "delivered":119179.919,
         "power":1
      },
      "PV":{
         "Consumed":4.493,
         "Delivered":145193.808,
         "Power":5187,
         "Eigenverbrauch":26013.886,
         "Aktueller Verbrauch":5262
      }
   }
}

But EVCC calls this URL http://192.168.0.107/cm?cmnd=Status
{
   "Status":{
      "Module":1,
      "DeviceName":"SML Meters",
      "FriendlyName":[
         "Tasmota"
      ],
      "Topic":"tasmota_DFFA50",
      "ButtonTopic":"0",
      "Power":0,
      "PowerOnState":3,
      "LedState":1,
      "LedMask":"FFFF",
      "SaveData":1,
      "SaveState":1,
      "SwitchTopic":"0",
      "ButtonRetain":0,
      "SwitchRetain":0,
      "SensorRetain":0,
      "PowerRetain":0,
      "InfoRetain":0,
      "StateRetain":0,
      "StatusRetain":0
   }
}

### Steps to reproduce

1.Just configure a new source in meters sections with type custom and source http
2. Configure a new Tasmota URL with "Status 8" in the request
3. check the trace
...

### Configuration details

```yaml
meters:
  - name: grid_tasmota
    type: custom
    power:
        source: http
        uri: http://192.168.0.107/cm?cmnd=Status%208
        jq: .StatusSNS.Total.power
        scale: 1
        timeout: 10s
  - name: pv_tasmota
    type: custom
    power:
        source: http
        uri: http://192.168.0.107/cm?cmnd=Status%208
        jq: .StatusSNS.PV.Power
        scale: 1 
        timeout: 10s
  - name: battery
    type: template
    template: victron-energy
    usage: battery
    host: 192.168.1.205
    port: 502
```

### Log details

```text
grid_tasmota
------------
[http  ] TRACE 2024/06/04 09:43:57 GET http://192.168.0.107/cm?cmnd=Status 8
[http  ] TRACE 2024/06/04 09:43:57 {"Status":{"Module":1,"DeviceName":"SML Meters","FriendlyName":["Tasmota"],"Topic":"tasmota_DFFA50","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
Power: strconv.ParseFloat: parsing "<nil>": invalid syntax
```

### What type of operating system are you running?

Linux

### Version

0.126.6

GrimmiMeloni, 2024-06-04:
Maybe a side effect of https://github.com/evcc-io/evcc/pull/14146

cc @andig

andig, 2024-06-04:
> http://192.168.0.107/cm?cmnd=Status%208

Ist erstmal falsch. M.E. müsste das `http://192.168.0.107/cm?cmnd=Status+8` heissen. Könntest Du das nochmal kurz probieren?

andig, 2024-06-04:
Fixed per https://github.com/evcc-io/evcc/pull/14146#issuecomment-2147652236

closed 2024-06-04 by andig

Neisi, 2024-06-04:
> > http://192.168.0.107/cm?cmnd=Status%208
> 
> Ist erstmal falsch. M.E. müsste das `http://192.168.0.107/cm?cmnd=Status+8` heissen. Könntest Du das nochmal kurz probieren?

Hatte ich gestern schon getestet, Ergebnis ist das gleiche.