evcc-io/evcc #14180: EVCC Keeps freezing and crashing

djfanatix, 2024-06-04:
### Describe the bug

EVCC UI Freezes and crashes

### Steps to reproduce

I cannot find anything relevant in the logs
It seems it starts fine and then it freezes.
The UI is also frozen, it does not change.
It initially fetches the values from modbus and homewizard, but then nothing (freeze) and a reboot after a while

I tried to remove circuit, no effect.

But I added a log anyhow.
EVCC 0.126.6
HA Addon

### Configuration details

```yaml
#uri: 192.168.1.30:7070 # uri for ui
interval: 5s # control cycle interval

# sponsor token enables optional features (request at https://cloud.evcc.io)
sponsortoken: xx
plant: xx
# log settings
log: trace
levels:
  core: trace
  tariffs: info
  lp-2: trace
  lp-1: trace
  mqtt: info
  circuit: trace
  tesla: trace
  custom: trace
  tasmota: trace
  template: trace
  go-e: trace
  

chargers:
- name: tasmotagarage
  integrateddevice: true
  icon: bike
  type: template
  template: tasmota
  channel: 1
  host: 192.168.1.96 # tasmota device ip address (local)
  standbypower: 10 # treat as charging above this 

- name: goe
  type: template
  template: go-e-v3
  host: 192.168.1.84 # IP-Adresse oder Hostname 

vehicles:
- name: my_car
  type: template
  template: tesla-command
  title: Tesla Model 3
  accessToken: xx
  refreshToken: xx
  vin: xx
  capacity: xx # Battery capacity in kWh (optional) 
  
- name: vanmoof
  type: custom
  title: Vanmoof Electrified S # display name for UI
[... 246 lines cut ...]
[sunspec] TRACE 2024/06/04 14:07:07 model 101 block 0 point DCW: 2289.3
[modbus] TRACE 2024/06/04 14:07:07 modbus: send 00 0e 00 00 00 06 01 03 f5 74 00 02
[modbus] TRACE 2024/06/04 14:07:07 modbus: recv 00 0e 00 00 00 07 01 03 04 80 00 c3 d2
[site  ] DEBUG 2024/06/04 14:07:07 pv power: 1868W
[modbus] TRACE 2024/06/04 14:07:07 modbus: send 00 0f 00 00 00 06 01 03 e1 74 00 02
[modbus] TRACE 2024/06/04 14:07:07 modbus: recv 00 0f 00 00 00 07 01 03 04 80 00 c3 d2
[modbus] TRACE 2024/06/04 14:07:07 modbus: send 00 10 00 00 00 06 01 03 e1 84 00 02
[modbus] TRACE 2024/06/04 14:07:07 modbus: recv 00 10 00 00 00 07 01 03 04 fe 83 42 c3
[site  ] DEBUG 2024/06/04 14:07:07 battery soc: 98%
[site  ] DEBUG 2024/06/04 14:07:07 battery power: 421W
[homewizard] TRACE 2024/06/04 14:07:07 GET http://192.168.1.109/api/v1/data
[homewizard] TRACE 2024/06/04 14:07:07 {"wifi_ssid":"Netwerk van L&P","wifi_strength":62,"smr_version":50,"meter_model":"Fluvius 253769484_A","unique_id":"3153414733313030363434343339","active_tariff":1,"total_power_import_kwh":2163.162,"total_power_import_t1_kwh":784.099,"total_power_import_t2_kwh":1379.063,"total_power_export_kwh":345.822,"total_power_export_t1_kwh":257.599,"total_power_export_t2_kwh":88.223,"active_power_w":-162,"active_power_l1_w":-376,"active_power_l2_w":22,"active_power_l3_w":191,"active_voltage_l1_v":239.1,"active_voltage_l2_v":238.7,"active_voltage_l3_v":237.2,"active_current_l1_a":1.97,"active_current_l2_a":0.33,"active_current_l3_a":1.87,"active_power_average_w":11,"montly_power_peak_w":5234,"montly_power_peak_timestamp":240602113000,"total_gas_m3":0,"gas_timestamp":240604140507,"gas_unique_id":"37464C4F32313233303736323530","external":[{"unique_id":"37464C4F32313233303736323530","type":"gas_meter","timestamp":240604140507,"value":0,"unit":"m3"}]}
[site  ] DEBUG 2024/06/04 14:07:07 grid meter: -162W
[site  ] DEBUG 2024/06/04 14:07:07 site power: 509W
[mqtt  ] TRACE 2024/06/04 14:08:00 recv vanmoof/status: 'A'
[modbus] TRACE 2024/06/04 14:08:07 modbus: closing connection due to idle timeout: 1m0.00095116s
```

### What type of operating system are you running?

Linux

### Version

0.126.6

djfanatix, 2024-06-04:
[evcc-20240604-140854-trace.log](https://github.com/user-attachments/files/15551177/evcc-20240604-140854-trace.log)

andig, 2024-06-04:
Please see https://github.com/evcc-io/evcc/issues/8009 for diagnosis.

djfanatix, 2024-06-04:
I deleted the addon and reinstalled, and then I disable the circuits and now it works again.
If I enable the circuits EVCC hangs after startup. 
Is something changed for the circuits since 0.126.4 or 0.126.5?

andig, 2024-06-04:
Are you saying going back to 0.126.4 works again? You'll need to share the config that leads to this.

djfanatix, 2024-06-05:
Ok I did do a lot of trials and now I got a backup working again.

See configs below *edited

So if I disable the 'circuit:main' in the Loadpoints, it works again.

I also tried with parent circuits in the Loadpoints, but this doesn't work

Config which was crashing:
`
chargers:
- name: tasmotagarage
  integrateddevice: true
  icon: bike
  type: template
  template: tasmota
  channel: 1
  host: 192.168.1.96 # tasmota device ip address (local)
  standbypower: 10 # treat as charging above this 

- name: goe
  type: template
  template: go-e-v3
  host: 192.168.1.84 # IP-Adresse oder Hostname 

vehicles:
- name: my_car
  type: template
  template: tesla-command
  title: Tesla Model 3

  
- name: vanmoof
  type: custom
  title: Vanmoof Electrified S # display name for UI
  capacity: 0.4 # kWh

# site describes the EVU connection, PV and home battery
site:
  title: Kuringen # display name for UI
  bufferSoC: # ignore home battery discharge above soc (empty to disable)
  residualPower: 250
  circuit: main
  maxGridSupplyWhileBatteryCharging: 100
  meters:
    grid: homewizard # grid meter
    pv:
    - solaredge 
    battery: storedge # battery meter

circuits:
  - name: main
    maxPower: 4800
    meter: homewizard

meters:
  - name: homewizard
    type: template
    template: homewizard
    usage: grid
    host: 192.168.1.109 # IP address or hostname 
  - name: solaredge
    type: template
    template: solaredge-hybrid
    usage: pv
    # Modbus TCP
    modbus: tcpip
    id: 1
    host: 192.168.1.120 # Hostname
    port: 1502 # Port
    timeout: 10s # optional
  - name: storedge
    type: template
    template: solaredge-hybrid
    usage: battery
    # Modbus TCP
    modbus: tcpip
    id: 1
    host: 192.168.1.120 # Hostname
    port: 1502 # Port
    timeout: 10s # optional 

loadpoints:
- title: Go-E  # display name for UI
  charger: goe # charger
  circuit: main
  vehicle: my_car

- title: Tasmota # display name for UI
  charger: tasmotagarage # charger
  vehicle: vanmoof
  circuit: main`

Config working:
`
chargers:
- name: tasmotagarage
  integrateddevice: true
  icon: bike
  type: template
  template: tasmota
  channel: 1
  host: 192.168.1.96 # tasmota device ip address (local)
  standbypower: 10 # treat as charging above this 

- name: goe
  type: template
  template: go-e-v3
  host: 192.168.1.84 # IP-Adresse oder Hostname 

vehicles:
- name: my_car
  type: template
  template: tesla-command
  title: Tesla Model 3

  
- name: vanmoof
  type: custom
  title: Vanmoof Electrified S # display name for UI
  capacity: 0.4 # kWh

# site describes the EVU connection, PV and home battery
site:
  title: Kuringen # display name for UI
  bufferSoC: # ignore home battery discharge above soc (empty to disable)
  residualPower: 250
  circuit: main
  maxGridSupplyWhileBatteryCharging: 100
  meters:
    grid: homewizard # grid meter
    pv:
    - solaredge 
    battery: storedge # battery meter

circuits:
  - name: main
    maxPower: 4800
    meter: homewizard

meters:
  - name: homewizard
    type: template
    template: homewizard
    usage: grid
    host: 192.168.1.109 # IP address or hostname 
  - name: solaredge
    type: template
    template: solaredge-hybrid
    usage: pv
    # Modbus TCP
    modbus: tcpip
    id: 1
    host: 192.168.1.120 # Hostname
    port: 1502 # Port
    timeout: 10s # optional
  - name: storedge
    type: template
    template: solaredge-hybrid
    usage: battery
    # Modbus TCP
    modbus: tcpip
    id: 1
    host: 192.168.1.120 # Hostname
    port: 1502 # Port
    timeout: 10s # optional 

loadpoints:
- title: Go-E  # display name for UI
  charger: goe # charger
#  circuit: main
  vehicle: my_car

- title: Tasmota # display name for UI
  charger: tasmotagarage # charger
  vehicle: vanmoof
#  circuit: main
 `

andig, 2024-06-06:
You MUST NOT add the same circuit twice, so not doing this on the loadpoint is correct. I think we should validate that a higher-level (site) circuit cannot be reused on a lower level (loadpoint).

/cc @premultiply

premultiply, 2024-06-08:
Correct. Should never be used twice in the circuit tree.

andig, 2024-06-08:
Not sure I follow. What should not be used twice in the tree? Question here is if a root circuit can have loadpoints assigned I think? I'm not entirely sure what we should expect.

andig, 2024-06-09:
Closing as config issue. #14244 will prevent this kind of mistake.

closed 2024-06-09 by andig