evcc-io/evcc #14187: Mercedes integration

ThiloGa, 2024-06-04:
### Describe the bug

Ich habe eine funktionierende Intergration über http request an HomeAssistant und wollte im Zuge der Version 126.6 mit dem Fix 14093 jetzt mal das MB Template ausprobieren.

evcc token läuft problemlos durch.

Hier finde ich Ausgabe der Token etwas unglücklich, könnte man meinen, dass die Konfig so aus sehen sollte:
vehicles:
  - name: my_car
    type: template
    template: mercedes
    title: EQV
    user: tg@[...].com
    region: EMEA
    token:
      access: abchjsafkjwfd....
      refresh: OFNIOzkv9Pz9lOskjdfbf...

Mit _richtiger_ Konfig:

evcc.exe vehicle my_car
[main  ] INFO 2024/06/04 21:06:11 evcc 0.126.6
[...]
[main  ] ERROR 2024/06/04 21:06:11 creating vehicle my_car failed: cannot create vehicle type 'template': cannot create vehicle type 'mercedes': cannot get vehicles: Get "https://bff.emea-prod.mobilesdk.mercedes-benz.com/v2/vehicles": unexpected status: 400 (Bad Request)

Selbiges im experimentellen WEB-UI

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
vehicles:
  - name: my_car
    type: template
    template: mercedes
    title: EQV
    user: tg@[...].com
    region: EMEA
    accessToken: KQooB1KT0sEvklvIAzv9LTsE....
    refreshToken: OFNIOzkv9Pz9....
```

### Log details

```text
[main  ] INFO 2024/06/04 21:06:11 evcc 0.126.6
[...]
[main  ] ERROR 2024/06/04 21:06:11 creating vehicle my_car failed: cannot create vehicle type 'template': cannot create vehicle type 'mercedes': cannot get vehicles: Get "https://bff.emea-prod.mobilesdk.mercedes-benz.com/v2/vehicles": unexpected status: 400 (Bad Request)
```

### What type of operating system are you running?

Linux

### Version

126.6

ThiloGa, 2024-06-04:
> > Hier finde ich Ausgabe der Token etwas unglücklich, könnte man meinen, dass die Konfig so aus sehen sollte:
> > vehicles:
> 
> Die Ursache liegt darin, dass die Ausgabe davon ausgeht, dass `type: mercedes` gewählt ist statt `template`.
> 
> Für weitere Analyse bräuchte es User/Passwort an [info@evcc.io](mailto:info@evcc.io)

Mir dünkt, dass die Beschreibung https://docs.evcc.io/docs/devices/vehicles#mercedes-benz dann nicht wirklich korrekt ist.

Mit 
```yaml
    name: my_car
    type: mercedes
    title: eqv2
    user: tg@[...].com
    region: EMEA
    tokens:
      access: eyJhbGciOiJkaXIiLCJlbmMiOiJBMjU2Q0AqXA4i-tXVD3jqwDn2-oqA.LR3CFv8L05rUbHR8luXEGOfXVL9pwZMpIwXYtaS8c7s
      refresh: NLKamTOpMcCcpujbvtBfvn`
```

tuts.

Danke

closed 2024-06-04 by ThiloGa

Matikubo, 2024-06-09:
@ThiloGa: Danke für den Post, denn die Beschreibung in der Dokumentation ist in der Tat etwas irreführend, und jetzt funktioniert die Mercedes Einbindung endlich 😀