evcc-io/evcc #14192: HTTP Plugin sets body on GET request

panzerdev, 2024-06-05:
### Describe the bug

Using the http plugin to change the state of a shelly via a GET call
Changed to POST it works

### Steps to reproduce

Using a custom charger and using the http plugin the do a GET request leads to errors because the Shelly device doesn't allow that. 

### Configuration details

```yaml
Not working:
charger:
  - name: pool
    type: custom
    enable:
      source: http
      uri: "http://192.168.2.236/rpc/Switch.Set?id=0&on={{.enable}}"

Working:
charger:
  - name: pool
    type: custom
    enable:
      source: http
      method: POST
      uri: http://192.168.2.236/rpc
      body: '{"id":1,"method":"Switch.Set","params":{"id":0,"on": {{.enable}}}}'
```

### Log details

```text
[http  ] TRACE 2024/06/04 22:55:41 GET http://192.168.2.236/rpc/Switch.Set?id=0&on=false
[http  ] TRACE 2024/06/04 22:55:41 
GET /rpc/Switch.Set?id=0&on=false HTTP/1.1
Host: 192.168.2.236
User-Agent: Go-http-client/1.1
Content-Length: 5
Accept-Encoding: gzip

false
[main  ] ERROR 2024/06/04 22:55:41 disable: Get "http://192.168.2.236/rpc/Switch.Set?id=0&on=false": read tcp 192.168.64.2:44650->192.168.2.236:80: read: connection reset by peer
```

### What type of operating system are you running?

Linux

### Version

b995404ceeab5e851531635c13495e3aad07730b

GrimmiMeloni, 2024-06-07:
@panzerdev I am not understanding why you consider this to be a bug in evcc. The description sounds to me like a user error (i.e. using GET where POST is required). What am I missing? 😅

panzerdev, 2024-06-07:
Do you see me setting the body anywhere? Also Shelly offers an GET endpoint for switching and this totally fine. Maybe you think that wrong, but wrong is too set a body on a GET request.

GrimmiMeloni, 2024-06-07:
All understood and agreed regarding GET/POST and Body/NoBody. 
But the example is a custom charger that is missing up GET/POST. That's why I am not following. Is this custom charger somewhere shown incorrectly in the docs?

panzerdev, 2024-06-07:
It's a http plugin and where it's used shouldn't matter. 

Also GET is and should be the default. I'm

By the way

https://docs.evcc.io/docs/reference/plugins

Have a look here
[image]

andig, 2024-06-07:
Full ack. PR welcome.

closed 2024-06-12 by andig