evcc-io/evcc #14193: Skoda Enyaq: Fehler 500 bei SOC limit Abfrage

ammann-ruchti, 2024-06-05:
### Describe the bug

Guten Tag zusammen,
seit heute Morgen ab dem Start des Ladevorgangs erhalte ich den Fehler 500 bei der Abfrage des SOC limit. Ist das euch bekannt? Die übrigen Abfragen funktionieren.

### Steps to reproduce

1. Geschieht während des Ladevorgangs
2.
3.
...

### Configuration details

```yaml
...
```

### Log details

```text
p-1 ] DEBUG 2024/06/05 08:43:57 soc estimated: 59.19% (vehicle: 56.00%)
[lp-1 ] DEBUG 2024/06/05 08:43:57 vehicle soc: 59%
[lp-1 ] ERROR 2024/06/05 08:43:57 vehicle soc limit: Get "https://api.connect.skoda-auto.cz/api/v1/charging/TMBJC7NY5NF029324/settings": unexpected status: 500 (Internal Server Error)
[lp-1 ] ERROR 2024/06/05 08:43:57 vehicle range: Get "https://api.connect.skoda-auto.cz/api/v1/charging/TMBJC7NY5NF029324/status": unexpected status: 500 (Internal Server Error)
[lp-1 ] DEBUG 2024/06/05 08:43:57 pv charge current: 3.28A = 6A + -2.72A (1880W @ 3p)
```

### What type of operating system are you running?

Linux

### Version

V0. 126.4

StefanSchoof, 2024-06-05:
500 heißt Serverfehler und da meine App auch den SoC nicht mehr aktualisiert, denke ich VW hat gerade Probleme.

ammann-ruchti, 2024-06-05:
Ja, ist mir schon klar. Die Abfrage in der MySkoda App funktioniert jedoch heute mal problemlos. Hat das mit der Umstellung bei Skoda zu tun, dass die "alten" Zugänge gekappt werden? Darum habe ich das Issue erstellt.

closed 2024-06-05 by ammann-ruchti