evcc-io/evcc #14197: ford-connect: error when generating token with "evcc token"

miniluma, 2024-06-05:
### Describe the bug

To generate the refreshToken for ford-connect, I ran `evcc token ev5 -c ./evcc.yaml -l trace` but I get an error.

### Steps to reproduce

1. in yaml, configure vehicole with template ford-connect
2. get ford developer account, request API access and get new accesstoken 
4. use clientid and secret in yaml
5. run evcc token ev5 -c ./evcc.yaml -l trace
6. enter accesstoken when asked
...

### Configuration details

```yaml
vehicles:
- name: ev5
  type: template
  template: ford-connect
  title: Mustang Mach-E # Wird in der Benutzeroberfläche angezeigt (optional)
  clientid: 30...
  clientsecret: RHO...
  accessToken: Uz2...
#  refreshToken: # Siehe https://docs.evcc.io/docs/devices/vehicles#ford-connect
  vin: WF... # Erforderlich, wenn mehrere Fahrzeuge des Herstellers vorhanden sind (optional)
  capacity: 75.7 # Akkukapazität in kWh (optional)
```

### Log details

```text
? Please enter your authorization code: Uz...=
[ford-connect] TRACE 2024/06/05 09:12:39 POST https://dah2vb2cprod.b2clogin.com/914d88b1-3523-4bf6-9be4-1b96b4f6f919/oauth2/v2.0/token?p=B2C_1A_signup_signin_common
[ford-connect] TRACE 2024/06/05 09:12:41 code=Uz...M%3D&code_verifier=M-9h1IIQGC_JCcSEQK4VzefS9qmFf95-HwC7BL9zj6E&grant_type=authorization_code&redirect_uri=https%3A%2F%2Flocalhost%3A3000
--
{"error":"invalid_grant","error_description":"AADB2C90090: The provided JWE is not a valid 5 segment token.\r\nCorrelation ID: 30f70f83-95d3-4ae6-92c0-ca2e75532fa2\r\nTimestamp: 2024-06-05 07:12:41Z\r\n"}
[ford-connect] TRACE 2024/06/05 09:12:41 POST https://dah2vb2cprod.b2clogin.com/914d88b1-3523-4bf6-9be4-1b96b4f6f919/oauth2/v2.0/token?p=B2C_1A_signup_signin_common
[ford-connect] TRACE 2024/06/05 09:12:42 client_id=30990062-9618-40e1-a27b-7c6bcb23658a&client_secret=RH...&code=Uz...M%3D&code_verifier=M-9h1IIQGC_JCcSEQK4VzefS9qmFf95-HwC7BL9zj6E&grant_type=authorization_code&redirect_uri=https%3A%2F%2Flocalhost%3A3000
--
{"error":"invalid_grant","error_description":"AADB2C90090: The provided JWE is not a valid 5 segment token.\r\nCorrelation ID: 6306d598-5141-4dcb-ad2a-df972ba1721b\r\nTimestamp: 2024-06-05 07:12:41Z\r\n"}
[main  ] FATAL 2024/06/05 09:12:42 oauth2: "invalid_grant" "AADB2C90090: The provided JWE is not a valid 5 segment token.\r\nCorrelation ID: 6306d598-5141-4dcb-ad2a-df972ba1721b\r\nTimestamp: 2024-06-05 07:12:41Z\r\n"
```

### What type of operating system are you running?

Linux

### Version

0.126.6 (b995404c)

Max112233445566, 2024-06-05:
Authorization code does not correspond to the accessToken. Please comment out accessToken in evcc.yaml and try again. If it still does not work, try to comment out clientid, clientsecret and vin as well.
Please take also a look at https://github.com/evcc-io/evcc/discussions/14117#discussioncomment-9654192

miniluma, 2024-06-05:
Authorization code != accessToken is the key. Thank you! @Max112233445566 

Token generation works,
YAML with token works after commenting out the VIN.

closed 2024-06-05 by miniluma