evcc-io/evcc #14200: No SOC on Ford Mach-e

ruffieuxh, 2024-06-05:
### Describe the bug

Hallo,

Seit zwei Wochen habe ich für meinen Ford Mach-e keinen SoC mehr in der OpenWB Anzeige.

SW: OpenWB V1.9 

Ich kann via App und Web Portal problemlos auf meinen Ford Account zugreifen. Zudem habe ich bereits erfolglos das Token erneuert.

Besten Dank für eine Prüfung des Problemes.

Heinz

### Steps to reproduce

1. EVCC in OpenWB konfigurieren
2. Fehlerlog prüfen
3.
...

### Configuration details

```yaml
SW: OpenWB V1.9 

Das File evcc.yaml kann ich auf OpenWB nicht finden.
```

### Log details

```text
Auszug aus dem SoC.log:

2024-06-05 13:13:25: PID: 25937: Lp1: Requesting SoC (LV0)
2024-06-05 13:13:31: PID: 25937: Lp1: Error from EVCC: 2024/06/05 13:13:31 rpc error: code = Unknown desc = cannot create vehicle type 'ford': login failed: unexpected status: 403 (Forbidden) (LV0)
```

### What type of operating system are you running?

Linux

### Version

Auf openWB kein evcc verfügbar

andig, 2024-06-06:
Ford changed api. You'll need client id, secret, access and refresh tokens. Are these configured in OpenWB?

/cc @mhrinas can they be configured if available?

ruffieuxh, 2024-06-06:
In openWB (V 1.9.304.0, wich is the latest nightly build) I can configure user, password, VIN and token. This worked until about 2-3 weeks ago. Did they change the api recently?

andig, 2024-06-07:
Yes…

foersst, 2024-06-07:
I've also created the configuration with id, secret and both tokens. 
I always get this error message
[image]

ruffieuxh, 2024-06-08:
> I've also created the configuration with id, secret and both tokens. I always get this error message
> 
> ![image](https://private-user-images.githubusercontent.com/171460331/337816267-69184e47-b869-42cd-bf16-19dc97cae724.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTc4MzAyNzcsIm5iZiI6MTcxNzgyOTk3NywicGF0aCI6Ii8xNzE0NjAzMzEvMzM3ODE2MjY3LTY5MTg0ZTQ3LWI4NjktNDJjZC1iZjE2LTE5ZGM5N2NhZTcyNC5qcGVnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDYwOCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA2MDhUMDY1OTM3WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9Mjk2OWQ4NjJiYmYxNDkwNTYxMDVkYTcxODJkNzA3OGU5MjE2OTZmOWQ1ZTk0MmZhMDExODJkYzc5YmFlNTE3MiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.J3fumFmuvxn8bdIVwxXUKeHM9iFuGfkYjot2eJ47I4c)
> 
> Is my yaml wrong or is there still a problem with ford.!

Is this with openWB? What SW version?

foersst, 2024-06-08:
> I've also created the configuration with id, secret and both tokens. 
> I always get this error message
> 
> [image]
> 
> Is my yaml wrong or is there still a problem with ford.!

I found the solution in another comment. Just delete out the VIN and now it's running.

foersst, 2024-06-08:
> Is this with openWB? What SW version?

In my case it's working now. The solution is to delete the whole line with VIN.

andig, 2024-06-08:
> Is my yaml wrong or is there still a problem with ford.!

What yaml?

foersst, 2024-06-09:
the problem was using the VIN. After taking it out, everything works fine.
[image]

foersst, 2024-06-09:
The issue was the VIN in my configuration. After taking it out, everything was fine.
[image]

andig, 2024-06-09:
Happy to analyse this if you want to send full config to info@evcc.io. Otherwise we should close here.

closed 2024-06-16 by andig