evcc-io/evcc #14203: ford-connect: show vehicle title instead of api nickname in ui vehicle list

Max112233445566, 2024-06-05:
### Describe the bug

UI -> Configuration -> Vehicles shows vehicle title from evcc.yaml
UI -> vehicle list shows attribute "nickname" from ford api

[image]

[image]

{
"status": "SUCCESS",
"vehicle": {
"vehicleId": "ff80....",
"make": "F",
"modelName": "Kuga - TD (CX482 EU)",
"modelYear": "2020",
"color": "Frozen White",
**"nickName": "Dennis",**

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
vehicles:
- type: template
  template: ford-connect
  title: Kuga
```

### Log details

```text
.
```

### What type of operating system are you running?

Linux

### Version

0.126.6

andig, 2024-06-06:
Does this only happen when vehicles are added from UI or also using yaml file?

Max112233445566, 2024-06-06:
I have not yet added a vehicle via the UI, only via the yaml file.

andig, 2024-06-06:
@Max112233445566  Could you please share the output of /api/state?

andig, 2024-06-06:
Also share your vehicle config. What you've posted above is incomplete.

andig, 2024-06-07:
@Max112233445566 no idea how this could happen. Could you share your full vehicle config with info@evcc.io for debugging?

Max112233445566, 2024-06-07:
Yes, in the UI vehicle list, the nickname <Dennis> is displayed, if there is a valid connection to the cloud. If not, the yaml vehicle title <Kuga> is displayed. In the UI configuration, there is always the correct value <Kuga> displayed. See also the screenshots in my initial posting.

closed 2024-06-08 by andig