evcc-io/evcc #14204: Give "tasmota-sml" template the option to set a custom jsonPrefix

osulivanDE, 2024-06-05:
Depending on which instructions users have followed to set up their Tasmota SML read head, it is possible that a wide variety of jsonPrefix variants have been set up there.

for example: 
+1,3,s,16,9600,**MT681**
Here is the documentation from Tasmota for this slug. Look for "jsonPrefix":
https://tasmota.github.io/docs/Smart-Meter-Interface/#meter-definition

The EVCC template “tasmota-sml” requires the configuration as follows with a fixed "SML" as jsonPrefix
+1,3,s,16,9600,**SML**

Docu here:
https://docs.evcc.io/docs/devices/meters#tasmota-sml-ir-lesekopf-f%C3%BCr-smarte-stromz%C3%A4hler-bspw-hichi

The problem is that many other integrations are dependent on this jsonPrefix and it might not be easy to modify to the from EVCC required "SML"

Battery storage, HomeAssistant, solar integrations etc. might depend on this. It would therefore be nice if EVCC could adapt to a custom slug as it is possible with the custom template where you can set this jsonPrefix yourself with "jq":

```
  - name: Tasmota
    type: custom
    power:
        source: http
        uri: http://192.168.2.95...
        jq: .StatusSNS.MT681.Power_curr
```

This would also make it easier for people to switch from the custom template to the tasmota-sml template.

@andig maybe this makes it more easy to push users to the correct configuration as you mentioned here:  https://github.com/evcc-io/evcc/pull/14146#issuecomment-2148995542

In my idea the configuration from tasmota-sml could look like this:

```
meters:
  - name: my_grid
    type: template
    template: tasmota-sml
    usage: grid
    host: 192.0.2.2 # IP-Adresse oder Hostname
    user: # Standard-User ist admin (optional)
    password: # Passwort des Benutzerkontos (bei führenden Nullen bitte in einfache Hochkommata setzen) (optional) 
    jsonPrefix: #  JSON Payload Prefix. Standard ist: SML (optional)
```

premultiply, 2024-06-05:
Ok, PR welcome!

andig, 2024-06-05:
Just ignore any prefix?

osulivanDE, 2024-06-05:
Even better since it makes it much more convenient for everyone.

premultiply, 2024-06-05:
> Just ignore any prefix?

Like this?
`select(.StatusSNS != null) | .. | .Power_curr? | select(type != "null")`

premultiply, 2024-06-05:
Sounds easy using jq but will need modification how the JSON parsing inside Tasmota implementation is done.

/cc @thierolm

thierolm, 2024-06-06:
I also thought about possible ways, to make the SML part of the Tasmota SML meter more flexible regarding the individual script flavours of the users, but
it's not only the prefix which can be defined individually but also each reading parameter, e.g. Power_curr, which will be provided in the Tasmota response JSON.
Trying to make the Tasmota SML template as flexible as possible, would lead to more configuration parameters.
So why making evcc template more flexible and complex, by having the flexibility already on Tasmota script side?

thierolm, 2024-06-07:
Idea: We are providing a `tasmota-sml-http` template, which configures a HTTP plugin config with many parameters to map the JSON tags and creates multiple HTTP requests to the SML device.

andig, 2024-06-07:
Lets not do this ;) The request here was prefix. Is that still a relevant topic?

thierolm, 2024-06-07:
> Is that still a relevant topic?

Not for me.

osulivanDE, 2024-06-07:
> Lets not do this ;) The request here was prefix. Is that still a relevant topic?

For me yes. Being flexible in the jsonPrefix would be great. 

@thierolm mentioned that the other parameters like **Power_curr** **Power_p1**, **Total_in**, ... might also be flexible since they are also variables on Tasmotas side. 
As far as i have seen diffreent setup guides for Tasmota-SML heads, those variables seem to be quite universal. Imho those do not have to be flexible.

osulivanDE, 2024-06-08:
Thanks to everyone. I totally understand your point @thierolm 
I for my side switched ESS, Solar and HA to the "SML" prefix and now use tasmota-sml

closed 2024-06-08 by osulivanDE

satchafunkilus, 2025-05-26:
I have the situation where I run multiple IR heads off of a single Tasmota device. While it's possible to rename one of the prefixes to SML, I can't do it for all 3. So it would be really nice to have an option to specify the prefix in the template, to be able to add multiple Tasmota sensors. For reference, this is how it looks like now:

`
>D
>B
->sensor53 r
>M 3
+1,4,s,0,9600,OG
+2,3,s,0,9600,SML
+3,2,s,0,9600,EG
1,77070100010800ff@1000,Total Verbrauch,KWh,Total_in,4
1,77070100020800ff@1000,Total Einspeisung,kWh,Total_out,4
1,77070100100700ff@1,Current consumption,W,Power_curr,0
2,77070100010800ff@1000,Gesamtverbrauch,KWh,Totalin,4
2,77070100020800ff@1000,Gesamtertrag,KWh,Totalout,4
2,77070100100700ff@1,BHKW_Leistung,W,Powercurr,0
3,77070100600100ff@#,Meter Nr,,Meter_ID,0
3,77070100010800ff@1000,Total consumption,KWh,Total_in,4
3,77070100100700ff@1,Current consumption,W,Power_curr,0
3,77070100000009ff@#,Meter Nr,,Meter_number,0
#  `

satchafunkilus, 2025-05-27:
Can you clarify what you mean by define multiple meters? How would I be able to differentiate between 3 meters on a single tasmota device in evcc? All i can define is the IP, so it will always take the meter with the SML prefix