evcc-io/evcc #14207: Opel: Doku unzureichend

andig, 2024-06-06:
Es wird nicht klar, woher die Tokens kommen:     accessToken: # Siehe https://docs.evcc.io/docs/devices/vehicles#opel
    refreshToken: # Siehe https://docs.evcc.io/docs/devices/vehicles#opel

andig, 2024-06-06:
@naltatis es scheint als würde die description aus dem Template fehlen? Nämlich:

> Benötigt `access` und `refresh` Tokens. Diese können über den Befehl `evcc token [name]` generiert werden.

naltatis, 2024-06-06:
Kontext? Experimental Config UI? Auf der Dokuseite steht dieser Text ja.
[image]

Glaube aber auch, dass wir den Satz an sich noch mal überarbeiten,bzw. ausführlicher machen sollten. Das sind ja drei Schritte: Fahrzeug ohne Token anlegen, Cli Befehl ausführen, Token ergänzen.

smokie128, 2024-06-27:
Konnte über die GUI das Fahrzeug nicht ohne Token anlegen, also dementsprechend auch kein token über die Onboard-Variante erstellen. Mit dem [PSA Token Helper](https://github.com/hurzhurz/psa-token-helper) ging es.

Hier beißt sich die Maus selbst in den Schwanz.

andig, 2024-06-27:
Eigentlich sollten sich auch „kaputte“ Fahrzeuge speichern lassen? @naltatis

closed 2024-07-09 by github-actions[bot]