evcc-io/evcc #14210: Wakeup wird nicht ausgelöst

cmock, 2024-06-06:
### Describe the bug

in der konstellation cFos PowerBrain Solar und Smart <span>#</span>1 wird das auto morgens, wenn genügend PV-leistung vorhanden ist, nicht aufgeweckt.

grundsätzlich ist die funktionalität offenbar in evcc vorhanden, siehe #11958 .

wenn ich jetzt an evcc vorbei ins modbus-register 8086 eine "5" schreibe, wacht das auto auf (evcc läuft und ich schreibe per `mbpoll` auf der shell in das register). soweit ich den code verstehe, wäre das auch das, was evcc über die `api.Resurrector` bei cFos-wallboxen macht: https://github.com/evcc-io/evcc/blob/b995404ceeab5e851531635c13495e3aad07730b/charger/cfos.go#L155-L159

was ich noch nicht verstanden habe, ist, unter welchen bedingungen der wakeup-timer deaktiviert wird; erwähnenswert ist hier, daß nach dem aktivieren der ladung bei schlafendem auto 1,3A und 5W gezogen werden.

nach "start charging" im log ist das auto weiterhin im schlafzustand, bis ich eben wie oben erwähnt händisch den CP-disconnect anstosse.

### Steps to reproduce

1. abends auto anstecken, modus "PV"
2. warten bis die sonne aufgeht
3. auto wird nicht aufgeweckt

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
# trial token, valid until 2024-06-18
sponsortoken: ...

network:
  schema: http
  host: rockpro.tahina.priv.at
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: ...

interval: 30s # control cycle interval

database:
  type: sqlite
  dsn: /rockpool/evcc/evcc.sqlite

mqtt:
  broker: raspi.tahina.priv.at:1883
  topic: evcc

# telemetry: true

meters:
  - name: grid
    type: custom
    power: # W
      source: calc
      add:
        - source: mqtt
          topic: smartmeter/power
          jq: .P
          timeout: 10s
        - source: mqtt
          topic: smartmeter/power
          jq: .PO
          timeout: 10s
          scale: -1
    energy: # kWh
      source: calc
      add:
        - source: mqtt
          topic: smartmeter/energy
          jq: .tPI
          timeout: 10s
        - source: mqtt
          topic: smartmeter/energy
          jq: .tPO
          timeout: 10s
          scale: -1
    currents:
      - source: mqtt
        topic: smartmeter/power
        jq: .I1
        timeout: 10s
      - source: mqtt
        topic: smartmeter/power
        jq: .I2
        timeout: 10s
      - source: mqtt
        topic: smartmeter/power
        jq: .I3
        timeout: 10s
  - name: cfos_meter
    type: template
    template: cfos
    usage: charge
    modbus: tcpip
    id: 1
    host: 192.168.57.21
    port: 4701
  - name: pv_production
    type: custom
    power:
      source: mqtt
      topic: pv_inverter/P_out
      timeout: 35s
    energy:
      source: mqtt
      topic: pv_inverter/energy_total
      scale: 0.001
      timeout: 35s
    currents:
      - source: mqtt
        topic: pv_inverter/L1_I
        timeout: 35s
      - source: mqtt
        topic: pv_inverter/L2_I
        timeout: 35s
      - source: mqtt
        topic: pv_inverter/L3_I
        timeout: 35s

site:
  title: Hausi
  meters:
    grid: grid
    pv:
      - pv_production

tariffs:
  currency: EUR
  grid:
    type: awattar
    region: at
    charges: 0.08043
    tax: 0.2
  feedin:
    type: fixed
    price: 0.05

loadpoints:
  - title: Auto
    charger: wallbox
    meter: cfos_meter
    priority: 2
  - title: Roller
    charger: roller_charger
    vehicle: roller
    priority: 1
    enable:
      threshold: -500
      delay: 1m
    disable:
      threshold: -50
      delay: 1m

chargers:
  - name: roller_charger
    type: template
    template: tasmota
    host: 192.168.57.12
    channel: 1
    standbypower: 15
  - name: wallbox
    type: template
    template: cfos
    host: 192.168.57.21

vehicles:
  - name: roller
    type: template
    template: offline
    title: Bimie Grazie
    capacity: 3
    phases: 1
    icon: moped
  - name: smart_1
    type: smart-hello
    title: Smart #1
    user: user@example.com
    password: SagIchNicht!
    vin: ...
    capacity: 62
    icon: car
    features: ["coarsecurrent"]
    identifiers:
      - 12345678
  - name: guest_ev
    type: template
    template: offline
    title: Gast-EV
    icon: generic
    identifiers:
      - 87654321

messaging:
  events:
    start:
      title: Laden gestartet
      msg: Laden gestartet, ${vehicleTitle} im Modus ${mode}
    stop:
      title: Laden beendet
      msg: Laden beendet, ${vehicleTitle}, ${chargedEnergy:%.1fk} kWh in ${chargeDuration}
  services:
    - type: ntfy
      uri: https://ntfy.sh/...
```

### Log details

```text
[site  ] DEBUG 2024/06/06 09:23:43 ----
[lp-1  ] DEBUG 2024/06/06 09:23:43 charge power: 0W
[lp-2  ] DEBUG 2024/06/06 09:23:43 charge power: 133W
[main  ] DEBUG 2024/06/06 09:23:43 lp Auto at prio 2 gets additional 133W from Roller at prio 1, total 133W
[site  ] DEBUG 2024/06/06 09:23:43 pv power: 1622W
[site  ] DEBUG 2024/06/06 09:23:43 grid meter: -1402W
[site  ] DEBUG 2024/06/06 09:23:43 grid currents: [1.93 2.32 2.1]A
[site  ] DEBUG 2024/06/06 09:23:43 giving loadpoint priority for additional: 133W
[site  ] DEBUG 2024/06/06 09:23:43 site power: -1535W
[lp-1  ] DEBUG 2024/06/06 09:23:43 !! session: chargeRater.chargedEnergy=2.9 - chargedAtStartup=0.0
[lp-1  ] DEBUG 2024/06/06 09:23:44 charge total import: 135.440kWh
[lp-1  ] DEBUG 2024/06/06 09:23:44 charger status: B
[lp-1  ] DEBUG 2024/06/06 09:23:44 pv charge current: 6.67A = 0A + 6.67A (-1535W @ 1p)
[lp-1  ] DEBUG 2024/06/06 09:23:44 site power -1535W <= 0W enable threshold
[lp-1  ] DEBUG 2024/06/06 09:23:44 pv enable in 0s
[lp-1  ] DEBUG 2024/06/06 09:23:44 pv enable timer elapsed
[lp-1  ] DEBUG 2024/06/06 09:23:44 charger enable
[lp-1  ] DEBUG 2024/06/06 09:23:44 wake-up timer: start
[site  ] DEBUG 2024/06/06 09:24:13 ----
[lp-1  ] DEBUG 2024/06/06 09:24:13 charge power: 5W
[lp-2  ] DEBUG 2024/06/06 09:24:13 charge power: 131W
[site  ] DEBUG 2024/06/06 09:24:13 pv power: 1649W
[site  ] DEBUG 2024/06/06 09:24:13 grid meter: -1435W
[site  ] DEBUG 2024/06/06 09:24:13 grid currents: [1.97 2.69 2.15]A
[site  ] DEBUG 2024/06/06 09:24:13 site power: -1435W
[lp-2  ] DEBUG 2024/06/06 09:24:13 !! session: chargeRater.chargedEnergy=0.7 - chargedAtStartup=0.0
[lp-2  ] DEBUG 2024/06/06 09:24:13 charge total import: 6.487kWh
[lp-2  ] DEBUG 2024/06/06 09:24:13 charger status: C
[lp-2  ] DEBUG 2024/06/06 09:24:14 pv charge current: 8.24A = 2A + 6.24A (-1435W @ 1p)
[site  ] DEBUG 2024/06/06 09:24:43 ----
[lp-1  ] DEBUG 2024/06/06 09:24:43 charge power: 5W
[lp-2  ] DEBUG 2024/06/06 09:24:43 charge power: 129W
[main  ] DEBUG 2024/06/06 09:24:43 lp Auto at prio 2 gets additional 129W from Roller at prio 1, total 129W
[site  ] DEBUG 2024/06/06 09:24:43 pv power: 1677W
[site  ] DEBUG 2024/06/06 09:24:43 grid meter: -1461W
[site  ] DEBUG 2024/06/06 09:24:43 grid currents: [2 2.72 2.19]A
[site  ] DEBUG 2024/06/06 09:24:43 giving loadpoint priority for additional: 129W
[site  ] DEBUG 2024/06/06 09:24:43 site power: -1590W
[lp-1  ] DEBUG 2024/06/06 09:24:43 !! session: chargeRater.chargedEnergy=2.9 - chargedAtStartup=0.0
[lp-1  ] DEBUG 2024/06/06 09:24:43 charge total import: 135.440kWh
[lp-1  ] DEBUG 2024/06/06 09:24:43 charger status: C
[lp-1  ] DEBUG 2024/06/06 09:24:44 charge start energy: 135kWh
[lp-1  ] INFO 2024/06/06 09:24:44 start charging ->
[lp-1  ] DEBUG 2024/06/06 09:24:44 wake-up timer: stop
[lp-1  ] DEBUG 2024/06/06 09:24:45 soc estimated: 74.64% (vehicle: 74.00%)
[lp-1  ] DEBUG 2024/06/06 09:24:45 vehicle soc: 75%
[lp-1  ] DEBUG 2024/06/06 09:24:45 vehicle range: 209km
[lp-1  ] DEBUG 2024/06/06 09:24:45 pv charge current: 12.9A = 6A + 6.91A (-1590W @ 1p)
[lp-1  ] DEBUG 2024/06/06 09:24:45 pv timer reset
[lp-1  ] DEBUG 2024/06/06 09:24:45 pv timer inactive
[lp-1  ] DEBUG 2024/06/06 09:24:45 max charge current: 12A
[site  ] DEBUG 2024/06/06 09:25:13 ----
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.126.6

andig, 2024-06-06:
Und was ist hier das issue?

cmock, 2024-06-06:
dass evcc das WakeUp() in cfos.go in meiner konstellation nicht aufzurufen scheint, obwohl es das sollte.

andig, 2024-06-06:
Warum auch- das Auto lädt ja?

cmock, 2024-06-06:
nein, das auto fängt eben erst zum laden an, wenn ich händisch das modbus-register beschreibe.

wenn ich ins log schaue, es geht um lp-1: 

um 09:23:44 sieht evcc genügend PV-überschuss und gibt das laden frei. 

um 09:24:44 glaubt evcc, daß der ladevorgang wirklich gestartet ist, das auto schläft aber weiter und zieht an der nun aktiven wallbox nur seine 5W/1.3A standby-leistung.

irgendwann später setz ich den `mbpoll`-befehl ab, der für 5 sek den CP abschaltet, das auto wacht auf und zieht erst dann wirklich die freigegebene leistung.

(lp-2 ist ein elektroroller an einer tasmota-steckdose und nur der vollständigkeit halber im log)

andig, 2024-06-07:
Das mag ja alles sein- wir brauchen aber ein Log vom Fehlerfall, nicht vom erfolgreichen Workaround! Hier gibts ja nix zu analysieren. Dafür bitte auch den Charger auf `trace` stellen.

andig, 2024-06-07:
> um 09:24:44 glaubt evcc, daß der ladevorgang wirklich gestartet ist, das auto schläft aber weiter und zieht an der nun aktiven wallbox nur seine 5W/1.3A standby-leistung.

Wenn das so sein sollte, dass **lügt** die Wallbox. Status C heisst, das Fahrzeug lädt. Bitte Ticket bei CFOS, das ist einfach grob falsch.

cmock, 2024-06-07:
ich glaub, das auto lügt, nicht die wallbox -- aber egal.

ich werd morgen früh mal auf trace stellen und ein update liefern.

andig, 2024-06-07:
Der Status kommt jdfls. von der Wallbox. Und der ist ausschlaggebend.

cmock, 2024-06-07:
dh der wakeup-timer wird gecanceled, wenn die wallbox status "C" berichtet? 

das ist in dieser konstellation halt problematisch, weil offenbar das ladegerät im auto zwar signalisiert, daß es bereit ist, aber irgendein steuergerät erst aufgeweckt werden muss, damit es ihm sagt, daß es auch strom ziehen darf.

ich mein, ich kann einen cronjob einrichten, der eine stunde nach sonnenaufgang das modbus-register beschreibt, aber mir wär eine lösung in evcc lieber. daß der wakup-timer erst stoppt, wenn die wallbox 50% der freigegebenen leistung meldet oder so...

andig, 2024-06-07:
> dh der wakeup-timer wird gecanceled, wenn die wallbox status "C" berichtet?

Natürlich, denn ab da wird geladen. Ladebereitschaft wäre Status B. Wie das schief gehen kann ist mir nicht klar.

/cc @premultiply bei der cfos bei jeden `enable` das Register zu schreiben ist ja auch keine Option, oder???

premultiply, 2024-06-07:
Nicht wirklich.
Normalerweise sollte so eine Box-Firmware das ja auch ohnehin völlig automatisch machen...

Bei Status C nochmal aufzuwecken ergibt jedenfalls keinen Sinn.

andig, 2024-06-07:
> Nicht wirklich.

Weil wir dann zu recht schlafende Fahrzeuge aufwecken oder weshalb?

> Bei Status C nochmal aufzuwecken ergibt jedenfalls keinen Sinn.

Schadet aber auch nicht- oder passiert da was?

premultiply, 2024-06-07:
Ja, die Ladung wird sofort abgebrochen, da die Freigabe entzogen bzw. virtuell abgesteckt wird.

Je nach Fahrzeug könnte das sogar zu einer Fehlermeldung führen.

cmock, 2024-06-07:
man könnte das verhalten ja optional machen, so wie "coarsecurrent".

denn wie die kommentare in #11958 zeigen, zeigen wohl verschiedene autos das verhalten.

ich schalt jetzt jedenfalls mal trace ein und start ein tcpdump, und dann schauen wir morgen früh, wie sich das verhält.

cmock, 2024-06-08:
OK, wir haben die geschichte inkl trace.

wirklich aufgewacht ist das auto heute erst, als die umschaltung auf 3P erfolgte; diesen fall behandelt die wallbox nämlich vollständig selber.

```
[site  ] DEBUG 2024/06/08 07:24:21 ----
[cfos  ] TRACE 2024/06/08 07:24:21 modbus: send 09 25 00 00 00 06 01 03 1f 7e 00 02
[cfos  ] TRACE 2024/06/08 07:24:22 modbus: recv 09 25 00 00 00 07 01 03 04 00 00 00 00
[lp-1  ] DEBUG 2024/06/08 07:24:22 charge power: 0W
[site  ] DEBUG 2024/06/08 07:24:22 pv power: 1778W
[site  ] DEBUG 2024/06/08 07:24:22 grid meter: -1704W
[site  ] DEBUG 2024/06/08 07:24:22 grid currents: [2.45 2.54 2.35]A
[site  ] DEBUG 2024/06/08 07:24:22 site power: -1704W
[site  ] DEBUG 2024/06/08 07:24:51 ----
[cfos  ] TRACE 2024/06/08 07:24:51 modbus: send 09 26 00 00 00 06 01 03 1f 7e 00 02
[cfos  ] TRACE 2024/06/08 07:24:51 modbus: recv 09 26 00 00 00 07 01 03 04 00 00 00 00
[lp-1  ] DEBUG 2024/06/08 07:24:51 charge power: 0W
[site  ] DEBUG 2024/06/08 07:24:52 pv power: 1808W
[site  ] DEBUG 2024/06/08 07:24:52 grid meter: -1731W
[site  ] DEBUG 2024/06/08 07:24:52 grid currents: [2.48 2.58 2.39]A
[site  ] DEBUG 2024/06/08 07:24:52 site power: -1731W
[cfos  ] TRACE 2024/06/08 07:24:52 modbus: send 09 27 00 00 00 06 01 03 1f 7a 00 04
[cfos  ] TRACE 2024/06/08 07:24:52 modbus: recv 09 27 00 00 00 0b 01 03 08 00 00 00 00 00 02 40 18
[lp-1  ] DEBUG 2024/06/08 07:24:52 charge total import: 147.480kWh
[cfos  ] TRACE 2024/06/08 07:24:52 modbus: send 09 28 00 00 00 06 01 03 1f 9c 00 01
[cfos  ] TRACE 2024/06/08 07:24:52 modbus: recv 09 28 00 00 00 05 01 03 02 00 01
[lp-1  ] DEBUG 2024/06/08 07:24:52 charger status: B
[cfos  ] TRACE 2024/06/08 07:24:52 modbus: send 09 29 00 00 00 06 01 03 1f a0 00 0f
[cfos  ] TRACE 2024/06/08 07:24:52 modbus: recv 09 29 00 00 00 21 01 03 1e 38 62 64 65 61 36 33 39 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[cfos  ] TRACE 2024/06/08 07:24:52 modbus: send 09 2a 00 00 00 06 01 03 1f 9e 00 01
[cfos  ] TRACE 2024/06/08 07:24:52 modbus: recv 09 2a 00 00 00 05 01 03 02 00 00
[lp-1  ] DEBUG 2024/06/08 07:24:52 pv charge current: 7.53A = 0A + 7.53A (-1731W @ 1p)
[lp-1  ] DEBUG 2024/06/08 07:24:52 site power -1731W <= 0W enable threshold
[lp-1  ] DEBUG 2024/06/08 07:24:52 pv enable in 0s
[lp-1  ] DEBUG 2024/06/08 07:24:52 pv enable timer elapsed
[cfos  ] TRACE 2024/06/08 07:24:52 modbus: send 09 2b 00 00 00 06 01 06 1f 9d 00 3c
[cfos  ] TRACE 2024/06/08 07:24:52 modbus: recv 09 2b 00 00 00 06 01 06 1f 9d 00 3c
[lp-1  ] DEBUG 2024/06/08 07:24:52 max charge current: 6A
[cfos  ] TRACE 2024/06/08 07:24:52 modbus: send 09 2c 00 00 00 06 01 06 1f 9e 00 01
[cfos  ] TRACE 2024/06/08 07:24:52 modbus: recv 09 2c 00 00 00 06 01 06 1f 9e 00 01
[lp-1  ] DEBUG 2024/06/08 07:24:52 charger enable
[lp-1  ] DEBUG 2024/06/08 07:24:52 wake-up timer: start
[site  ] DEBUG 2024/06/08 07:25:21 ----
[cfos  ] TRACE 2024/06/08 07:25:21 modbus: send 09 2d 00 00 00 06 01 03 1f 7e 00 02
[cfos  ] TRACE 2024/06/08 07:25:22 modbus: recv 09 2d 00 00 00 07 01 03 04 00 00 00 05
[lp-1  ] DEBUG 2024/06/08 07:25:22 charge power: 5W
[site  ] DEBUG 2024/06/08 07:25:22 pv power: 1836W
[site  ] DEBUG 2024/06/08 07:25:22 grid meter: -1757W
[site  ] DEBUG 2024/06/08 07:25:22 grid currents: [2.53 2.92 2.43]A
[site  ] DEBUG 2024/06/08 07:25:22 site power: -1757W
[site  ] DEBUG 2024/06/08 07:25:51 ----
[cfos  ] TRACE 2024/06/08 07:25:51 modbus: send 09 2e 00 00 00 06 01 03 1f 7e 00 02
[cfos  ] TRACE 2024/06/08 07:25:51 modbus: recv 09 2e 00 00 00 07 01 03 04 00 00 00 05
[lp-1  ] DEBUG 2024/06/08 07:25:51 charge power: 5W
[site  ] DEBUG 2024/06/08 07:25:52 pv power: 2059W
[site  ] DEBUG 2024/06/08 07:25:52 grid meter: -1964W
[site  ] DEBUG 2024/06/08 07:25:52 grid currents: [2.82 3.18 2.73]A
[site  ] DEBUG 2024/06/08 07:25:52 site power: -1964W
[cfos  ] TRACE 2024/06/08 07:25:52 modbus: send 09 2f 00 00 00 06 01 03 1f 7a 00 04
[cfos  ] TRACE 2024/06/08 07:25:52 modbus: recv 09 2f 00 00 00 0b 01 03 08 00 00 00 00 00 02 40 18
[lp-1  ] DEBUG 2024/06/08 07:25:52 charge total import: 147.480kWh
[cfos  ] TRACE 2024/06/08 07:25:52 modbus: send 09 30 00 00 00 06 01 03 1f 9c 00 01
[cfos  ] TRACE 2024/06/08 07:25:52 modbus: recv 09 30 00 00 00 05 01 03 02 00 02
[lp-1  ] DEBUG 2024/06/08 07:25:52 charger status: C
[cfos  ] TRACE 2024/06/08 07:25:52 modbus: send 09 31 00 00 00 06 01 03 1f 7a 00 04
[cfos  ] TRACE 2024/06/08 07:25:52 modbus: recv 09 31 00 00 00 0b 01 03 08 00 00 00 00 00 02 40 18
[lp-1  ] DEBUG 2024/06/08 07:25:52 charge start energy: 147kWh
[lp-1  ] INFO 2024/06/08 07:25:52 start charging ->
[lp-1  ] DEBUG 2024/06/08 07:25:52 wake-up timer: stop
[cfos  ] TRACE 2024/06/08 07:25:53 modbus: send 09 32 00 00 00 06 01 03 1f a0 00 0f
[cfos  ] TRACE 2024/06/08 07:25:53 modbus: recv 09 32 00 00 00 21 01 03 1e 38 62 64 65 61 36 33 39 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[lp-1  ] DEBUG 2024/06/08 07:25:53 soc estimated: 36.00% (vehicle: 36.00%)
[lp-1  ] DEBUG 2024/06/08 07:25:53 vehicle soc: 36%
[lp-1  ] DEBUG 2024/06/08 07:25:53 vehicle range: 97km
[cfos  ] TRACE 2024/06/08 07:25:53 modbus: send 09 33 00 00 00 06 01 03 1f 9e 00 01
[cfos  ] TRACE 2024/06/08 07:25:53 modbus: recv 09 33 00 00 00 05 01 03 02 00 01
[lp-1  ] DEBUG 2024/06/08 07:25:53 pv charge current: 14.5A = 6A + 8.54A (-1964W @ 1p)
[lp-1  ] DEBUG 2024/06/08 07:25:53 pv timer reset
[lp-1  ] DEBUG 2024/06/08 07:25:53 pv timer inactive
[cfos  ] TRACE 2024/06/08 07:25:53 modbus: send 09 34 00 00 00 06 01 06 1f 9d 00 8c
[cfos  ] TRACE 2024/06/08 07:25:53 modbus: recv 09 34 00 00 00 06 01 06 1f 9d 00 8c
[lp-1  ] DEBUG 2024/06/08 07:25:53 max charge current: 14A
```

andig, 2024-06-08:
```
[lp-1  ] DEBUG 2024/06/08 07:24:52 charger enable
[lp-1  ] DEBUG 2024/06/08 07:25:52 charger status: C
```

Aus deinen Minilogschnipseln ist wenig zu sehen, außer dass 1 Minute nach Ladefreigabe die Ladung beginnt. Bis dahin wurde `1f96` nicht geschrieben. Ob das noch passiert wäre lässt sich nicht sagen.

andig, 2024-06-08:
@premultiply ich sehe im Logfile kein

    wake-up charger

premultiply, 2024-06-08:
Wozu auch?
Das Laden (Status C) beginnt ja ganz offensichtlich vorher ohne externe Nachhilfe.

andig, 2024-06-08:
Schon klar- aber nach 1min. Nach 30s hätte das Wakeup erfolgen sollen, oder?

premultiply, 2024-06-08:
Ich hatte angenommen, dass das ggf. am 30 Sek-Intervall hier liegt, sprich der neue Status just eintrifft bevor der Wake-up ausgelöst wird, oder?

andig, 2024-06-08:
Möglich, bräuchte mehr Log.

cmock, 2024-06-08:
der logschnipsel ist schon bewußt so gewählt -- nach dem ende passiert nix, 5W leistungsaufnahme am charger durch das schlafende auto, bis dann einige zeit später genug PV-leistung ansteht für die umschaltung auf 3P, die dann auch das auto aufweckt.

im anhang das gesamte log...
[evcc.log](https://github.com/user-attachments/files/15747730/evcc.log)

andig, 2024-06-08:
Verstehe ich nicht. Nach 1min beginnt die Ladung doch? Das bringt mich wieder an die Stelle von oben:

> Wenn das so sein sollte, dann lügt die Wallbox. Status C heisst, das Fahrzeug lädt. Bitte Ticket bei CFOS, das ist einfach grob falsch.

In dem Fall: kein Todo bei evcc!

cmock, 2024-06-08:
irgendwie diskutieren wir hier im kreis.

meine enduser-perspektive ist, daß ich das auto abends anstecke und mir erwarte, daß es bei sonnenaufgang anfängt zu laden, egal, wer in dem dreieck aus autohersteller, wallbox-hersteller und evcc jetzt "schuld hat" am derzeitigen unbefriedigenden status.

technisch gesehen vermute ich, daß physisch sehr wohl ein status C vorliegt; ich würd allerdings nur sehr ungern an der offenen wallbox unter spannung mit dem oszi nachschauen, was in dem fall wirklich auf CP passiert. der umstand, daß ja leistung aufgenommen wird, wenn auch nur 5W, zeigt, daß das relais durchschaltet und läßt es plausibel erscheinen, daß das ladegerät im auto auf steckerseite soweit wach ist und das autonom handhabt. 

damit das dann aber auch auf DC-seite was produktives macht, braucht es offenbar (nicht nur beim smart <span>#<span>1, siehe #11958) noch den wakeup-zyklus, wie besprochen.

ich versteh auch euren standpunkt, da nicht für jeden fall special cases einbauen zu wollen, aber irgendwie müssen wir halt eine lösung finden, und ich fände es schade, wenn die "ich schreib mir das wieder mal selber, weil die bestehenden lösungen halt doch nicht funktionieren" lauten würde, weil an sich ist evcc schon sehr schön und funktional.

andig, 2024-06-08:
> technisch gesehen vermute ich

Es gibt Standards für sowas. Entweder der wird hier verletzt oder nicht. Wenn das klar ist können wir weiter schauen.

#11958 ist ebenfalls cfos. Daher:

> Nochmal ganz klar: die Frage hier ist: wie und warum kann eine WB Status C signalisieren wenn physisch kein Status C herrscht? Wir kennen keine andere Box die so ein Verhalten zu zeigen scheint.

Also bitte an den Hersteller wenden!

andig, 2024-06-08:
https://evsim.gonium.net/#wie-kommuniziert-das-elektroauto-mit-der-ladesäule

cmock, 2024-06-08:
ich soll mich jetzt also mit dem oszi zwischen CP und PE hängen und schauen, ob da das auto wirklich status C signalisiert in diesem schlafzustand?

was würde das für einen unterschied machen bezüglich der lösung des problems, ob die wallbox lügt oder das auto? soll ich dann ein ticket bei Smart aufmachen, wenn das auto wirklich C signalisiert?

mal ganz konkret: wenn ich einen pullrequest schicken würde, der ein "feature: forceWakeup" umsetzt, hätte das eine chance, angenommen zu werden?

andig, 2024-06-08:
> soll ich dann ein ticket bei Smart aufmachen, wenn das auto wirklich C signalisiert?

Was spricht dagegen? Wenn ein E-Auto an einer normgerechten WB nicht lädt ist das ein Mangel. Wenn die WB sich nicht normgerecht verhält ebenso.

> wenn ich einen pullrequest schicken würde, der ein "feature: forceWakeup" umsetzt, hätte das eine chance, angenommen zu werden?

Wir haben ein funktionierendes Wakeup soweit bekannt. Solange nicht klar ist, warum es noch eins braucht und wo das überhaupt hin gehört (Charger? Auto?)- nein. Wir wollen evcc nicht mit Optionen für alle Arten von Sondersituation vollpflastern. Außerdem ist es a) kein Massenphänomen und b) per Workaround lösbar.

premultiply, 2024-06-08:
> wenn ich einen pullrequest schicken würde, der ein "feature: forceWakeup" umsetzt, hätte das eine chance, angenommen zu werden?

Auf keinen Fall. Wenn das Auto meldet laden (Status C) muss es das auch tun.
Wenn man trotz Status "Laden" nach 30 Sekunden nochmal abstecken muss dass es auch wirklich lädt ist irgendwas ganz grundlegend verkehrt.

Und es wäre nicht das erste Fahrzeug wo der Hersteller die Software nachbessern musste und dies auch getan hat.

closed 2024-06-08 by andig

julred, 2024-07-10:
Vermutlich auch interessant für dieses Issue. Mit der neusten Beta Firmware der cfos Wallbox (2.1.8) ist die Unterstützung EV40 EV-Ready mit reingekommen. Das Feature soll das Aufwecken der Fahrzeuge realisieren.

Zitat cfos:

> EV-Ready ist eine Spezifikation, die hauptsächlichem von französichen
> Auto-Herstellern verwendet wird.
> 
> Nach der technischen Spezifikation
> https://www.asefa-cert.com/medias/ev-ready_certification_technical_reference_document_v2.0a.pdf
> 
> soll zum Aufwecken die CP-Leitung für 3 Sekunden auf -12 V geschaltet
> werden (EV40).
> 
> Mit der angegeben Option wird dies statt dem simulierten Ausstecken
> durchgeführt. Dies wirkt sich auch auf das von EVCC verwendete Kommando
> zum Aufwecken aus.