evcc-io/evcc #14215: Idee: Template für Fake-Grid-Meter, wenn nur PV-Meter vorhanden ist

VolkerK62, 2024-06-06:
Wenn nur ein PV-Meter vorhanden ist, wird in den FAQ empfohlen, für den "üblichen" Hausverbrauch `residualpower` zu nutzen.
Das hat mich auf die Idee gebracht ein Fake-Grid-Meter zu basteln.
Ich denke, das Ganze könnte man auch in ein Template gießen in dem der "übliche" Hausverbrauch eingetragen werden kann (evtl. per API änderbar?).

```
- name: grid_fake
  type: custom
  power:
    source: calc
    add:
    - source: http   # Erzeugung
      uri: localhost:7070/api/state
      jq: .result.pvPower
      scale: -1
    - source: const  # Hausverbrauch
      value: 500
    - source: http   # Ladeleistungen
      uri: localhost:7070/api/state
      jq: .result.loadpoints | map(.chargePower) | add
```

VolkerK62, 2024-06-07:
Aktuell wird im UI im Balken keine Erzeugung angezeigt, wenn keine Ladung läuft.
Wenn die Ladung läuft, gibt es erhebliche (Werte) Differenzen zwischen IN und OUT.
Zweck ist ein ähnliches look&feel Ergebnis zu bekommen, als wenn ein Grid Meter vorhanden wäre.

Natürlich könnte man das auch nur als Info in die FAQ aufnehmen und wer es braucht, kann es nachbauen.
Ein Template wäre da allerdings bequemer und fehlerunanfälliger.
Den Hausverbrauch per API ändern zu können, wäre dabei nur die Kirsche auf der Torte.

andig, 2024-06-07:
> Aktuell wird im UI im Balken keine Erzeugung angezeigt, wenn keine Ladung läuft.

Das klingt nach einem Fehler. @naltatis ?

> Wenn die Ladung läuft, gibt es erhebliche (Werte) Differenzen zwischen IN und OUT.

Mit einem fake grid meter wären die dann ja nur an anderer Stelle, oder?

naltatis, 2024-06-07:
Ein Grid Meter zu haben, was konstanten Verbrauch simuliert, finde ich gut. Das können wir auch zu Demo-Zwecken und für das Onboarding verwenden. Sonnengeführtes Laden geht ja prinzipiell auch ohne Grid-Meter.

@andig Die Errechnung von Hausverbrauch und Preis funktioniert aktuell nicht ohne Grid-Meter. Da werden keine Daten geliefert. Siehe auch Konversation im Slack https://evccgroup.slack.com/archives/C01321PUJAD/p1717606343931499 Hier müssten wir die fehlende Werte errechnen (wie @VolkerK62 im Template vorschlägt).

[image]

Glaube geschickter wäre es, wenn wir es nativ einbauen. Das Fake-Meter ist ja nicht self-contained, sondern muss dann ja mit der Adresse von evcc selbst konfiguriert werden, um an Produktion und Ladeleistung zu kommen.

naltatis, 2024-06-07:
Hab gerade noch mal in den Code geschaut. Mir ist noch nicht ganz klar, warum wir in diesem Fall keine Grid-Daten sehen. Berechnet wird das auf jeden Fall:

https://github.com/evcc-io/evcc/blob/master/core/site.go#L620-L623

Ich schau mir mal an wo das verloren geht.

naltatis, 2024-06-07:
Ok, Ursache ist, dass wir den errechneten Grid-Power-Wert nicht publishen. Mit dem PR ist das behoben. Hausverbrauch ist dann natürlich als 0 angezeigt, aber damit kann man sicher erst einmal arbeiten.