evcc-io/evcc #14217: UI wird nach Update auf Version 0.126.6 nicht mehr geladen

mattbalzer, 2024-06-06:
### Describe the bug

Nach Update auf der Version 0.126.6 in Home Assistant wird das Add-On nicht mehr geladen. 
Folgende Fehlermeldung steht im Protokoll:
[site  ] ERROR 2024/06/06 22:41:57 grid meter: strconv.ParseFloat: parsing "<nil>": invalid syntax

und 

Request failed with status code 502.
GET http://192.168.1**.1**:8123/api/hassio_ingress/TVvbuGF39rESsl4U2JFzt4CS9vdSeAse1tgQNDRyb44/api/auth/status

An der Konfiguration wurde nichts verändert.

### Steps to reproduce

1. Update bei Home Assistant auf Version 0.126.6
2. Start evcc. Keine Benutzeroberfläche vorhanden
...

### Configuration details

```yaml
meters:
  - name: grid1
    type: custom
    power:
      source: http
      uri: http://192.168.1**.**/cm?cmnd=Status%208
      jq: .StatusSNS.MT681.Power_cur

  - type: template
    template: shelly-pro-3em
    usage: pv
    host: 192.168.1**.**
    name: pv1

site:
  title: Gromberg
  meters:
    grid: grid1
    pv:
      - pv1
```

### Log details

```text
[site  ] INFO 2024/06/06 22:48:26 site config:
[site  ] INFO 2024/06/06 22:48:26   meters:      grid ✓ pv ✓ battery ✗
[site  ] INFO 2024/06/06 22:48:26     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2024/06/06 22:48:26     pv 1:      power ✓ energy ✓ currents ✓
[site  ] INFO 2024/06/06 22:48:26   vehicles:
[site  ] INFO 2024/06/06 22:48:26     vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
[lp-1  ] INFO 2024/06/06 22:48:26 loadpoint 1:
[lp-1  ] INFO 2024/06/06 22:48:26   mode:        minpv
[lp-1  ] INFO 2024/06/06 22:48:26   charger:     power ✓ energy ✗ currents ✓ phases ✗ wakeup ✗
[lp-1  ] INFO 2024/06/06 22:48:26   meters:      charge ✓
[lp-1  ] INFO 2024/06/06 22:48:26     charge:    power ✓ energy ✗ currents ✓
[lp-1  ] DEBUG 2024/06/06 22:48:26 phase timer inactive
[lp-1  ] DEBUG 2024/06/06 22:48:26 pv timer inactive
[lp-1  ] INFO 2024/06/06 22:48:26 vehicle updated: unknown -> Audi Q4 Sportback e-tron
[lp-1  ] DEBUG 2024/06/06 22:48:26 set charge mode: minpv
[site  ] DEBUG 2024/06/06 22:48:26 ----
[lp-1  ] DEBUG 2024/06/06 22:48:26 charge power: 0W
[lp-1  ] DEBUG 2024/06/06 22:48:26 charge currents: [0 0 0]A
[site  ] DEBUG 2024/06/06 22:48:26 pv power: 0W
[site  ] ERROR 2024/06/06 22:48:27 grid meter: strconv.ParseFloat: parsing "<nil>": invalid syntax
[site  ] DEBUG 2024/06/06 22:48:56 ----
[lp-1  ] DEBUG 2024/06/06 22:48:56 charge power: 0W
[lp-1  ] DEBUG 2024/06/06 22:48:56 charge currents: [0 0 0]A
[site  ] DEBUG 2024/06/06 22:48:56 pv power: 0W
[site  ] ERROR 2024/06/06 22:48:56 grid meter: strconv.ParseFloat: parsing "<nil>": invalid syntax
[site  ] DEBUG 2024/06/06 22:49:26 ----
[lp-1  ] DEBUG 2024/06/06 22:49:26 charge power: 0W
[lp-1  ] DEBUG 2024/06/06 22:49:26 charge currents: [0 0 0]A
[site  ] DEBUG 2024/06/06 22:49:27 pv power: 0W
[site  ] ERROR 2024/06/06 22:49:27 grid meter: strconv.ParseFloat: parsing "<nil>": invalid syntax
[site  ] DEBUG 2024/06/06 22:49:56 ----
[lp-1  ] DEBUG 2024/06/06 22:49:56 charge power: 0W
[lp-1  ] DEBUG 2024/06/06 22:49:56 charge currents: [0 0 0]A
[site  ] DEBUG 2024/06/06 22:49:56 pv power: 0W
[site  ] ERROR 2024/06/06 22:49:57 grid meter: strconv.ParseFloat: parsing "<nil>": invalid syntax
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.126.6

andig, 2024-06-06:
Fixed in master

closed 2024-06-06 by andig

mattbalzer, 2024-06-06:
Danke für die schnelle Antwort. D.h. bei der nächste Version korrigiert?
Leider habe ich evcc neu installiert, d.h. vorher deinstalliert und alle meine Ladevorgänge sind nicht mehr vorhanden. Gibt es eine Möglichkeit die wieder zu bekommen?