evcc-io/evcc #14219: Aktivierung 1P/3P Umschaltung für KSE wBX16 Charger

immofan, 2024-06-07:
Die KSE wBX16 Wallbox wird ja schon unterstütz. Die 1P/3P Umschaltung kann die Box. Der Code ist wohl (teilweise??) schon vorhanden. Die Funktion wurde aber wieder heraus genommen.  

Ich kann gerne beim Testen helfen. Wallbox, PV-Anlage und Auto vorhanden. 
Die Funktion wäre aufgrund der größe meiner PV-Anlage sehr hilfreich. Danke schon mal.

premultiply, 2024-06-07:
Hat sich denn an der Situation was geändert?
Wenn ja, dann finde mal bitte heraus wie das per Modbus funktioniert.

immofan, 2024-06-09:
Mit meinen eingeschränkten Kenntnissen in diesem Bereich scheint es so zu sein, dass die Umschaltung erst ab einem gewissen Softwarestand implementiert ist. Das Register 0x35 konnte ich mit dem alten Softwarestand noch nicht mal abfragen. Nach einem Update geht das problemlos. Ergebnis: 0 (=3 phasig)

Nur schreiben kann ich in das Register nicht. Beim Versuch eine 1 (=1phasig)  in das Register zur schreiben, kommt der Fehler: "Invalid replay error!"
In andere Register kann ich schreiben (z.B. Ladestromstärke). 

An der Stelle fehlen mir die Kenntnisse, in welchem Format dort gearbeitet wird. Auch ist mir nicht bekannt, ob evtl. vorbereitende Aufrufe erforderlich sind...

Testbedingungen: Auto (VW ID 4) angesteckt, Laden via EVCC auf aus gestellt. Denke mal, es ist nicht gut während einem Ladevorgang umzuschalten. 

Was kann ich probieren? Tipps?

premultiply, 2024-06-10:
_Originally postedr by @4rz0 in https://github.com/evcc-io/evcc/issues/3824#issuecomment-2155851630_

immofan, 2024-06-10:
Softwareversion bei meiner Box lt. Register 0x30 ist 0x3243.
Ich habe mit dem Hersteller Kontakt aufgenommen. Mal sehen was die zu meinem Fehler sagen.

immofan, 2024-06-10:
Vom Hersteller habe ich folgende Auskunft bekommen. Für mich ist das schlecht und war aus der Dokumentation nicht herauszulesen. Für die Käufer der neueren Version ist das eine positive Auskunft:

"Leider ist die Phasenumschaltung bei dieser Hardwareversion noch nicht unterstützt. Den entsprechenden Hinweis finden Sie in der Bedienungsanleitung der Wallbox unter Punkt 9:  Umschaltung zwischen 1/3-Phasen Laden
 
Konkret bedeutet der ausgelesene Wert, dass es sich bei Ihrer Wallbox um die Version C(0x43)2(0x32). Generell gilt, dass Wallboxen der Hardwareversion C die Phasen Umschaltung nicht unterstützen. Die Umstellung auf die neue Hardwareversion fand bei uns in Q1/22 statt.

premultiply, 2024-06-10:
Und ab welcher Version genau wird es unterstützt?

PR #14284 (2024-06-14): KSE: Add 1p3p, voltages and total energy
Fixes #14219

1p3p requires at least HW Revision "R".

Please test and report back the output of `evcc charger --diagnose --log trace`.