evcc-io/evcc #14222: Auswertung Ladeenergie: minimale Summendifferenz bei Netzbezug

An-Wi, 2024-06-07:
### Describe the bug

Ich versuche schon eine ganze Weile die Ursache einer ca. 12,7-kWh-Differenz bei mir bei der Summenbildung vom Netzbezug zu finden, die im Vergleich zu den CSV-Daten auftritt. 
Die Summenbildung der "Sonne" passte zunächst augenscheinlich, wurde mir aber bei genauerer Analyse der CSV-Datei verständlicher.

Aus mir nicht bekannten Gründen habe ich 3 Einträge in der Ladehistorie, die keine Endzeit und keinen Solar-%-Eintrag aber einen Wert für die geladene Energie besitzen:
23.12.2023 12:39
16.12.2023 10:10
14.10.2023 10:13 (nicht im Screenshot: geladene Energie ist sehr gering)
[image]

Diese 3 Einträge werden vermutlich bei der Summenbildung nicht berücksichtigt, wodurch die besagte Differenz bei mir entsteht...

Da das hier "Jammern auf sehr hohem Niveau" ist, habe ich mich bemüht, die Ursache zu finden und daher bis jetzt mit der Meldung gewartet.

Der richtig Umgang mit solchen unvollständigen Datensätzen ist wahrscheinlich "ignorieren", oder?
Frage: Könnte ich das bei mir "heilen", da ich sowohl die fehlende Endzeit als auch die fehlende Solar-% "manuell ermitteln" könnte?

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
-
```

### Log details

```text
-
```

### What type of operating system are you running?

Windows

### Version

0.121.0 - 0.126.6

naltatis, 2024-06-07:
Danke für die Analyse. Generell gibt es schon einen Prozess, der unvollständige/abgebrochene Einträge bei einem Neustart vervollständigt. Siehe hier
https://github.com/evcc-io/evcc/blob/b995404ceeab5e851531635c13495e3aad07730b/core/session/db.go#L56-L81
Wenn ich das richtig sehe, korrigieren wir hier aber "nur" die Energiemenge und Zählerstände, nicht aber die `finished` Zeit. Das scheint bei deinen Einträge auch passiert zu sein.

/cc @GrimmiMeloni

naltatis, 2024-06-07:
Bzgl. Abweichung Auswertung im UI: Bei den fehlerhaften Einträgen fehlt ja nicht nur die Endzeit, sondern auch Details wie Preis und Sonnenanteil. Demnach würde ich das jetzige Verhalten, unvollständige Einträge zu ignorieren auch als korrekt ansehen. Konsequenterweise sollten wir sie dann auch in der CSV nicht mit exportieren.

Grundproblem ist aus meiner Sicht eher das Speicherverhalten. Wir sollten Werte wie Ladedauer, Sonnenanteil, ... einfach in regelmäßigen Abstände (alle 15min würd ja schon reichen) in die DB schreiben. Dann wären solche Abweichungen, wenn sie denn mal passieren, immer relativ klein und wir könnten die Logik von oben (rückwirkendes Errechnen von Energiemenge) entfernen und beim Neustart einfach alle "Unfinished" Ladevorgänge auf die aktuelle Zeit setzen.

Zur Tragweite: Ich hab mal meine Datenbank überprüft. Dort habe ich knapp 1000 Einträge. Davon ist in diesem Jahr einer ohne Endzeit. Im letzten Jahr waren es vier. Aber das hängt natürlich auch vom technischen Setup und den genutzten Geräten ab.

naltatis, 2024-06-07:
> Meldet sich das System auf diesem Weg noch "sauber ab" und schreibt den Ladevorgang in die Datenbank zurück?

Ja. Nur wenn du hart den Stecker ziehst, geht was verloren. Neustart via UI oder OS schreibt vorher noch in die DB.

StefanSchoof, 2024-06-07:
Mit [0.121.3](https://github.com/evcc-io/evcc/releases/tag/0.121.3) wurde [Session log: close unfinished sessions on startup](https://github.com/evcc-io/evcc/pull/10246). Das ist vom Release ist vom 28.10.23. Je nachdem wann das Update auf die Version gemacht wurde, kann das der Fix gewesen sein.

GrimmiMeloni, 2024-06-07:
> Wenn ich das richtig sehe, korrigieren wir hier aber "nur" die Energiemenge und Zählerstände, nicht aber die `finished` Zeit. Das scheint bei deinen Einträge auch passiert zu sein.

Ja, guter Punkt. Wir haben uns damals in der Diskussion primär auf die Energiewerte konzentriert.

Mir scheint als wäre das hier aber bereits geklärt, bzw. wenn überhaupt sind die fehlenden Endstände nicht das Problem? Oder ist in der Sache jetzt doch noch was zu tun (z.B. das Schließen von offenen Sessions beim Start um einen Endzeitpunkt erweitern)?

naltatis, 2024-06-07:
Ich würde sagen, wir haben hier aktuell kein (relevantes/akutes) Problem. Gerade, weil die Unstimmigkeiten vor dem Fix waren.
Daher würd ich vorschlagen, es erstmal beim Status quo zu belassen und das Issue zu schließen.

closed 2024-06-07 by naltatis

An-Wi, 2024-06-09:
Wenn der Fix bereits von 28.10.23 war und ich noch 2 Einträge mit fehlende Endzeiten in Dezember 23 hatte, dann könnte nach wie vor das Problem bestehen, denn ich spiele immer zeitnah (innerhalb 1-2 Tagen) die Updates ein. Möglicherweise kann aber auch ein weiteres Fix nach Dez. 2023 das Problem endgültig behoben haben... von dieser Seite sehe ich hier zunächst keinen Handlungsbedarf.

Zu meiner 2. Frage nochmal: Wie kann ich die DB manuell nachbearbeiten und die 3 Problemfälle bei mir händisch korrigieren?

An-Wi, 2024-06-13:
Issue auch aus meiner Sicht abgeschlossen. Behalte es trotzdem im Auge und melde mich, falls erneut unvollständige Einträge auftauchen sollten.

Meine DB konnte ich Dank "DB Browser for SQLite" nun korrigieren - meine Abweichungen sind somit nicht mehr existent. 
Danke euch für den schnellen Support.