evcc-io/evcc #14226: Tesla fleet API error 429 Too Many Requests

Raudi1, 2024-06-07:
### Describe the bug

Plötzlich geht bei mir die Tesla Anbindung nicht mehr. Kommt immer `429 Too Many Requests` egal ob EVCC SOC abfragt, oder Ladegeschwindigkeit sendet. Hat bisher ohne jegliche Änderung meinerseits problemlos funktioniert. Über die Tesla APP und auch Home Assistant läuft alles ohne Probleme.

### Steps to reproduce

Tesla anstecken/als Fahrzeug auswählen

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: warn
levels:
  wario: trace
  db: trace

# unique installation id
plant: xxx

interval: 15s # control cycle interval

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: true

sponsortoken: xxx

meters:
  - type: template
    template: shelly-pro-3em
    usage: grid
    host: 192.168.1.131
    name: Strom

  - name: Balkonsolar
    type: template
    template: hoymiles-opendtu
    usage: pv
    host: 192.168.1.126

chargers:
  - type: template
    template: twc3
    host: 192.168.1.35
    name: twc

  - name: gemini
    type: template
    template: go-e-v3
    host: 192.168.1.149

  - name: handtuchwaermer
    type: template
    template: shelly
    host: 192.168.1.125
    standbypower: 15 # Leistung oberhalb des angegebenen Wertes wird als Ladeleistung gewertet (optional)
    integrateddevice: true
    icon: heater
    heating: true

vehicles:
  - name: wario
    type: template
    template: tesla
    title: Wario
    accessToken: xxx
    refreshToken: xxx
    vin: xxx
    capacity: 60
    mincurrent: 1
    maxcurrent: 16
    control: true

  - name: Teslamate
    type: template
    template: teslamate
    title: Wario-Teslamate
    id: 1
    capacity: 60

loadpoints:
  - title: TWC3
    charger: twc
    vehicle: wario
    mode: pv
    priority: 2

  - title: Go-e
    charger: gemini
    vehicle: wario
    mode: pv
    priority: 1

  - title: Handtuchwaermer
    charger: handtuchwaermer
    priority: 0
    guardduration: 1m

site:
  title: Heideweg 20
  meters:
    grid: Strom
    pv: Balkonsolar
tariffs:
  grid:
    type: tibber
    token: xxx
  co2:
    type: grünstromindex
    zip: xxx

mqtt:
  broker: localhost:1883
  topic: evcc # root topic for publishing, set empty to disable publishing
  user: MQTT
  password: xxx
```

### Log details

```text
[db    ] TRACE 2024/06/07 12:34:36 UPDATE `sessions` SET `created`="2024-06-07 12:09:20.11",`finished`="0000-00-00 00:00:00",`loadpoint`="TWC3",`identifier`="",`vehicle`="Wario",`odometer`=NULL,`meter_start_kwh`=NULL,`meter_end_kwh`=NULL,`charged_kwh`=0,`charge_duration`=NULL,`solar_percentage`=NULL,`price`=NULL,`price_per_kwh`=NULL,`co2_per_kwh`=NULL WHERE `id` = 412 1 <nil>
[lp-1  ] ERROR 2024/06/07 12:34:50 vehicle odometer: 429 Too Many Requests
[lp-1  ] ERROR 2024/06/07 12:34:50 vehicle soc: 429 Too Many Requests
[lp-1  ] ERROR 2024/06/07 12:34:50 charger get max current: 429 Too Many Requests
[lp-1  ] ERROR 2024/06/07 12:34:50 vehicle soc: 429 Too Many Requests
[lp-1  ] ERROR 2024/06/07 12:34:50 charger get max current: 429 Too Many Requests
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.126.6

Dirk77777, 2024-06-07:
Gleiches Problem (bei mir Docker mit 0.126.6). Gefühlt wechselt sich dieser Fehler mit #14229 ab.

Raudi1, 2024-06-07:
Ist möglich, dass Tesla was an der API umgestellt hat. Evtl. limitieren sie jetzt die Commands pro Tag und Fahrzeug.
https://developer.tesla.com/docs/tesla-fleet-api#membership-levels

Dirk77777, 2024-06-07:
Also wenn das stimmt, nur 50 Commands pro Tag und Fahrzeug, dann ist es vorbei mit dem PV-Überschussladen.

Raudi1, 2024-06-07:
Ja, sieht wohl leider so aus. Größere Pakete könnten natürlich helfen, aber die gibt es ja noch nichtmal.
Allerdings ginge es evtl. über BLE. https://github.com/teslamotors/vehicle-command
Wäre eh besser als über eine Cloud Lösung gehen zu müssen.

premultiply, 2024-06-07:
Oder einfach lokal wie vorgesehen über die Wallbox? 🤷🏻‍♂️

row2hot, 2024-06-07:
Gleiches Problem hier, ich hab auch die Meldung:  429 Too Many Requests

Raudi1, 2024-06-07:
> Oder einfach lokal wie vorgesehen über die Wallbox? 🤷🏻‍♂️

Die gehen leider nur bis 6A runter. Meine PV hat aber nur 1,75kWp. Damit habe ich aber trotzdem im letzten Monat 90% nur durch PV geladen.

andig, 2024-06-07:
@Raudi1 OT: wie machst du das?

Raudi1, 2024-06-07:
> @Raudi1 OT: wie machst du das?

Bei gutem Wetter hauptsächlich das Fahrrad genommen und fast immer wenn Sonne war ab 1A 1-phasig über EVCC geladen. Waren im letzten Monat ca. 100kWh laut EVCC.

Am elegantesten wäre es denke ich den Tesla BLE Support in ESPHome oder so zu integrieren und das dann über EVCC zu steuern.
Es gibt auch schon Versuche in die Richtung: https://github.com/esphome/feature-requests/issues/2630

andig, 2024-06-07:
>  und fast immer wenn Sonne war ab 1A 1-phasig über EVCC geladen. 

@Raudi1 nochmal OT: schon klar- aber wie? `charger: twc3` mit `minCurrent: 1`?

andig, 2024-06-07:
Bei evcc, insbesondere auch serverseitig bzgl. Tesla Service hat sich nichts geändert, es gab auch keinen Neustart. Im Logfile kann ich erkennen, dass es viele Fahrzeuge gibt wo das problemlos zu laufen scheint obwohl wir tw. mehrere Requests pro Sekunde in Richtung Tesla schicken.

StevieC121176, 2024-06-07:
> > und fast immer wenn Sonne war ab 1A 1-phasig über EVCC geladen.
> 
> @Raudi1 nochmal OT: schon klar- aber wie? `charger: twc3` mit `minCurrent: 1`?

laut Konfiguration oben: ja

```
chargers:
  - type: template
    template: twc3
    host: 192.168.1.35
    name: twc

vehicles:
  - name: wario
    type: template
    template: tesla
    title: Wario
    accessToken: xxx
    refreshToken: xxx
    vin: xxx
    capacity: 60
    mincurrent: 1
    maxcurrent: 16
    control: true
```

Raudi1, 2024-06-07:
> > und fast immer wenn Sonne war ab 1A 1-phasig über EVCC geladen.
> 
> @Raudi1 nochmal OT: schon klar- aber wie? `charger: twc3` mit `minCurrent: 1`?

Genau. Kann auch gerne noch mehr Daten von EVCC und Teslamate teilen, falls Interesse besteht.

> Bei evcc, insbesondere auch serverseitig bzgl. Tesla Service hat sich nichts geändert, es gab auch keinen Neustart. Im Logfile kann ich erkennen, dass es viele Fahrzeuge gibt wo das problemlos zu laufen scheint obwohl wir tw. mehrere Requests pro Sekunde (für unterschiedliche Fahrzeuge) in Richtung Tesla schicken.
> 
> See [teslamotors/vehicle-command#255](https://github.com/teslamotors/vehicle-command/issues/255)

Ich habe heute Nacht bei meinem Tesla ein Update auf 2024.14.9 gemacht. Ggfs. hängt es damit zusammen. Oder es wird einfach nur sukzessiv umgestellt.
Jedenfalls reicht das aktuelle Limit des einzigen verfügbaren Membership Levels von Tesla nicht. Habe nachdem EVCC nicht mehr steuern konnte dann von Hand in Home Assistant gesteuert. Das ging nach einer Weile dann auch nicht mehr...
Ich denke mittelfristig wird es nur über BLE vernünftig gehen. Lokal ist mir eh lieber als Cloud. Diese Umstellung ist mal wieder ein gutes Beispiel warum.

Raudi1, 2024-06-08:
Heute geht es wieder. Die Frage ist nur für wie lange bevor das Limit wieder erreicht wird.

FraBoCH, 2024-06-08:
Same issue on my side, also since updating my Model Y to 2024.14.9. Yesterday evening, no way to control anything, 429 errors at any request. This morning it was working again. To mitigate the issue for the time being, I’ve increased the refresh interval in yaml config to 60 instead of 30.

ugoetzen, 2024-06-08:
I think there is an issue with the Tesla API. Yesterday I got the 429 errors and nothing worked at all.
Today after evcc was working as aspected for a while, I now got these errors and the charging power cannot adjusted anymore. Reading the API seems to still working.
[lp-1 ] WARN 2024/06/08 13:37:09 charger logic error: current mismatch (got 6A, expected 16A)
[lp-1 ] DEBUG 2024/06/08 13:37:09 pv charge current: 0A = 6A + -8.79A (6065W @ 3p)
[lp-1 ] DEBUG 2024/06/08 13:37:09 projected site power 6065W >= 0W disable threshold
[lp-1 ] DEBUG 2024/06/08 13:37:09 pv disable timer start: 3m0s
[lp-1 ] DEBUG 2024/06/08 13:37:09 pv disable in 3m0s
[lp-1 ] ERROR 2024/06/08 13:37:19 max charge current 3A: Post "https://tesla.evcc.io/api/1/vehicles/XP7YGCEL7NB00XXXX/command/set_charging_amps": context deadline exceeded (Client.Timeout exceeded while awaiting headers)

FraBoCH, 2024-06-08:
I observe something similar: I’m getting timeouts from EVCC proxy: \"https://tesla.evcc.io/api/1/vehicles/XXXXXXX/command/set_charging_amps\": net/http: request canceled (Client.Timeout exceeded while awaiting headers).
I have no issue changing the amp & starting/stopping charge from the Tesla mobile app. 
I’m also on a tesla wallbox 3 with a Tesla Y.

Shuflel, 2024-06-08:
> I observe something similar: I’m getting timeouts from EVCC proxy: [...]

Having the exact same problem with our Model 3.

andig, 2024-06-08:
@all could you please stop posting "me too"?

Raudi1, 2024-06-08:
Ich habe jetzt mal die Bluetooth Option getestet. Funktioniert super!
Installationsschritte:
Auf dem Raspberry PI Zero W raspberry OS lite installiert und SSH+WLAN konfiguriert.
```
sudo apt update && sudo apt install -y wget git build-essential
wget https://go.dev/dl/go1.20.14.linux-armv6l.tar.gz
tar -xvf go1.20.14.linux-armv6l.tar.gz
mkdir -p ~/.local/share && mv go ~/.local/share
echo 'export GOPATH=$HOME/.local/share/go' >> ~/.bashrc
echo 'export PATH=$HOME/.local/share/go/bin:$PATH' >> ~/.bashrc
git clone https://github.com/teslamotors/vehicle-command.git
cd vehicle-command
go get ./...
go build ./...
go install ./...
openssl ecparam -genkey -name prime256v1 -noout > private.pem
openssl ec -in private.pem -pubout > public.pem
sudo setcap 'cap_net_admin=eip' "$(which tesla-control)"
tesla-control -vin YOUR_VIN -ble add-key-request public.pem owner cloud_key
```
Dann muss man einfach die Keycard auf die Konsole halten und bestätigen. Achtung: Es taucht vorher im Tesla keine Meldung auf!
Im Anschluss kann man Kommandos wie über die cloud API schicken. Beispiel:
`tesla-control -ble -vin xxx -key-file private.pem charging-start`

Jetzt wäre nur meine Frage, wie man das am besten in EVCC einbindet? MQTT, REST...? Am besten wäre es natürlich das ganze in den bestehenden Tesla Code einzubinden und redundant zu gestalten. Dann könnte man wenn vorhanden und erreichbar über BLE gehen, und sonst über die Cloud.

andig, 2024-06-08:
Wie du willst, aber dafür bitte neue Diskussion. Template welcome.

Raudi1, 2024-06-08:
> Wie du willst, aber dafür bitte neue Diskussion. Template welcome.

https://github.com/evcc-io/evcc/discussions/14252

Wenn sich ein Entwickler dafür finden könnte, wäre das eine super Sache. Ich helfe auch so gut es geht, aber ich kann leider keinen sauberen Code schreiben.

user1272349087234, 2024-06-09:
Heute kommt permanent dieser Fehler: Garage: max charge current 3A: Post \"https://tesla.evcc.io/api/1/vehicles/LRW3E7EK1NC510698/command/set_charging_amps\": context deadline exceeded (Client.Timeout exceeded while awaiting headers)

Stacykid, 2024-06-09:
Bei mir auch das gleiche Problem, meine M3 hat von ca. 50-80% problemlos geladen, dann war wohl das Tageslimit an Request erreicht und EVCC regelte nicht mehr. Er bliebt bei der zuletzt eingestellten Ladeleistung.

Da wird Tesla wieder etwas an der API umgeschraubt haben, was ein Müll.....

phoet, 2024-06-09:
https://github.com/teslamate-org/teslamate/issues/3928#issuecomment-2141012015

> The official API mentions a limit of 1 call every 5 minutes: that's actually 300 calls per day on vehicle_data then gets a 429 response with a retry-after header of 86400 seconds.

Man müsste das intervall also auf 300 Sekunden stellen. Wäre ja nicht so schlimm, wenn man unterschiedliche update intervalle konfigurieren könnte.

haroldboom, 2024-06-09:
> Ich habe jetzt mal die Bluetooth Option getestet. Funktioniert super! Installationsschritte: Auf dem Raspberry PI Zero W raspberry OS lite installiert und SSH+WLAN konfiguriert.
> 
> ```
> sudo apt update && sudo apt install -y wget git build-essential
> wget https://go.dev/dl/go1.20.14.linux-armv6l.tar.gz
> tar -xvf go1.20.14.linux-armv6l.tar.gz
> mkdir -p ~/.local/share && mv go ~/.local/share
> echo 'export GOPATH=$HOME/.local/share/go' >> ~/.bashrc
> echo 'export PATH=$HOME/.local/share/go/bin:$PATH' >> ~/.bashrc
> git clone https://github.com/teslamotors/vehicle-command.git
> cd vehicle-command
> go get ./...
> go build ./...
> go install ./...
> openssl ecparam -genkey -name prime256v1 -noout > private.pem
> openssl ec -in private.pem -pubout > public.pem
> sudo setcap 'cap_net_admin=eip' "$(which tesla-control)"
> tesla-control -vin YOUR_VIN -ble add-key-request public.pem owner cloud_key
> ```
> 
> Dann muss man einfach die Keycard auf die Konsole halten und bestätigen. Achtung: Es taucht vorher im Tesla keine Meldung auf! Im Anschluss kann man Kommandos wie über die cloud API schicken. Beispiel: `tesla-control -ble -vin YOUR_VIN -key-file private.pem charging-start`
> 
> Jetzt wäre nur meine Frage, wie man das am besten in EVCC einbindet? MQTT, REST...? Am besten wäre es natürlich das ganze in den bestehenden Tesla Code einzubinden und redundant zu gestalten. Dann könnte man wenn vorhanden und erreichbar über BLE gehen, und sonst über die Cloud.

This would be a great solution for me, not relying on the cloud because I also use Teslamate which is using some of the 300 quota. Would this be hard to implement? Should another issue be created? @andig

infinus, 2024-06-09:
> [teslamate-org/teslamate#3928 (comment)](https://github.com/teslamate-org/teslamate/issues/3928#issuecomment-2141012015)
> 
> > The official API mentions a limit of 1 call every 5 minutes: that's actually 300 calls per day on vehicle_data then gets a 429 response with a retry-after header of 86400 seconds.
> 
> Man müsste das intervall also auf 300 Sekunden stellen. Wäre ja nicht so schlimm, wenn man unterschiedliche update intervalle konfigurieren könnte.

So for now set 

interval: 300s  to comply with current API limitations until fix is found.

yipikaye, 2024-06-10:
Can confirm it does seem to be max requests in a 24hr period. After 24hrs it works for 30min or so then stops. 

I'm thinking to get one of these and be done with it
TWC3 is no good

https://evse.com.au/product/ocular-iq-solar-7-kw-w-5m-cable/

andig, 2024-06-10:
See https://github.com/evcc-io/evcc/discussions/14252 for BLE

Breakyman, 2024-06-10:
Gleiches Problem bei meinem MY: 
ERROR 2024/06/10 08:52:34 max charge current 3A: Post "https://tesla.evcc.io/api/1/vehicles/XP7YGCEL2PBXXXXXX/command/set_charging_amps": net/http: request canceled (Client.Timeout exceeded while awaiting headers)

phoet, 2024-06-10:
Vielleicht dieses Issue locken, bis es eine praktische Lösung gibt?

wimaha, 2024-06-13:
If you are looking for a simple solution to use BLE instead of the Fleet API to bypass the rate limits, this is a functional solution that works with EVCC: https://github.com/wimaha/TeslaBleHttpProxy

Maddimax, 2024-06-13:
So am I understanding it right, that the Tesla control just does not work at the moment? Worth mentioning in the documentation maybe?

Likemike24, 2024-06-28:
Wir hier eigentlich noch eine EVCC-eigene Lösung entwickelt, um das wieder hinzubekommen? 
Tronity z.B. hat den Datenabruf zwischenzeitlich von 3 Minuten wieder auf 30 Sekunden umgestellt ("In Zusammenarbeit mit Tesla"). Man musste dafür aber einen neuen Zugriffschlüssel für das Fahrzeug erstellen.

phoet, 2024-06-28:
Bekommen die das mit, wenn man die Org erwähnt @TRONITY_IO @tronity @sebastian-tronity  ?

Stacykid, 2024-06-28:
> Bekommen die das mit, wenn man die Org erwähnt @TRONITY_IO @tronity @sebastian-tronity ?

Aktuell ruht der See, ich hoffe das EVCC auch an etwas bastelt, solange ist es nicht wirklich gut brauchbar.

sebastian-tronity, 2024-06-28:
Hier am besten Tesla anschreiben.

andig, 2024-06-28:
@Stacykid wir *können* nichts basteln da Tesla das Problem ist. Leider haben wir auf unsere Anfrage bisher keine Antwort erhalten. Wäre vielleicht eine gute Gelegenheit, Dich als Kunde von Tesla dafür einzusetzen, dass uns für unsere ClientID `08ff827fa31b-4ea6-bed1-817be0ab40c8` ein Angebot für einen hochvolumigeren Plan gemacht wird.

D4rkH0ur, 2024-06-28:
> @Stacykid wir *können* nichts basteln da Tesla das Problem ist. Leider haben wir auf unsere Anfrage bisher keine Antwort erhalten. Wäre vielleicht eine gute Gelegenheit, Dich als Kunde von Tesla [...]

Ich habe mal eine Email gesendet. Mal schauen ob es eine Antwort gibt 😅

Stacykid, 2024-06-28:
Welche Mailadresse wäre das denn? Würde da auch hinschreiben.

D4rkH0ur, 2024-06-28:
> wäre

fleetapisupport@tesla.com

Stacykid, 2024-06-28:
so, auch mal nett hingeschrieben.....

pssturges, 2024-06-29:
Interesting development that has implications for this issue if there is any substance to it
https://www.facebook.com/groups/chargehqbeta/permalink/851971010123659/

phoet, 2024-06-29:
@pssturges what exactly is to see under that private permalink?

pssturges, 2024-06-29:
Ah sorry. Didn’t realise it’s a private group

[image]
[image]

haroldboom, 2024-06-29:
I hope this is true because it would be the silver bullet for us

Likemike24, 2024-06-30:
Neues von der Tronity Front:
`heute freuen wir uns, eine neue Art der Fahrzeugintegration vorstellen zu dürfen. Diese haben wir in Zusammenarbeit mit Tesla in Tronity implementiert und werden die betroffenen Nutzer schrittweise für das neue Vorgehen freischalten.

Was ist neu und warum ist dies auch für Nicht-Tesla-Fahrer spannend?

In der aktuellen Implementierung werden die Fahrzeugdaten jeweils von den Servern der Fahrzeughersteller abgerufen. Dabei sendet das Fahrzeug die Daten an den Fahrzeughersteller, und TRONITY kann im Namen der Nutzer die Daten entsprechend abrufen und verarbeiten. 

Im Gegensatz dazu ist in der Zusammenarbeit mit Tesla nun eine Integration in TRONITY möglich, bei der das Fahrzeug die Daten direkt an TRONITY sendet! Dabei entscheidet das Fahrzeug, wann Daten gesendet werden, und es ist somit in der Regel ein Datenzugriff alle 30 Sekunden möglich 🤯 und nicht wie aktuell alle 3-5 Minuten.

Daneben gibt es viele weitere Vorteile. Zum Beispiel kann das Fahrzeug das eigene Energiemanagement berücksichtigen, wodurch kein unnötiger Batterieverlust in der Datenübertragung entsteht.
`

andig, 2024-06-30:
Tronity war so nett uns auch direkt zu antworten. In kurz: das hilf leider nichts weil darüber keine Kommandos an das Fahrzeug geschickt werden können- und genau das ist das Problem. Dabei hilft jetzt allerdings https://github.com/evcc-io/evcc/pull/14616.

Likemike24, 2024-06-30:
OK, schade

haroldboom, 2024-07-01:
> Ah sorry. Didn’t realise it’s a private group
> 
> ![IMG_1835](https://private-user-images.githubusercontent.com/32161518/344392903-80ee4ddf-bdb7-4bdb-a447-10fe893ff8d6.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTk3OTUzMDYsIm5iZiI6MTcxOTc5NTAwNiwicGF0aCI6Ii8zMjE2MTUxOC8zNDQzOTI5MDMtODBlZTRkZGYtYmRiNy00YmRiLWE0NDctMTBmZTg5M2ZmOGQ2LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA3MDElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNzAxVDAwNTAwNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTc5MjNhYTJmNmZkNGQxZTE3NWZlMzM1MTRmYzI0Njc5YmZkMGJmZTM2M2I4ZDM5NDcwYTViNzg5YTMzN2I1NzImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.5vnVuTCcnmS92f2TFx82AnquN-Zgj9gumrJ0kbW2Rz8) ![IMG_1836](https://private-user-images.githubusercontent.com/32161518/344392906-209274e1-aaac-4f70-bf44-a83f8a8bf2e7.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTk3OTUzMDYsIm5iZiI6MTcxOTc5NTAwNiwicGF0aCI6Ii8zMjE2MTUxOC8zNDQzOTI5MDYtMjA5Mjc0ZTEtYWFhYy00ZjcwLWJmNDQtYTgzZjhhOGJmMmU3LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA3MDElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNzAxVDAwNTAwNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTQ4YTQwNTNjYzM0MDRhZGQ1MzJjMzU1MjE5M2MxM2E4NGJjODIyYTY2MmQ2M2JjMWQ2NmU4YjcxZGU1ZjI4NzkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.dkrAKkn6BLsq4gKeSXcSDPG6CRBv6wOolw3vRklyK-E)

Hi @pssturges can you post any updates from that thread here? I cant find anything online about Tesla implementing OCPP, I though it would be bigger news than that

pssturges, 2024-07-01:
No, unfortunately

andig, 2024-07-03:
Not considered since we don't need it. We're getting the state via TWC.

Btw: beautiful dashboard!

tapwork, 2024-07-03:
thanks ;-) 

Right. Wallgen is also an option. understand you always get the charge state from the charger. makes sense

andig, 2024-07-06:
Excellent, thank you all for your patience!

spubanz, 2024-07-09:
Hi,
first off, thanks for the solution with the BLE command proxy - works like a charme for our Model 3!
However, since our other car - a legacy Model X does not support the BLE key, I'm pretty much still facing the API call limit issue with that one.
Are there still efforts to get in touch with Tesla about other plans that allow more API calls? That would help all the legacy Modle S & X drivers.
Thanks again!

andig, 2024-07-10:
Yes, but Tesla doesn‘t answer at this time…

User47873S, 2024-07-11:
Ich schlage vor Tesla mit der Anfrage zu überfluten das die APIs wieder freischalten. Das muss teurer und aufwendiger sein die Kunden Beschwerden zu bearbeiten als die Serverkosten oder warum es gedrosselt wurde. Am besten mit Papier Briefen damit kein bot es erledigen kann.

spubanz, 2024-07-12:
Hi again,
in the mean time I found a workarround that runs on HomeAssistant and the old API, thus it works without BLE and on all Tesla vehicles.
I'm using the Tesla Custom Integration ([https://github.com/alandtse/tesla](url)) in combination with the REST API and simply set up a custom vehicle in evcc.
```yaml
vehicles:
  - name: modelX
    type: custom
    title: Hotblack Desiato
    capacity: 100
    icon: car
    chargeenable: # start/stop charging
        source: http
        uri: "http://MY_HA_IP:8123/api/services/switch/{{if .chargeenable}}turn_on{{else}}turn_off{{end}}"
        method: POST
        headers:
            - authorization: Bearer MY_HA_ACCESS_TOKEN
            - content-type: application/json
        body: '{"entity_id": "switch.MY_CAR_charger"}'
        jq: .state
    maxcurrent: # set charger max current (A)
        source: http
        uri: http://MY_HA_IP:8123/api/services/number/set_value
        method: POST
        headers:
            - authorization: Bearer MY_HA_ACCESS_TOKEN
            - content-type: application/json
        body: '{"entity_id": "number.MY_CAR_charging_amps", "value": "{{.maxcurrent}}"}'
        jq: .state | tonumber
    wakeup: # vehicle wake up command
        source: http
        uri: http://MY_HA_IP:8123/api/services/button/press
        method: POST
        headers:
            - authorization: Bearer MY_HA_ACCESS_TOKEN
            - content-type: application/json
        body: '{"entity_id": "button.MY_CAR_wake_up"}'
        jq: .state
    status: # get vehicle connection status
        source: combined
        plugged: # plugged in?
            source: http
            uri: http://MY_HA_IP:8123/api/states/binary_sensor.MY_CAR_charger
            method: GET
            headers:
                - authorization: Bearer MY_HA_ACCESS_TOKEN
                - content-type: application/json
            jq: .state
        charging: # charging?
            source: http
            uri: http://MY_HA_IP:8123/api/states/binary_sensor.MY_CAR_charging
            method: GET
            headers:
                - authorization: Bearer MY_HA_ACCESS_TOKEN
                - content-type: application/json
            jq: .state
    soc: # battery soc (%)
        source: http
        uri: http://MY_HA_IP:8123/api/states/sensor.MY_CAR_battery
        method: GET
        headers:
            - authorization: Bearer MY_HA_ACCESS_TOKEN
            - content-type: application/json
        jq: .state | tonumber
    range: # electric range (km)
        source: http
        uri: http://MY_HA_IP:8123/api/states/sensor.MY_CAR_range
        method: GET
        headers:
            - authorization: Bearer MY_HA_ACCESS_TOKEN
            - content-type: application/json
        jq: .state | tonumber
    limitsoc: # charge limit set by car (%)
        source: http
        uri: http://MY_HA_IP:8123/api/states/number.MY_CAR_charge_limit
        method: GET
        headers:
            - authorization: Bearer MY_HA_ACCESS_TOKEN
            - content-type: application/json
        jq: .state | tonumber
    odometer: # milage (km)
        source: http
        uri: http://MY_HA_IP:8123/api/states/sensor.MY_CAR_odometer
        method: GET
        headers:
            - authorization: Bearer MY_HA_ACCESS_TOKEN
            - content-type: application/json
        jq: .state | tonumber
```
The important pice is that you use the custom integration whithout the fleet API.

Maybe that helps someone out there.

Cheers!

andig, 2024-07-12:
> The important pice is that you use the custom integration whithout the fleet API.

That seems pretty impossible?

spubanz, 2024-07-12:
> > The important pice is that you use the custom integration whithout the fleet API.
> 
> That seems pretty impossible?

Well, when you install the plug in in home assistant it shows you a checkmark box "use fleet API" that you can leave unchecked. If you do so you can enter a refresh token on the next page that you can generate as described in the installation instructions. 

With that it works for me. Frankly I'm not able to tell what really goes on in the background since I'm not that deep in the topic.

[16 more replies]