evcc-io/evcc #14228: Offline-Vehicle mit RFID-Identifiers werden falsch/gar nicht erkannt

TobiasHuber1980, 2024-06-07:
### Describe the bug

Ich möchte die Wallbox per RDID freischalten, in evcc soll dann das dem Identifier entsprechende Offline-Vehicle (=Mitarbeiter) angezeigt werden.

### Steps to reproduce

1. Keba Wallbox mit RFID-Karte von "employee1" freischalten, Auto anstecken.
2. evcc erkennt zuerst unknown und ändert dann automatisch auf ein anderes Auto.
3. [lp-3  ] INFO 2024/06/07 12:44:58 vehicle updated: unknown -> VW ID.4 (Erich Huber)
...

### Configuration details

```yaml
vehicles:
  - name: car1
    type: template
    template: vw
    title: VW ID.5 (Tobias Huber)
    user: xxx
    password: xxx
    vin: WVGZZZE2ZPPxxx
    capacity: 82
    timeout: 10s
    identifiers:
    - 04115E12841191

  - name: car2
    type: template
    template: bmw
    title: BMW i3
    user: xxx
    password: xxx
    vin: WBY8P210707xxx
    capacity: 42
    identifiers:
    - 0498661A9B1190

  - name: car3
    type: template
    template: vw
    title: VW ID.4 (Erich Huber)
    user: xxx
    password: xxx
    vin: WVGZZZE2ZPExxx
    capacity: 82
    timeout: 10s
    identifiers:
    - 04AA091A9B1190

  - name: employee1
    title: Hubert Hoffmann
    type: template
    template: offline
    identifiers:
    - 0402571A9B1191
```

### Log details

```text
[main  ] INFO 2024/06/07 12:43:48 evcc 0.126.6
[main  ] INFO 2024/06/07 12:43:48 using config file: /etc/evcc.yaml
[main  ] INFO 2024/06/07 12:43:48 starting ui and api at :7070
[db    ] INFO 2024/06/07 12:43:51 using sqlite database: /var/lib/evcc/evcc.db
[lp-1  ] DEBUG 2024/06/07 12:43:55 charge total import: 852.669kWh
[lp-2  ] DEBUG 2024/06/07 12:43:55 charge total import: 3081.727kWh
[lp-3  ] DEBUG 2024/06/07 12:43:55 charge total import: 44.263kWh
[site  ] WARN 2024/06/07 12:43:55 vehicle 'Hubert Hoffmann' does not support automatic detection
[site  ] INFO 2024/06/07 12:43:55 site config:
[... 191 lines cut ...]
[site  ] DEBUG 2024/06/07 12:48:25 grid meter: -33009W
[site  ] DEBUG 2024/06/07 12:48:25 site power: -33009W
[lp-1  ] DEBUG 2024/06/07 12:48:25 charge total import: 852.669kWh
[lp-1  ] DEBUG 2024/06/07 12:48:25 charger status: A
[site  ] DEBUG 2024/06/07 12:48:55 ----
[lp-1  ] DEBUG 2024/06/07 12:48:55 charge power: 0W
[lp-1  ] DEBUG 2024/06/07 12:48:55 charge currents: [0 0 0]A
[lp-2  ] DEBUG 2024/06/07 12:48:55 charge power: 0W
[lp-2  ] DEBUG 2024/06/07 12:48:55 charge currents: [0 0 0]A
[lp-3  ] DEBUG 2024/06/07 12:48:55 charge power: 10365W
[lp-3  ] DEBUG 2024/06/07 12:48:55 charge currents: [16 15.5 14.5]A
[site  ] DEBUG 2024/06/07 12:48:55 pv power: 45630W
[site  ] DEBUG 2024/06/07 12:48:55 grid meter: -19163W
[site  ] DEBUG 2024/06/07 12:48:55 site power: -19163W
[lp-2  ] DEBUG 2024/06/07 12:48:55 charge total import: 3081.727kWh
[lp-2  ] DEBUG 2024/06/07 12:48:55 charger status: A
```

### What type of operating system are you running?

Linux

### Version

0.126.6

0x3d13f, 2024-06-07:
Ich habe bei meiner Keba festgestellt, dass nicht die ganze Kennung des RFID Tags bei evcc ankommt.
z.B. musst ich folgende Änderung vornehmen (man sieht, dass nur der erste Teil der RFID Kennung in die evcc.yaml muss:
```
identifiers:
  - "047dafba"
  #- "047dafba7f823100"  
```

TobiasHuber1980, 2024-06-07:
ich teste montag....

GrimmiMeloni, 2024-06-07:
Der Hinweis von @0x3d13f passt auch zum Verhalten. Wenn nur die ersten 8 Zeichen berücksichtigt werden, haben MA 1 und 3 die gleiche RFID Kennung.

andig, 2024-06-07:
Falls etwas vom Charger käme sollte im Logfile ein

    charger vehicle id: xyz

zu sehen sein- ist hier aber nicht zu erkennen. Im Zweifel mal die WB auf `trace` stellen und nochmal probieren.

premultiply, 2024-06-07:
Da scheint Keba vielleicht auch Mist gemacht haben beim Modbus-Layout:
https://github.com/evcc-io/evcc/blob/c35566cd48b6c9a5f2ec3fcc8dc564bfbc37ef21/charger/keba-modbus.go#L264

Es sind via nur 2 Register (=32 Bit) vorgesehen.

andig, 2024-06-07:
Also 1: schauen ob die überhaupt stimmen und 2: schauen ob `evcc charger` etwas sieht und 3: schauen ob `Identify()` überhaupt aufgerufen wird.

TobiasHuber1980, 2024-06-10:
Habe meinen! Fehler gefunden. Die Einrückungen in der yaml bei den Identifiers waren nicht korrekt. 

So ist es korrekt:
```
 identifiers:
   - 040c121a
```

```
[lp-2 ] DEBUG 2024/06/10 11:44:00 charger vehicle id: 040c121a
[lp-2 ] INFO 2024/06/10 11:44:00 vehicle updated: unknown -> WS-HF99E
[lp-2 ] DEBUG 2024/06/10 11:44:00 pv charge current: 11.5A = 0A + 11.5A (-7921W @ 3p)
[lp-2 ] DEBUG 2024/06/10 11:44:00 site power -7921W <= 0W enable threshold
[lp-2 ] DEBUG 2024/06/10 11:44:00 pv enable in 0s
[lp-2 ] DEBUG 2024/06/10 11:44:00 pv enable timer elapsed
[lp-2 ] DEBUG 2024/06/10 11:44:00 max charge current: 6A
[lp-2 ] DEBUG 2024/06/10 11:44:00 charger enable
[lp-2 ] DEBUG 2024/06/10 11:44:00 wake-up timer: start
[lp-2 ] DEBUG 2024/06/10 11:44:29 charge power: 4019W
[lp-2 ] DEBUG 2024/06/10 11:44:29 charge currents: [5.93 5.99 5.99]A
[lp-2 ] DEBUG 2024/06/10 11:44:59 charge power: 4006W
[lp-2 ] DEBUG 2024/06/10 11:44:59 charge currents: [5.91 5.98 5.96]A
[lp-2 ] DEBUG 2024/06/10 11:45:29 charge power: 4027W
[lp-2 ] DEBUG 2024/06/10 11:45:29 charge currents: [5.91 6 5.96]A
[lp-2 ] DEBUG 2024/06/10 11:45:30 charge total import: 3082.246kWh
[lp-2 ] DEBUG 2024/06/10 11:45:30 charger status: C
[lp-2 ] DEBUG 2024/06/10 11:45:30 charge start energy: 3.08e+03kWh
[lp-2 ] INFO 2024/06/10 11:45:30 start charging ->
```

andig, 2024-06-10:
Schwer zu glauben. In dem Fall hätte evcc nicht starten dürfen.

TobiasHuber1980, 2024-06-10:
tut es aber :-) - evcc startet problemlos mit:

```
  - name: fahrschule0
    title: WS-HF120E
    type: template
    template: offline
    identifiers:
    - 0402571a

  - name: fahrschule1
    title: WS-HF121E
    type: template
    template: offline
    identifiers:
    - 04a21e1a
```

andig, 2024-06-10:
Das sieht für mich auch richtig aus. Ein

    evcc vehicle

sollte zeigen ob alles korrekt erkannt wurde. Gibts da einen Unterschied zwischen den Varianten?

TobiasHuber1980, 2024-06-10:
so kommen die identifiers in evcc auch an, also `[lp-2 ] DEBUG 2024/06/10 11:44:00 charger vehicle id: 040c121a`
```
 identifiers:
   - 040c121a
```

so nicht
```
 identifiers:
 - 040c121a
```

`evcc vehicle` zeigt bei beiden varianten den identifier an.

andig, 2024-06-10:
> evcc vehicle zeigt bei beiden varianten den identifier an.

Dann sind die Varianten auch identisch. Es gibt keine andere Möglichkeit. Denn verglichen wird nicht gegen die yaml, sondern gegen die evcc-interne Repräsentation.

premultiply, 2024-06-14:
Dann bewirbt und implementiert diese Box das RFID-Feature wohl (derzeit?) nicht via Modbus.

premultiply, 2024-06-14:
https://www.keba.com/download/x/dea7ae6b84/kecontactp30modbustcp_pgen.pdf

Register 1016

`evcc charger --log trace --diagnose` sollte Klarheit bringen.

TobiasHuber1980, 2024-06-14:
> https://www.keba.com/download/x/dea7ae6b84/kecontactp30modbustcp_pgen.pdf
> 
> Register 1016
> 
> `evcc charger --log trace --diagnose` sollte Klarheit bringen.

```
pi@raspberrypi:~ $ evcc charger --log trace --diagnose
[main  ] INFO 2024/06/14 11:15:29 evcc 0.127.1
[main  ] INFO 2024/06/14 11:15:29 using config file: /etc/evcc.yaml
[db    ] INFO 2024/06/14 11:15:33 using sqlite database: /home/pi/.evcc/evcc.db
[db    ] TRACE 2024/06/14 11:15:33 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2024/06/14 11:15:33 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/06/14 11:15:33 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/06/14 11:15:33 SELECT * FROM `settings` 10 <nil>
[db    ] TRACE 2024/06/14 11:15:33 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db    ] TRACE 2024/06/14 11:15:33 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db    ] TRACE 2024/06/14 11:15:33 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2024/06/14 11:15:33 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/06/14 11:15:33 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/06/14 11:15:33 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2024/06/14 11:15:33 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "config_details" AND sql IS NOT NULL order by type = "table" desc 2 <nil>
[db    ] TRACE 2024/06/14 11:15:33 SELECT * FROM `config_details` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/06/14 11:15:33 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT ""fk_configs_details"" %" OR sql LIKE "%CONSTRAINT fk_configs_details %" OR sql LIKE "%CONSTRAINT `fk_configs_details`%" OR sql LIKE "%CONSTRAINT [fk_configs_details]%" OR sql LIKE "%CONSTRAINT         fk_configs_details      %") -1 <nil>
[db    ] TRACE 2024/06/14 11:15:33 SELECT count(*) FROM sqlite_master WHERE type = "index" AND tbl_name = "config_details" AND name = "idx_unique" -1 <nil>
[db    ] TRACE 2024/06/14 11:15:33 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT ""fk_devices_details"" %" OR sql LIKE "%CONSTRAINT fk_devices_details %" OR sql LIKE "%CONSTRAINT `fk_devices_details`%" OR sql LIKE "%CONSTRAINT [fk_devices_details]%" OR sql LIKE "%CONSTRAINT         fk_devices_details      %") -1 <nil>
[db    ] TRACE 2024/06/14 11:15:33 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%""device_id"" %" OR sql LIKE "%device_id %" OR sql LIKE "%`device_id`%" OR sql LIKE "%[device_id]%" OR sql LIKE "%   device_id       %") -1 <nil>
[db    ] TRACE 2024/06/14 11:15:33 SELECT * FROM `configs` WHERE `configs`.`class` = 1 ORDER BY id 0 <nil>
[salia ] TRACE 2024/06/14 11:15:33 GET http://192.168.1.10/api
[keba  ] TRACE 2024/06/14 11:15:34 modbus: send 00 01 00 00 00 06 ff 03 03 f8 00 02
[keba  ] TRACE 2024/06/14 11:15:34 modbus: send 00 01 00 00 00 06 ff 03 03 f8 00 02
[keba  ] TRACE 2024/06/14 11:15:34 modbus: recv 00 01 00 00 00 07 ff 03 04 00 04 cb 09
[keba  ] TRACE 2024/06/14 11:15:34 modbus: send 00 02 00 00 00 06 ff 03 06 0e 00 02
[keba  ] TRACE 2024/06/14 11:15:34 modbus: recv 00 01 00 00 00 07 ff 03 04 00 04 a3 94
[keba  ] TRACE 2024/06/14 11:15:34 modbus: send 00 02 00 00 00 06 ff 03 06 0e 00 02
[keba  ] TRACE 2024/06/14 11:15:34 modbus: recv 00 02 00 00 00 07 ff 03 04 00 00 00 00
[keba  ] TRACE 2024/06/14 11:15:34 modbus: send 00 03 00 00 00 06 ff 03 06 42 00 02
[keba  ] TRACE 2024/06/14 11:15:34 modbus: recv 00 02 00 00 00 07 ff 03 04 00 00 00 00
[keba  ] TRACE 2024/06/14 11:15:34 modbus: send 00 03 00 00 00 06 ff 03 06 42 00 02
[keba  ] TRACE 2024/06/14 11:15:34 modbus: recv 00 03 00 00 00 07 ff 03 04 00 00 00 00
[keba  ] TRACE 2024/06/14 11:15:34 modbus: recv 00 03 00 00 00 07 ff 03 04 00 00 00 00
[salia ] TRACE 2024/06/14 11:15:34 {"device":{"product":"2310007","modelname":"Salia PLCC Slave","hardware_version":"1.0","software_version":"2.1.0","vcs_version":"V0R5e","hostname":"salia","mac_address":"00:01:87:13:31:4a","serial":"101257802","uuid":"5491ad62-022a-4356-a32c-00018713314a","internal_id":"564874","ip_lo":"127.0.0.1","ip_br0:fallback":"169.254.12.53","ip_br0":"192.168.1.10"},"secc":{"port0":{"salia":{"chargemode":"manual","keyswitch":"0","heartbeat":"alive","mem":"322380","uptime":"22:25","thermal":"65061","load":"0.82","thermalmax":"69317","thermalmin":"0","modbus":{"status":"running"},"socketminamp":"6","socketmaxamp":"16","mains_batt_power":"0","mains":"25937.197","mains_l1_power":"7253.128","mains_l2_power":"9542.294","mains_l1_current":"33.376167","mains_l2_current":"43.669155","mains_batt_soc":"0","mains_l3_current":"42.803944","mains_l3_power":"9141.775","LM":{"internal":{"status":"internal lm not activated"}},"intctrl_limit":"16","chargedata":"-","ecoplus":"1","mains_type":"janitza","authmode":"free","mains_available":"1","mains_batt_status":"","mains_status":"Ok","mains_batt_available":"0","tc_enabled":"0","firmwareprogress":"0","firmwarestate":"idle","status":"manual+ -\/-\ [... cut ...] nt":"0","rfid":{"available":"0","authorizereq":""},"ev_present":"0","charging":"0","ready_for_slac":"0","cable_current_limit":"-1","ventilation":{"state":{"actual":"0"},"available":"0"},"ocpp":{"state":"Available"},"high_level":{"failure":"0"},"backend":{"availability":{"target":"operative"}},"charging_stop":"1"}}}
Power:          0W
Energy:         853.6kWh
Current L1..L3: 0A 0A 0A
Charge status:  A
Enabled:        false

Diagnostic dump:
Model name: Salia PLCC Slave
Software version: 2.1.0
```

TobiasHuber1980, 2024-06-14:
Keba P30 c-serie ist korrekt, daten stimmen mit Register 1016 überein.
`Product:        314121`

Keba P30 x-serie ist was faul.
`Product:        304020` würde bedeuten kein RFID, was aber komisch ist, denn ich kann ja die Box mit den RFID-Karten freischalten und sehe auch die Ladevorgänge im Kaba-Portal.

GrimmiMeloni, 2024-06-14:
Noch ein Hinweis dazu. Je nach Implementierung ist ein RFID Token nur ein "kurzlebiges" Event.
Die Easee z.B. unterstützt RFID, aber man kann evcc charger nicht dazu verwenden die RFID IDs zu ermitteln. (Es sei denn man schafft GENAU den Zeitpunkt zu syncen in dem evcc den Charger einmalig abfragt und gleichzeitig das Token präsentiert wird.)
Vielleicht ist es bei der Keba ähnlich. Ggf. mal im laufenden evcc den o.g. Trace auf der X-series anwerfen, und dann das Token präsentieren. Vielleicht sieht man dann was.

premultiply, 2024-06-14:
> Keba P30 x-serie ist was faul.
Product:        304020 würde bedeuten kein RFID, was aber komisch ist, denn ich kann ja die Box mit den RFID-Karten freischalten und sehe auch die Ladevorgänge im Kaba-Portal.

Dann erstmal ein Fall für den Keba-Support.
Wahrscheinlich ein Bug in der Firmware.

TobiasHuber1980, 2024-06-14:
> > Keba P30 x-serie ist was faul.
> > Product:        304020 würde bedeuten kein RFID, was aber komisch ist, denn ich kann ja die Box mit den RFID-Karten freischalten und sehe auch die Ladevorgänge im Kaba-Portal.
> 
> Dann erstmal ein Fall für den Keba-Support. Wahrscheinlich ein Bug in der Firmware.

Hab mal bei Keba ein Ticket erstellt.

premultiply, 2024-06-14:
Ich mache mal hier erstmal zu, da wir an der Stelle nichts tun können. Wenn Keba die Firmware gefixt hat sollte das automatisch funktionieren.
Sollten sonstige neue Erkenntnisse folgen gerne dann gerne hier ergänzen.

closed 2024-06-14 by premultiply

TobiasHuber1980, 2024-06-19:
So, ich bin ein klein wenig weiter...
**Keba per UDP** (P30 x-serie)
```
- type: template
  template: keba
  host: 192.168.1.31
  name: wallbox3
```
```
pi@raspberrypi:~ $ evcc charger
[main  ] INFO 2024/06/19 11:23:43 evcc 0.127.2 (5eaa90ad)
[main  ] INFO 2024/06/19 11:23:43 using config file: /etc/evcc.yaml
[db    ] INFO 2024/06/19 11:23:48 using sqlite database: /home/pi/.evcc/evcc.db
wallbox3
--------
Power:          0W
Energy:         230.1kWh
Current L1..L3: 0A 0A 0A
Charge status:  A
Enabled:        false
Identifier:     048b691284119000
```

**Keba per Modbus tcpip:** (P30 x-serie)
```
- type: template
  template: keba-modbus
  host: 192.168.1.31
  name: wallbox3
  modbus: tcpip
  id: 255
  port: 502
```
```
pi@raspberrypi:~ $ sudo systemctl stop evcc
pi@raspberrypi:~ $ evcc charger
[main  ] INFO 2024/06/19 11:22:32 evcc 0.127.2 (5eaa90ad)
[main  ] INFO 2024/06/19 11:22:32 using config file: /etc/evcc.yaml
[db    ] INFO 2024/06/19 11:22:37 using sqlite database: /home/pi/.evcc/evcc.db
wallbox3
--------
Power:          0W
Energy:         230.1kWh
Current L1..L3: 0A 0A 0A
Charge status:  A
Enabled:        false

mit udp scheint es zu funktionieren, mit tcp nicht ......
```

andig, 2024-06-19:
Vielleicht hat die X-Serie kein RFID? Sonst einfach mal das Register auslesen und schauen was drin steht. Insebsondere auch das Produkt Feature Register 1016.

premultiply, 2024-06-24:
> Hat das wiederum Auswirkungen auf evcc?

Ja. Anhand des Produktcodes werden die Features und damit die API-Schnittstellen aktiviert/deaktiviert.

> dies wurde bereits in einer neuen PDC Version 3.10.57 gefixed

Dann sollte ein Firmwareupdate das Problem lösen.

0x3d13f, 2024-06-24:
In der aktuell bei Keba downloadbären Version 1.17.2 für die P30 x-series ist laut Release Notes nur die PDC Version 3.10.53 enthalten.

TobiasHuber1980, 2024-06-24:
> In der aktuell bei Keba downloadbaren Version 1.17.2 für die P30 x-series ist laut Release Notes nur die PDC Version 3.10.53 enthalten.

das ist (leider) korrekt. ich habe bei keba nach der neueren firmware angefragt...

TobiasHuber1980, 2024-06-25:
Die Version 3.10.57 gibt es aktuell noch nicht und wird zukünftig auf der Downloadseite verfügbar sein. Zeitangaben haben wir hierzu bedauerlicherweise nicht.

TobiasHuber1980, 2024-07-04:
**Update:**
Keba hat Version 1.8.0 veröffentlicht, das Update hat ungewöhnlich lange gedauert (ca. 60 Minuten)

**Ergebnis:**
Die P30 x-Serie übermittelt Register 1016 jetzt korrekt: ` Product: 304021`

```
Diagnostic dump:
[keba  ] TRACE 2024/07/04 16:52:43 modbus: send 00 0c 00 00 00 06 ff 03 03 f6 00 02
[keba  ] TRACE 2024/07/04 16:52:43 modbus: recv 00 0c 00 00 00 07 ff 03 04 01 a9 d7 bc
        Serial: 27908028
[keba  ] TRACE 2024/07/04 16:52:43 modbus: send 00 0d 00 00 00 06 ff 03 03 fa 00 02
[keba  ] TRACE 2024/07/04 16:52:43 modbus: recv 00 0d 00 00 00 07 ff 03 04 03 0a 39 00
        Firmware:       3.10.57
[keba  ] TRACE 2024/07/04 16:52:43 modbus: send 00 0e 00 00 00 06 ff 03 03 f8 00 02
[keba  ] TRACE 2024/07/04 16:52:43 modbus: recv 00 0e 00 00 00 07 ff 03 04 00 04 a3 95
        Product:        304021
[keba  ] TRACE 2024/07/04 16:52:43 modbus: send 00 0f 00 00 00 06 ff 03 06 0e 00 02
[keba  ] TRACE 2024/07/04 16:52:43 modbus: recv 00 0f 00 00 00 07 ff 03 04 00 00 00 00
        Phases source:  0
[keba  ] TRACE 2024/07/04 16:52:43 modbus: send 00 10 00 00 00 06 ff 03 06 10 00 02
[keba  ] TRACE 2024/07/04 16:52:43 modbus: recv 00 10 00 00 00 07 ff 03 04 00 00 00 00
        Phases state:   0
[keba  ] TRACE 2024/07/04 16:52:43 modbus: send 00 11 00 00 00 06 ff 03 06 42 00 02
[keba  ] TRACE 2024/07/04 16:52:43 modbus: recv 00 11 00 00 00 07 ff 03 04 00 00 00 00
        Failsafe timeout:       0s
```
jetzt funktioniert rfid & modbus tcpip :-)

premultiply, 2024-07-05:
Jetzt fehlt nur noch die zweite Hälfte der RFID-Kennung via Modbus...