evcc-io/evcc #14237: Tesla Client timeout and and current mismatch errors

infinus, 2024-06-07:
### Describe the bug

Recently updated Tesla to 2024.14.9 version and EVCC is having errors not sure if related but updated last night and this morning having issues.  

1st issue: I can see the vehicle status but cannot control charging. Vehicle is charging at 17A per Tesla app. 

I get a charger logic error: current mismatch error and http: request canceled (Client.Timeout exceeded while awaiting headers)

### Steps to reproduce

1. update Tesla
2.Try to control charging
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: trace

# unique installation id
plant: 5e866f7fa74360b3e83390f7677cc63bcaa6a0e5542a0cdccffec90af47b531c

interval: 10s # control cycle interval

sponsortoken: *Removed*

meters:
- type: template
  template: enphase
  usage: grid
  host: 192.168.1.225
  token: *removed*
  name: grid1
- type: template
  template: enphase
  usage: pv
  host: 192.168.1.225
  token: *removed*
  name: pv2

chargers:
- type: template
  template: twc3
  host: 192.168.1.108
  name: TWC

vehicles:
- name: Infinus
  type: template
  template: tesla
  accessToken: *removed*
  refreshToken: *removed*
  vin: 5YJ3E1EC6MF849726
  capacity: 74 # kWh
  control: true

loadpoints:
- title: Home
  charger: TWC
  vehicle: Infinus
  mode: pv

tariffs:
[... 50 lines cut ...]
[lp-1 ] WARN 2024/06/07 11:49:38 charger logic error: current mismatch (got 14A, expected 17A)
[lp-1 ] DEBUG 2024/06/07 11:49:38 pv charge current: 11.6A = 14A + -2.42A (557W @ 1p)
[tesla ] TRACE 2024/06/07 11:49:38 POST https://tesla.evcc.io/api/1/vehicles/5YJ3E1EC6MF849726/command/set_charging_amps
[tesla ] TRACE 2024/06/07 11:49:48 {"charging_amps": 11}
[lp-1 ] ERROR 2024/06/07 11:49:48 max charge current 11A: Post "https://tesla.evcc.io/api/1/vehicles/5YJ3E1EC6MF849726/command/set_charging_amps": net/http: request canceled (Client.Timeout exceeded while awaiting headers)`

Tesla Trace:

`[tesla ] TRACE 2024/06/07 11:48:12 POST https://tesla.evcc.io/api/1/vehicles/5YJ3E1EC6MF849726/command/set_charging_amps
[tesla ] TRACE 2024/06/07 11:48:22 {"charging_amps": 6}
[tesla ] TRACE 2024/06/07 11:48:25 POST https://tesla.evcc.io/api/1/vehicles/5YJ3E1EC6MF849726/command/set_charging_amps
[tesla ] TRACE 2024/06/07 11:48:35 {"charging_amps": 5}
[tesla ] TRACE 2024/06/07 11:48:38 GET https://fleet-api.prd.na.vn.cloud.tesla.com/api/1/vehicles/5YJ3E1EC6MF849726/vehicle_data
[tesla ] TRACE 2024/06/07 11:48:38 {"response":{"id":1493095239254069,"user_id":1761349,"vehicle_id":164144749364,"vin":"5YJ3E1EC6MF849726","color":null,"access_type":"OWNER","granular_access":{"hide_private":false},"tokens":["6e5293892ef311ff","0162d4c292ea8bed"],"state":"online","in_service":false,"id_s":"1493095239254069","calendar_enabled":true,"api_version":76,"backseat_token":null,"backseat_token_updated_at":null,"ble_autopair_enrolled":false,"charge_state":{"battery_heater_on":false,"battery_level":60,"battery_range":161.73,"charge_amps":17,"charge_current_request":17,"charge_current_request_max":48,"charge_enable_request":true,"charge_energy_added":2.01,"charge_limit_soc":90,"charge_limit_soc_max":100,"charge_limit_soc_min":50,"charge_limit_soc_std":80,"charge_miles_added_ideal":8.0,"charge_miles_added_rated":8.0,"charge_port_cold_weather_mode":false,"charge_port_color":"\u003cinvalid\u003e","charge_port_door_open":true,"charge_port_latch":"Engaged","charge_rate":14.9,"charger_actual_current":17,"charger_phases":1,"charger_pilot_current":48,"charger_power":4,"charger_voltage":241,"charging_state":"Charging","conn_charge_cable":"SAE","est_battery_range":160.53,"fast_charger_brand":"\u003cinvalid\u003e","fast_charger_present":false,"fast_charger_type":"ACSingleWireCAN","ideal_battery_range":161.73,"max_range_charge_counter":0,"minutes_to_full_charge":305,"not_enough_power_to_heat":null,"off_peak_charging_enabled":false,"off_peak_charging_times":"all_week","off_peak_hours_end_time":360,"preconditioning_enabled":false,"preconditioning_times":"all_week","scheduled_charging_mode":"Off","scheduled_charging_pending":false,"scheduled_charging_start_time":null,"scheduled_charging_start_time_app":0,"scheduled_departure_time":1697457600,"scheduled_departure_time_minutes":300,"supercharger_session_trip_planner":false,"time_to_full_charge":5.08,"timestamp":1717786118531,"trip_charging":false,"usable_battery_level":60,"user_charge_enable_request":null},"climate_state":{"allow_cabin_overheat_protection":true,"auto_seat_climate_left":false,"auto_seat_climate_right":false,"battery_heater":false,"battery_heater_no_power":null,"cabin_overheat_protection":"FanOnly","cabin_overheat_protection_actively_cooling":false,"climate_keeper_mode":"off","cop_activation_temperature":"High","defrost_mode":0,"driver_temp_setting":15.0,"fan_status":0,"hvac_auto_request":"On","inside_temp":36.9,"is_auto_conditioning_on":false,"is_climate_on":false,"is_front_defroster_on":false,"is_preconditioning":false,"is_rear_defroster_on":false,"left_temp_direction":0,"max_avail_temp":28.0,"min_avail_temp":15.0,"outside_temp":39.5,"passenger_temp_setting":15.0,"remote_heater_control_enabled":false,"right_temp_direction":0,"seat_heater_left":0,"seat_heater_rear_center":0,"seat_heater_rear_left":0,"seat_heater_rear_right":0,"seat_heater_right":0,"side_mirror_heaters":false,"supports_fan_only_cabin_overheat_protection":true,"timestamp":1717786118531,"wiper_blade_heater":false},"drive_state":{"power":-4,"shift_state":null,"speed":null,"timestamp":1717786118531},"gui_settings":{"gui_24_hour_time":false,"gui_charge_rate_units":"kW","gui_distance_units":"mi/hr","gui_range_display":"Rated","gui_temperature_units":"F","gui_tirepressure_units":"Psi","show_range_units":false,"timestamp":1717786118531},"parked_accessory":{"aux_park_lamps":"NaPremium","badge_version":0,"can_accept_navigation_requests":true,"can_actuate_trunks":true,"car_special_type":"base","car_type":"model3","charge_port_type":"US","cop_user_set_temp_supported":true,"dashcam_clip_save_supported":true,"default_charge_to_max":false,"driver_assist":"TeslaAP3","ece_restrictions":false,"efficiency_package":"M32021","eu_vehicle":false,"exterior_color":"PearlWhite","exterior_trim":"Black","exterior_trim_override":"","has_air_suspension":false,"has_ludicrous_mode":false,"has_seat_cooling":false,"headlamp_type":"Premium","interior_trim_type":"White2","key_version":2,"motorized_charge_port":true,"paint_color_override":"253,255,231,0.7,0.04","performance_package":"Performance","plg":true,"pws":true,"rear_drive_unit":"PM216MOSFET","rear_seat_heaters":1,"rear_seat_type":0,"rhd":false,"roof_color":"RoofColorGlass","seat_type":null,"sentry_preview_supported":true,"spoiler_type":"Passive","sun_roof_installed":null,"supports_qr_pairing":false,"third_row_seats":"None","timestamp":1717786118532,"trim_badging":"p74d","use_range_badging":true,"utc_offset":-25200,"webcam_selfie_supported":true,"webcam_supported":true,"wheel_type":"UberTurbine20Gunpowder"},"vehicle_config":{"aux_park_lamps":"NaPremium","badge_version":0,"can_accept_navigation_requests":true,"can_actuate_trunks":true,"car_special_type":"base","car_type":"model3","charge_port_type":"US","cop_user_set_temp_supported":true,"dashcam_clip_save_supported":true,"default_charge_to_max":false,"driver_assist":"TeslaAP3","ece_restrictions":false,"efficiency_package":"M32021","eu_vehicle":false,"exterior_color":"PearlWhite","exterior_trim":"Black","exterior_trim_override":"","has_air_suspension":false,"has_ludicrous_mode":false,"has_seat_cooling":false,"headlamp_type":"Premium","interior_trim_type":"White2","key_version":2,"motorized_charge_port":true,"paint_color_override":"253,255,231,0.7,0.04","performance_package":"Performance","plg":true,"pws":true,"rear_drive_unit":"PM216MOSFET","rear_seat_heaters":1,"rear_seat_type":0,"rhd":false,"roof_color":"RoofColorGlass","seat_type":null,"sentry_preview_supported":true,"spoiler_type":"Passive","sun_roof_installed":null,"supports_qr_pairing":false,"third_row_seats":"None","timestamp":1717786118532,"trim_badging":"p74d","use_range_badging":true,"utc_offset":-25200,"webcam_selfie_supported":true,"webcam_supported":true,"wheel_type":"UberTurbine20Gunpowder"},"vehicle_state":{"api_version":76,"autopark_state_v3":"unavailable","calendar_supported":true,"car_version":"2024.14.9 0b82f700aebe","center_display_state":0,"dashcam_clip_save_available":true,"dashcam_state":"Recording","df":0,"dr":0,"fd_window":0,"f [... cut ...] alse,"tpms_soft_warning_rl":false,"tpms_soft_warning_rr":false,"valet_mode":false,"valet_pin_needed":false,"vehicle_name":"Infinus","vehicle_self_test_progress":0,"vehicle_self_test_requested":false,"webcam_available":true}}}
[tesla ] TRACE 2024/06/07 11:48:38 POST https://tesla.evcc.io/api/1/vehicles/5YJ3E1EC6MF849726/command/set_charging_amps
[tesla ] TRACE 2024/06/07 11:48:48 {"charging_amps": 13}`
```

### What type of operating system are you running?

Docker container

### Version

0.126.6

TobiasHuber1980, 2024-06-07:
Siehe https://github.com/evcc-io/evcc/issues/14226

infinus, 2024-06-07:
I do not see error 429 Too Many Requests in any of my logs so I'm not sure if it is related.

Likemike24, 2024-06-08:
Had the same issue and short after that I got 429 too many requests. It must be related.

infinus, 2024-06-08:
You are right maybe an hour or two after this started I got the 429 also.

closed 2024-06-08 by andig