evcc-io/evcc #14241: Passwort

Mestro82, 2024-06-08:
### Describe the bug

Hallo zusammen,
ich habe eine Passwort mit dem Zeichen @ verwendet. Allerdings kann ich mich damit nicht einloggen und auch Set oder Reset funktioniert nicht. Auch im Log kann ich den Vorgang Reset nicht sehen. Führt das "Sonderzeichen" zu Problemen?

### Steps to reproduce

1. Set um neues PW einzugeben (mehrmals)
2. Reset (mehrmals) -> es kommt keine Abfrage bei GUI Start

### Configuration details

```yaml
# unique installation id
plant: abc

network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: evcc.local
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 30s # control cycle interval. Interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval

# database configuration for persisting charge sessions and settings
# database:
#   type: sqlite
#   dsn: <path-to-db-file>

# sponsor token enables optional features (request at https://sponsor.evcc.io)
# sponsortoken:

# telemetry enables aggregated statistics
#
# Telemetry allows collecting usage data (grid and green energy, charge power).
# Data is aggregated, no individual charging sessions are tracked. The collected,
# anonymous data can be retrieved using https://api.evcc.io.
#
# See https://github.com/evcc-io/evcc/pull/4343 or details.
#
# For time being, this is only available to sponsors, hence data is associated with
# the sponsor token's identity.
#
# telemetry: true

# log settings
log: info
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  cache: error
  db: error

# modbus proxy for allowing external programs to reuse the evcc modbus connection
# each entry will start a proxy instance at the given port speaking Modbus TCP and
# relaying to the given modbus downstream device (either TCP or RTU, RS485 or TCP)
modbusproxy:
  #  - port: 5200
  #    uri: solar-edge:502
  #    # rtu: true
  #    # readonly: true # use `deny` to raise modbus errors

# meter definitions
[... 211 lines cut ...]
Jun 08 07:42:01 raspi evcc[47725]: [lp-1  ] DEBUG 2024/06/08 07:42:01 set smart cost limit: 5
Jun 08 07:42:01 raspi evcc[47725]: [lp-1  ] DEBUG 2024/06/08 07:42:01 phase timer inactive
Jun 08 07:42:01 raspi evcc[47725]: [lp-1  ] DEBUG 2024/06/08 07:42:01 pv timer inactive
Jun 08 07:42:01 raspi evcc[47725]: [lp-1  ] INFO 2024/06/08 07:42:01 vehicle updated: unknown -> VW e-Up!
Jun 08 07:42:01 raspi evcc[47725]: [site  ] DEBUG 2024/06/08 07:42:01 ----
Jun 08 07:42:01 raspi evcc[47725]: [lp-1  ] DEBUG 2024/06/08 07:42:01 charge power: 0W
Jun 08 07:42:02 raspi evcc[47725]: [site  ] DEBUG 2024/06/08 07:42:02 pv power: 1795W
Jun 08 07:42:02 raspi evcc[47725]: [site  ] DEBUG 2024/06/08 07:42:02 battery soc: 28%
Jun 08 07:42:02 raspi evcc[47725]: [site  ] DEBUG 2024/06/08 07:42:02 battery power: -708W
Jun 08 07:42:02 raspi evcc[47725]: [site  ] DEBUG 2024/06/08 07:42:02 grid meter: -0W
Jun 08 07:42:02 raspi evcc[47725]: [site  ] DEBUG 2024/06/08 07:42:02 site power: -608W
Jun 08 07:42:02 raspi evcc[47725]: [lp-1  ] DEBUG 2024/06/08 07:42:02 charger status: B
Jun 08 07:42:02 raspi evcc[47725]: [lp-1  ] INFO 2024/06/08 07:42:02 car connected
Jun 08 07:42:02 raspi evcc[47725]: [lp-1  ] DEBUG 2024/06/08 07:42:02 pv timer elapse
Jun 08 07:42:02 raspi evcc[47725]: [lp-1  ] DEBUG 2024/06/08 07:42:02 pv timer inactive
Jun 08 07:42:02 raspi evcc[47725]: [site  ] DEBUG 2024/06/08 07:42:02 set battery mode: normal
```

### What type of operating system are you running?

Linux

### Version

0.126.6

naltatis, 2024-06-08:
Hm, ich kann das nicht reproduzieren. Sonderzeichen sind unproblematisch. Ich hab das gerade noch mal mit `@`, anderen Sondernzeichen und Emoji 😄 ausprobiert, sollte keinen Unterschied machen.

> auch Set oder Reset funktioniert nicht.

Kannst du mal genauer beschreiben, was bei dir passiert, welche Schritte du gehst und welches Verhalten du beobachtest?

Zur Sicherheit: Du verwendest evcc direkt. Nicht in einem iframe oder durch irgend eine Art Reverse Proxy?

naltatis, 2024-06-08:
> Reset (mehrmals) -> es kommt keine Abfrage bei GUI Start

Die Abfrage kommt erst, wenn du in die Einstellungen (Menü > Konfiguration) gehst. Die Haupt-UI ist aktuell immer ohne Authentifizierung aufrufbar.

Mestro82, 2024-06-08:
Wenn ich die Konfiguration aufrufe kommt nur das Fenster „Passwort“ aber nicht „neues Passwort“ vergeben… ich verwende EVCC direkt (bin auch direkt auf dem Raspi)… hatte dann im Verdacht dass wenn ich noch an der yaml datei bin, diese geschlossen… aber weder Set noch Reset funktionieren… gibt es einem
Log der das vll aufnimmt?

Mestro82, 2024-06-08:
EVCC vorher gestoppt… EVCC vorher gestartet… beides mal beides probieren…

Wenn ihr geht ihr bei Reset vor?

naltatis, 2024-06-08:
Was helfen würde ist, wenn du mal einen Screenshot des Network tabs deiner Browser devtools machen könntest.

naltatis, 2024-06-08:
Zu 'evcc passwort set|reset': dafür sollte der evcc process immer vorher angehalten werden damit noch parallel in die gleiche db geschrieben wird.

Mestro82, 2024-06-08:
> Was helfen würde ist, wenn du mal einen Screenshot des Network tabs deiner Browser devtools machen könntest.

Da bräuchte ich etwas Unterstützung was ich huer genau machen muss

naltatis, 2024-06-09:
Spannend ist für diesen Fall dieser Request:

[image]

An die Chrome Dev-Tools kommst du so: https://developer.chrome.com/docs/devtools/open?hl=de

Mestro82, 2024-06-10:
[image]
[image]

naltatis, 2024-06-10:
Danke für die Screenshots. Kannst du einmal erklären, was du genau gemacht hast und was passiert ist?
Hauptseite aufgerufen. Bei Navigation auf Konfiguration einen Passwort-Dialog bekommen. Richtiges Passwort eingegeben und eine Ablehnung (ungültiges passwort) in der UI bekommen?

Mestro82, 2024-06-10:
Ja genau.. dann unsicher gewesen wg dem PW.. dann "reset" versucht.. verwundert dass keien Abfrage bzgl neuem Passwort kommt.. dann "set" mit neuem Passwort... weiterhin die Fehlermeldung.. kann ich EVCC komplett zurücksetzen?

naltatis, 2024-06-13:
@Mestro82 Du kannst du Datenbank-Datei löschen (o. umbenennen). Der Ort siehst du beim Hochfahren im Log.

TobiasSt78, 2024-06-13:
Ich habe das gleiche Problem auch ohne Sonderzeichen. Das eingegebene Passwort wird nicht erkannt, password reset und password set hat beides keine Auswirkung. den evcc Dienst hatte ich vorher gestoppt.
Gleiches Symptom: Es kommt zwar die Passwortabfrage aber es wird kein Passwort akzeptiert. 

Die DevTools sehen bei mir ähnlich unspektakulär aus:
[image]

`tobias@pi5htr:~ $ sudo systemctl stop evcc
tobias@pi5htr:~ $ evcc password set
[main  ] INFO 2024/06/13 21:57:33 evcc 0.126.6
[main  ] INFO 2024/06/13 21:57:33 using config file: /etc/evcc.yaml
[db    ] INFO 2024/06/13 21:57:33 using sqlite database: /home/tobias/.evcc/evcc.db
[mqtt  ] INFO 2024/06/13 21:57:33 connecting evcc-1649255431 at tcp://localhost:1883
[eebus ] INFO 2024/06/13 21:57:33 Local SKI:  1d6492b269e9503fb3d979a2d0887cdd5bbe70db
? Password [? for help] *******************
tobias@pi5htr:~ $ sudo systemctl start evcc
tobias@pi5htr:~ $ sudo evcc password reset
[main  ] INFO 2024/06/13 21:58:54 evcc 0.126.6
[main  ] INFO 2024/06/13 21:58:54 using config file: /etc/evcc.yaml
[db    ] INFO 2024/06/13 21:58:54 using sqlite database: /root/.evcc/evcc.db
[mqtt  ] INFO 2024/06/13 21:58:54 connecting evcc-1890168441 at tcp://localhost:1883
[eebus ] INFO 2024/06/13 21:58:54 Local SKI:  1d6492b269e9503fb3d979a2d0887cdd5bbe70db
? Are you sure? Yes
tobias@pi5htr:~ $ sudo systemctl start evcc
`

Mestro82, 2024-06-14:
Ich habe festgestellt, wenn die YAML einen Fehler produziert, wird wieder ein neues Passwort abgefragt. 

Ich hatte nun final meinen Raspi neu aufgesetzt und jetzt funktioniert auch mein Passwort mit @ problemlos

closed 2024-06-14 by Mestro82

naltatis, 2024-06-14:
@Mestro82 ich konnte es jetzt reproduzieren. Danke für die Hinweise.