evcc-io/evcc #14250: Sofar Solar ME3000-SP Template

luekii, 2024-06-08:
### Describe the bug

Hallo,
ich habe einen Sofar Solar ME3000-SP  (Speicher Wechselrichter) via USB-RS485 Adapter an einem Raspi mit evcc v0.126.6 angebunden.
Die Kommunikation funktioniert soweit, nur sind die Werte teilweise fehlerhaft.

battery3
--------
Power: 30W
Soc:   90%

Der Soc Wert stimmt, aber Power entspricht nicht der Lade- bzw. Entladeleistung, sondern dem Ladestrom der Akkus in Ampere.

Ich denke Mal, dass das Template "sofarsolar" für den ME3000-SP nicht passt. Das Template "sofarsolar-g3" habe ich auch getestet, das funktioniert garnicht.

Beste Grüße 
Markus 

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
...
```

### Log details

```text
...
```

### What type of operating system are you running?

Linux

### Version

0.126.6

andig, 2024-06-08:
Gibts denn eine Doku für Dein Gerät? Ohne gibts keine neuen Erkenntnisse und damit wontfix.

luekii, 2024-06-09:
Es scheint wohl diese Doku zu sein, das ist das Einzige was ich dazu finde.

https://www.google.com/url?sa=t&source=web&rct=j&opi=89978449&url=https://community.openenergymonitor.org/uploads/short-url/dF8j79PsvtCS0xQ5aPxHXO4HZDE.pdf&ved=2ahUKEwiZo4P1j8yGAxWt9AIHHRccB4QQFnoECA4QAQ&usg=AOvVaw3vgnBRGH3PFd6Bj8C6_Pfv

andig, 2024-06-09:
Dummer Fehler. Closed in https://github.com/evcc-io/evcc/commit/ebaa42f405b64b119e91228f50d55db619228967. Komisch, dass das vor Dir noch niemand bemerkt hat...

luekii, 2024-06-17:
Hallo nochmal,
grundsätzlich funktioniert es jetzt, danke dafür!
Allerdings scheint da noch etwas vertauscht zu sein, denn sobald der Speicher endlädt wird die entsprechende Leistung bei "Batterie lädt" angezeigt und umgekehrt.
Oder kann ich da etwas falsch gemacht haben?

Grüße Markus

premultiply, 2024-06-17:
Dann wäre auch noch das Vorzeichen falsch.

luekii, 2024-06-17:
[site ] DEBUG 2024/06/17 21:33:46 ----
[lp-1 ] DEBUG 2024/06/17 21:33:46 charge power: 0W
[site ] DEBUG 2024/06/17 21:33:46 battery soc: 96%
[site ] DEBUG 2024/06/17 21:33:46 battery power: -270W
[site ] DEBUG 2024/06/17 21:33:46 grid meter: 2W
[site ] DEBUG 2024/06/17 21:33:46 grid powers: [215 121 -334]W
[site ] DEBUG 2024/06/17 21:33:46 grid currents: [1.86 0.73 -1.71]A
[site ] DEBUG 2024/06/17 21:33:46 pv power: 0W
[site ] DEBUG 2024/06/17 21:33:46 site power: -268W

Hier noch ein Auszug aus dem Log, wenn der Speicher entlädt.

closed 2024-06-18 by andig