evcc-io/evcc #14256: aktuelles nightly startet nicht

VolkerK62, 2024-06-09:
### Describe the bug

Das Nightly mit http://github.com/evcc-io/evcc/pull/13319 bleibt beim Start hängen. Keine Fehlermeldung.
Nightly vom Vortag funktioniert ohne Probleme.

### Steps to reproduce

.

### Configuration details

```yaml
network:
  schema: http
  host: evcc.local
  port: 7070

database:
  type: sqlite
  dsn: /home/wsl/.evcc/evcc.db

interval: 10s

sponsortoken: xxx

telemetry: false

log: trace #debug

#####################################################################
meters:
- name: pv
  type: template
  template: solarlog
  usage: pv
  host: 192.168.178.23
  port: 502 

- name: grid
  type: template
  template: shelly-3em
  usage: grid
  host: 192.168.178.28

#################################################################
chargers:
- name: E-Bikes
  type: template
  template: fritzdect
  ## in Garage
  uri: http://fritz.box
  user: xxx
  password: xxx
  ain: xxx
  standbypower: 10
  integrateddevice: true
  icon: bike

#######################################################################
vehicles:
- name: bike
  type: template
  template: offline
  title: E-Bikes
  icon: bike
  capacity: 0.625
  phases: 1

#########################################################################
loadpoints:
- title: E-Bikes
  charger: E-Bikes

########################################################################
site:
- title: Haus
  meters:
    grid: grid
    pv: pv
  residualpower: 500

########################################################################
tariffs:
  currency: EUR
  grid:
    type: fixed #template #fixed # custom
    price: 0.354  # Preis je kWh Bezug

  feedin:
    type: fixed
    price: 0.1095  # Preis je kWh Einspeisung
      
######################################################################
mqtt:
  broker: 192.168.178.25:1883
  topic: wsl

#####################################################################
```

### Log details

```text
wsl@DESKTOP-7RU7VSU:~$ evcc
[main  ] INFO 2024/06/09 05:47:35 evcc 0.126.6 (133ebf4d)
[main  ] INFO 2024/06/09 05:47:35 using config file: /etc/evcc.yaml
[db    ] INFO 2024/06/09 05:47:36 using sqlite database: /home/wsl/.evcc/evcc.db
[db    ] TRACE 2024/06/09 05:47:36 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2024/06/09 05:47:36 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/06/09 05:47:36 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/06/09 05:47:36 SELECT * FROM `settings` 31 <nil>
[mqtt  ] INFO 2024/06/09 05:47:36 connecting evcc-1863941694 at tcp://192.168.178.25:1883
[mqtt  ] DEBUG 2024/06/09 05:47:36 tcp://192.168.178.25:1883 connected
[db    ] TRACE 2024/06/09 05:47:36 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db    ] TRACE 2024/06/09 05:47:36 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db    ] TRACE 2024/06/09 05:47:36 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2024/06/09 05:47:36 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/06/09 05:47:36 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/06/09 05:47:36 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2024/06/09 05:47:36 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "config_details" AND sql IS NOT NULL order by type = "table" desc 2 <nil>
[db    ] TRACE 2024/06/09 05:47:36 SELECT * FROM `config_details` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/06/09 05:47:36 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT ""fk_configs_details"" %" OR sql LIKE "%CONSTRAINT fk_configs_details %" OR sql LIKE "%CONSTRAINT `fk_configs_details`%" OR sql LIKE "%CONSTRAINT [fk_configs_details]%" OR sql LIKE "%CONSTRAINT                                                            fk_configs_details       %") -1 <nil>
[db    ] TRACE 2024/06/09 05:47:36 SELECT count(*) FROM sqlite_master WHERE type = "index" AND tbl_name = "config_details" AND name = "idx_unique" -1 <nil>
[db    ] TRACE 2024/06/09 05:47:36 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT ""fk_devices_details"" %" OR sql LIKE "%CONSTRAINT fk_devices_details %" OR sql LIKE "%CONSTRAINT `fk_devices_details`%" OR sql LIKE "%CONSTRAINT [fk_devices_details]%" OR sql LIKE "%CONSTRAINT                                                            fk_devices_details       %") -1 <nil>
[db    ] TRACE 2024/06/09 05:47:36 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%""device_id"" %" OR sql LIKE "%device_id %" OR sql LIKE "%`device_id`%" OR sql LIKE "%[device_id]%" OR sql LIKE "%     device_id       %") -1 <nil>
[main  ] INFO 2024/06/09 05:47:36 listening at :7070
[db    ] TRACE 2024/06/09 05:47:36 SELECT * FROM `configs` WHERE `configs`.`class` = 2 ORDER BY id 0 <nil>
[db    ] TRACE 2024/06/09 05:47:36 SELECT * FROM `configs` WHERE `configs`.`class` = 1 ORDER BY id 0 <nil>
[db    ] TRACE 2024/06/09 05:47:36 SELECT * FROM `configs` WHERE `configs`.`class` = 3 ORDER BY id 0 <nil>
[db    ] TRACE 2024/06/09 05:47:36 SELECT * FROM `configs` WHERE `configs`.`class` = 5 ORDER BY id 0 <nil>
```

### What type of operating system are you running?

Linux

### Version

0.126.6 (133ebf4d)

andig, 2024-06-09:
Bei mir auch 😫

andig, 2024-06-09:
Closed in https://github.com/evcc-io/evcc/commit/0c318f97161012002350ace22ee81a6e23bec1a0

closed 2024-06-09 by andig