evcc-io/evcc #14261: Epic: Load management enhancements

andig, 2024-06-09:
- [ ] Fail-safe mode on errors (https://github.com/evcc-io/evcc/pull/14249)
- [ ] Honor priorities (https://github.com/evcc-io/docs/pull/567)
- [ ] Support planner (https://github.com/evcc-io/docs/pull/567)

VolkerK62, 2024-06-11:
Frage: könnte man hier auch noch eine Schieflast-Überwachung mit reinnehmen?

premultiply, 2024-06-11:
Hier noch nicht, da das eine genaue Zuordnung der Phasen der ersten einzelnen Circuits und Loadpoints erfordert.

VolkerK62, 2024-06-16:
Bin mir nicht ganz sicher, ob das hierhin passt, oder ein eigenes Issue ist.
Wenn das LM "eingreift", dann werden mA-Werte als "max charge current" gesendet, auch wenn die Wallbox nur ganze A kann.
```
Jun 15 09:32:42 raspberrypi evcc[8014]: [site  ] DEBUG 2024/06/15 09:32:42 ----
Jun 15 09:32:43 raspberrypi evcc[8014]: [lp-1  ] DEBUG 2024/06/15 09:32:43 charge power: 4599W
Jun 15 09:32:43 raspberrypi evcc[8014]: [lp-1  ] DEBUG 2024/06/15 09:32:43 charge currents: [20.3 0 0]A
Jun 15 09:32:43 raspberrypi evcc[8014]: [lp-2  ] DEBUG 2024/06/15 09:32:43 charge power: 0W
Jun 15 09:32:43 raspberrypi evcc[8014]: [lp-3  ] DEBUG 2024/06/15 09:32:43 charge power: 0W
Jun 15 09:32:43 raspberrypi evcc[8014]: [lp-4  ] DEBUG 2024/06/15 09:32:43 charge power: 0W
Jun 15 09:32:43 raspberrypi evcc[8014]: [lp-5  ] DEBUG 2024/06/15 09:32:43 charge power: 0W
Jun 15 09:32:43 raspberrypi evcc[8014]: [lp-6  ] DEBUG 2024/06/15 09:32:43 charge power: 0W
Jun 15 09:32:43 raspberrypi evcc[8014]: [circuit-carport] DEBUG 2024/06/15 09:32:43 power: 4595W
Jun 15 09:32:43 raspberrypi evcc[8014]: [circuit-carport] WARN 2024/06/15 09:32:43 over current detected: 20.3A > 20A
Jun 15 09:32:43 raspberrypi evcc[8014]: [circuit-main] DEBUG 2024/06/15 09:32:43 power: 2455.9000000000005W
Jun 15 09:32:43 raspberrypi evcc[8014]: [circuit-main] DEBUG 2024/06/15 09:32:43 current: 19.07A
Jun 15 09:32:43 raspberrypi evcc[8014]: [site  ] DEBUG 2024/06/15 09:32:43 pv power: 4577W
Jun 15 09:32:43 raspberrypi evcc[8014]: [site  ] DEBUG 2024/06/15 09:32:43 battery soc: 29%
Jun 15 09:32:43 raspberrypi evcc[8014]: [site  ] DEBUG 2024/06/15 09:32:43 battery power: 1022W
Jun 15 09:32:43 raspberrypi evcc[8014]: [site  ] DEBUG 2024/06/15 09:32:43 grid meter: 2496W
Jun 15 09:32:43 raspberrypi evcc[8014]: [site  ] DEBUG 2024/06/15 09:32:43 grid powers: [-1351 -423 4270]W
Jun 15 09:32:44 raspberrypi evcc[8014]: [site  ] DEBUG 2024/06/15 09:32:44 grid currents: [-5.89 -2.09 19.1]A
Jun 15 09:32:44 raspberrypi evcc[8014]: [site  ] DEBUG 2024/06/15 09:32:44 aux power 1: 0W
Jun 15 09:32:44 raspberrypi evcc[8014]: [site  ] DEBUG 2024/06/15 09:32:44 aux power: 0W
Jun 15 09:32:44 raspberrypi evcc[8014]: [site  ] DEBUG 2024/06/15 09:32:44 site power: 3668W
Jun 15 09:32:44 raspberrypi evcc[8014]: [lp-1  ] DEBUG 2024/06/15 09:32:44 charge voltages: [226 0 0]V
Jun 15 09:32:44 raspberrypi evcc[8014]: [lp-1  ] DEBUG 2024/06/15 09:32:44 detected connected phases: 1p
Jun 15 09:32:44 raspberrypi evcc[8014]: [lp-1  ] DEBUG 2024/06/15 09:32:44 detected active phases: 1p
Jun 15 09:32:44 raspberrypi evcc[8014]: [lp-1  ] DEBUG 2024/06/15 09:32:44 !! session: chargeRater.chargedEnergy=0.1 - chargedAtStartup=0.0
Jun 15 09:32:44 raspberrypi evcc[8014]: [lp-1  ] DEBUG 2024/06/15 09:32:44 charge total import: 1865.400kWh
Jun 15 09:32:44 raspberrypi evcc[8014]: [lp-1  ] DEBUG 2024/06/15 09:32:44 charger status: C
Jun 15 09:32:44 raspberrypi evcc[8014]: [lp-1  ] DEBUG 2024/06/15 09:32:44 soc estimated: 75.12% (vehicle: 75.00%)
Jun 15 09:32:44 raspberrypi evcc[8014]: [lp-1  ] DEBUG 2024/06/15 09:32:44 vehicle soc: 75%
Jun 15 09:32:44 raspberrypi evcc[8014]: [lp-1  ] DEBUG 2024/06/15 09:32:44 vehicle soc limit: 100%
Jun 15 09:32:44 raspberrypi evcc[8014]: [lp-1  ] DEBUG 2024/06/15 09:32:44 vehicle range: 345km
Jun 15 09:32:44 raspberrypi evcc[8014]: [circuit-carport] DEBUG 2024/06/15 09:32:44 validate current: 21.2A -> 20.9A <= 20A at 20.3A: capped at 20.9A
Jun 15 09:32:44 raspberrypi evcc[8014]: [lp-1  ] DEBUG 2024/06/15 09:32:44 !! active phases: 1p = min(1p measured 0p vehicle 1p physical 0p charger)
Jun 15 09:32:44 raspberrypi evcc[8014]: [lp-1  ] DEBUG 2024/06/15 09:32:44 max charge current: 20.9A
Jun 15 09:32:47 raspberrypi evcc[8014]: [site  ] DEBUG 2024/06/15 09:32:47 ----
```

VolkerK62, 2024-06-30:
> Add circuits api to support SteuVE

Noch ein Problem entdeckt, oder ich habe einen Gedankenfehler?
2 Ladepunkte hängen am selben Kabelanschluss. Dieser Anschluss hat kein Meter.
Jeder Ladepunkt (hat sein eigenes Meter) bekommt seinen eigenen circuit. (Annahme von mir, dass darüber später die Regelung bzgl SteuVE läuft).
Beide Ladepunkte sind dem circuit "loadpoints" zugeordnet (wegen der Maximalleistung des Anschlusses).
Der circuit "loadpoints" ist somit nur eine Zusammenfassung der beiden Ladepunkte. 
Es kommt der Fehler `circuit loadpoints has no meter an no loadpoint assigned`.

```
circuits:
  - name: main
    title: Hauptsicherung
    maxCurrent: 48
  - name: loadpoints
    title: Ladepunkte
    maxCurrent: 16
    parent: main
  - name: lp1
    title: LP1
    maxCurrent: 16
    parent: loadpoints
  - name: lp2
    title: LP2
    maxCurrent: 16
    parent: loadpoints
```

VolkerK62, 2024-06-30:
>hier brauchst Du das ja nicht weil Du einfach beide Boxen an den loadpoints Circuit hängen kannst?

aktuell ja. 
Aber der Hinweis auf "SteuVE" hat mich auf diesen Gedanken gebracht. 
Diese Regulierung betrifft ja jede Wallbox einzeln und wenn die Regelung über den circuit laufen soll (was ja Sinn macht), dann braucht jede Wallbox einen eigenen.
Aber, stimmt, ist noch nicht relevant.

premultiply, 2024-06-30:
Die Dimmung erfolgt pauschal für alle SteuVE am Anschluss über die _netzrelevante_ Leistungsaufnahme.
Im Fall eines EMS gibt es dazu eine vglw. komplizierte Formel für die Summe.
Es wird dabei leider nicht nur einfach `n * 4200 kW` als SteuVE-Limit vorgegeben.

VolkerK62, 2024-06-30:
OK, "Gleichzeitigkeitsfaktor" ist das Stichwort. Danke

jeffborg, 2024-07-18:
It would be good to allow use of plugins to set the values for circuit. Not sure if that would be a start to help support SteuVE

Roeland54, 2024-07-20:
I have created an english version of the documentation page for loadbalancing. (https://github.com/evcc-io/docs/pull/584)

But I I have noticed a difference in behavior and how the documentation describes how it should work. This statement: "Alle Ladepunkte ohne explizite Stromkreiszuordnung werden diesem Hauptstromkreis zugeordnet. " I conclude from this that if I have only one circuit "main" I do not need to specify "circuit" on the loadpoint configuration. But if I try this loadmanagement does not work on mij loadpoints. I need to specify "circuit: main" on the loadpoints and then it works. 

So is this a bug in loadmanagement or just wrong documentation?

Roeland54, 2024-07-23:
In my use-case these warnings are not needed. Overcurrent is expected when we start cooking so it happens almost every day. 
I am wondering if these warning have any use to other people? Otherwise maybe they could be disabled by default?

[image]

TweetsOfNiklas, 2024-07-26:
Ich habe auch noch ein Problem in Kombination von Lastmanagement und dem Homewizard P1 Meter gefunden. Der Homewizard gibt egal ob Einspeisung oder Bezug eine positive Amperezahl aus. Bedeutet wenn Einspeisung größer als die Limitierung ist, lädt gar nichts mehr. Soll ich dafür ein eigenes Issue erstellen?

`[circuit-main] WARN 2024/07/26 11:15:46 over current detected: 21.7A > 10A
[site ] DEBUG 2024/07/26 11:15:46 pv power: 16079W
[site ] ERROR 2024/07/26 11:15:46 battery 1 soc: outdated
[site ] DEBUG 2024/07/26 11:15:46 battery soc: 0%
[site ] DEBUG 2024/07/26 11:15:46 battery power: 190W
[site ] DEBUG 2024/07/26 11:15:46 grid meter: -14162W
[site ] DEBUG 2024/07/26 11:15:46 grid currents: [21.7 21 16.6]A
[site ] DEBUG 2024/07/26 11:15:46 site power: -13822W
[lp-2 ] DEBUG 2024/07/26 11:15:46 charge voltages: [241 241 242]V
[lp-2 ] DEBUG 2024/07/26 11:15:46 detected connected phases: 3p
[lp-2 ] DEBUG 2024/07/26 11:15:47 charge total import: 1083.804kWh
[lp-2 ] DEBUG 2024/07/26 11:15:47 charger status: B
[circuit-main] DEBUG 2024/07/26 11:15:47 validate current: 21.7A + (0A -> 16A) > 10A capped at 0A
[lp-2 ] DEBUG 2024/07/26 11:15:47 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
[site ] DEBUG 2024/07/26 11:15:55 ----
[lp-1 ] DEBUG 2024/07/26 11:15:55 charge power: 0W
[lp-1 ] DEBUG 2024/07/26 11:15:55 charge currents: [0 0 0]A
[lp-2 ] DEBUG 2024/07/26 11:15:55 charge power: 5W
[lp-2 ] DEBUG 2024/07/26 11:15:56 charge currents: [0 0.0442 0]A
[lp-3 ] DEBUG 2024/07/26 11:15:56 charge power: 0W
[lp-3 ] DEBUG 2024/07/26 11:15:56 charge currents: [0 0 0]A
[circuit-outdoor] DEBUG 2024/07/26 11:15:56 power: 5.3418W
[circuit-outdoor] DEBUG 2024/07/26 11:15:56 current: 0.0442A
[circuit-main] DEBUG 2024/07/26 11:15:56 power: -13970W`

Roeland54, 2024-07-26:
This is a homewizard issue not a load management issue. So maybe better to open a new issue. 
What you see is strange because I see negative currents in the messages from the homewizard api and the docs also show negative values are possible for current. If you open a separate issue we can discus it further. (if possible use english that would make it easier for me).

andig, 2024-07-28:
> But I I have noticed a difference in behavior and how the documentation describes how it should work

@Roeland54 lets move this discussion to https://github.com/evcc-io/docs/issues/587

RenatusRo, 2024-10-23:
> It would be good to allow use of plugins to set the values for circuit. Not sure if that would be a start to help support SteuVE

Yes, would very much like to have a restful/mqtt api.

2 use cases in the frame of the flemish capacity tariff/peak shaving:

- if the intended max already has been surpassed (by whatever reason) we can lift the limit for the remaining days of the month - it has to be paid for anyway
- if we monitor the current quarter hourly peak, we can decide at minute 10 of 15 to give more power to the circuit for the remaining 5 minutes without hitting the intended limit (not really netzdienlich, but we are very liberalistic here).

see https://github.com/evcc-io/evcc/discussions/16784

andig, 2024-10-23:
This would be https://github.com/evcc-io/evcc/pull/16809. In deviation from the other plugins and given that max power/currents already are config values, this PR works by allowing the plugin to override the current value. If plugin fails, the configured value is acts as fallback.

hyperbart, 2024-10-31:
@andig Is the EVCC API documentation automated to be updated? How can I determin which endpoints and values can be used to modify this?

hyperbart, 2024-10-31:
Allright, best to wait then for the documentation, correct?

andig, 2024-10-31:
Sorry- for what?

hyperbart, 2024-10-31:
I was/am under the assumption/understanding the recent pull request https://github.com/evcc-io/evcc/pull/16809 allows us to set the maxPower on the circuit externally via the HTTP API already(?).

I wanted to start experimenting with this but couldn't find the documentation yet, but maybe I have misunderstood the current status of said functionality.

andig, 2024-10-31:
Just add https://github.com/evcc-io/evcc/pull/16809/files#diff-e558b29f57bfef72b6614f8a345c713e6fe64cd1e9032e99a7b7b998a0442271R49-R50. Docs to follow in https://github.com/evcc-io/docs/issues/658. That's not an "api" but simply configuration of the hems device.

VolkerK62, 2024-10-31:
Example:
```
circuits: # Loadmanagement
- name: main
  title: Hausanschluss
  maxCurrent: 35
  GetMaxCurrent:
    source: mqtt
    topic: Haus/circuit/maxcurrent
  maxpower: 24000
  GetMaxPower:
    source: mqtt
    topic: Haus/circuit/maxpower
  meter: grid
```

encbladexp, 2025-03-05:
Am I missing something, or don't we have an API for Loadmanagement yet?

Use case: I would like to limit the power during the day to the limit of my inverter, to avoid that my cars charge more as my PV could (theoretically) do. So my wish would be to limit the power output of all loadpoints in sum to e.g. 8 kW, even if smartcharging is used and 2 cars are plugged in at the same time.

However, during the night: I don't care, so both loadpoints should be able to use all available power (because energy is cheap) and I maybe need my car on the next day.

Simplified: Charge slower (but not too slow when only one car is connected) when by car is during the day at home, but charge fast during the night.

I know that EVCC should not die on feature creep, therefore: an API for Loadmanagement would be enough, so I could do my complex ideas on my own, but without worrying and playing around with 1P/3P charging and currents, as it would be needed right now (AFAIK).

jeffborg, 2025-03-05:
I've been doing this as above now for months no issue.

https://github.com/evcc-io/evcc/issues/14261#issuecomment-2449611775

Have demand tariff so ensuring a max power draw is critical during certain hours. If this fails limits to 2.4kw if the mqtt fails.

This does not need an api you can even use JavaScript in place of mqtt here and code up the times if that suites your use case.

encbladexp, 2025-03-06:
> This does not need an api you can even use JavaScript in place of mqtt here and code up the times if that suites your use case.

I didn't think about that in the first place, will use an HTTP source as I have an energy management backend written in Python anyways, still would prefer having an API in mid/long term.

encbladexp, 2025-03-09:
We could not discuss back and forth what an API should be, but that will not be helpful. ;)

But: GetMaxPower with an HTTP source will work for me, so I am fine.

hybe2022, 2025-06-03:
Leider wurde der Punkt Multiple HEMS fail to configure #19216](https://github.com/evcc-io/evcc/issues/19216) noch nicht in der Liste des ersten Beitrags als offener Punkt hinzugefügt. Das andere Ticket wurde allerdings bereits geschlossen und auf dieses hier sie Duplikat gesetzt.

humi208, 2025-06-14:
https://github.com/evcc-io/evcc/issues/21834#issuecomment-2972902886

@andig 
As I am not part of the team I am not allowed to edit the first comment of this ticket and add the mentioned failure to the description of this ticket.

naltatis, 2025-08-22:
@jori0001 nur zum Verständnis. Welche Infos sind dir hier wichtig?
1. 
2. Zuordnung von Ladepunkt zu Circuit (die ändert sich ja nicht)
3. Info wenn Leistung am Ladepunkt durch LM begrenzt ist
4. Übersicht über die Auslastung (also auch wenn noch alles grün ist)

jori0001, 2025-08-22:
@naltatis also aktuell würden mich die Punkte 2 und 3 (ja auch die Werte, wenn alles im grünen Bereich ist) interessieren. Ich könnte mir aber auch vorstellen, dass Punkt 1 interessant sein könnte, wenn man mehr als einen Circuit hat.
Habe allerdings keine gute Idee, wie man das schön auf dem Dashboard darstellen kann. ggf. mittels teils eingefärbten Loadpoints, um anhand der Farbe besser sehen zu können, welche Loadpoints durch einen Circuit "gruppiert" sind... wobei, soll ja auch nicht so bunt werden... Ihr habt da bestimmt ne Idee... ;)

pergolafabio, 2025-09-13:
Correct, is that being worked on?

andig, 2025-09-13:
For [Fast mode doesn't start at auto-switching phases ](https://github.com/evcc-io/evcc/issues/14827) it's not quite as simple since phase switching properly done requires timers. At time the load management is applied, all timers are already evaluated. 

What we could try as next best thing might be to pre-check the total load management limit in fast mode and- if limit is below min 3p power (or below max 1p power)- switch to 1p immediately. That would not guarantee that fast mode can be enabled at all, but it would at least give a chance. I would also ensure that we see no flip-flopping of the phase switching since both having fast charging selected and load management limit being applied should be pretty static.

/cc @premultiply @mfuchs1984

andig, 2025-09-13:
> One thing not really documented well is the use of the "planner" tariff. If you simply leave out the peak/demand times it works beautifully!

/cc @naltatis @Maschga

andig, 2025-09-13:
> This is a really hard problem to solve and for us in Australia on a demand tariff it's not worth scheduling any charging during the demand window.

@jeffborg lets not get into an OT discussion. The planner already ignores highest cost slots.

jeffborg, 2025-09-13:
> > This is a really hard problem to solve and for us in Australia on a demand tariff it's not worth scheduling any charging during the demand window.
> 
> @jeffborg lets not get into an OT discussion. The planner already ignores highest cost slots.

The problem is the slots are not the highest cost. If I pulled 7kw during this time it would almost cost an extra $100 in the month even though the per kWh rate might be 5c. I've had to remove them via the planner tariff so they are not even considered for cheap charging or planner.

pergolafabio, 2025-09-13:
> Tbo, I was only thinking about power limits, not current limits. That would help the Belgium "piektarief". Current limits have nothing to do with phase switching obviously.

Correct, it would help a lot for the "piektarief" in Belgium, would be great if evcc could switch to 1 phase charging when a max power limit is reached...

andig, 2025-09-13:
> The problem is the slots are not the highest cost. If I pulled 7kw during this time it would almost cost an extra $100 in the month even though the per kWh rate might be 5c. I've had to remove them via the planner tariff so they are not even considered for cheap charging or planner.

I see. Indeed, you can use the `planner` tariff as additional cost function if you need another cost function beyond grid, co2 and feed-in.

andig, 2025-09-13:
@pergolafabio @jeffborg could you please give https://github.com/evcc-io/evcc/pull/23571 a try?

andig, 2025-09-14:
The SHM vs. HEMS conflict has been resolved in https://github.com/evcc-io/evcc/pull/23190.

jeffborg, 2025-09-14:
@andig I've integrated #23571 have charge plan for tomorrow. but I'm not sure it's value for me as I'm only a single phase system, and at the moment we have no demand periods for 2 months so my dynamic circuit limit is always 15kw

andig, 2025-09-14:
Seems you're not the target audience, but also not experiencing a current issue with this ;) Sorry for the confusion.

jeffborg, 2025-09-14:
@andig if I limit the circuit temporarily then the charge plan does adjust, so #23571 hasn't broken things for the single phase case.

pergolafabio, 2025-09-15:
how is this delay configured, just started installing evcc

```
circuits:
  - name: main
    title: Hoofdkring
    # maxCurrent: 16 # 63A (optional)
    maxPower: 6500 # 30kW (optional)
    meter: grid

loadpoints:
  - title: Pergola - Quintens
    charger: alfen
    mode: minpv # charge mode (off, now, minpv, pv)
    circuit: main
    phases: 0
    minCurrent: 6
    maxCurrent: 16
```

andig, 2025-10-30:
Added above

MLang2008, 2025-11-24:
> > Die beiden LP hängen 1-phasig an L1/L3, so dass auf jeder Phase problemlos 16A gezogen werden können. EVCC denkt aber, es würde 32A im Lastmanagement ziehen.
> 
> Solange die beiden Boxen nicht die gleiche Zuordnung haben wird dies reibungslos und erwartungsgemäß funktionieren. Im schlimmsten Fall braucht es noch einen externen Zähler für den `circuit` um die reale Phasennutzung zu erkennen.

Kannst Du erläutern, was Du mit "Zuordnung" meinst?
Tatsächlich funktioniert es bei mir nicht, obwohl 2 unterschiedliche Phasen belastet werden.

mfuchs1984, 2026-09-17:
I think the last checkbox `Phase switching under load management` is done with https://github.com/evcc-io/evcc/pull/32353

andig, 2026-09-17:
@RenatusRo is peak management still an issue here considering https://github.com/evcc-io/evcc/pull/32881

RenatusRo, 2026-09-17:
@andig 
A very good step, but it doesn't tick all the boxes yet:
 
[Follow the peak](https://github.com/evcc-io/evcc/issues/21970) is covered in the sense of 'Set a acceptible static peak'. Following the (measured, monthly resetting) peak would need:
- setting a static start value Pt (planned target, e.g. 4.5 kW) on the 1st of each month
- monitoring the measured peak Pm from the P1 meter and raising Pt whenever Pm > Pt

At the moment this is only achievable through external automation feeding the getMaxPower plugin.

[delay for enforcing maxPower limit in circuit scenario when used for peak capacity shaving](https://github.com/evcc-io/evcc/issues/24122)

What matters for billing is the 15-minute average. A coffee machine switching on is not a reason to stop charging the car (if minimal charging power is hit for some minutes) — enforcing the limit second by second is too strict. In this scenario the circuit limit is not a fuse, but a soft cost factor.

andig, 2026-09-18:
>Following the (measured, monthly resetting) peak would need

I'm not convinced we're aiming for that. More the other way round: we're already giving you the lowest peak possible for given demand. Wdyt?

djfanatix, 2026-10-10:
Just an enhancement: the documentation says:

_Home battery [grid charging](https://docs.evcc.io/en/features/battery#gridcharge) is not considered. When the home battery actively charges from the grid, this consumption is not included in load management calculations. However, during active [§ 14a power reduction](https://docs.evcc.io/en/external-limit) signals, grid charging is automatically paused._

Could this be reconsidered? especially with the Optimizer-auto mode? So that grid charging peak power for batteries is included or could be enabled in load management.

Otherwise, When optimizer-auto is enabled, there is no way to disable grid charging.

longtom99, 2026-10-13:
Adding **Austria** to the list of countries with power-based grid fees.

From 2027, Austria introduces a new grid fee structure for all households (SNE-G-V by E-Control, the national regulator).
Each grid operator must switch between 1 Jan 2027 and 1 Jul 2028 at the latest, so this will affect basically every evcc user in Austria.

**How it works:**
- A new *Leistungspreis* (power price) is billed on the **highest 15-minute average grid import per calendar month**, at any time of day.
- According to E-Control's simulation there are two tiers, roughly €34/kW/year up to 10 kW and about double above 10 kW (final values pending).
- The share of power-based grid costs will rise step by step to about 50 %.

This is essentially the same mechanism as the Flemish capacity tariff (#21970).
An 11 kW wallbox plus a heat pump and base load easily creates monthly peaks of 15 kW or more, which will cost several hundred euros per year.

Time-of-use grid rates (summer 10–16, winter 23–05) can already be modelled with fixed tariff zones.
What is missing is native **monthly peak awareness**:
- Track the highest 15-min average grid import of the current month.
- Allow charging up to that peak without restriction, since anything below it costs nothing extra.
- Reset to a configurable baseline on the 1st of each month, with an optional hard cap (e.g. 10 kW tier boundary).
- Ideally, use a 15-min rolling average instead of instantaneous power before throttling (see #24122).

I understand the current recommendation is to set circuit `maxPower` externally via MQTT/HTTP (e.g. from Home Assistant), and that works.
But with Belgium, Norway, Australia and now Austria all moving to peak-based grid fees, a built-in solution would help a lot of users who don't run a separate automation platform.

Sources:
- [E-Control press release, 12 Oct 2026](https://www.ots.at/presseaussendung/OTS_20261012_OTS0085/e-control-neue-regeln-fuer-die-energiewende)
- [Summary of the reform (German)](https://i-magazin.com/e-control-netzentgelte-reform-2027/)

djfanatix, 2026-10-13:
_Track the highest 15-min average grid import of the current month._

This is possible with a lot of P1 meters such as dsrm, homewizard,.. so getting the data is easy.

TweetsOfNiklas, 2026-10-14:
The minimum in Austria will be 2kW. So it would be great to set it to 2kW in the summer months when charging is done almost completely by the sun. Currentlywith setting maxPower charging would not start with surplus when energy is going to the battery.
For this, also see this:
https://github.com/evcc-io/evcc/issues/33187

longtom99, 2026-10-14:
@TweetsOfNiklas good point. In #33187 it was explained that circuit `maxPower` intentionally acts as the hard limit of the grid connection and uses the raw meter value, so it can't really serve both purposes.

For peak-based tariffs it would help to have a separate **soft grid import target** (e.g. 2 kW in summer, or the current monthly peak), which only limits *grid* import and still lets PV/battery surplus charging start (respecting `prioritySoc`/`bufferSoc`), while `maxPower` stays the hard fuse.

@djfanatix agreed, the data is easy to get. In Austria the smart meter customer interface (e.g. via SHRDZM) provides grid power, so evcc could also calculate the 15-min average and the monthly peak itself.

Tommatheussen, 2026-10-14:
Belgium uses a minimum peak of 2.5kW, so a configurable (or fixed per country?) soft target is needed.
Additionally, the P1 DSMR (at least in Belgium) exposes 2 properties:
- Max peak this month
- Current peak (resets every 15 minutes)

These could be useful to extract and use for this purpose

longtom99, 2026-10-14:
@Tommatheussen good point, so the baseline should be configurable (2 kW AT, 2.5 kW BE). Using the meter's peak registers where available (DSMR) and falling back to evcc calculating the 15-min average from grid power would cover meters that don't expose them.

hyperbart, 2026-10-14:
@Tommatheussen : Yes, DSMR provides this and I use a small script to extract that value and push it to a file which evcc reads. "Current peak" I am not aware of. Which register would that be?
@longtom99 : Yes, you need 2 variabeles in this story, one variable which is the "acceptableMonthPeak" which is the peak you are willing to start at every month, if this is not set it should default to a default value and you need a variable which gets populated with the value from the meter so it can follow the peak.

@longtom99 : do you also have a smart meter which exposes these values in AT? DSMR spec?

Tommatheussen, 2026-10-14:
> Yes, DSMR provides this and I use a small script to extract that value and push it to a file which evcc reads. "Current peak" I am not aware of. Which register would that be?

Looks like `1-0:1.4.0` should be the register, found on [page 32 of the specs](https://partner.fluvius.be/sites/fluvius/files/2025-09/digital-metering-system-emucs-p1-v2-1.pdf) as well as confirmed to be implemented by [`dsmr_parser`](https://github.com/ndokter/dsmr_parser/blob/9898b362407fe63f3aca0be2f6b2b99072be6296/dsmr_parser/obis_references.py#L108)

andig, 2026-10-14:
I‘ve created https://github.com/evcc-io/evcc/issues/34840 And would like to track the capacity tariff discussion there.