evcc-io/evcc #14275: ERROR: charge total import: write tcp xxx.xxx.xxx.xxx:50934->xxx.xxx.xxx.xxx:502: write: connection reset by peer

MiniBlister, 2024-06-10:
### Describe the bug

Since 3 days I get the following connection error messages: 

ERROR: charge total import: write tcp xxx.xxx.xxx.xxx:50934->xxx.xxx.xxx.xxx:502: write: connection reset by peer
This occures every 30 seconds which is the intervall I used. Ths system was running with 10 seconds in the past w/o an issue.

My setup:
- CAR: Audi Etron
- Charger: Keba P30 / modbus sponsor version
- Fronius: Meter

### Steps to reproduce

What I did so far:
1. Update and restart all Network components (Fritzbox)
2. Update Keba Software to 1.17.2
3. Disconnect the Audi and drove the car for a few kilometers
4. Checked all network settings again. Ping + trace rt do not show any issue to get to the charger (2.48 ms ping time)
5. Disconnect Keba from Power and Reset the Charger
6. Used Wireless connections vs wired connection to Keba Charger
7. Restart EVCC Docker
8. Rebuild EVCC Docker with latest image
9. Restart Docker Hardware
...

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: evcc.local
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 30s # control cycle interval

# database configuration for persisting charge sessions and settings
# database:
#   type: sqlite
#   dsn: <path-to-db-file>

# sponsor token enables optional features (request at https://sponsor.evcc.io)
# trial token, valid until 2023-10-14
sponsortoken: xxx

# telemetry: true

# log settings
log: debug
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  cache: error
  db: error
  Keba: trace

# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
# for documentation see https://docs.evcc.io/docs/devices/meters
meters:
  - name: GEN24grid
    type: template
[... 157 lines cut ...]
[lp-1 ] DEBUG 2024/06/10 07:49:04 charge power: 0W
[lp-1 ] DEBUG 2024/06/10 07:49:04 charge currents: [0 0 0]A
[lp-1 ] DEBUG 2024/06/10 07:49:06 !! session: chargeRater.chargedEnergy=0.1 - chargedAtStartup=0.0
[lp-1 ] ERROR 2024/06/10 07:49:06 charge total import: write tcp 192.168.178.15:54614->192.168.178.50:502: write: connection reset by peer
[lp-1 ] DEBUG 2024/06/10 07:49:06 charger status: B
[lp-1 ] DEBUG 2024/06/10 07:49:06 limitSoc reached: 88.0% > 85%
[lp-1 ] DEBUG 2024/06/10 07:49:34 charge power: 0W
[lp-1 ] DEBUG 2024/06/10 07:49:34 charge currents: [0 0 0]A
[lp-1 ] DEBUG 2024/06/10 07:49:36 !! session: chargeRater.chargedEnergy=0.1 - chargedAtStartup=0.0
[lp-1 ] ERROR 2024/06/10 07:49:36 charge total import: write tcp 192.168.178.15:35642->192.168.178.50:502: write: connection reset by peer
[lp-1 ] DEBUG 2024/06/10 07:49:36 charger status: B
[lp-1 ] DEBUG 2024/06/10 07:49:36 limitSoc reached: 88.0% > 85%
[lp-1 ] DEBUG 2024/06/10 07:50:05 charge power: 0W
[lp-1 ] DEBUG 2024/06/10 07:50:05 charge currents: [0 0 0]A
[lp-1 ] DEBUG 2024/06/10 07:50:06 !! session: chargeRater.chargedEnergy=0.1 - chargedAtStartup=0.0
[lp-1 ] ERROR 2024/06/10 07:50:06 charge total import: write tcp 192.168.178.15:57416->192.168.178.50:502: write: connection reset by peer
```

### What type of operating system are you running?

Linux

### Version

0.126.6

andig, 2024-06-10:
> Since 3 days I get the following connection error messages:

What did you change 3 days ago?

MiniBlister, 2024-06-10:
This is when I updated to 0.126.6
Since than I did everything mentioend above w/o getting the issue fixed.

andig, 2024-06-10:
Please try downgrade.

MiniBlister, 2024-06-10:
just went down all the way to 0.125.0 ---no change

andig, 2024-06-10:
Then the SW change is not the cause. What else did change? Please open a Ticket with KEBA!

MiniBlister, 2024-06-10:
the only change I did with could have an impact:

I installed the following HACS:
https://github.com/marq24/ha-evcc

MiniBlister, 2024-06-10:
[image]

this is just what Home Assistant receives from EVCC via MQTT. Strange. It seems like the value is sometimes availabel and sometimes not. Even today the value was available

MiniBlister, 2024-06-10:
this helped me a lot and I found the issue.
I am running Home Aassistant and EVCC in docker containers and modbus connections to the wallbox from both dockers. After stopping the modbus connection from Home Assistant I did not get an error anymore.

closed 2024-06-10 by MiniBlister