evcc-io/evcc #14280: LG ESS per Modbus / SunSpec

cvg-de, 2024-06-10:
Aktuell gibt es eine gut funktionierende Anbindung der LG ESS Home 8/10.

Zusätzlich kann die aber auch SunSpec, wenn man in den Installateur-Einstellungen die "Fernsteuerung" freigibt. Das SunSpec-Dokument  (V 1.3 von 2021) habe ich vorliegen und hier angehängt, und die Register meiner LG ESS Home 8 kann ich per qModMaster auch ohne weiteres auslesen und erhalte sinnvolle, zur Beschreibung passende Werte.
Ist es interessant die ESS zusätzlich oder ausschließlich darüber anzusteuern? Wenn künftige Features wie "Batterie ausschalten beim Schnellladen" über ModBus gesteuert werden wäre vermutlich eine Vereinheitlichung hilfreich.

[(Home 8,10) LG ESS sunspec ver 1.3-1.pdf](https://github.com/user-attachments/files/15777206/Home.8.10.LG.ESS.sunspec.ver.1.3-1.pdf)

premultiply, 2024-06-10:
Gibt es denn Werte oder Funktionen die man nur via SunSpec bekommt?

Kannst das Gerät ja mal als generische Sunspec-Meter in allen Facetten anlegen und mit `evcc meter` vergleichen.

cvg-de, 2024-06-15:
Probiere ich in den nächsten Wochen mal aus, wenn ich freie Rechenzeit habe. Melde mich.

closed 2024-06-30 by andig