evcc-io/evcc #14283: Ladevorgang wird nicht mehr automatisch gestartet. Go-e Gemini flex

Xtremefly, 2024-06-10:
### Describe the bug

Verwendet wird ein Cupra born in Verbindung des Go-e Gemini flex Charger. Wird das Auto geladen und man stoppt z.B. den Ladevorgang über den "Aus" Button über die evcc GUI wird der Ladevorgang gestoppt. Will man dann z.b. am nächste Tag das Auto wieder aufladen startet dann der Ladevorgang nicht mehr. Es startet auch nicht wenn man in dem Modus "schnell" Laden möchte.
Firmware Version Go-e : 56.1 (In der Version 54.7 gleiches Problem)
Über die API von dem Go-e charger habe ich herausgefunden das es um die Hardwareversion 4 sich handelt. 
In der Go-e App ist nur die API V1+2 aktiviert, sowie die Option "Ausstecken simulieren" wie in der Dokumentation beschrieben.
Leider weiß ich nicht wo der Fehler ist. Auch wenn ich auf Gastfahrzeug umschalte habe ich das gleiche Verhalten. Muss in der Go-e App noch etwas spezielles eingestellt werden?

### Steps to reproduce

1. Auto laden
2. Ladevorgang stoppen über Button "Aus" oder über die App vom Fahrzeug
3. Kabel abziehen
4. Sobald dann das Kabel wieder angesteckt wird startet der Ladevorgang nicht mehr
5. Erst nach einem Restart des Go-e charger
...

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: evcc.local
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 30s # control cycle interval. Interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval

# database configuration for persisting charge sessions and settings
# database:
#   type: sqlite
#   dsn: <path-to-db-file>

# sponsor token enables optional features (request at https://sponsor.evcc.io)
sponsortoken: <meinToken>

# log settings
log: info
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  cache: error
  db: error
  goe: trace

# modbus proxy for allowing external programs to reuse the evcc modbus connection
# each entry will start a proxy instance at the given port speaking Modbus TCP and
# relaying to the given modbus downstream device (either TCP or RTU, RS485 or TCP)
modbusproxy:
  #  - port: 5200
  #    uri: solar-edge:502
  #    # rtu: true
  #    # readonly: true # use `deny` to raise modbus errors

meters:
  - name: my_grid
    type: template
    template: fronius-gen24
    usage: grid
    host: 192.168.62.80 # IP-Adresse oder Hostname
    port: 502 # Port (optional)
  - name: my_pv
    type: template
[... 236 lines cut ...]
[lp-1  ] DEBUG 2024/06/10 18:58:11 pv timer elapse
[lp-1  ] DEBUG 2024/06/10 18:58:11 pv timer inactive
[lp-1  ] DEBUG 2024/06/10 18:58:11 charge total import: 67.446kWh
[lp-1  ] INFO 2024/06/10 18:58:11 start charging ->
[lp-1  ] DEBUG 2024/06/10 18:58:11 wake-up timer: stop

Neustart von Go-e Charger (go-e trace)
[go-e  ] TRACE 2024/06/10 18:57:51 GET http://192.168.178.38/api/status?filter=alw,car,eto,nrg,wh,trx,cards
[go-e  ] TRACE 2024/06/10 18:57:52 {"alw":false,"eto":67445,"car":1,"wh":709.8155,"trx":null,"nrg":[231.88,231.57,233.12,1.86,0,0,0,0,0,0,0,0,0,0,0,0],"cards":[{"name":"n/a","energy":0,"cardId":true},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false}]}
[go-e  ] TRACE 2024/06/10 18:58:11 GET http://192.168.178.38/api/status?filter=alw,car,eto,nrg,wh,trx,cards
[go-e  ] TRACE 2024/06/10 18:58:11 {"alw":true,"eto":67446,"car":2,"wh":1.412859,"trx":null,"nrg":[230.64,232.5,233.12,1.55,5.966,0,0,1339.42,0,0,0,1339.42,97.34171,0,0,100],"cards":[{"name":"n/a","energy":0,"cardId":true},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false}]}
[go-e  ] TRACE 2024/06/10 18:58:12 GET http://192.168.178.38/api/status?filter=alw,car,eto,nrg,wh,trx,cards
[go-e  ] TRACE 2024/06/10 18:58:17 GET http://192.168.178.38/api/status?filter=alw,car,eto,nrg,wh,trx,cards
[go-e  ] TRACE 2024/06/10 18:58:18 {"alw":true,"eto":67449,"car":2,"wh":4.04501,"trx":null,"nrg":[230.64,232.5,233.43,1.55,5.966,0,0,1340.265,0,0,0,1340.265,97.40312,0,0,100],"cards":[{"name":"n/a","energy":0,"cardId":true},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false}]}
[go-e  ] TRACE 2024/06/10 18:58:18 GET http://192.168.178.38/api/set?psm=1
[go-e  ] TRACE 2024/06/10 18:58:18 {"psm":true}
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.126.6

Sillium007, 2024-06-11:
Die Wallbox sieht bzw meldet dass kein Auto angeschlossen ist (oder evcc interpretiert es so) `charger status: A` -> https://evsim.gonium.net/#der-control-pilot-cp

Laden aktiviert wird in allem was Du geschickt hast nur einmal:
`[go-e  ] TRACE 2024/06/10 18:45:41 GET http://192.168.178.38/api/set?frc=2`

Was ich aber am ganzen Log nicht verstehe. Deine Konfiguration sagt dass der GoE die 192.168.62.90 hat. Im Log kommt die aber nicht vor. Stattdessen kommt die 192.168.178.38. Die anscheinend auch ein GoE ist (sonst würden ja keine "korrekten" Antworten kommen).

Ich würde die v1 Api ausschalten. Brauchst für nichts und die Detektion der Box in Evcc nimmt nicht die v1 wenn mal kurz ein Aussetzer ist beim Aufstart.

Xtremefly, 2024-06-11:
Hallo, 

oh hab gesehen, dass ich noch die alte Konfig hochgeladen habe. Hatte die Wallbox, evcc hardware und Wechselrichter in einem separaten Vlan und habe diese in mein Hauptlan Netz umgezogen, weil ich testen wollte ob da vielleicht ein Problem ist. Zur neuen Konfig sind nur die IP Adressen anderst.

Ich versuche dann heute mal noch die API V1 auszuschalten.
Aber d.h. der Ladevorgang startet nicht, weil die Wallbox das Auto nicht erkennt?

Kann auch gerne mal das komplette Log hochladen, hab gestern nur Auszüge hochgeladen, dass es übersichtlicher ist.