evcc-io/evcc #14296: Absturz nach Nightly Update 126.6

Kludi2019, 2024-06-11:
### Describe the bug

Hallo alle zusammen,
ich habe heute nightly auf 126.6 upgedatet und evcc startet nicht mehr. 126.6 release downgrade funktioniert. Als Fehlermeldung kommt folgendes:

[main ] INFO 2024/06/11 17:02:16 evcc 0.126.6 (https://github.com/evcc-io/evcc/commit/4acdc033bddd6c7cd818896c20b3c473b6a907cd)
[main ] INFO 2024/06/11 17:02:16 using config file: /etc/evcc.yaml
[main ] INFO 2024/06/11 17:02:16 listening at :7070
[main ] FATAL 2024/06/11 17:02:16 failed parsing config file: decoding failed due to the following error(s): cannot parse 'EEBus.Certificate.Public[0]' as uint: strconv.ParseUint: parsing "-----BEGIN CERTIFICATE-----\nMIIBhzCCAS2gAwIBAgIBATAKBggqhkjOPQQDAjAzMQswCQYDVQQGEwJERTENMAsG\nA1UEChMERVZDQzEVMBMGA1UEAwwMRVZDQ19IRU1TXzAxMB4XDTIyMTAzMTA5MzQx\nMFoXDTMyMTAyODA5MzQxMFowMzELMAkGA1UEBhMCREUxDTALBgNVBAoTBEVWQ0Mx\nFTATBgNVBAMMDEVWQ0NfSEVNU18wMTBZMBMGByqGSM49AgEGCCqGSM49AwEHA0IA\nBG9LfY4+xVjYp/fXS7yX66rW68il3KmxzC9Hqw02dbiACDIa6b7SPHniw6vSSDG0\nKgrzLNR6c6fXaOqsRCax/BmjMjAwMA8GA1UdEwEB/wQFMAMBAf8wHQYDVR0OBBYE\nFCNlU4HENAR0BV4oWM6B8o8QrM9UMAoGCCqGSM49BAMCA0gAMEUCIQDE9HHTldX/\nN2CSc9SxXT0qsUKYehin6OiebycLNUEfkwIgQIhWCZH1CaNmCLK0ClxEqzrfx2Xo\nougJGx5M07FNtqE=\n-----END CERTIFICATE-----\n": invalid syntax cannot parse 'EEBus.Certificate.Private[0]' as uint: strconv.ParseUint: parsing "-----BEGIN EC PRIVATE KEY-----\nMHcCAQEEIL0mhWia5hk/EDBVPMXzsK+G2ScD58mnFybVMre1i3WCoAoGCCqGSM49\nAwEHoUQDQgAEb0t9jj7FWNin99dLvJfrqtbryKXcqbHML0erDTZ1uIAIMhrpvtI8\neeLDq9JIMbQqCvMs1Hpzp9do6qxEJrH8GQ==\n-----END EC PRIVATE KEY-----\n": invalid syntax
[main ] FATAL 2024/06/11 17:02:16 will attempt restart in: 15m0s

Gruß Gerhard

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: 6d0cd584651b8821c9f9c4c885404a80de75220ae4fed45689c3c651a401e207

interval: 10s # control cycle interval

meters:
#- type: template
  #template: sonnenbatterie 
  #usage: grid  
  #host: 192.168.178.33  
  #port: 8080  
  #name: grid1

- type: template
  template: tasmota
  usage: grid
  host: 192.168.178.61
  user: xxxx
  password: xxxxxxxxx
  name: grid1
- type: template
  template: sonnenbatterie
  usage: pv
  host: 192.168.178.33
  port: 8080
  name: pv2
- type: template
  template: sonnenbatterie
  usage: battery
  host: 192.168.178.33
  port: 8080
  name: battery3

- type: template
  name: Mobilladegerat_Pwr
  template: shelly-1pm
  user: xxxx
  password: xxxxx
  usage: charge
  host: 192.168.178.26

chargers:
- type: template
  template: ellipro
  ski: EE38 62DB 0242 7083 xxxx xxxx 
  ip: 192.168.178.74
  name: wallbox

- type: template
  template: shelly
  host: 192.168.178.26
  standbypower: 5
  user: xxxx
  password: xxxxx
  name: Mobilladegeraet

vehicles:
- type: template
  template: fiat
  title: Paulchen
  user: xxxxxx
  password: 'xxxxx'
  vin: xxxxxxx
  capacity: 42
  name: ev4

- type: template
  template: id
  title: Id5
  user: xxxxxx
  password: 'xxxxxx'
  vin: WVGxxxxxxx
  capacity: 77
  timeout: 10s
  name: ev1

loadpoints:
- title: Elli
  charger: wallbox
  mode: off
  
- title: mobil
  charger: Mobilladegeraet
  meter: Mobilladegerat_Pwr
  mode: off
  
mqtt:
  broker: 192.168.178.32:1887
  topic: evcc
  clientid: foo
  user: xxx
  password: 'xxxxxx'

tariffs:
  currency: EUR
  grid:
    type: fixed
    price: 0.368

  feedin:
    type: fixed
    price: 0.082

site:
  - title: Im Schxxxxx
    meters:
      grid: grid1
      pv: pv2
      battery: battery3
    
eebus:
  certificate:
      private: |
          -----BEGIN EC PRIVATE KEY-----
          MHcCAQEEIL0mhWia5hk/EDBVPMXzsK+G2ScD58mnFybVMre1i3WCoAoGCCqGSM49
          AwEHoUQDQgAEb0t9jj7FWNin99dLvJfrqtbryKXcqbHML0erDTZ1uIAIMhrpvtI8
          eeLDq9JIMbQqCvMs1Hpzp9do6qxEJrH8GQ==
          -----END EC PRIVATE KEY-----
      public: |
          -----BEGIN CERTIFICATE-----
          MIIBhzCCAS2gAwIBAgIBATAKBggqhkjOPQQDAjAzMQswCQYDVQQGEwJERTENMAsG
          A1UEChMERVZDQzEVMBMGA1UEAwwMRVZDQ19IRU1TXzAxMB4XDTIyMTAzMTA5MzQx
          MFoXDTMyMTAyODA5MzQxMFowMzELMAkGA1UEBhMCREUxDTALBgNVBAoTBEVWQ0Mx
          FTATBgNVBAMMDEVWQ0NfSEVNU18wMTBZMBMGByqGSM49AgEGCCqGSM49AwEHA0IA
          BG9LfY4+xVjYp/fXS7yX66rW68il3KmxzC9Hqw02dbiACDIa6b7SPHniw6vSSDG0
          KgrzLNR6c6fXaOqsRCax/BmjMjAwMA8GA1UdEwEB/wQFMAMBAf8wHQYDVR0OBBYE
          FCNlU4HENAR0BV4oWM6B8o8QrM9UMAoGCCqGSM49BAMCA0gAMEUCIQDE9HHTldX/
          N2CSc9SxXT0qsUKYehin6OiebycLNUEfkwIgQIhWCZH1CaNmCLK0ClxEqzrfx2Xo
          ougJGx5M07FNtqE=
          -----END CERTIFICATE-----
```

### Log details

```text
[main ] INFO 2024/06/11 17:02:16 evcc 0.126.6 (https://github.com/evcc-io/evcc/commit/4acdc033bddd6c7cd818896c20b3c473b6a907cd)
[main ] INFO 2024/06/11 17:02:16 using config file: /etc/evcc.yaml
[main ] INFO 2024/06/11 17:02:16 listening at :7070
[main ] FATAL 2024/06/11 17:02:16 failed parsing config file: decoding failed due to the following error(s): cannot parse 'EEBus.Certificate.Public[0]' as uint: strconv.ParseUint: parsing "-----BEGIN CERTIFICATE-----\nMIIBhzCCAS2gAwIBAgIBATAKBggqhkjOPQQDAjAzMQswCQYDVQQGEwJERTENMAsG\nA1UEChMERVZDQzEVMBMGA1UEAwwMRVZDQ19IRU1TXzAxMB4XDTIyMTAzMTA5MzQx\nMFoXDTMyMTAyODA5MzQxMFowMzELMAkGA1UEBhMCREUxDTALBgNVBAoTBEVWQ0Mx\nFTATBgNVBAMMDEVWQ0NfSEVNU18wMTBZMBMGByqGSM49AgEGCCqGSM49AwEHA0IA\nBG9LfY4+xVjYp/fXS7yX66rW68il3KmxzC9Hqw02dbiACDIa6b7SPHniw6vSSDG0\nKgrzLNR6c6fXaOqsRCax/BmjMjAwMA8GA1UdEwEB/wQFMAMBAf8wHQYDVR0OBBYE\nFCNlU4HENAR0BV4oWM6B8o8QrM9UMAoGCCqGSM49BAMCA0gAMEUCIQDE9HHTldX/\nN2CSc9SxXT0qsUKYehin6OiebycLNUEfkwIgQIhWCZH1CaNmCLK0ClxEqzrfx2Xo\nougJGx5M07FNtqE=\n-----END CERTIFICATE-----\n": invalid syntax cannot parse 'EEBus.Certificate.Private[0]' as uint: strconv.ParseUint: parsing "-----BEGIN EC PRIVATE KEY-----\nMHcCAQEEIL0mhWia5hk/EDBVPMXzsK+G2ScD58mnFybVMre1i3WCoAoGCCqGSM49\nAwEHoUQDQgAEb0t9jj7FWNin99dLvJfrqtbryKXcqbHML0erDTZ1uIAIMhrpvtI8\neeLDq9JIMbQqCvMs1Hpzp9do6qxEJrH8GQ==\n-----END EC PRIVATE KEY-----\n": invalid syntax
[main ] FATAL 2024/06/11 17:02:16 will attempt restart in: 15m0s
```

### What type of operating system are you running?

Linux

andig, 2024-06-11:
> 126.6 release downgrade funktioniert.

Downgrade wohin?

Kludi2019, 2024-06-11:
Auf stable 126.6

andig, 2024-06-11:
Danke, das hilft. Problem kommt mit 554ce1db6025ee202c6ae22cf873432835901ea2, #13319

/cc @naltatis

andig, 2024-06-11:
Liegt vmtl. an https://github.com/evcc-io/evcc/pull/13319/files#diff-7067a7e1be04be21d22a91abd819dc9e483810e9b1cd320e5c7c7cfa25a07b76L49-L59, ich sehe allerdings nicht, warum.

andig, 2024-06-11:
Ich kanns leider nicht greifen. 2 Testfälle hinzu gefügt, beide funktionieren. Deine Config funktioniert nicht. Warum ist mir ein absolutes Rätsel.

Kludi2019, 2024-06-11:
Kann ich dabei noch etwas tun?

GrimmiMeloni, 2024-06-12:
Ich kann das Problem mit der gezeigten Konfiguration auf dem Branch von Andig (#14298) nicht nachstellen. Habe dazu den EEBus Block aus der oben gezeigten Konfiguration in die demo.yaml kopiert. Das läuft einfach durch. Ich habe auch die Tests in #14298 einmal lokal mit den gezeigten Certs bestückt. Auch diese laufen durch.

@andig hast Du es nachgestellt bekommen? Und wenn ja, wie?

andig, 2024-06-12:
Das Problem kommt bei mir, wenn eine in die DB migrierte Leer-Konfiguration zurück gelesen wird:

```yaml
uri: ""
shipid: ""
interfaces: []
certificate:
    public: []
    private: []
```

Die sieht eigentlich gültig aus. Warum das nicht geht muss ich noch raus finden.

andig, 2024-06-12:
I still can't repro it:

```go
func TestEEbusConfig(t *testing.T) {
	conf := `
eebus:
  uri: "foo"
  shipid: ""
  interfaces: []
  certificate:
    private: []
    public: []
`

	var res globalconfig.All
	require.NoError(t, yaml.Unmarshal([]byte(conf), &res))
	require.Equal(t, "foo", res.EEBus.URI)

	viper.SetConfigType("yaml")
	require.NoError(t, viper.ReadConfig(strings.NewReader(conf)))
	viper.UnmarshalExact(&res)
	require.Equal(t, "foo", res.EEBus.URI)
}
```

andig, 2024-06-12:
Think I understand what's going on. go-yaml doesn't like to unmarshal string into []byte. In previous version this was done at a later step by mapstructure which did handle this.

closed 2024-06-12 by andig