evcc-io/evcc #14304: Ladeplan startet nach Neustart sofort obwohl Startzeit noch nicht erreicht

An-Wi, 2024-06-11:
### Describe the bug

Zunächst wurde das Hostsystem und somit auch evcc neu gestartet. Danach waren die Vehicle-Daten nicht verfügbar. Der zuvor eingestellte Ladeplan war inaktiv bzw. nicht sichtbar.
Daraufhin habe ich über die UI evcc manuell neu gestartet. Vehicle-Daten waren abrufbar aber auch das Laden wurde sofort gestartet obwohl der Ladplan den Start erst in 3h errechnet hatte! => siehe Logs

Erst ein Switch auf "Aus" beendete das ungewollte Laden und anschließendes Setzen von Modus "PV" beachtet auch wieder den eingestellten Ladeplan und wartet nun brav bis 02:06 Uhr.

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
-
```

### Log details

```text
[main  ] INFO 2024/06/11 23:02:00 evcc 0.126.6
[main  ] INFO 2024/06/11 23:02:00 using config file: c:\evcc\evcc.yaml
[main  ] INFO 2024/06/11 23:02:00 starting ui and api at :7070
[db    ] INFO 2024/06/11 23:02:00 using sqlite database: C:\Users\User\.evcc\evcc.db
[db    ] TRACE 2024/06/11 23:02:00 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2024/06/11 23:02:00 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/06/11 23:02:00 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/06/11 23:02:00 SELECT * FROM `settings` 31 <nil>
[db    ] TRACE 2024/06/11 23:02:00 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db    ] TRACE 2024/06/11 23:02:00 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db    ] TRACE 2024/06/11 23:02:00 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2024/06/11 23:02:00 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/06/11 23:02:00 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/06/11 23:02:00 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2024/06/11 23:02:00 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "config_details" AND sql IS NOT NULL order by type = "table" desc 2 <nil>
[db    ] TRACE 2024/06/11 23:02:00 SELECT * FROM `config_details` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/06/11 23:02:00 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT ""fk_configs_details"" %" OR sql LIKE "%CONSTRAINT fk_configs_details %" OR sql LIKE "%CONSTRAINT `fk_configs_details`%" OR sql LIKE "%CONSTRAINT [fk_configs_details]%" OR sql LIKE "%CONSTRAINT 	fk_configs_details	%") -1 <nil>
[db    ] TRACE 2024/06/11 23:02:00 SELECT count(*) FROM sqlite_master WHERE type = "index" AND tbl_name = "config_details" AND name = "idx_unique" -1 <nil>
[db    ] TRACE 2024/06/11 23:02:00 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT ""fk_devices_details"" %" OR sql LIKE "%CONSTRAINT fk_devices_details %" OR sql LIKE "%CONSTRAINT `fk_devices_details`%" OR sql LIKE "%CONSTRAINT [fk_devices_details]%" OR sql LIKE "%CONSTRAINT 	fk_devices_details	%") -1 <nil>
[db    ] TRACE 2024/06/11 23:02:00 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%""device_id"" %" OR sql LIKE "%device_id %" OR sql LIKE "%`device_id`%" OR sql LIKE "%[device_id]%" OR sql LIKE "%	device_id	%") -1 <nil>
[db    ] TRACE 2024/06/11 23:02:03 SELECT * FROM `configs` WHERE `configs`.`class` = 2 ORDER BY id 0 <nil>
[db    ] TRACE 2024/06/11 23:02:03 SELECT * FROM `configs` WHERE `configs`.`class` = 1 ORDER BY id 0 <nil>
[db    ] TRACE 2024/06/11 23:02:03 SELECT * FROM `configs` WHERE `configs`.`class` = 5 ORDER BY id 0 <nil>
[db    ] TRACE 2024/06/11 23:02:03 SELECT * FROM `config_details` WHERE `config_details`.`config_id` = 1 3 <nil>
[db    ] TRACE 2024/06/11 23:02:03 SELECT * FROM `configs` WHERE `configs`.`class` = 3 ORDER BY id 1 <nil>
[nissan] TRACE 2024/06/11 23:02:03 POST https://prod.eu2.auth.kamereon.org/kauth/json/realms/root/realms/a-ncb-prod/authenticate
[nissan] TRACE 2024/06/11 23:02:03 {"authId":"eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJvdGsiOiI5bnI4MnUzZmkwdjJtYTh1Ympua3I2YjMyaCIsInJlYWxtIjoiL2EtbmNiLXByb2QiLCJzZXNzaW9uSWQiOiJiWndOM0p3SnNrTXZRYXk5R2JXM25nZWJYdncuKkFBSlRTUUFDTURJQUFsTkxBQnhoWjFWd1JVdHpaV0pST1cxdlJrUldiMlE0WldOWk4zSnNaR2M5QUFSMGVYQmxBQWxKVGw5TlJVMVBVbGtBQWxNeEFBSTRNZy4uKiJ9.7QN6xaz5HoGEENUXZAkXbVZ9fFdJ3TtoTfTtlPMYqr0","template":"","stage":"LDAP1","header":"Sign in","callbacks":[{"type":"NameCallback","output":[{"name":"prompt","value":"User Name:"}],"input":[{"name":"IDToken1","value":""}]},{"type":"PasswordCallback","output":[{"name":"prompt","value":"Password:"}],"input":[{"name":"IDToken2","value":""}]}]}
[nissan] TRACE 2024/06/11 23:02:03 POST https://prod.eu2.auth.kamereon.org/kauth/json/realms/root/realms/a-ncb-prod/authenticate
[nissan] TRACE 2024/06/11 23:02:03 {"authId":"eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJvdGsiOiI5bnI4MnUzZmkwdjJtYTh1Ympua3I2YjMyaCIsInJlYWxtIjoiL2EtbmNiLXByb2QiLCJzZXNzaW9uSWQiOiJiWndOM0p3SnNrTXZRYXk5R2JXM25nZWJYdncuKkFBSlRTUUFDTURJQUFsTkxBQnhoWjFWd1JVdHpaV0pST1cxdlJrUldiMlE0WldOWk4zSnNaR2M5QUFSMGVYQmxBQWxKVGw5TlJVMVBVbGtBQWxNeEFBSTRNZy4uKiJ9.7QN6xaz5HoGEENUXZAkXbVZ9fFdJ3TtoTfTtlPMYqr0","template":"","stage":"LDAP1","header":"Sign in","callbacks":[{"type":"NameCallback","output":[{"name":"prompt","value":"User Name:"}],"input":[{"name":"IDToken1","value":"***"}]},{"type":"PasswordCallback","output":[{"name":"prompt","value":"Password:"}],"input":[{"name":"IDToken2","value":"***"}]}]}
--
{"tokenId":"0dLYZ-KfWC3qUyjcH_dkZpRPGGo.*AAJTSQACMDIAAlNLABxzWUVTcVliYXU5VHMrSmQzNWoxN0Y1ZHNaMzg9AAR0eXBlAANDVFMAAlMxAAI4Mg..*","successUrl":"/kauth/console","realm":"/a-ncb-prod"}
[nissan] TRACE 2024/06/11 23:02:03 GET https://prod.eu2.auth.kamereon.org/kauth/oauth2/a-ncb-prod/authorize?client_id=a-ncb-nc-android-prod&nonce=sdfdsfez&redirect_uri=org.kamereon.service.nci%3A%2Foauth2redirect&response_type=code&scope=openid+profile+vehicles
[nissan] TRACE 2024/06/11 23:02:03 POST https://prod.eu2.auth.kamereon.org/kauth/oauth2/a-ncb-prod/access_token
[nissan] TRACE 2024/06/11 23:02:03 client_id=a-ncb-nc-android-prod&client_secret=6GKIax7fGT5yPHuNmWNVOc4q5POBw1WRSW39ubRA8WPBmQ7MOxhm75EsmKMKENem&code=XGWplpu5GVmuLVMYfGpXAhRtRrc&grant_type=authorization_code&redirect_uri=org.kamereon.service.nci%3A%2Foauth2redirect
--
{"access_token":"fJii9F-0IU5nl34Io-XGZiUNpuY","refresh_token":"8wnXrzmiG1pk6UD6AMdLCBhs2v0","scope":"openid profile vehicles","id_token":"eyJ0eXAiOiJKV1QiLCJraWQiOiJtTUdBbVdLOUNSNHJwMEZWNVFYdUlvOUtBM3c9IiwiYWxnIjoiUlMyNTYifQ.eyJGSUQiOiJLQU1FUkVPTi1lNTRkNGNjZC1kNzFlLTRlZDgtYTQ2Zi05NWRhZDY4ODY2NmQiLCJhdF9oYXNoIjoieEdkTnFJZVNiY3ZzV1JhVHdfbDVTUSIsImNvdW50cnkiOiJERSIsInN1YiI6IktBTUVSRU9OLWU1NGQ0Y2NkLWQ3MWUtNGVkOC1hNDZmLTk1ZGFkNjg4NjY2ZCIsImF1ZGl0VHJhY2tpbmdJZCI6ImI0ZjQ3ZWUzLTgwMjktNDQzNi1iMmFhLWU4MjkwNDMwNWFmOC01MDIzMjA3NTEiLCJLSUQiOiJlNTRkNGNjZC1kNzFlLTRlZDgtYTQ2Zi05NWRhZDY4ODY2NmQiLCJpc3MiOiJodHRwczovL3Byb2QuZXUyLmF1dGgua2FtZXJlb24ub3JnOjQ0My9rYXV0aC9vYXV0aDIvYS1uY2ItcHJvZCIsInRva2VuTmFtZSI6ImlkX3Rva2VuIiwidmVoaWNsZXMiOnsidmVoaWNsZXMiOlt7InZpbiI6ImQ4Yjk5OTZhLTY1MDQtNDczOC05ODkyLTUwNWEyMzhkNjRiMSIsImd3IjoiQVZOIiwicm9sZSI6Ik9XTkVSIn1dfSwiZ2l2ZW5fbmFtZSI6IkFuZHJlYXMiLCJub25jZSI6InNkZmRzZmV6IiwiYXVkIjoiYS1uY2ItbmMtYW5kcm9pZC1wcm9kIiwiY19oYXNoIjoiSXdTVDZiS1VOa0JpVVk1aXYtTExMQSIsImFjciI6ImluZmluaXRpIiwib3JnLmZvcmdlcm9jay5vcGVuaWRjb25uZWN0Lm9wcyI6IjJ1cEZfZFZSSFdlYnY3M0NuXzludW4wSzQyVSIsImF6cCI6ImEtbmNiLW5jLWFuZHJvaWQtcHJvZCIsImF1dGhfdGltZSI6MTcxODEzOTcyMywibmFtZSI6ImFuLndpcnRoQGdteC5kZSIsInJlYWxtIjoiL2EtbmNiLXByb2QiLCJleHAiOjE3MTgxNDMzMjMsInRva2VuVHlwZSI6IkpXVFRva2VuIiwiZmFtaWx5X25hbWUiOiJXaXJ0aCIsImlhdCI6MTcxODEzOTcyM30.hdqsrmH3uf6fWQjGwsl8OMxYF1SdbkB9pQWSYFQ-XXrAJ-KT6_CJBt-oe2RUh7Kl3GwXPhoeb4aZisCWTihMqMAUUM9dQPAHu-c8BB3NPnddu1eHOtrPTe7qgY2zcdEKmHq7TOOFd9keyB1AeUczWLW1IsKNZRfNJpwV3tKOMo996NdwtcuorDHUr2xNPFMJnJb0gVaAsVe8b0jZJ-TOi7YScUKxIyN_UceWayD_nQOW6c0TvgomI99uJz-tywVbhHf7WClmlFGD5Ni9zAgWDiJStS13i_sCW8f_AGIChsQtRAdebatWhCQAiiG6JNe0YDEkQ99KGnUHHzVXdRfxGQ","token_type":"Bearer","expires_in":21599,"nonce":"sdfdsfez"}
[nissan] TRACE 2024/06/11 23:02:03 GET https://alliance-platform-usersadapter-prod.apps.eu2.kamereon.io/user-adapter/v1/users/current
[... cut ...]
[lp-1  ] DEBUG 2024/06/11 23:03:45 soc estimated: 42.24% (vehicle: 42.00%)
[lp-1  ] DEBUG 2024/06/11 23:03:45 vehicle soc: 42%
[lp-1  ] DEBUG 2024/06/11 23:03:45 vehicle range: 235km
[lp-1  ] DEBUG 2024/06/11 23:03:45 plan: charge 5h48m19s between 2024-06-12 02:01:41 +0200 CEST until 2024-06-12 07:50:00 +0200 CEST (power: 11040W, avg cost: 0.280)
[lp-1  ] DEBUG 2024/06/11 23:03:45 plan: continuing until end of slot
[site  ] DEBUG 2024/06/11 23:04:05 ----
```

### What type of operating system are you running?

Windows

### Version

0.126.6

andig, 2024-06-12:
> Ladeplan startet nach Neustart sofort obwohl Startzeit noch nicht erreicht

Wie passt das mit dem

> [lp-1  ] DEBUG 2024/06/11 23:02:06 plan: charge 9h30m15s between 2024-06-11 23:02:06 +0200 CEST until 2024-06-12 07:50:00 +0200 CEST (overruns by 42m21s, power: 11040W, avg cost: 0.280)

zusammen?

An-Wi, 2024-06-12:
Die Berechnung ist falsch! Sehe ich leider erst jetzt diese Logzeile -> 20 Sekunden später kommt die korrekte Berechnung!

`[lp-1  ] DEBUG 2024/06/11 23:02:26 plan: charge 5h49m36s between 2024-06-12 02:00:24 +0200 CEST until 2024-06-12 07:50:00 +0200 CEST (power: 11040W, avg cost: 0.280)`

Wie kann die Berechnung 20s vorher noch 9h30m liefern? Was ging da in der Berechnung schief?
Und warum stoppt der Plan dann nicht augenblicklich sondern läuft munter weiter?
Deswegen evtl.? `[lp-1  ] DEBUG 2024/06/11 23:03:45 plan: continuing until end of slot`

andig, 2024-06-12:
Es gibt an der Stelle wohl noch keinen soc. Also alles richtig!

An-Wi, 2024-06-12:
Dieses Verhalten findest du "korrekt"? 🙁
Besser wäre doch, den Plan ohne SOC gar nicht erst zu initiieren, sondern noch zu warten bzw. in dem "Rundlauf ohne SOC" einfach auszusetzen.
Ähnlich lief es zuvor, als gar keine Vehicle-Daten vorhanden waren - da war der Plan "unsichtbar". Was ist dann da anders?

andig, 2024-06-12:
Besser als ein leeres Auto. Was sagt dir, dass in der nächsten Runde ein Soc kommt? Nach 15min wird wieder abgeschaltet.

An-Wi, 2024-06-12:
Ah, interessant. Das mit den 15min wusste ich nicht. Da war ich nach 6min wohl etwas zu ungeduldig:
`[main  ] INFO 2024/06/11 23:02:00 starting ui and api at :7070`
`...`
`[lp-1  ] DEBUG 2024/06/11 23:08:31 set charge mode: off`

Trotzdem würde ich es als Problem ansehen, wenn bereits 6 Sekunden nach Start von evcc der "plan:" berechnet und ausgeführt wird. Das zu diesem Zeitpunkt noch nicht alle notwendigen Daten vorliegen (gerade von der Nissan API, die teilweise mehrere Login-Wiederholungen benötigt), ist mit relativ hoher Wahrscheinlichkeit öfter mal der Fall und tritt vermutlich nicht nur bei mir so auf... es fühlt sich irgendwie "instabil" an - mal tritt es auf, mal nicht... könnte mir vorstellen, dass es für die meisten User verwirrend ist, wenn evcc so unterschiedlich agiert. 🤷‍♂️

andig, 2024-06-12:
Yo. Zum Glück läuft evcc ja gerne problemlos wochenlang durch 👍🏻

An-Wi, 2024-06-13:
Mh, das ist toll für evcc, aber auf einem Windows-Host nicht so einfach umsetzbar...
Wäre es ein Kompromiss und einfach umsetzbar, wenn der Planer den 1. (und ggf. 2.) Rundlauf nach Start abwartet bzw. erst 30-60s nach dem Start von evcc aktiv wird?
Oder sprechen andere interne Abläufe dagegen?

Mir ist klar, dass wir hier von einer Kleinigkeit reden, daher kurz mal zum Hintergrund meiner eigentlichen Problematik: 
Meine Hauptsicherung hat historisch bedingt momentan (noch) 35A pro Phase und neben der Wallbox und sonstigen Haushaltsgeräten hängt u.a. auch ein Durchlauferhitzer mit dran, der gern mal bis 24kW zieht (also gut 30A pro Phase)... Bei ca. 43A fliegt _nachweislich_ eine Phasensicherung raus. Im besten Fall meldet die Wallbox "nur" einen Fehler (fehlende Phase), lädt dann aber auch nicht mehr weiter (=> "leeres Auto") - im schlimmsten Fall stirbt evcc den "Stromtot", wenn der Windows-Host an der rausgeflogenen Phase hängt, was auch ein "leeres Auto" bedeutet. 
Insofern versuche ich die Ladeplanung entsprechend in Zeiten zu verlegen, in denen der Durchlauferhitzer nicht parallel arbeiten sollte...

premultiply, 2024-06-13:
☝️Lastmanagement-Feature benutzen

andig, 2024-06-13:
> Mh, das ist toll für evcc, aber auf einem Windows-Host nicht so einfach umsetzbar...

Oder root cause lösen :)

An-Wi, 2024-06-13:
> Oder root cause lösen :)

Das liegt leider nicht in meiner Macht... warte seit gut 6 Monaten auf meinen Elektriker für diese 5-Minuten-Sache... ☕

<hr>

> ☝️Lastmanagement-Feature benutzen

@premultiply Klingt spannend - wo finde ich das? Kurzer Doku-Hinweis reicht...
Hab leider nur das hier zum Thema "Lastmanagement" gefunden:
_Kann ich mehrere Wallboxen nutzen?
Mehrere Wallboxen und damit Ladepunkte können in evcc verwendet werden. Es ist jedoch heute noch kein **Lastmanagement** möglich. Das ist auf unserer langen Liste für die weitere Entwicklung._

andig, 2024-06-13:
> Das liegt leider nicht in meiner Macht...

Dein Elektriker installiert auch Linux?