evcc-io/evcc #14307: em2go startet immer bei 11kW - PV Modus - Regelt nicht schnell genug - bricht ab

kasn24, 2024-06-12:
### Describe the bug

Hallo,

das Problem besteht komischerweise nur morgens bei "aufgehen" der Sonne. 70kWh Ladestand erreicht, 4kW Überschuss (bereits 1kW einphasig hätten gereicht, aber da war der Akku noch nicht voll genug) 
Das Laden startet, die Wallbox zieht direkt auf 11kW hoch auch schneller als in evcc angezeigt wird. Das packt meistens auch mein Akku nicht und wir gehen in den Netzbezug. 
Statt runter zu regeln oder sogar die Phasen zu wechseln, wird einfach abgebrochen.

Ich habe noch nichts derartiges gefunden, dass die Wallbox direkt mit 3 Phasen und der niedrigsten Amperezahl startet.
Die ebenfalls konfigurierte SMA Wallbox macht dies ohne Probleme. 

Da kommen wir zu Problem 2, wo ich versuche zeitnah Logs nachzuliefern. 
Die SMA Wallbox lädt den MG4 1-phasig bis voll, lädt dann konstant weiter mit ca. 10W und schaltet nicht ab. Dies führt zum "klacken" vom Ladegerät des MG4 und ist mit Sicherheit nicht gesund. Das hat die SMA Wallbox gesteuert über den HomeManager nicht gemacht. 

Ich hoffe ihr könnt mir weiterhelfen. 

Viele Grüße

### Steps to reproduce

1. Überschuss 4kW
2. PV Laden
3. em2Go Wallbox startet mit 11kW
4. Bricht Ladevorgang ab, statt runter zu regeln
5. beginnt wieder von vorn
...

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: 192.168.187.182
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 30s # control cycle interval. Interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval

# database configuration for persisting charge sessions and settings
# database:
#   type: sqlite
#   dsn: <path-to-db-file>

# sponsor token enables optional features (request at https://sponsor.evcc.io)
# sponsortoken:
# trial token, valid until 2024-05-19
sponsortoken: XXXX
plant: XXXXX
# telemetry enables aggregated statistics
#
# Telemetry allows collecting usage data (grid and green energy, charge power).
# Data is aggregated, no individual charging sessions are tracked. The collected,
# anonymous data can be retrieved using https://api.evcc.io.
#
# See https://github.com/evcc-io/evcc/pull/4343 or details.
#
# For time being, this is only available to sponsors, hence data is associated with
# the sponsor token's identity.
#
# telemetry: true

# log settings
log: info
levels:
[... 561 lines cut ...]
[site  ] DEBUG 2024/06/12 08:44:17 grid meter: -63W
[site  ] DEBUG 2024/06/12 08:44:17 grid powers: [563 -294 -332]W
[site  ] DEBUG 2024/06/12 08:44:17 grid currents: [2.65 -1.4 -1.51]A
[site  ] DEBUG 2024/06/12 08:44:17 site power: -1405W
[lp-2  ] DEBUG 2024/06/12 08:44:17 !! session: chargeRater.chargedEnergy=0.1 - chargedAtStartup=0.0
[lp-2  ] DEBUG 2024/06/12 08:44:17 charge total import: 28.900kWh
[lp-2  ] DEBUG 2024/06/12 08:44:17 charger status: B
[lp-2  ] INFO 2024/06/12 08:44:17 stop charging <-
[lp-2  ] DEBUG 2024/06/12 08:44:17 pv timer reset
[lp-2  ] DEBUG 2024/06/12 08:44:17 pv timer inactive
[lp-2  ] DEBUG 2024/06/12 08:44:17 charge total import: 28.900kWh
[lp-2  ] DEBUG 2024/06/12 08:44:17 !! session: start=28.900 stop=28.900 chargedEnergy=0.100
[lp-2  ] DEBUG 2024/06/12 08:44:17 pv charge current: 6.11A = 0A + 6.11A (-1405W @ 1p)
[lp-2  ] DEBUG 2024/06/12 08:44:17 site power -1405W <= 0W enable threshold
[lp-2  ] DEBUG 2024/06/12 08:44:17 pv enable timer start: 1m0s
[lp-2  ] DEBUG 2024/06/12 08:44:17 pv enable in 1m0s
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.126.6

kasn24, 2024-06-12:
Die Logs zu Problem 2

`[site  ] DEBUG 2024/06/12 10:58:16 ----
[lp-1  ] DEBUG 2024/06/12 10:58:16 charge power: 2008W
[lp-1  ] DEBUG 2024/06/12 10:58:16 charge currents: [8.7 -0 -0]A
[lp-2  ] DEBUG 2024/06/12 10:58:16 charge power: 0W
[lp-2  ] DEBUG 2024/06/12 10:58:17 charge currents: [0 0 0]A
[site  ] DEBUG 2024/06/12 10:58:17 pv power: 2516W
[site  ] DEBUG 2024/06/12 10:58:17 battery soc: 100%
[site  ] DEBUG 2024/06/12 10:58:17 battery power: 0W
[site  ] DEBUG 2024/06/12 10:58:17 grid meter: -117W
[site  ] DEBUG 2024/06/12 10:58:17 grid powers: [1298 -684 -731]W
[site  ] DEBUG 2024/06/12 10:58:17 grid currents: [5.91 -3.07 -3.19]A
[site  ] DEBUG 2024/06/12 10:58:17 site power: -17W
[lp-2  ] DEBUG 2024/06/12 10:58:17 !! session: chargeRater.chargedEnergy=0.1 - chargedAtStartup=0.0
[lp-2  ] DEBUG 2024/06/12 10:58:17 charge total import: 28.900kWh
[lp-2  ] DEBUG 2024/06/12 10:58:17 charger status: A
[site  ] DEBUG 2024/06/12 10:58:46 ----
[lp-1  ] DEBUG 2024/06/12 10:58:46 charge power: 2002W
[lp-1  ] DEBUG 2024/06/12 10:58:46 charge currents: [8.7 -0 -0]A
[lp-2  ] DEBUG 2024/06/12 10:58:46 charge power: 0W
[lp-2  ] DEBUG 2024/06/12 10:58:46 charge currents: [0 0 0]A
[site  ] DEBUG 2024/06/12 10:58:47 pv power: 2528W
[site  ] DEBUG 2024/06/12 10:58:47 battery soc: 100%
[site  ] DEBUG 2024/06/12 10:58:47 battery power: 0W
[site  ] DEBUG 2024/06/12 10:58:47 grid meter: -152W
[site  ] DEBUG 2024/06/12 10:58:47 grid powers: [1319 -737 -734]W
[site  ] DEBUG 2024/06/12 10:58:47 grid currents: [6 -3.28 -3.21]A
[site  ] DEBUG 2024/06/12 10:58:47 site power: -52W
[lp-1  ] DEBUG 2024/06/12 10:58:47 detected active phases: 1p
[lp-1  ] DEBUG 2024/06/12 10:58:47 !! session: chargeRater.chargedEnergy=1.6 - chargedAtStartup=0.0
[lp-1  ] DEBUG 2024/06/12 10:58:47 charger status: C
[lp-1  ] DEBUG 2024/06/12 10:58:47 pv charge current: 9.7A = 9.48A + 0.227A (-52W @ 1p)
[lp-1  ] DEBUG 2024/06/12 10:58:47 max charge current: 9.7A
[site  ] DEBUG 2024/06/12 10:59:16 ----
[lp-1  ] DEBUG 2024/06/12 10:59:16 charge power: 2045W
[lp-1  ] DEBUG 2024/06/12 10:59:16 charge currents: [8.88 -0 -0]A
[lp-2  ] DEBUG 2024/06/12 10:59:16 charge power: 0W
[lp-2  ] DEBUG 2024/06/12 10:59:16 charge currents: [0 0 0]A
[site  ] DEBUG 2024/06/12 10:59:16 pv power: 2520W
[site  ] DEBUG 2024/06/12 10:59:17 battery soc: 100%
[site  ] DEBUG 2024/06/12 10:59:17 battery power: 993W
[site  ] DEBUG 2024/06/12 10:59:17 grid meter: 106W
[site  ] DEBUG 2024/06/12 10:59:17 grid powers: [2245 -1072 -1067]W
[site  ] DEBUG 2024/06/12 10:59:17 grid currents: [10.5 -4.65 -4.6]A
[site  ] DEBUG 2024/06/12 10:59:17 site power: 1199W
[lp-2  ] DEBUG 2024/06/12 10:59:17 !! session: chargeRater.chargedEnergy=0.1 - chargedAtStartup=0.0
[lp-2  ] DEBUG 2024/06/12 10:59:17 charge total import: 28.900kWh
[lp-2  ] DEBUG 2024/06/12 10:59:17 charger status: A
[site  ] DEBUG 2024/06/12 10:59:46 ----
[lp-1  ] DEBUG 2024/06/12 10:59:46 charge power: 2046W
[lp-1  ] DEBUG 2024/06/12 10:59:46 charge currents: [8.87 -0 -0]A
[lp-2  ] DEBUG 2024/06/12 10:59:46 charge power: 0W
[lp-2  ] DEBUG 2024/06/12 10:59:46 charge currents: [0 0 0]A
[site  ] DEBUG 2024/06/12 10:59:47 pv power: 2534W
[site  ] DEBUG 2024/06/12 10:59:47 battery soc: 100%
[site  ] DEBUG 2024/06/12 10:59:47 battery power: 475W
[site  ] DEBUG 2024/06/12 10:59:47 grid meter: -43W
[site  ] DEBUG 2024/06/12 10:59:47 grid powers: [1792 -920 -915]W
[site  ] DEBUG 2024/06/12 10:59:47 grid currents: [8.79 -4.02 -3.96]A
[site  ] DEBUG 2024/06/12 10:59:47 site power: 532W
[lp-1  ] DEBUG 2024/06/12 10:59:47 detected active phases: 1p
[lp-1  ] DEBUG 2024/06/12 10:59:47 !! session: chargeRater.chargedEnergy=1.6 - chargedAtStartup=0.0
[lp-1  ] DEBUG 2024/06/12 10:59:47 charger status: C
[lp-1  ] DEBUG 2024/06/12 10:59:47 pv charge current: 7.39A = 9.7A + -2.31A (532W @ 1p)
[lp-1  ] DEBUG 2024/06/12 10:59:47 max charge current: 7.39A
[site  ] DEBUG 2024/06/12 11:00:16 ----
[lp-1  ] DEBUG 2024/06/12 11:00:16 charge power: 1W
[lp-1  ] DEBUG 2024/06/12 11:00:16 charge currents: [0.513 -0 -0]A
[lp-2  ] DEBUG 2024/06/12 11:00:16 charge power: 0W
[... 116 lines cut ...]
[lp-1  ] DEBUG 2024/06/12 11:03:47 detected active phases: 1p
[lp-1  ] DEBUG 2024/06/12 11:03:47 !! session: chargeRater.chargedEnergy=1.7 - chargedAtStartup=0.0
[lp-1  ] DEBUG 2024/06/12 11:03:47 charger status: C
[lp-1  ] DEBUG 2024/06/12 11:03:47 pv charge current: 12.3A = 3.06A + 9.24A (-2125W @ 1p)
[lp-1  ] DEBUG 2024/06/12 11:03:47 max charge current: 12.3A
[site  ] DEBUG 2024/06/12 11:04:16 ----
[lp-1  ] DEBUG 2024/06/12 11:04:16 charge power: 1W
[lp-1  ] DEBUG 2024/06/12 11:04:16 charge currents: [0.514 -0 -0]A
[lp-2  ] DEBUG 2024/06/12 11:04:16 charge power: 0W
[lp-2  ] DEBUG 2024/06/12 11:04:16 charge currents: [0 0 0]A
[site  ] DEBUG 2024/06/12 11:04:16 pv power: 3410W
[site  ] DEBUG 2024/06/12 11:04:17 battery soc: 100%
[site  ] DEBUG 2024/06/12 11:04:17 battery power: 0W
[site  ] DEBUG 2024/06/12 11:04:17 grid meter: -2979W
[site  ] DEBUG 2024/06/12 11:04:17 grid powers: [-970 -981 -1028]W
[site  ] DEBUG 2024/06/12 11:04:17 grid currents: [-4.57 -4.29 -4.44]A
[site  ] DEBUG 2024/06/12 11:04:17 site power: -2879W
[lp-2  ] DEBUG 2024/06/12 11:04:17 !! session: chargeRater.chargedEnergy=0.1 - chargedAtStartup=0.0
[lp-2  ] DEBUG 2024/06/12 11:04:17 charge total import: 28.900kWh
[lp-2  ] DEBUG 2024/06/12 11:04:17 charger status: A
[site  ] DEBUG 2024/06/12 11:04:46 ----
[lp-1  ] DEBUG 2024/06/12 11:04:46 charge power: 307W
[lp-1  ] DEBUG 2024/06/12 11:04:46 charge currents: [0.515 -0 -0]A
[lp-2  ] DEBUG 2024/06/12 11:04:46 charge power: 0W
`

RenatusRo, 2024-06-12:
was mir zu problem 1 eigenartig erscheint: ich sehe nirgendwo etwas von "max charge current:" im log - weil da wird der errechnete strom (pv charge current) an die wallbox gesendet. deswegen haut die mit 11kW rein, sobald das enable kommt, weil der errechnete ueberschuss niemals uebermittel wurde. (und dann geht sie natuerlich wieder aus wegen netzbezug).
just my thoughts, no solution ...

kasn24, 2024-06-12:
Jetzt im Tagesbetrieb wo ein ständiger Sonnen/Wolken Wechsel ist und auch die Küche unter Volllast läuft, funktioniert das drosseln und Phasenumschalten

```
[site  ] DEBUG 2024/06/12 13:47:16 ----
[lp-1  ] DEBUG 2024/06/12 13:47:16 charge power: 0W
[lp-1  ] DEBUG 2024/06/12 13:47:16 charge currents: [-0 -0 -0]A
[lp-2  ] DEBUG 2024/06/12 13:47:16 charge power: 11100W
[lp-2  ] DEBUG 2024/06/12 13:47:16 charge currents: [15.9 15.9 16]A
[site  ] DEBUG 2024/06/12 13:47:17 pv power: 13248W
[site  ] DEBUG 2024/06/12 13:47:17 battery soc: 98%
[site  ] DEBUG 2024/06/12 13:47:17 battery power: 164W
[site  ] DEBUG 2024/06/12 13:47:17 grid meter: -11W
[site  ] DEBUG 2024/06/12 13:47:17 grid powers: [-634 1258 -636]W
[site  ] DEBUG 2024/06/12 13:47:17 grid currents: [-3.11 5.48 -2.87]A
[site  ] DEBUG 2024/06/12 13:47:17 site power: 253W
[lp-2  ] DEBUG 2024/06/12 13:47:17 detected active phases: 3p
[lp-2  ] DEBUG 2024/06/12 13:47:17 !! session: chargeRater.chargedEnergy=0.6 - chargedAtStartup=0.0
[lp-2  ] DEBUG 2024/06/12 13:47:17 charge total import: 28.900kWh
[lp-2  ] DEBUG 2024/06/12 13:47:17 charger status: C
[lp-2  ] DEBUG 2024/06/12 13:47:17 vehicle api refresh
[lp-2  ] DEBUG 2024/06/12 13:47:17 pv charge current: min 6A > 5.63A (253W @ 3p, battery: true)
[site  ] DEBUG 2024/06/12 13:47:46 ----
[lp-1  ] DEBUG 2024/06/12 13:47:46 charge power: 0W
[lp-1  ] DEBUG 2024/06/12 13:47:46 charge currents: [-0 -0 -0]A
[lp-2  ] DEBUG 2024/06/12 13:47:46 charge power: 11100W
[lp-2  ] DEBUG 2024/06/12 13:47:47 charge currents: [15.9 15.9 16]A
[site  ] DEBUG 2024/06/12 13:47:47 pv power: 10411W
[site  ] DEBUG 2024/06/12 13:47:47 battery soc: 98%
[site  ] DEBUG 2024/06/12 13:47:47 battery power: 1637W
[site  ] DEBUG 2024/06/12 13:47:47 grid meter: -582W
[site  ] DEBUG 2024/06/12 13:47:47 grid powers: [-211 -169 -202]W
[site  ] DEBUG 2024/06/12 13:47:47 grid currents: [-1.82 -1.33 -1.29]A
[site  ] DEBUG 2024/06/12 13:47:47 site power: 1155W
[lp-1  ] DEBUG 2024/06/12 13:47:47 !! session: chargeRater.chargedEnergy=1.7 - chargedAtStartup=0.0
[lp-1  ] DEBUG 2024/06/12 13:47:47 charger status: B
[site  ] DEBUG 2024/06/12 13:48:16 ----
[lp-1  ] DEBUG 2024/06/12 13:48:16 charge power: 0W
[lp-1  ] DEBUG 2024/06/12 13:48:16 charge currents: [-0 -0 -0]A
[lp-2  ] DEBUG 2024/06/12 13:48:16 charge power: 11200W
[lp-2  ] DEBUG 2024/06/12 13:48:16 charge currents: [16 15.9 16]A
[site  ] DEBUG 2024/06/12 13:48:16 pv power: 11347W
[site  ] DEBUG 2024/06/12 13:48:17 battery soc: 98%
[site  ] DEBUG 2024/06/12 13:48:17 battery power: 1816W
[site  ] DEBUG 2024/06/12 13:48:17 grid meter: 1374W
[site  ] DEBUG 2024/06/12 13:48:17 grid powers: [-284 1931 -273]W
[site  ] DEBUG 2024/06/12 13:48:17 grid currents: [-1.94 8.38 -1.48]A
[site  ] DEBUG 2024/06/12 13:48:17 site power: 3290W
[lp-2  ] DEBUG 2024/06/12 13:48:17 detected active phases: 3p
[lp-2  ] DEBUG 2024/06/12 13:48:17 !! session: chargeRater.chargedEnergy=0.8 - chargedAtStartup=0.0
[lp-2  ] DEBUG 2024/06/12 13:48:17 charge total import: 28.900kWh
[lp-2  ] DEBUG 2024/06/12 13:48:17 charger status: C
[lp-2  ] DEBUG 2024/06/12 13:48:17 vehicle api refresh
[lp-2  ] DEBUG 2024/06/12 13:48:17 pv charge current: min 6A > 1.23A (3290W @ 3p, battery: true)
[site  ] DEBUG 2024/06/12 13:48:46 ----
[lp-1  ] DEBUG 2024/06/12 13:48:46 charge power: 0W
[lp-1  ] DEBUG 2024/06/12 13:48:46 charge currents: [-0 -0 -0]A
[lp-2  ] DEBUG 2024/06/12 13:48:46 charge power: 11100W
[lp-2  ] DEBUG 2024/06/12 13:48:47 charge currents: [15.9 15.8 15.9]A
[site  ] DEBUG 2024/06/12 13:48:47 pv power: 17920W
[site  ] DEBUG 2024/06/12 13:48:47 battery soc: 98%
[site  ] DEBUG 2024/06/12 13:48:47 battery power: -2832W
[site  ] DEBUG 2024/06/12 13:48:47 grid meter: -2926W
[site  ] DEBUG 2024/06/12 13:48:47 grid powers: [-1449 -33 -1444]W
[site  ] DEBUG 2024/06/12 13:48:47 grid currents: [-6.25 -0.834 -6.41]A
[site  ] DEBUG 2024/06/12 13:48:47 site power: -5658W
[lp-1  ] DEBUG 2024/06/12 13:48:47 !! session: chargeRater.chargedEnergy=1.7 - chargedAtStartup=0.0
[lp-1  ] DEBUG 2024/06/12 13:48:47 charger status: B
[site  ] DEBUG 2024/06/12 13:49:16 ----
[lp-1  ] DEBUG 2024/06/12 13:49:16 charge power: 0W
[lp-1  ] DEBUG 2024/06/12 13:49:16 charge currents: [-0 -0 -0]A
[... 146 lines cut ...]
[site  ] DEBUG 2024/06/12 13:53:17 grid currents: [6.98 3.14 -9.3]A
[site  ] DEBUG 2024/06/12 13:53:17 site power: 806W
[lp-2  ] DEBUG 2024/06/12 13:53:17 detected active phases: 1p
[lp-2  ] DEBUG 2024/06/12 13:53:17 !! session: chargeRater.chargedEnergy=1.2 - chargedAtStartup=0.0
[lp-2  ] DEBUG 2024/06/12 13:53:17 charge total import: 28.900kWh
[lp-2  ] DEBUG 2024/06/12 13:53:17 charger status: C
[lp-2  ] DEBUG 2024/06/12 13:53:17 vehicle api refresh
[lp-2  ] DEBUG 2024/06/12 13:53:17 pv charge current: 12.5A = 16A + -3.51A (806W @ 1p)
[lp-2  ] DEBUG 2024/06/12 13:53:17 max charge current: 12.5A
[site  ] DEBUG 2024/06/12 13:53:46 ----
[lp-1  ] DEBUG 2024/06/12 13:53:46 charge power: 0W
[lp-1  ] DEBUG 2024/06/12 13:53:46 charge currents: [-0 -0 -0]A
[lp-2  ] DEBUG 2024/06/12 13:53:46 charge power: 2800W
[lp-2  ] DEBUG 2024/06/12 13:53:46 charge currents: [12 0 0]A
[site  ] DEBUG 2024/06/12 13:53:46 pv power: 6316W
[site  ] DEBUG 2024/06/12 13:53:46 battery soc: 99%
[site  ] DEBUG 2024/06/12 13:53:46 battery power: -503W
[site  ] DEBUG 2024/06/12 13:53:47 grid meter: 72W
[site  ] DEBUG 2024/06/12 13:53:47 grid powers: [999 927 -1854]W
[site  ] DEBUG 2024/06/12 13:53:47 grid currents: [4.51 4.21 -7.9]A
[site  ] DEBUG 2024/06/12 13:53:47 site power: -331W
[lp-1  ] DEBUG 2024/06/12 13:53:47 !! session: chargeRater.chargedEnergy=1.7 - chargedAtStartup=0.0
[lp-1  ] DEBUG 2024/06/12 13:53:47 charger status: B

```

RenatusRo, 2024-06-12:
ja, jetzt sieht man ja auch, dass da immer "max charge current:" steht, also die box auch echt angesteuert wird.

und dass passiert, wenn sich der zur verfuegung stehende ladestrom aendert und er groesser als effectiveMinCurrent ist:
```
	// set current
	if chargeCurrent != lp.chargeCurrent && chargeCurrent >= lp.effectiveMinCurrent() {
           .....

```
eine der beiden bedingungen muss bei dir am morgen da im wege stehen.

kasn24, 2024-06-12:
und es geht munter weiter - die Küche beendet den Betrieb und wir haben 5-16kW Überschuss zusätzlich zu den 3,7kW einphasigen laden. Keine Anzeichen auf dreiphasig umzuschalten über mehr als 10min. Erst ein manuelles "aus" und wieder auf "PV" ermöglicht den 3-phasigen Betrieb

```
[site  ] DEBUG 2024/06/12 14:42:16 ----
[lp-1  ] DEBUG 2024/06/12 14:42:16 charge power: 0W
[lp-1  ] DEBUG 2024/06/12 14:42:16 charge currents: [-0 -0 -0]A
[lp-2  ] DEBUG 2024/06/12 14:42:16 charge power: 3600W
[lp-2  ] DEBUG 2024/06/12 14:42:16 charge currents: [15.7 0.2 0.2]A
[site  ] DEBUG 2024/06/12 14:42:16 pv power: 19847W
[site  ] DEBUG 2024/06/12 14:42:17 battery soc: 100%
[site  ] DEBUG 2024/06/12 14:42:17 battery power: 0W
[site  ] DEBUG 2024/06/12 14:42:17 grid meter: -15954W
[site  ] DEBUG 2024/06/12 14:42:17 grid powers: [-2810 -6559 -6584]W
[site  ] DEBUG 2024/06/12 14:42:17 grid currents: [-14.3 -28.9 -29.1]A
[site  ] DEBUG 2024/06/12 14:42:17 site power: -15854W
[lp-2  ] DEBUG 2024/06/12 14:42:17 detected active phases: 1p
[lp-2  ] DEBUG 2024/06/12 14:42:17 !! session: chargeRater.chargedEnergy=3.7 - chargedAtStartup=0.0
[lp-2  ] DEBUG 2024/06/12 14:42:17 charge total import: 32.100kWh
[lp-2  ] DEBUG 2024/06/12 14:42:17 charger status: C
[lp-2  ] DEBUG 2024/06/12 14:42:17 pv charge current: 84.9A = 16A + 68.9A (-15854W @ 1p)
[site  ] DEBUG 2024/06/12 14:42:46 ----
[lp-1  ] DEBUG 2024/06/12 14:42:46 charge power: 0W
[lp-1  ] DEBUG 2024/06/12 14:42:46 charge currents: [-0 -0 -0]A
[lp-2  ] DEBUG 2024/06/12 14:42:46 charge power: 3600W
[lp-2  ] DEBUG 2024/06/12 14:42:46 charge currents: [15.8 0.2 0.2]A
[site  ] DEBUG 2024/06/12 14:42:47 pv power: 11154W
[site  ] DEBUG 2024/06/12 14:42:47 battery soc: 100%
[site  ] DEBUG 2024/06/12 14:42:47 battery power: 0W
[site  ] DEBUG 2024/06/12 14:42:47 grid meter: -7231W
[site  ] DEBUG 2024/06/12 14:42:47 grid powers: [84 -3651 -3664]W
[site  ] DEBUG 2024/06/12 14:42:47 grid currents: [1.51 -15.5 -15.6]A
[site  ] DEBUG 2024/06/12 14:42:47 site power: -7131W
[lp-1  ] DEBUG 2024/06/12 14:42:47 !! session: chargeRater.chargedEnergy=1.7 - chargedAtStartup=0.0
[lp-1  ] DEBUG 2024/06/12 14:42:47 charger status: B
[site  ] DEBUG 2024/06/12 14:43:16 ----
[lp-1  ] DEBUG 2024/06/12 14:43:16 charge power: 0W
[lp-1  ] DEBUG 2024/06/12 14:43:16 charge currents: [-0 -0 -0]A
[lp-2  ] DEBUG 2024/06/12 14:43:16 charge power: 3600W
[lp-2  ] DEBUG 2024/06/12 14:43:16 charge currents: [15.7 0.2 0.2]A
[site  ] DEBUG 2024/06/12 14:43:16 pv power: 10629W
[site  ] DEBUG 2024/06/12 14:43:16 battery soc: 100%
[site  ] DEBUG 2024/06/12 14:43:16 battery power: 0W
[site  ] DEBUG 2024/06/12 14:43:16 grid meter: -6599W
[site  ] DEBUG 2024/06/12 14:43:16 grid powers: [296 -3417 -3478]W
[site  ] DEBUG 2024/06/12 14:43:16 grid currents: [2.08 -14.6 -14.8]A
[site  ] DEBUG 2024/06/12 14:43:16 site power: -6499W
[lp-2  ] DEBUG 2024/06/12 14:43:16 detected active phases: 1p
[lp-2  ] DEBUG 2024/06/12 14:43:17 !! session: chargeRater.chargedEnergy=3.7 - chargedAtStartup=0.0
[lp-2  ] DEBUG 2024/06/12 14:43:17 charge total import: 32.100kWh
[lp-2  ] DEBUG 2024/06/12 14:43:17 charger status: C
[lp-2  ] DEBUG 2024/06/12 14:43:17 pv charge current: 44.3A = 16A + 28.3A (-6499W @ 1p)
[site  ] DEBUG 2024/06/12 14:43:46 ----
[lp-1  ] DEBUG 2024/06/12 14:43:46 charge power: 0W
[lp-1  ] DEBUG 2024/06/12 14:43:46 charge currents: [-0 -0 -0]A
[lp-2  ] DEBUG 2024/06/12 14:43:46 charge power: 3500W
[lp-2  ] DEBUG 2024/06/12 14:43:47 charge currents: [15.3 0.2 0.2]A
[site  ] DEBUG 2024/06/12 14:43:47 pv power: 10447W
[site  ] DEBUG 2024/06/12 14:43:47 battery soc: 100%
[site  ] DEBUG 2024/06/12 14:43:47 battery power: 0W
[site  ] DEBUG 2024/06/12 14:43:47 grid meter: -6493W
[site  ] DEBUG 2024/06/12 14:43:47 grid powers: [252 -3341 -3403]W
[site  ] DEBUG 2024/06/12 14:43:47 grid currents: [1.89 -14.2 -14.5]A
[site  ] DEBUG 2024/06/12 14:43:47 site power: -6393W
[lp-1  ] DEBUG 2024/06/12 14:43:47 !! session: chargeRater.chargedEnergy=1.7 - chargedAtStartup=0.0
[lp-1  ] DEBUG 2024/06/12 14:43:47 charger status: B
[site  ] DEBUG 2024/06/12 14:44:16 ----
[lp-1  ] DEBUG 2024/06/12 14:44:16 charge power: 0W
[lp-1  ] DEBUG 2024/06/12 14:44:16 charge currents: [-0 -0 -0]A
[lp-2  ] DEBUG 2024/06/12 14:44:16 charge power: 3600W
[lp-2  ] DEBUG 2024/06/12 14:44:16 charge currents: [15.7 0.2 0.2]A
[... 178 lines cut ...]
[site  ] DEBUG 2024/06/12 14:48:17 grid powers: [55 26 19]W
[site  ] DEBUG 2024/06/12 14:48:17 grid currents: [1.66 1.08 0.924]A
[site  ] DEBUG 2024/06/12 14:48:17 site power: 1108W
[lp-2  ] DEBUG 2024/06/12 14:48:17 !! session: chargeRater.chargedEnergy=0.1 - chargedAtStartup=0.0
[lp-2  ] DEBUG 2024/06/12 14:48:17 charge total import: 32.700kWh
[lp-2  ] DEBUG 2024/06/12 14:48:17 charger status: C
[lp-2  ] INFO 2024/06/12 14:48:17 start charging ->
[lp-2  ] DEBUG 2024/06/12 14:48:17 wake-up timer: stop
[lp-2  ] DEBUG 2024/06/12 14:48:17 pv charge current: min 6A > 1.18A (1108W @ 1p, battery: true)
[site  ] DEBUG 2024/06/12 14:48:46 ----
[lp-1  ] DEBUG 2024/06/12 14:48:46 charge power: 0W
[lp-1  ] DEBUG 2024/06/12 14:48:46 charge currents: [-0 -0 -0]A
[lp-2  ] DEBUG 2024/06/12 14:48:46 charge power: 11200W
[lp-2  ] DEBUG 2024/06/12 14:48:47 charge currents: [16 15.9 16]A
[site  ] DEBUG 2024/06/12 14:48:47 pv power: 10774W
[site  ] DEBUG 2024/06/12 14:48:47 battery soc: 100%
[site  ] DEBUG 2024/06/12 14:48:47 battery power: 852W
[site  ] DEBUG 2024/06/12 14:48:47 grid meter: -170W
[site  ] DEBUG 2024/06/12 14:48:47 grid powers: [-44 -62 -64]W
[site  ] DEBUG 2024/06/12 14:48:47 grid currents: [-1.67 -1.13 -0.974]A
[site  ] DEBUG 2024/06/12 14:48:47 site power: 782W
[lp-1  ] DEBUG 2024/06/12 14:48:47 !! session: chargeRater.chargedEnergy=1.7 - chargedAtStartup=0.0
[lp-1  ] DEBUG 2024/06/12 14:48:47 charger status: B

```

andig, 2024-06-12:
Bevor wir ein Suchspiel machen- was ist der relevante Teil des Logs?

andig, 2024-06-12:
Soweit ich sehe, soll hier 1p Laden starten:

```
[lp-2  ] DEBUG 2024/06/12 08:38:17 pv charge current: 7.35A = 0A + 7.35A (-1692W @ 1p)
[lp-2  ] DEBUG 2024/06/12 08:38:17 site power -1692W <= 0W enable threshold
```

Da die Box keine Phasenumschaltung hat: wie soll das funktionieren? Wo kommt das 1p her?

RenatusRo, 2024-06-12:
> Da die Box keine Phasenumschaltung hat: wie soll das funktionieren? Wo kommt das 1p her?

Hat sie doch aber?:
```
template: em2go-home
products:
  - brand: EM2GO
    description:
      generic: Home
capabilities: ["1p3p", "mA"]
```

kasn24, 2024-06-12:
Ja sie hat eine Phasenumschaltung und sie schaltet auch teilweise um. Aber eben im ersten Beispiel nicht nach unten, dort regelt sie die Last nicht mal auf 4,6kW bei 3p sondern stoppt einfach nur bevor der Spaß wieder von vorne los geht.
Und am Nachmittag geht sie nicht mehr von 1p auf 3p hoch.

andig, 2024-06-12:
Aso ja, die home. Mir sind das zuviele Logs durcheinander. Können wir mal auf ein relevantes schauen und auf den relevanten Ladepunkt? Wie sieht das aus?

andig, 2024-06-12:
This is confusing:

```
[lp-2  ] DEBUG 2024/06/12 08:39:16 charge currents: [16 16 16.1]A
[lp-2  ] DEBUG 2024/06/12 08:39:17 pv charge current: 0A = 6A + -33A (7591W @ 1p)
```

I don't see how we can end up with 1p detected here:

```go
func (lp *Loadpoint) ActivePhases() int {
	physical := lp.GetPhases()
	vehicle := lp.getVehiclePhases()
	measured := lp.getMeasuredPhases()
	charger := lp.getChargerPhysicalPhases()

	active := min(expect(vehicle), expect(physical), expect(measured), expect(charger))

	// sanity check - we should not assume less active phases than actually measured
	if measured > 0 && active < measured {
		lp.log.WARN.Printf("phase mismatch between %dp measured for %dp vehicle and %dp charger", measured, vehicle, physical)
	}

	return active
}
```

Einer der Werte muss 1 sein, vmtl. sogar der measured Wert. Nur wie kann das sein? Wenn uns nix Besseres einfällt muss da nochmal ein Logging rein.

/cc @GrimmiMeloni

kasn24, 2024-06-12:
> Logging ist morgen im nightly, dann bitte neu testen (nicht auf HA...)

Aufforderung an mich ? 
Ich kann morgen früh keine Logs exportieren, erst Freitag früh wieder. Nicht auf HomeAssistant testen? Sondern?

GrimmiMeloni, 2024-06-14:
@kasn24 Das Log ist entweder unvollständig, oder zu kurz. Die entscheidende Information bzgl. der Phasenerkennung fehlt. Bitte mal ein Log teilen, das vor dem Ladestart bzw. Einstecken des Fahrzeugs beginnt, und auch den Ladevorgang mit den 11kW für ein paar Minuten aufzeichnet.

kasn24, 2024-06-14:
Sorry, beim letzten Mal waren die Logs zu lang... 
Ich versuche es gleich zu reproduzieren, sobald etwas mehr Sonne raus kommt. 
Zur Verständniskorrektur, wie in den obigen Logs auch: es wird nicht über mehrere Minuten mit 11kW geladen. Es wird der Ladevorgang abgebrochen, bevor gedrosselt wird. 

Mir scheint es aus Sicht eines Laien, die Wallbox erhält nur den Befehl "start" und geht selbstständig mit der Leistung hoch ohne eine Maximalgrenze von evcc zu bearbeiten. 
evcc merkt es stimmt was nicht und es stoppt einfach, bevor er runter regeln kann.

kasn24, 2024-06-14:
so habe die Wallbox von Aus PV gestellt um 08:50 Uhr. 
das "stop charging <-" passiert automatisch, da mache ich nichts. In den folgenden Logs wurde das ganz 3 mal simuliert, bis ich den Modus der Wallbox wieder auf Aus gestellt hat.

Gefiltert auf LP-2
```
[lp-2  ] DEBUG 2024/06/14 08:48:51 charge power: 0W
[lp-2  ] DEBUG 2024/06/14 08:48:52 charge currents: [0 0 0]A
[lp-2  ] DEBUG 2024/06/14 08:48:52 charge total import: 50.000kWh
[lp-2  ] DEBUG 2024/06/14 08:48:52 charger status: B
[lp-2  ] DEBUG 2024/06/14 08:49:21 charge power: 0W
[lp-2  ] DEBUG 2024/06/14 08:49:22 charge currents: [0 0 0]A
[lp-2  ] DEBUG 2024/06/14 08:49:51 charge power: 0W
[lp-2  ] DEBUG 2024/06/14 08:49:52 charge currents: [0 0 0]A
[lp-2  ] DEBUG 2024/06/14 08:49:52 charge total import: 50.000kWh
[lp-2  ] DEBUG 2024/06/14 08:49:52 charger status: B
[lp-2  ] **DEBUG 2024/06/14 08:50:09 set charge mode: pv**
[lp-2  ] DEBUG 2024/06/14 08:50:09 charge power: 0W
[lp-2  ] DEBUG 2024/06/14 08:50:09 charge currents: [0 0 0]A
[lp-2  ] DEBUG 2024/06/14 08:50:10 charge total import: 50.000kWh
[lp-2  ] DEBUG 2024/06/14 08:50:10 charger status: B
[lp-2  ] DEBUG 2024/06/14 08:50:10 available power 1681W < 4140W min 3p threshold
**[lp-2  ] DEBUG 2024/06/14 08:50:10 phase scale1p in 0s
[lp-2  ] DEBUG 2024/06/14 08:50:10 switched phases: 1p**
[lp-2  ] DEBUG 2024/06/14 08:50:10 phase timer inactive
[lp-2  ] DEBUG 2024/06/14 08:50:10 pv charge current: 7.31A = 0A + 7.31A (-1681W @ 1p)
[lp-2  ] DEBUG 2024/06/14 08:50:10 site power -1681W <= 0W enable threshold
[lp-2  ] DEBUG 2024/06/14 08:50:10 pv enable in 0s
[lp-2  ] DEBUG 2024/06/14 08:50:10 pv enable timer elapsed
[lp-2  ] DEBUG 2024/06/14 08:50:10 max charge current: 6A
[lp-2  ] DEBUG 2024/06/14 08:50:10 charger enable
[lp-2  ] DEBUG 2024/06/14 08:50:10 wake-up timer: start
[lp-2  ] DEBUG 2024/06/14 08:50:21 charge power: 0W
[lp-2  ] DEBUG 2024/06/14 08:50:22 charge currents: [0.5 0.2 0.2]A
[lp-2  ] DEBUG 2024/06/14 08:50:51 charge power: 11100W
[lp-2  ] DEBUG 2024/06/14 08:50:52 charge currents: [15.9 15.9 16]A
[lp-2  ] DEBUG 2024/06/14 08:50:52 charge total import: 50.000kWh
[lp-2  ] DEBUG 2024/06/14 08:50:52 charger status: C
[lp-2  ] INFO 2024/06/14 08:50:52 start charging ->
[lp-2  ] DEBUG 2024/06/14 08:50:52 wake-up timer: stop
[lp-2  ] DEBUG 2024/06/14 08:50:52 pv charge current: 0A = 6A + -41A (9436W @ 1p)
[lp-2  ] DEBUG 2024/06/14 08:50:52 projected site power 9436W >= 0W disable threshold
[lp-2  ] DEBUG 2024/06/14 08:50:52 pv disable in 0s
[lp-2  ] DEBUG 2024/06/14 08:50:52 pv disable timer elapsed
[lp-2  ] DEBUG 2024/06/14 08:50:52 charger disable
[lp-2  ] DEBUG 2024/06/14 08:50:52 wake-up timer: stop
[lp-2  ] DEBUG 2024/06/14 08:51:21 charge power: 0W
[lp-2  ] DEBUG 2024/06/14 08:51:22 charge currents: [0 0 0]A
[lp-2  ] DEBUG 2024/06/14 08:51:51 charge power: 0W
[lp-2  ] DEBUG 2024/06/14 08:51:52 charge currents: [0 0 0]A
[lp-2  ] DEBUG 2024/06/14 08:51:52 charge total import: 50.000kWh
[lp-2  ] DEBUG 2024/06/14 08:51:52 charger status: B
[lp-2  ] INFO 2024/06/14 08:51:52 stop charging <-
[lp-2  ] DEBUG 2024/06/14 08:51:52 charge total import: 50.000kWh
[lp-2  ] DEBUG 2024/06/14 08:51:52 !! session: start=50.000 stop=50.000 chargedEnergy=0.000
[lp-2  ] DEBUG 2024/06/14 08:51:52 pv charge current: 11.2A = 0A + 11.2A (-2568W @ 1p)
[lp-2  ] DEBUG 2024/06/14 08:51:52 site power -2568W <= 0W enable threshold
[lp-2  ] DEBUG 2024/06/14 08:51:52 pv enable in 0s
[lp-2  ] DEBUG 2024/06/14 08:51:52 pv enable timer elapsed
[lp-2  ] DEBUG 2024/06/14 08:51:52 charger enable
[lp-2  ] DEBUG 2024/06/14 08:51:52 wake-up timer: start
[lp-2  ] DEBUG 2024/06/14 08:52:21 charge power: 11100W
[lp-2  ] DEBUG 2024/06/14 08:52:22 charge currents: [16 15.9 16]A
[lp-2  ] DEBUG 2024/06/14 08:52:51 charge power: 11100W
[lp-2  ] DEBUG 2024/06/14 08:52:52 charge currents: [15.9 15.9 15.9]A
[lp-2  ] DEBUG 2024/06/14 08:52:52 charge total import: 50.000kWh
[lp-2  ] DEBUG 2024/06/14 08:52:52 charger status: C
[lp-2  ] INFO 2024/06/14 08:52:52 start charging ->
[lp-2  ] DEBUG 2024/06/14 08:52:52 wake-up timer: stop
[lp-2  ] DEBUG 2024/06/14 08:52:52 pv charge current: 0A = 6A + -37.5A (8628W @ 1p)
[lp-2  ] DEBUG 2024/06/14 08:52:52 projected site power 8628W >= 0W disable threshold
[... 302 lines cut ...]
[site  ] DEBUG 2024/06/14 08:54:52 battery power: 6582W
[site  ] DEBUG 2024/06/14 08:54:52 grid meter: -9122W
[site  ] DEBUG 2024/06/14 08:54:52 grid powers: [-3024 -3048 -3050]W
[site  ] DEBUG 2024/06/14 08:54:52 grid currents: [-13.1 -13.1 -13.1]A
[site  ] DEBUG 2024/06/14 08:54:52 site power: -2440W
[lp-2  ] DEBUG 2024/06/14 08:54:52 charge total import: 50.000kWh
[lp-2  ] DEBUG 2024/06/14 08:54:52 charger status: B
[lp-2  ] INFO 2024/06/14 08:54:52 stop charging <-
[lp-2  ] DEBUG 2024/06/14 08:54:52 charge total import: 50.000kWh
[lp-2  ] DEBUG 2024/06/14 08:54:52 !! session: start=50.000 stop=50.000 chargedEnergy=0.000
[site  ] DEBUG 2024/06/14 08:55:21 ----
[lp-1  ] DEBUG 2024/06/14 08:55:21 charge power: 0W
[lp-1  ] DEBUG 2024/06/14 08:55:21 charge currents: [-0 -0 -0]A
[lp-2  ] DEBUG 2024/06/14 08:55:21 charge power: 0W
[lp-2  ] DEBUG 2024/06/14 08:55:22 charge currents: [0 0 0]A
[site  ] DEBUG 2024/06/14 08:55:22 pv power: 2933W
[site  ] DEBUG 2024/06/14 08:55:22 battery soc: 82%
[site  ] DEBUG 2024/06/14 08:55:22 battery power: -2716W
[site  ] DEBUG 2024/06/14 08:55:22 grid meter: 3W
[site  ] DEBUG 2024/06/14 08:55:22 grid powers: [-2 8 -3]W
[site  ] DEBUG 2024/06/14 08:55:22 grid currents: [-1.02 0.586 -0.472]A
[site  ] DEBUG 2024/06/14 08:55:22 site power: -2613W
[lp-1  ] DEBUG 2024/06/14 08:55:22 !! session: chargeRater.chargedEnergy=2.7 - chargedAtStartup=2.2
[lp-1  ] DEBUG 2024/06/14 08:55:22 charger status: B
```

RenatusRo, 2024-06-14:
sieht so aus, als wuerde evcc alles richtig machen: fragt 1p/6A
```
[lp-2  ] DEBUG 2024/06/14 08:50:10 switched phases: 1p
[lp-2  ] DEBUG 2024/06/14 08:50:10 max charge current: 6A
[lp-2  ] DEBUG 2024/06/14 08:50:10 charger enable
[lp-2  ] DEBUG 2024/06/14 08:50:10 wake-up timer: start
```
aber wallbox macht 3p/16A
```
[lp-2  ] DEBUG 2024/06/14 08:50:51 charge power: 11100W
[lp-2  ] DEBUG 2024/06/14 08:50:52 charge currents: [15.9 15.9 16]A
```

vielleicht noch mal alles mit settings
```
log: debug
levels:
  modbus: trace
  em2go-home: trace
```

kasn24, 2024-06-14:
> 
> vielleicht noch mal alles mit settings
> 
> ```
> log: debug
> levels:
>   modbus: trace
>   em2go-home: trace
> ```
Soll ich das in den Settings unter lp-2 hinzufügen?

Eben hatten wir das Spiel, lp-2 steigt mit 11kW ein (obwohl weniger sinnvoll wäre). Daraufhin wird lp-1 gedrosselt und lp-2 bricht wieder ab.
Dann wo genug Sonne war, war lp-2 lang genug am laden um gedrosselt zu werden und sich im Laden endlich dem Überschuss anzupassen. Es muss also immer erstmal ein paar Sekunden laden, damit die der lp-2 die Einstellungen von evcc annimmt. Erstmal macht er was er will, sobald er den Startbefehl bekommt.

```
[lp-2  ] DEBUG 2024/06/14 11:24:52 !! session: chargeRater.chargedEnergy=0.1 - chargedAtStartup=0.0
[lp-2  ] DEBUG 2024/06/14 11:24:52 charge total import: 50.000kWh
[lp-2  ] DEBUG 2024/06/14 11:24:52 charger status: B
[lp-2  ] DEBUG 2024/06/14 11:24:52 pv charge current: 10A = 0A + 10A (-6915W @ 3p)
[lp-2  ] DEBUG 2024/06/14 11:24:52 site power -6915W <= 0W enable threshold
[lp-2  ] DEBUG 2024/06/14 11:24:52 pv enable in 48s
[lp-2  ] DEBUG 2024/06/14 11:24:52 pv enable timer remaining: 48s
[lp-2  ] DEBUG 2024/06/14 11:25:21 charge power: 0W
[lp-2  ] DEBUG 2024/06/14 11:25:22 charge currents: [0 0 0]A
[lp-2  ] DEBUG 2024/06/14 11:25:51 charge power: 0W
[lp-2  ] DEBUG 2024/06/14 11:25:52 charge currents: [0 0 0]A
[lp-2  ] DEBUG 2024/06/14 11:25:52 !! session: chargeRater.chargedEnergy=0.1 - chargedAtStartup=0.0
[lp-2  ] DEBUG 2024/06/14 11:25:52 charge total import: 50.000kWh
[lp-2  ] DEBUG 2024/06/14 11:25:52 charger status: B
[lp-2  ] DEBUG 2024/06/14 11:25:52 pv charge current: 16.8A = 0A + 16.8A (-11576W @ 3p)
[lp-2  ] DEBUG 2024/06/14 11:25:52 site power -11576W <= 0W enable threshold
[lp-2  ] DEBUG 2024/06/14 11:25:52 pv enable in 0s
[lp-2  ] DEBUG 2024/06/14 11:25:52 pv enable timer elapsed
[lp-2  ] DEBUG 2024/06/14 11:25:52 charger enable
[lp-2  ] DEBUG 2024/06/14 11:25:52 wake-up timer: start
[lp-2  ] DEBUG 2024/06/14 11:26:31 charge power: 11100W
[lp-2  ] DEBUG 2024/06/14 11:26:32 charge currents: [15.9 15.9 16]A
[lp-2  ] DEBUG 2024/06/14 11:26:51 charge power: 11100W
[lp-2  ] DEBUG 2024/06/14 11:26:52 charge currents: [15.9 15.9 16]A
[lp-2  ] DEBUG 2024/06/14 11:26:52 !! session: chargeRater.chargedEnergy=0.1 - chargedAtStartup=0.0
[lp-2  ] DEBUG 2024/06/14 11:26:52 charge total import: 50.000kWh
[lp-2  ] DEBUG 2024/06/14 11:26:52 charger status: C
[lp-2  ] INFO 2024/06/14 11:26:52 start charging ->
[lp-2  ] DEBUG 2024/06/14 11:26:52 wake-up timer: stop
[lp-2  ] DEBUG 2024/06/14 11:26:52 pv charge current: 7.58A = 6A + 1.58A (-1092W @ 3p)
[lp-2  ] DEBUG 2024/06/14 11:26:52 pv timer reset
[lp-2  ] DEBUG 2024/06/14 11:26:52 pv timer inactive
[lp-2  ] DEBUG 2024/06/14 11:26:52 max charge current: 7.58A
[lp-2  ] DEBUG 2024/06/14 11:27:21 charge power: 4800W
[lp-2  ] DEBUG 2024/06/14 11:27:22 charge currents: [7 6.9 7]A
[lp-2  ] DEBUG 2024/06/14 11:27:51 charge power: 4900W
[lp-2  ] DEBUG 2024/06/14 11:27:52 charge currents: [7 7 7]A
[lp-2  ] DEBUG 2024/06/14 11:27:52 detected active phases: 3p
[lp-2  ] DEBUG 2024/06/14 11:27:52 !! session: chargeRater.chargedEnergy=0.2 - chargedAtStartup=0.0
[lp-2  ] DEBUG 2024/06/14 11:27:52 charge total import: 50.000kWh
[lp-2  ] DEBUG 2024/06/14 11:27:52 charger status: C
[lp-2  ] DEBUG 2024/06/14 11:27:52 pv charge current: 7.75A = 7.58A + 0.171A (-118W @ 3p)
[lp-2  ] DEBUG 2024/06/14 11:27:52 max charge current: 7.75A
[lp-2  ] DEBUG 2024/06/14 11:28:21 charge power: 4800W
[lp-2  ] DEBUG 2024/06/14 11:28:22 charge currents: [7 6.9 7]A
[lp-2  ] DEBUG 2024/06/14 11:28:51 charge power: 4900W
[lp-2  ] DEBUG 2024/06/14 11:28:52 charge currents: [7 7 7]A
[lp-2  ] DEBUG 2024/06/14 11:28:52 detected active phases: 3p
[lp-2  ] DEBUG 2024/06/14 11:28:52 !! session: chargeRater.chargedEnergy=0.2 - chargedAtStartup=0.0
[lp-2  ] DEBUG 2024/06/14 11:28:52 charge total import: 50.000kWh
[lp-2  ] DEBUG 2024/06/14 11:28:52 charger status: C
[lp-2  ] DEBUG 2024/06/14 11:28:52 pv charge current: 7.64A = 7.75A + -0.109A (75W @ 3p)
[lp-2  ] DEBUG 2024/06/14 11:28:52 max charge current: 7.64A

```

RenatusRo, 2024-06-14:
bei den gobalen settings ganz oben.
aber mit der reaktionszeit - good point: stell doch mal 
`interval: 90s`
ein. 
regelung funktioniert immer nur so gut, wie das langsamste glied in der kette.
(wir haben hier eine em2go pro - die hat anfangs auch gezickt)

GrimmiMeloni, 2024-06-14:
90s ist aber heftig. Bei 2 Loadpoints wird der einzelne LP nur alle 3 Minuten geregelt. Das könnte bereits zu lang für den PV Timer sein (je nach Setting). Ggf. also eher mit 45s Interval probieren. 
Aber ganz ehrlich, wenn ein Charger innerhalb von 1 Minuten nicht reagiert, ist das nicht normal.

kasn24, 2024-06-14:
Hmm.. könnte ein Netzwerkproblem die Ursache sein oder würde evcc dies bemerken?

RenatusRo, 2024-06-14:
sollte es ein allgemeineres em2go problem sein (von wegen langsam oder ignorant?)
-> https://github.com/evcc-io/evcc/discussions/14361

kasn24, 2024-06-14:
Danke an alle Beteiligten und Mitlesenden! 
Ich werde es nach Release versuchen so schnell wie möglich zu testen und Feedback zu geben. Auch wie es sich dann mit der Phasenumschalten 1p zu 3p verhält.

Bzgl. dem oben genannten Problem mit der SMA Wallbox und MG die nicht abschalten, eröffne ich am besten ein eigenes Thema oder kennt hier jemand was vergleichbares? habe noch nichts passendes gefunden

andig, 2024-06-15:
@kasn24 kannst Du mal bitte probieren, ob Du- wenn die Umschaltung nicht geklappt hat- diese auf der Kommandozeile erzwingen kannst? Also mittels:

    evcc charger --phases 1|3

JensD71, 2024-06-16:
Habe das gleich problem. Der em2go home setzt beim Starten des Ladens immer den Max. Strom (16A) als Sollstrom ein.
Da beim Regeln nur ein neuer Wert in den em2go geschrieben wird wenn es eine Änderung gab, bleibt dieser Wert im Regelbetrieb teilweise sehr lange bestehen.
Lösung aus meiner Sicht: - Entweder Sollwert erst nach dem Starten des Ladens in den em2go schreiben oder Sollwert auch bei nicht geändertem Wert in den em2go schreiben.

GrimmiMeloni, 2024-06-16:
> Da beim Regeln nur ein neuer Wert in den em2go geschrieben wird wenn es eine Änderung gab, bleibt dieser Wert im Regelbetrieb teilweise sehr lange bestehen.

Korrekt. Charger mit "Eigenleben" sind immer unschön. Deshalb prüft der Loadpoint auch in jedem Intervall den vom Charger gemeldeten aktuellen Wert, sofern `CurrentGetter` auf dem Charger implementiert ist. Für den em2go ist das aktuell nicht der Fall, müßte also jemand nachziehen, dann wird evcc das nachregeln.

kasn24, 2024-06-16:
Noch könnte ich ihn zurückgeben. Bin ich echt am überlegen. Finde den ständigen Netzbezug echt nervig.
Bräuchte nur ne andere im ähnlichen Preissegment mit CEE Stecker. GoE vielleicht...

andig, 2024-06-16:
Also Strom nachlesen. Funktioniert die Phasenumschaltung?

JensD71, 2024-06-16:
Kann ich noch nicht testen, hab aktuell erst 1 Phase. ;-)

andig, 2024-06-16:
Strom kannst Du morgen im Nightly nachtesten. Jetzt sollten wir hoffentlich sehen, dass da mit 16A geladen wird und einschreiten.

andig, 2024-06-16:
> Noch könnte ich ihn zurückgeben. 

@kasn24 siehe oben. Ein Test würde helfen.

kasn24, 2024-06-16:
> @kasn24 siehe oben. Ein Test würde helfen.

Tut mir leid ich stehe glaube etwas auf dem Schlauch. 
Was testen? Die Änderung im Code? Nightly? - was heißt das? 

Konsolen Befehl kann ich keine geben, da mein evcc im HomeAssistant läuft.

andig, 2024-06-16:
Tja, dann wirds nix. Machen wir hier zu 😥

closed 2024-06-16 by andig

JensD71, 2024-06-16:
sieht gut aus.Danke.

  nicht schnell genug - bricht ab (Issue #14307)

kasn24, 2024-06-16:
Naja liegt jetzt woran? Weil mir die Begriffe noch nichts sagen? Dann gib mir irgendwo einen Link wo das vielleicht was sagt? 

Wegen fehlender Konsole? Es gibt ja noch welche mit dem gleichen Problem. Falls das sonst keiner kann, würde ich versuchen die kommende Woche evcc umzuziehen. 

Den korrigierten Code testen? Da muss ich wohl auf das nächste evcc Release warten?! 

Aber nur halbe Sätze antworten, lässt mich nicht nur mich dumm sterben, sondern löst auch nicht das Problem für zukünftige Nutzer der gleichen Wallbox. 

Ich bin ja willig zu testen, aber bin wie gesagt neu bei evcc und auch etwas abhängig von Sonne und Verfügbarkeit der Autos um die Szenarien zu testen.

JensD71, 2024-06-16:
update zu v0.127.2

kasn24, 2024-06-17:
Update erfolgt, Test leider immer noch negativ. 
Interval auf 45s

Kurzfassung: 
```
[lp-2 ] WARN 2024/06/17 08:11:48 charger logic error: current mismatch (got 16A, expected 6A)
``` 

Langfassung

```
[lp-2  ] DEBUG 2024/06/17 08:08:02 charge power: 0W
[lp-2  ] DEBUG 2024/06/17 08:08:02 charge currents: [0 0 0]A
[lp-2  ] DEBUG 2024/06/17 08:08:47 charge power: 0W
[lp-2  ] DEBUG 2024/06/17 08:08:47 charge currents: [0 0 0]A
[lp-2  ] DEBUG 2024/06/17 08:08:47 !! session: chargeRater.chargedEnergy=0.2 - chargedAtStartup=0.1
[lp-2  ] DEBUG 2024/06/17 08:08:48 charge total import: 85.700kWh
[lp-2  ] DEBUG 2024/06/17 08:08:48 charger status: B
[lp-2  ] INFO 2024/06/17 08:08:48 stop charging <-
[lp-2  ] DEBUG 2024/06/17 08:08:48 pv timer reset
[lp-2  ] DEBUG 2024/06/17 08:08:48 pv timer inactive
[lp-2  ] DEBUG 2024/06/17 08:08:48 charge total import: 85.700kWh
[lp-2  ] DEBUG 2024/06/17 08:08:48 !! session: start=85.700 stop=85.700 chargedEnergy=0.100
[lp-2  ] DEBUG 2024/06/17 08:08:48 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[lp-2  ] DEBUG 2024/06/17 08:08:48 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[lp-2  ] DEBUG 2024/06/17 08:08:48 pv charge current: 6.33A = 0A + 6.33A (-1455W @ 1p)
[lp-2  ] DEBUG 2024/06/17 08:08:48 site power -1455W <= 0W enable threshold
[lp-2  ] DEBUG 2024/06/17 08:08:48 pv enable timer start: 1m0s
[lp-2  ] DEBUG 2024/06/17 08:08:48 pv enable in 1m0s
[lp-2  ] DEBUG 2024/06/17 08:09:32 charge power: 0W
[lp-2  ] DEBUG 2024/06/17 08:09:32 charge currents: [0 0 0]A
[lp-2  ] DEBUG 2024/06/17 08:10:17 charge power: 0W
[lp-2  ] DEBUG 2024/06/17 08:10:17 charge currents: [0 0 0]A
[lp-2  ] DEBUG 2024/06/17 08:10:17 !! session: chargeRater.chargedEnergy=0.2 - chargedAtStartup=0.1
[lp-2  ] DEBUG 2024/06/17 08:10:18 charge total import: 85.700kWh
[lp-2  ] DEBUG 2024/06/17 08:10:18 charger status: B
[lp-2  ] DEBUG 2024/06/17 08:10:18 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[lp-2  ] DEBUG 2024/06/17 08:10:18 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[lp-2  ] DEBUG 2024/06/17 08:10:18 pv charge current: 6.03A = 0A + 6.03A (-1387W @ 1p)
[lp-2  ] DEBUG 2024/06/17 08:10:18 site power -1387W <= 0W enable threshold
[lp-2  ] DEBUG 2024/06/17 08:10:18 pv enable in 0s
[lp-2  ] DEBUG 2024/06/17 08:10:18 pv enable timer elapsed
[lp-2  ] DEBUG 2024/06/17 08:10:18 max charge current: 6A
[lp-2  ] DEBUG 2024/06/17 08:10:18 charger enable
[lp-2  ] DEBUG 2024/06/17 08:10:18 wake-up timer: start
[lp-2  ] DEBUG 2024/06/17 08:11:02 charge power: 11100W
[lp-2  ] DEBUG 2024/06/17 08:11:02 charge currents: [15.9 15.9 16]A
[lp-2  ] DEBUG 2024/06/17 08:11:47 charge power: 11000W
[lp-2  ] DEBUG 2024/06/17 08:11:47 charge currents: [15.9 15.8 15.9]A
[lp-2  ] DEBUG 2024/06/17 08:11:47 !! session: chargeRater.chargedEnergy=0.2 - chargedAtStartup=0.1
[lp-2  ] DEBUG 2024/06/17 08:11:48 charge total import: 85.700kWh
[lp-2  ] DEBUG 2024/06/17 08:11:48 charger status: C
[lp-2  ] INFO 2024/06/17 08:11:48 start charging ->
[lp-2  ] DEBUG 2024/06/17 08:11:48 wake-up timer: stop
[lp-2  ] WARN 2024/06/17 08:11:48 charger logic error: current mismatch (got 16A, expected 6A)
[lp-2  ] DEBUG 2024/06/17 08:11:48 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[lp-2  ] DEBUG 2024/06/17 08:11:48 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[lp-2  ] DEBUG 2024/06/17 08:11:48 pv charge current: 0A = 16A + -41.5A (9552W @ 1p)
[lp-2  ] DEBUG 2024/06/17 08:11:48 projected site power 9552W >= 0W disable threshold
[lp-2  ] DEBUG 2024/06/17 08:11:48 pv disable in 0s
[lp-2  ] DEBUG 2024/06/17 08:11:48 pv disable timer elapsed
[lp-2  ] DEBUG 2024/06/17 08:11:48 charger disable
[lp-2  ] DEBUG 2024/06/17 08:11:48 wake-up timer: stop
[lp-2  ] DEBUG 2024/06/17 08:12:32 charge power: 0W
[lp-2  ] DEBUG 2024/06/17 08:12:32 charge currents: [0 0 0]A
[lp-2  ] DEBUG 2024/06/17 08:13:17 charge power: 0W
[lp-2  ] DEBUG 2024/06/17 08:13:17 charge currents: [0 0 0]A
[lp-2  ] DEBUG 2024/06/17 08:13:18 !! session: chargeRater.chargedEnergy=0.2 - chargedAtStartup=0.1
[lp-2  ] DEBUG 2024/06/17 08:13:18 charge total import: 85.700kWh
[lp-2  ] DEBUG 2024/06/17 08:13:18 charger status: B
[lp-2  ] INFO 2024/06/17 08:13:18 stop charging <-
[lp-2  ] DEBUG 2024/06/17 08:13:18 pv timer reset
[lp-2  ] DEBUG 2024/06/17 08:13:18 pv timer inactive
[lp-2  ] DEBUG 2024/06/17 08:13:18 charge total import: 85.700kWh
[lp-2  ] DEBUG 2024/06/17 08:13:18 !! session: start=85.700 stop=85.700 chargedEnergy=0.100
[lp-2  ] DEBUG 2024/06/17 08:13:18 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[lp-2  ] DEBUG 2024/06/17 08:13:18 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[lp-2  ] DEBUG 2024/06/17 08:13:18 pv charge current: 8.47A = 0A + 8.47A (-1948W @ 1p)
[lp-2  ] DEBUG 2024/06/17 08:13:18 site power -1948W <= 0W enable threshold
[lp-2  ] DEBUG 2024/06/17 08:13:18 pv enable timer start: 1m0s
[lp-2  ] DEBUG 2024/06/17 08:13:18 pv enable in 1m0s
[lp-2  ] DEBUG 2024/06/17 08:14:02 charge power: 0W
[lp-2  ] DEBUG 2024/06/17 08:14:02 charge currents: [0 0 0]A
```

GrimmiMeloni, 2024-06-17:
Das liegt an Deiner Konfiguration.
Man kann im Log schön sehen, daß der Disable Timer <= 2* 45s Interval ist.

```
[lp-2  ] DEBUG 2024/06/17 08:10:18 charger enable
...
[lp-2  ] DEBUG 2024/06/17 08:11:48 pv disable in 0s
[lp-2  ] DEBUG 2024/06/17 08:11:48 pv disable timer elapsed
[lp-2  ] DEBUG 2024/06/17 08:11:48 charger disable
```

Anders ausgedrückt, Dein Charger hat ein Eigenleben. Bis evcc das erste Mal korrigierend eingreifen könnte ist Dein PV Timer bereits wieder abgelaufen und es wird direkt abgeschaltet.

Die Lösung ist ein sinnvolles Verhältnis zwischen Interval und Disable Timer zu wählen. Bei einer Box mit Eigenleben würde ich persönlich auf kürzere Intervalle setzen. Zusätzlich noch den Faktor 2 durch die zwei Loadpoints beachten. Wenn Deine anderen Meter aus abkönnen, würde ich auf 15s runter gehen. Dann hast Du den jeden Loadpoint alle 30s geregelt.
Alternativ kannst Du den PV Disable Timer verlängern.

kasn24, 2024-06-17:
Alles klar, vielen dank. Weiter oben wurde eine längere Zeit von 45-90sec empfohlen.
Ich habe es auf 15s angepasst und der Ladevorgang konnte korrekt starten (em2go selbstständig auf 11kW) hoch und dann aber runter geregelt.

Jetzt werden die Befehle zum Phasenwechsel aber von der em2go ignoriert:
```
[lp-2 ] WARN 2024/06/17 08:56:35 ignoring inconsistent phases: 1p < 3p observed active
[lp-2 ] DEBUG 2024/06/17 08:56:35 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[lp-2 ] DEBUG 2024/06/17 08:56:35 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[lp-2 ] DEBUG 2024/06/17 08:56:35 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[lp-2 ] DEBUG 2024/06/17 08:56:35 pv charge current: min 6A > 0A (1987W @ 1p, battery: true)
[lp-2 ] DEBUG 2024/06/17 08:56:49 charge power: 4200W
[lp-2 ] DEBUG 2024/06/17 08:56:50 charge currents: [6 6 6]A
[lp-2 ] DEBUG 2024/06/17 08:57:04 charge power: 4200W
[lp-2 ] DEBUG 2024/06/17 08:57:05 charge currents: [6 6 6]A
[lp-2 ] DEBUG 2024/06/17 08:57:05 detected active phases: 3p
[lp-2 ] DEBUG 2024/06/17 08:57:05 charge total import: 85.700kWh
[lp-2 ] DEBUG 2024/06/17 08:57:05 charger status: C
```

GrimmiMeloni, 2024-06-17:
Das ist ein bekanntes Problem.
@andig hier ist ein weiterer Fall von dem inconsistent Phases Problem. Ich bleibe dabei - meiner Ansicht nach ist die Logik zum Abgleich falsch, wenn der Loadpoint den Charger übersteuert/ignoriert. Mehr Details im alten Thread:

> Generell kann man sagen, wenn immer der Loadpoint weniger Phasen annimmt, als in Wahrheit aktiv sind, wird durch die aktuelle Implementierung ein runterschalten der Phasen verhindert, und man kommt deshalb in jedem Intervall wieder den "inconsistent phases" Fehler.

https://github.com/evcc-io/evcc/issues/12773#issuecomment-1987263008

andig, 2024-06-17:
> Naja liegt jetzt woran? Weil mir die Begriffe noch nichts sagen? Dann gib mir irgendwo einen Link wo das vielleicht erklärt wird.

@kasn24 Du sagtest:

> Konsolen Befehl kann ich keine geben, da mein evcc im HomeAssistant läuft.

Da genau das zu tun war geschlossen. Nicht mehr und nicht weniger. Wenns keine neuen Erkenntnisse mehr geben kann gibts für uns auch nichts mehr zu tun. Man kann evcc wunderbar auch ohne HA laufen lassen, scheint aber kaum jemand machen zu wollen.

andig, 2024-06-17:
> Das ist ein bekanntes Problem.

@GrimmiMeloni ich verstehe leider nur Bahnhof. M.E. korrigieren wir inkonsistente Phasen in `syncCharger`, jedenfalls wenn `api.PhaseGetter` vorhanden ist:

```go
		// sync phases
		phases := lp.GetPhases()
		if ps, ok := lp.charger.(api.PhaseGetter); ok && enabled && shouldBeConsistent && phases > 0 {
			if chargerPhases, err := ps.GetPhases(); err == nil {
				if chargerPhases != phases {
					lp.log.WARN.Printf("charger logic error: phases mismatch (got %d, expected %d)", chargerPhases, phases)
					lp.setPhases(chargerPhases)
				}
			} else if !errors.Is(err, api.ErrNotAvailable) {
				return fmt.Errorf("charger get phases: %w", err)
			}
		}
```

`api.PhaseGetter` gibt es hier nicht. Zu testen war, ob das setzen von 1p über funktioniert. Falls nein nutzt uns ja auch ein sync nix.

Wenn Du etwas ganz anderes meinst gerne neues (grundsätzliches) Issue.

kasn24, 2024-06-17:
War halt erstmal bequem aufm HA. 
Ändere ich die aktuelle Woche vielleicht mal, wenn ich Zeit habe. Einen Raspi habe ich hier noch liegen. 

Aber ein halber Satz mehr Kommunikation wäre trotzdem super gewesen, haben doch einige mehr hier geschrieben, dass nicht unbedingt klar war, was an wen geht :) 

Nochmals danke für euren Mühen bisher!

andig, 2024-06-17:
Klarer als in https://github.com/evcc-io/evcc/issues/14307#issuecomment-2170114074 kann ich es nicht schreiben :(

kasn24, 2024-06-19:
@andig 
> ```
> evcc charger --phases 1|3
> ```

ich habe evcc Nightly auf dem Raspi installiert. Leider kann ich deinen Befehl gerade nicht umsetzen. Wie ist die Syntax richtig?
Was muss ich hinter charger anfügen? Ich habe lp2, ChargerName und LoadpointName probiert. 

```
evcc charger lp2 --phases 1
[main  ] INFO 2024/06/19 17:34:26 evcc 0.127.2 (5eaa90ad)
[main  ] INFO 2024/06/19 17:34:26 using config file: /home/ks/evcc.yaml
[db    ] INFO 2024/06/19 17:34:27 using sqlite database: /home/ks/.evcc/evcc.db

``` 
Das bringt leider gar keine Wirkung im evcc und ich sehe auch nichts davon in den Logs.

Danke!

andig, 2024-06-19:
@kasn24 Dein Charger heisst `em2goWilly` soweit ich sehe, nicht `lp2`? Gerne noch ein `--log trace` mit anhängen.

kasn24, 2024-06-20:
Ja damit und mit allen anderen möglichen Namen hatte ich es ja erfolglos getestet. 

```
$ evcc charger em2goWilly --phases 1 --log trace
[main  ] INFO 2024/06/20 09:52:24 evcc 0.127.2 (5eaa90ad)
[main  ] INFO 2024/06/20 09:52:24 using config file: /home/ksn/evcc.yaml
[db    ] INFO 2024/06/20 09:52:25 using sqlite database: /home/ksn/.evcc/evcc.db
[db    ] TRACE 2024/06/20 09:52:25 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2024/06/20 09:52:25 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/06/20 09:52:25 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/06/20 09:52:25 SELECT * FROM `settings` 0 <nil>
[db    ] TRACE 2024/06/20 09:52:25 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db    ] TRACE 2024/06/20 09:52:25 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db    ] TRACE 2024/06/20 09:52:25 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2024/06/20 09:52:25 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/06/20 09:52:25 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/06/20 09:52:25 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2024/06/20 09:52:25 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "config_details" AND sql IS NOT NULL order by type = "table" desc 2 <nil>
[db    ] TRACE 2024/06/20 09:52:25 SELECT * FROM `config_details` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/06/20 09:52:25 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT ""fk_configs_details"" %" OR sql LIKE "%CONSTRAINT fk_configs_details %" OR sql LIKE "%CONSTRAINT `fk_configs_details`%" OR sql LIKE "%CONSTRAINT [fk_configs_details]%" OR sql LIKE "%CONSTRAINT 	fk_configs_details	%") -1 <nil>
[db    ] TRACE 2024/06/20 09:52:25 SELECT count(*) FROM sqlite_master WHERE type = "index" AND tbl_name = "config_details" AND name = "idx_unique" -1 <nil>
[db    ] TRACE 2024/06/20 09:52:25 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT ""fk_devices_details"" %" OR sql LIKE "%CONSTRAINT fk_devices_details %" OR sql LIKE "%CONSTRAINT `fk_devices_details`%" OR sql LIKE "%CONSTRAINT [fk_devices_details]%" OR sql LIKE "%CONSTRAINT 	fk_devices_details	%") -1 <nil>
[db    ] TRACE 2024/06/20 09:52:25 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%""device_id"" %" OR sql LIKE "%device_id %" OR sql LIKE "%`device_id`%" OR sql LIKE "%[device_id]%" OR sql LIKE "%	device_id	%") -1 <nil>
[db    ] TRACE 2024/06/20 09:52:25 SELECT * FROM `configs` WHERE `configs`.`class` = 1 ORDER BY id 0 <nil>
``` 

und die Logs ausm evcc selbst auch ohne Ergebnis

```
[lp-2 ] DEBUG 2024/06/20 09:52:04 !! active phases: 3p = min(3p measured 0p vehicle 3p physical 0p charger)
[lp-2 ] DEBUG 2024/06/20 09:52:04 pv charge current: 8.91A = 9A + -0.0935A (64W @ 3p)
[lp-2 ] DEBUG 2024/06/20 09:52:05 max charge current: 8A
[lp-2 ] DEBUG 2024/06/20 09:52:18 charge power: 5500W
[lp-2 ] DEBUG 2024/06/20 09:52:19 charge currents: [8 8 8]A
[lp-2 ] DEBUG 2024/06/20 09:52:33 charge power: 5600W
[lp-2 ] DEBUG 2024/06/20 09:52:34 charge currents: [8 8 8]A
[lp-2 ] DEBUG 2024/06/20 09:52:34 detected active phases: 3p
[lp-2 ] DEBUG 2024/06/20 09:52:34 charge total import: 124.800kWh
[lp-2 ] DEBUG 2024/06/20 09:52:34 charger status: C
[lp-2 ] DEBUG 2024/06/20 09:52:34 soc estimated: 80.00% (vehicle: 80.00%)
[lp-2 ] DEBUG 2024/06/20 09:52:34 vehicle soc: 80%
[lp-2 ] DEBUG 2024/06/20 09:52:34 vehicle soc limit: 85%
[lp-2 ] DEBUG 2024/06/20 09:52:34 vehicle range: 405km
[lp-2 ] DEBUG 2024/06/20 09:52:34 !! active phases: 3p = min(3p measured 0p vehicle 3p physical 0p charger)
[lp-2 ] DEBUG 2024/06/20 09:52:34 !! active phases: 3p = min(3p measured 0p vehicle 3p physical 0p charger)
[lp-2 ] DEBUG 2024/06/20 09:52:35 pv charge current: 9.42A = 8A + 1.42A (-977W @ 3p)
[lp-2 ] DEBUG 2024/06/20 09:52:35 max charge current: 9A
[lp-2 ] DEBUG 2024/06/20 09:52:48 charge power: 6300W
[lp-2 ] DEBUG 2024/06/20 09:52:49 charge currents: [9 9 9]A
[lp-2 ] DEBUG 2024/06/20 09:53:04 charge power: 6300W
[lp-2 ] DEBUG 2024/06/20 09:53:04 charge currents: [9 9 9]A
[lp-2 ] DEBUG 2024/06/20 09:53:04 detected active phases: 3p
[lp-2 ] DEBUG 2024/06/20 09:53:04 charge total import: 124.800kWh
[lp-2 ] DEBUG 2024/06/20 09:53:04 charger status: C
[lp-2 ] DEBUG 2024/06/20 09:53:04 soc estimated: 80.00% (vehicle: 80.00%)
[lp-2 ] DEBUG 2024/06/20 09:53:04 vehicle soc: 80%
[lp-2 ] DEBUG 2024/06/20 09:53:04 vehicle soc limit: 85%
[lp-2 ] DEBUG 2024/06/20 09:53:04 vehicle range: 405km
[lp-2 ] DEBUG 2024/06/20 09:53:05 !! active phases: 3p = min(3p measured 0p vehicle 3p physical 0p charger)
```

kasn24, 2024-06-20:
Oh ja er hat auf die falsche Konfig zugegriffen. Hab sie gelöscht; jetzt nimmt er die richtige... nehme ich den Charger als Namen:

```
ksn@raspevcc:~ $ evcc charger em2goWilly --phases 1 --log trace
[main  ] INFO 2024/06/20 11:19:28 evcc 0.127.2 (5eaa90ad)
[main  ] INFO 2024/06/20 11:19:28 using config file: /etc/evcc.yaml
[mqtt  ] INFO 2024/06/20 11:19:29 connecting evcc-691240258 at tcp://192.168.178.XXX:1883
[mqtt  ] DEBUG 2024/06/20 11:19:29 tcp://192.168.178.XXX:1883 connected
[main  ] FATAL 2024/06/20 11:19:39 cannot create charger 'em2goWilly': cannot create charger type 'template': cannot create charger type 'em2go-home': read tcp 192.168.178.YYY:33490->192.168.178.ZZZ:502: i/o timeout
``` 

Nehme ich den Loadpoint Namen:

```
ksn@raspevcc:~ $ evcc charger em2go11kw --phases 1 --log trace
[main  ] INFO 2024/06/20 11:20:23 evcc 0.127.2 (5eaa90ad)
[main  ] INFO 2024/06/20 11:20:23 using config file: /etc/evcc.yaml
[mqtt  ] INFO 2024/06/20 11:20:24 connecting evcc-1480268699 at tcp://192.168.178.XXX:1883
[mqtt  ] DEBUG 2024/06/20 11:20:24 tcp://192.168.178.XXX:1883 connected
```

Mich verwundern die IPs.... er connectet mit XXX auf die IP meines HomeAssistant MQTT ?
YYY ist das aktuelle EVC und ZZZ die Wallbox....

RenatusRo, 2024-06-20:
der erste call ist o.k.
`read tcp 192.168.178.YYY:33490->192.168.178.ZZZ:502: i/o timeout` liegt nicht an evcc.
- laeuft noch ein evcc als HA addon?, oder sonstwas
- netzwerk?
- stimmt ZZZ
- box neu gestartet (und dann ausreichend lang warten)

(XXX steht vermutlich genau so in der config als mqtt)

kasn24, 2024-06-20:
- nein evcc HA ist deaktiviert
- sonst läuft alles
- ja
- done (Raspi und Wallbox)

Ja, mich wundert der MQTT Aufruf hier aber. Danach ist die MQTT des aktiven EVCC nämlich abgebrochen. Vorhin und nun erneut. Der Fatal liest sich als würde er einen neuen Charger erstellen wollen. Daher auch der Timeout. Weil evcc unter YYY ja bereits darauf zugreift. 
Im Normalen Ordner sowie im etc Ordner ausgeführt kommt weiterhin das Ergebnis mit dem Timeout. Weiterhin als Normaler User: "user@raspevcc:/etc $ evcc charger em2goWilly --phases 1 --log trace"

Versucht mal anders auszuführen:
```
sudo systemctl evcc charger em2goWilly --phases 1 --log trace
systemctl: unrecognized option '--phases'
```

RenatusRo, 2024-06-20:
was ist der 'aktive EVCC'?  
Es darf nur einenen geben.
Wenn du `evcc charger em2go11kw --phases 1 --log trace` machst, darf kein anderer evcc laufen.

kasn24, 2024-06-20:
Deswegen sage ich ja, der EVCC auf HA ist deaktiviert. Da läuft nichts mehr.

RenatusRo, 2024-06-20:
und auf dem raspi laeuft  jetzt nicht heimlich ein evcc als service? 
vorsichtshalber erst
`sudo systemctl evcc stop`
und dann
`evcc charger em2go11kw --phases 1 --log trace`

kasn24, 2024-06-20:
Jetzt bin ich massiv verwirrt. 

Auf dem Raspi ist evcc gemäß Anleitung installiert. 
Wie soll da zwei mal evcc gleichzeitig laufen?
Nach stoppen ist kein evcc über die Oberfläche erreichbar. 
Status sagt es gibt nen evcc.service der seit dem Befehl aus ist. Aber jetzt läuft natürlich auch nichts mehr

```
user@raspevcc:~ $ sudo systemctl stop evcc
user@raspevcc:~ $ evcc charger em2goWilly --phases 1 --log trace
[main  ] INFO 2024/06/20 14:24:58 evcc 0.127.2 (5eaa90ad)
[main  ] INFO 2024/06/20 14:24:58 using config file: /etc/evcc.yaml
[mqtt  ] INFO 2024/06/20 14:24:59 connecting evcc-130798090 at tcp://192.168.178.XXX:1883
[mqtt  ] DEBUG 2024/06/20 14:24:59 tcp://192.168.178.XXX:1883 connected
user@raspevcc:~ $ sudo systemctl status evcc
○ evcc.service - evcc
     Loaded: loaded (/lib/systemd/system/evcc.service; enabled; preset: enabled)
     Active: inactive (dead) since Thu 2024-06-20 14:22:19 CEST; 3min 48s ago
   Duration: 2min 14.474s
    Process: 964 ExecStart=/usr/bin/evcc (code=exited, status=0/SUCCESS)
   Main PID: 964 (code=exited, status=0/SUCCESS)
        CPU: 2.521s

Jun 20 14:22:16 raspevcc evcc[964]: [site  ] DEBUG 2024/06/20 14:22:16 grid pow>
Jun 20 14:22:16 raspevcc evcc[964]: [site  ] DEBUG 2024/06/20 14:22:16 grid cur>
Jun 20 14:22:16 raspevcc evcc[964]: [site  ] DEBUG 2024/06/20 14:22:16 site pow>
Jun 20 14:22:17 raspevcc evcc[964]: [lp-2  ] DEBUG 2024/06/20 14:22:17 charge t>
Jun 20 14:22:17 raspevcc evcc[964]: [lp-2  ] DEBUG 2024/06/20 14:22:17 charger >
Jun 20 14:22:17 raspevcc evcc[964]: [lp-2  ] DEBUG 2024/06/20 14:22:17 next soc>
Jun 20 14:22:19 raspevcc systemd[1]: Stopping evcc.service - evcc...
Jun 20 14:22:19 raspevcc systemd[1]: evcc.service: Deactivated successfully.
Jun 20 14:22:19 raspevcc systemd[1]: Stopped evcc.service - evcc.
Jun 20 14:22:19 raspevcc systemd[1]: evcc.service: Consumed 2.521s CPU time.

user@raspevcc:~ $ 
```

RenatusRo, 2024-06-20:
gemäß Anleitung installiert ist im prinzip gut.
deswegen lief der auch im hintergrund als 1. instanz.
und wenn du dann auf der command line die 2te instanz gestartet hast, konnte die sich nicht mehr mit dem modbus von der em2go verbinden.
wenn der service aus ist, haben wir erstmal saubere verhaeltnisse.
deswegen kommt jetzt bei `evcc charger em2goWilly --phases 1 --log trace` auch kein fehler mehr von wegen `cannot create charger type 'template'`

passiert dann weiter nichts mehr?
mal
`evcc charger em2go11kw --phases 1 --log trace`
und 
`evcc charger em2go11kw --phases 3 --log trace`
ausprobieren

RenatusRo, 2024-06-20:
muss evcc charger em2goWilly sein.

kasn24, 2024-06-20:
Achso mir war nicht bewusst, dass ich mit dem Befehl eine neue Instanz starte, ich dachte ich mische in die aktuell laufende Instanz ein mit dem Befehl den Charger im aktiven Ladevorgang zum Phasenwechsel zu bewegen.

Nein bei beiden Befehlen ist nach wenigen Sekunden Schluss. Also nachdem MQTT Erfolg vermeldet.

andig, 2024-06-20:
Jetzt muss ich mal stop sagen. `evcc charger` kann man- falls die Box das zulässt- wunderbar parallel zu laufender Instanz absetzen.

> sudo systemctl evcc charger em2goWilly --phases 1 --log trace

@kasn24 bitte nicht raten. Davon war nie die Rede! Das ist kein Service sondern einfach nur ein kleiner Befehl.

> Wenn du evcc charger em2go11kw --phases 1 --log trace machst, darf kein anderer evcc laufen.

@RenatusRo bitte nicht raten. Das mag sein, gerade bei Modbus ist das aber oft gar kein Problem. Wir wollen ja gerade raus finden, ob die Umschaltung klappt während der Service das nicht kann. NICHT möglich ist das nur wenn die Box nur einen TCP/Modbus Client zulässt.

> Hab sie gelöscht; jetzt nimmt er die richtige... 

@kasn24 damit wir nicht raten: wie sieht /etc/evcc.yaml aus? Ist die korrekt?

> read tcp 192.168.178.YYY:33490->192.168.178.ZZZ:502: i/o timeout

Das geht tatsächlich nicht. Der einfache Test nutzt also nix. Anonymisieren der IPs kann du dir sparen, in Dein Netz kann ohnehin niemand rein schauen.

An dieser Stelle: ich gebs auf. em2go scheint manchmal die Phasen nicht umzuschalten, trace log beweist das. Da hilft nur Ticket beim Hersteller.

kasn24, 2024-06-20:
Raten musste ich, da ich noch kein Wiki diesbezüglich gefunden habe.
Vielleicht wäre gerade für Fehlersuchen eine kleine Anleitung Wiki für evcc Anfänger sinnvoll. :)

Ja die etc/evcc.yaml passt, die ist wie ganz oben aus HA nur die IPs der Instanz und für MQTT angepasst. Und ja die IPs passen auch alle, ohne irgendwelche Befehle läuft das System ja. Nur der von dir genannte Befehl eben nicht. und auch hier musste ich dann halt raten nach "Du verwendest eine andere Konfigdatei- user vs service. Bitte beim Aufruf die richtige angeben!"

Aber ist ja auch egal, da für dich erledigt. Danke trotzdem!

andig, 2024-06-20:
Es gibt halt nix zu prüfen da das neben dem Service bei der Box nicht möglich ist. Also Ticket beim Hersteller :(. Mit dem Ergebnis gerne wieder melden.

kasn24, 2024-06-20:
Ist schon in der mache.

[21 more replies]