evcc-io/evcc #14314: Umlaute in der yaml -  invalid leading UTF-8 octet

TobiasHuber1980, 2024-06-12:
### Describe the bug

Umlaute in der yaml führen zu 

```
While parsing config
yaml: invalid leading UTF-8 octet
```

keine Ahnung ob das ein bug ist :-) mit "mueller" geht´s natürlich

### Steps to reproduce

1. -
2. - 
3. -
...

### Configuration details

```yaml
- name: car2
    type: template
    template: vw
    title: VW ID.3 (Rosi Müller)
```

### Log details

```text
While parsing config
yaml: invalid leading UTF-8 octet
```

### What type of operating system are you running?

Linux

### Version

0.126.6

StefanSchoof, 2024-06-12:
Ist die Yaml den als utf8 gespeichert oder irgendein anderes Format?

TobiasHuber1980, 2024-06-13:
mit notepad++ noch mal alles auf utf-8 gesetzt, jetzt gut!

closed 2024-06-13 by TobiasHuber1980