evcc-io/evcc #14321: EVCC not starting after update to 0.127.0

pandamouse84, 2024-06-12:
### Describe the bug

EVCC is not starting after having updated to 0.127.0

### Steps to reproduce

1. Update to 0.127.0
2. Have a look at the logs

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

# unique installation id
plant: XXX

interval: 30s # control cycle interval

meters:
  - type: template
    template: varta
    usage: grid
    host: XXX.XXX.XXX.XXX
    port: 502
    name: gridVarta

chargers:
  - name: wallboxElli
    type: template
    template: ellipro
    ip: XXX.XXX.XXX.XXX
    ski: XXX

loadpoints:
  - title: Carport
    charger: wallboxElli

tariffs:
  co2:
    type: grünstromindex
    zip: XXXXX
  currency: EUR
  grid:
    type: fixed
    price: 0.2999

site:
  title: Mein Zuhause
  meters:
    grid: gridVarta
  residualPower: 100

eebus:
  certificate:
    private: |
      -----BEGIN EC PRIVATE KEY-----
Xxx
      -----END EC PRIVATE KEY-----
    public: |
      -----BEGIN CERTIFICATE-----
Xxx
      -----END CERTIFICATE-----
```

### Log details

```text
Using config file: /config/custom_yaml/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/custom_yaml/evcc.yaml'
[main  ] INFO 2024/06/12 22:03:14 evcc 0.127.0
[main  ] INFO 2024/06/12 22:03:14 using config file: /config/custom_yaml/evcc.yaml
[db    ] INFO 2024/06/12 22:03:14 using sqlite database: /data/evcc.db
[main  ] INFO 2024/06/12 22:03:14 listening at :7070
[main  ] FATAL 2024/06/12 22:03:14 cannot create charger 'wallboxElli': cannot create charger type 'template': cannot create charger type 'eebus': eebus not configured
[main  ] FATAL 2024/06/12 22:03:14 will attempt restart in: 15m0s
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.127.0

pandamouse84, 2024-06-12:
btw.

> will attempt restart in: 15m0s

From the logs it looks like he is trying to restart every 3 minutes. Maybe you want to put the correct var in that text.

pandamouse84, 2024-06-12:
Referring to the changelog, I would check the fix applied for EEBUS maybe the bug is in there.

kloeppb, 2024-06-12:
Same error here just upgraded from working system version 0.126.6

Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2024/06/12 22:14:13 evcc 0.127.0
[main  ] INFO 2024/06/12 22:14:13 using config file: /config/evcc.yaml
[db    ] INFO 2024/06/12 22:14:14 using sqlite database: /data/evcc.db
[mqtt  ] INFO 2024/06/12 22:14:14 connecting evcc-2062920207 at tcp://localhost:1883
[main  ] INFO 2024/06/12 22:14:14 listening at :7070
[main  ] FATAL 2024/06/12 22:14:17 cannot create charger 'connect_pro': cannot create charger type 'template': cannot create charger type 'eebus': eebus not configured
[main  ] FATAL 2024/06/12 22:14:17 will attempt restart in: 15m0s

MarcusLindemann, 2024-06-12:
Same issue here after upgrading working system from 0.126.6 to 0.127 on Raspbian bullseye.
```
[main ] INFO 2024/06/12 22:39:21 evcc 0.127.0
[main ] INFO 2024/06/12 22:39:21 using config file: /etc/evcc.yaml
[db ] INFO 2024/06/12 22:39:25 using sqlite database: /var/lib/evcc/evcc.db
[mqtt ] INFO 2024/06/12 22:39:26 connecting evcc-354576817 at tcp://xxxxx:1883
[mqtt ] DEBUG 2024/06/12 22:39:26 tcp://xxxxx:1883 connected
[main ] INFO 2024/06/12 22:39:26 listening at :7070
[main ] FATAL 2024/06/12 22:39:26 cannot create charger 'wallbox2': cannot create charger type 'template': cannot create charger type 'eebus': eebus not configured
[main ] FATAL 2024/06/12 22:39:26 will attempt restart in: 15m0s 
```

cschnatz, 2024-06-13:
Same issue too after upgrading from 0.126.6 to 0.127

`Jun 13 06:35:07 csc-evcc evcc[273886]: [main  ] FATAL 2024/06/13 06:35:07 cannot create charger 'Links': cannot create charger type 'template': cannot create charger type 'eebus': eebus not configured
`

The config itself contains the eebus configuration

```eebus:
  certificate:
    public: |
      -----BEGIN CERTIFICATE-----
    XYZ
      -----END CERTIFICATE-----

    private: |
      -----BEGIN EC PRIVATE KEY-----
      XYZ
      -----END EC PRIVATE KEY-----```

dennisfisch, 2024-06-13:
Same here, upgrading from working 0.125.0 to 0.127.0

pencoe, 2024-06-13:
Just to add my voice: same here after upgrading from 0.126.5 to 0.127.0

mattbalzer, 2024-06-13:
Same here, upgrading from not working 0.126.6 to 0.127.0

VolkerK62, 2024-06-13:
Ich denke, ihr könnt aufhören mit "bei mir auch". ;-)

closed 2024-06-13 by andig

Marty56, 2024-06-14:
Funktioniert immer noch nicht.

```
[main  ] INFO 2024/06/13 05:40:57 evcc 0.127.0
[main  ] INFO 2024/06/13 05:40:57 using config file: /config/mydockerdata/evcc/yaml/evcc.yaml
[db    ] INFO 2024/06/13 05:40:58 using sqlite database: /config/mydockerdata/evcc/sql/evcc.db
[mqtt  ] INFO 2024/06/13 05:40:58 connecting evcc at tcp://192.168.10.132:1884
[main  ] INFO 2024/06/13 05:40:58 listening at :7070
[main  ] FATAL 2024/06/13 05:40:58 cannot create charger 'wallbox': cannot create charger type 'template': cannot create charger type 'eebus': eebus not configured
[main  ] FATAL 2024/06/13 05:40:58 will attempt restart in: 15m0s
Using config file: /config/mydockerdata/evcc/yaml/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/config/mydockerdata/evcc/sql/evcc.db evcc --config /config/mydockerdata/evcc/yaml/evcc.yaml'
[main  ] INFO 2024/06/14 08:02:14 evcc 0.127.1
[main  ] INFO 2024/06/14 08:02:14 using config file: /config/mydockerdata/evcc/yaml/evcc.yaml
[db    ] INFO 2024/06/14 08:02:14 using sqlite database: /config/mydockerdata/evcc/sql/evcc.db
[mqtt  ] INFO 2024/06/14 08:02:14 connecting evcc at tcp://192.168.10.132:1884
[eebus ] INFO 2024/06/14 08:02:14 Local SKI:  d7138aeb02b47ae2a6f10eebeb83e3d87dca63b1
[main  ] INFO 2024/06/14 08:02:14 listening at :7070
[main  ] FATAL 2024/06/14 08:03:44 cannot create charger 'wallbox': cannot create charger type 'template': cannot
```

Uhrenradio, 2024-06-14:
Scheint aber kein prinzipielles Problem mehr mit EEBus zu sein: Bei mir startet das Image 0.127.1 wieder sauber auf (Das war mal ein schneller BugFix). 
Wenn Logs und YAML weiter helfen kann ich die auf Anfrage gerne posten...

andig, 2024-06-14:
I/o timeout hat nix hiermit zu tun.

Marty56, 2024-06-14:
wenn ich heute Abend Zuhause bin, schalte ich die Wallbox vom Strom.
Mal sehn, ob es danach geht. Das hatte ich früher schon mal.
Die Elli Box ist echt Scheiße.

eny34, 2024-06-14:
Bei mir funktioniert alles. Auch mit Elli Box, 0.127.1 und Enyaq über MySkoda. 
Danke für den schnellen fix.

dennisfisch, 2024-06-14:
Dito - alles super und vielen Dank für das Release. Klasse, dass das so schnell gefixed wurde! :D

Oberfranke, 2024-06-14:
Bei mir kommt auch mit dem Update 127.1  die Meldung, dass der Server nicht konfiguriert ist

pandamouse84, 2024-06-14:
@Marty56: Restart via the Elli App should fix the problem too ;)

pandamouse84, 2024-06-14:
@Oberfranke: Can you provide a screenshot or log, sound to me like a different problem.