evcc-io/evcc #14324: Apsystems EZ1 - invalid Syntax

lstefan123, 2024-06-13:
### Describe the bug

Bei Integration eines Apsystems-EZ1 Microwechselrichters (über UI), erhalte ich im Logfile folgende Fehler:

[site ] ERROR 2024/06/13 07:51:29 pv 2 power: strconv.ParseFloat: parsing "<nil>": invalid syntax
[site ] ERROR 2024/06/13 07:51:30 pv 2 power: strconv.ParseFloat: parsing "<nil>": invalid syntax
[site ] ERROR 2024/06/13 07:51:32 pv 2 energy: strconv.ParseFloat: parsing "<nil>": invalid syntax
[site ] ERROR 2024/06/13 07:51:34 pv 2 power: strconv.ParseFloat: parsing "<nil>": invalid syntax
[site ] ERROR 2024/06/13 07:51:36 pv 2 energy: strconv.ParseFloat: parsing "<nil>": invalid syntax
[site ] ERROR 2024/06/13 07:51:39 pv 2 energy: strconv.ParseFloat: parsing "<nil>": invalid syntax

Release 0.127.0 (Fehler trat auch schon bei 0.126.* auf)

Trotz Fehlermeldung werden die Daten aber bereitgestellt. 
Die Daten werden auch per MQTT veröffentlicht: evcc/site/pv/2/power, evcc/site/pv/2/energy

### Steps to reproduce

1. Wechselrichter über UI konfiguriert 
2. EVCC neu gestartet
3. 
...

### Configuration details

```yaml
Konfiguration des Wechselrichters ist über UI unter Angabe der IP-Adresse erfolgt
```

### Log details

```text
[site ] ERROR 2024/06/13 07:51:29 pv 2 power: strconv.ParseFloat: parsing "<nil>": invalid syntax
[site ] ERROR 2024/06/13 07:51:30 pv 2 power: strconv.ParseFloat: parsing "<nil>": invalid syntax
[site ] ERROR 2024/06/13 07:51:32 pv 2 energy: strconv.ParseFloat: parsing "<nil>": invalid syntax
[site ] ERROR 2024/06/13 07:51:34 pv 2 power: strconv.ParseFloat: parsing "<nil>": invalid syntax
[site ] ERROR 2024/06/13 07:51:36 pv 2 energy: strconv.ParseFloat: parsing "<nil>": invalid syntax
[site ] ERROR 2024/06/13 07:51:39 pv 2 energy: strconv.ParseFloat: parsing "<nil>": invalid syntax
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.127.0 (2fb309e6)

andig, 2024-06-13:
Bitte 

    evcc meter --log trace

andig, 2024-06-13:
Könntest Du das bitte formatieren? Danke :)

andig, 2024-06-15:
Das ist alles sehr merkwürdig. `apsystems-ez1` nutzt http. Dein Log enthält aber keinen einzigen http Request.

lstefan123, 2024-06-19:
Fehler war woanders. PV2 war doch nicht das AP-Systems Balkonkraftwerk, sondern ein altes Balkonkraftwerk, das seine Daten über MQTT aus einem Shelly erhalten hatte. Ich habe PV2 in der evcc.yaml entfernt und stattdessen über UI den Shelly ordentlich konfiguriert. Jetzt sieht es wieder fein aus. Sorry!!!

closed 2024-06-19 by lstefan123

hardoverflow, 2024-08-16:
Ich häng mich hier mal rein, da ich mit dem APsystems sporadisch folgende Fehlermeldung im Log sehe:

```
[http ] TRACE 2024/08/16 06:57:42 {"data":{"p1":11,"e1":0.00697,"te1":23.23756,"p2":10,"e2":0.00684,"te2":22.82682},"message":"SUCCESS","deviceId":"E07000097974"}
[http ] TRACE 2024/08/16 06:57:42 GET http://10.189.21.216:8050/getOutputData
[http ] TRACE 2024/08/16 06:57:42 {"data":{"p1":11,"e1":0.00697,"te1":23.23756,"p2":10,"e2":0.00685,"te2":22.82682},"message":"SUCCESS","deviceId":"E07000097974"}
[http ] TRACE 2024/08/16 06:58:11 GET http://10.189.21.216:8050/getOutputData
[http ] TRACE 2024/08/16 06:58:12 {"data":{},"message":"FAILED","deviceId":"E07000097974"}
[http ] TRACE 2024/08/16 06:58:41 GET http://10.189.21.216:8050/getOutputData
[http ] TRACE 2024/08/16 06:58:42 {"data":{"p1":11,"e1":0.00715,"te1":23.23778,"p2":11,"e2":0.00703,"te2":22.82696},"message":"SUCCESS","deviceId":"E07000097974"}
[http ] TRACE 2024/08/16 06:58:42 GET http://10.189.21.216:8050/getOutputData
[http ] TRACE 2024/08/16 06:58:42 {"data":{"p1":11,"e1":0.00716,"te1":23.23778,"p2":11,"e2":0.00703,"te2":22.82696},"message":"SUCCESS","deviceId":"E07000097974"}
[http ] TRACE 2024/08/16 06:59:11 GET http://10.189.21.216:8050/getOutputData
```

Sehr selten, aber es kommt halt doch schon mehrmals am Tag vor, wird der Request zwar beantwortet, liefert aber als Antwort "FAILED" mit einem leeren "data" Hash. Der nächste Request ist dann aber wieder erfolgreich. Ließe sich bei der message "FAILED", ein Retry mit einbauen?

andig, 2024-08-16:
Aktuell machen wir für PV kein Retry, das können wir aber einbauen. Macht aber nur für Leistung Sinn, weil das der einzige wirklich relevante Wert ist.

lstefan123, 2024-08-16:
Ich habe jetzt am EZ1 einen Zigbee(MQTT) Zwischenstecker. Die Unzuverlässigkeit der Datenlieferung hat ihr Ursache nicht bei EVCC. Die Daten bekommt auch HOMEASSISTANT nicht besser. Deswegen hatte das Thema keine Prio mehr.

hardoverflow, 2024-08-17:
> Aktuell machen wir für PV kein Retry, das können wir aber einbauen. Macht aber nur für Leistung Sinn, weil das der einzige wirklich relevante Wert ist.

Retry auf "Power"-only macht Sinn. Würde eine Implementierung sehr begrüßen. Diese wäre ja dann sicherlich generisch.

andig, 2024-08-18:
Tatsächlich macht PV schon einen Retry...