evcc-io/evcc #14330: 127.0 maxGridSupplyWhileBatteryCharging does not work anymore

djfanatix, 2024-06-13:
### Describe the bug

Since the update to 127.0, the setting of maxGridSupplyWhileBatteryCharging does not work anymore.
I have a hybrid inverter, so the DC Power is a lot more then the AC capability of the inverter.

EVCC is ignoring the setting and is importing from the grid

### Steps to reproduce

When restoring a backup to 126.6, is works fine again.

### Configuration details

```yaml
site:
  title: Kuringen # display name for UI
  bufferSoC: # ignore home battery discharge above soc (empty to disable)
  residualPower: 200
  circuit: main
  maxGridSupplyWhileBatteryCharging: 100
  meters:
    grid: homewizard # grid meter
circuits:
  - name: main
    maxPower: 4800
    meter: homewizard
  - name: goe
    maxPower: 9000
    #meter:
    parent: main

loadpoints:
- title: Go-E  # display name for UI
  charger: goe # charger
  circuit: goe

  vehicle: my_car
  # vehicles: # use if multiple vehicles allowed to charge on this loadpoint
    # - e-Up
  mode: pv
  #resetOnDisconnect: true # set defaults when vehicle disconnects
  soc: # battery soc (%)
    # polling defines usage of the vehicle APIs
    # Modifying the default settings it NOT recommended. It MAY deplete your vehicle's battery
    # or lead to vehicle manufacturer banning you from API use. USE AT YOUR OWN RISK.
    poll:
      # poll mode defines under which condition the vehicle API is called:
      #   charging: update vehicle ONLY when charging (this is the recommended default)
      #   connected: update vehicle when connected (not only charging), interval defines how often
      #   always: always update vehicle regardless of connection state, interval defines how often (only supported for single vehicle)
      mode: charging
      # poll interval defines how often the vehicle API may be polled if NOT charging
     # interval: 60m
    estimate: false # set true to interpolate between api updates
  enable: # pv mode enable behavior
    delay: 1m # threshold must be exceeded for this long
    threshold:  # grid power threshold (in Watts, negative=export). If zero, export must exceed minimum charge power to enable
  disable: # pv mode disable behavior
    delay: 5m # threshold must be exceeded for this long
    threshold: 150 # maximum import power (W)
  guardDuration: 5m # switch charger contactor not more often than this (default 10m)
```

### Log details

```text
[site ] DEBUG 2024/06/13 11:51:06 pv power: 6222W
[site ] DEBUG 2024/06/13 11:51:06 battery soc: 72%
[site ] DEBUG 2024/06/13 11:51:06 battery power: -2160W
[site ] DEBUG 2024/06/13 11:51:06 grid meter: 2644W
[site ] DEBUG 2024/06/13 11:51:06 site power: 684W
[lp-1 ] DEBUG 2024/06/13 11:51:06 detected active phases: 1p
[lp-1 ] DEBUG 2024/06/13 11:51:06 !! session: chargeRater.chargedEnergy=12.6 - chargedAtStartup=12.0
[lp-1 ] DEBUG 2024/06/13 11:51:06 charge total import: 513.056kWh
[lp-1 ] DEBUG 2024/06/13 11:51:06 charger status: C
[lp-1 ] DEBUG 2024/06/13 11:51:06 vehicle soc: 70%
[lp-1 ] DEBUG 2024/06/13 11:51:06 vehicle soc limit: 100%
[lp-1 ] DEBUG 2024/06/13 11:51:06 vehicle range: 300km
[lp-1 ] DEBUG 2024/06/13 11:51:06 pv charge current: 23A = 26A + -2.97A (684W @ 1p)
[lp-1 ] DEBUG 2024/06/13 11:51:06 max charge current: 23A
[site ] DEBUG 2024/06/13 11:51:10 ----
[lp-1 ] DEBUG 2024/06/13 11:51:10 charge power: 5494W
[lp-1 ] DEBUG 2024/06/13 11:51:10 charge currents: [22.9 0 0]A
[lp-2 ] DEBUG 2024/06/13 11:51:10 charge power: 0W
[circuit-goe] DEBUG 2024/06/13 11:51:10 power: 5493.848145W
[circuit-goe] DEBUG 2024/06/13 11:51:10 current: 22.91399956A
[circuit-main] DEBUG 2024/06/13 11:51:11 power: 1912W
[circuit-main] DEBUG 2024/06/13 11:51:11 current: 0A
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

127.0

naltatis, 2024-06-13:
@andig ich hab mal nen Fix gemacht. Wir müssen das genau so wie ResidualPower behandeln. Also keine Dopplung.

closed 2024-06-13 by andig

tthommes, 2025-05-30:
Hallo,

ich habe ein ähnliches Problem mit der Version 0.203.6:

Meine PV Anlage hat 13 kWp installiert, der Hybridwechselrichter kann aber nur 10 kW AC und schiebt den Rest direkt per DC in die Hausbatterie. Der Parameter `maxGridSupplyWhileBatteryCharging 50` unter `site:` wird leider ignoriert und hat keinen Effekt.

andig, 2025-05-30:
Schau bitte ins Logfile. Den Parameter gibt es nicht mehr und die Meldung ist eindeutig:

    `MaxGridSupplyWhileBatteryCharging` is deprecated- use `maxACPower` in pv configuration instead