evcc-io/evcc #14336: tariff `energy-charts-api`: costs und tax werden nicht berücksichtigt

VolkerK62, 2024-06-13:
Angaben bei `tax` und `costs` spiegeln sich nicht im Preis wider

Config mit 0 ct/%
```
tariffs:
  currency: EUR
  grid:
    type: template
    template: energy-charts-api
    bzn: DE-LU # Gebotszonen - https://api.energy-charts.info/#/prices/day_ahead_price_price_get
    costs: 0.0 # Zusätzlicher fester Aufschlag pro kWh (z.B. 0.05 für 5 Cent) (optional)
    tax: 0.0 # Zusätzlicher prozentualer Aufschlag (z.B. 0.2 für 20%) (optional) 
```
Output `evcc tariff` mit 0 ct/% 
```
grid:
From                   To                     Price/Cost
2024-06-13 00:00:00    2024-06-13 01:00:00    0.099
2024-06-13 01:00:00    2024-06-13 02:00:00    0.093
2024-06-13 02:00:00    2024-06-13 03:00:00    0.084
2024-06-13 03:00:00    2024-06-13 04:00:00    0.085
2024-06-13 04:00:00    2024-06-13 05:00:00    0.084
2024-06-13 05:00:00    2024-06-13 06:00:00    0.099
2024-06-13 06:00:00    2024-06-13 07:00:00    0.122
2024-06-13 07:00:00    2024-06-13 08:00:00    0.131
2024-06-13 08:00:00    2024-06-13 09:00:00    0.115
2024-06-13 09:00:00    2024-06-13 10:00:00    0.084
2024-06-13 10:00:00    2024-06-13 11:00:00    0.069
2024-06-13 11:00:00    2024-06-13 12:00:00    0.062
2024-06-13 12:00:00    2024-06-13 13:00:00    0.053
2024-06-13 13:00:00    2024-06-13 14:00:00    0.047
2024-06-13 14:00:00    2024-06-13 15:00:00    0.043
2024-06-13 15:00:00    2024-06-13 16:00:00    0.054
2024-06-13 16:00:00    2024-06-13 17:00:00    0.077
2024-06-13 17:00:00    2024-06-13 18:00:00    0.080
2024-06-13 18:00:00    2024-06-13 19:00:00    0.107
2024-06-13 19:00:00    2024-06-13 20:00:00    0.140
2024-06-13 20:00:00    2024-06-13 21:00:00    0.188
2024-06-13 21:00:00    2024-06-13 22:00:00    0.164
2024-06-13 22:00:00    2024-06-13 23:00:00    0.108
2024-06-13 23:00:00    2024-06-14 00:00:00    0.085
```

Config mit 0.99 ct/%
```
tariffs:
  currency: EUR
  grid:
    type: template
    template: energy-charts-api
    bzn: DE-LU # Gebotszonen - https://api.energy-charts.info/#/prices/day_ahead_price_price_get
    costs: 0.99 # Zusätzlicher fester Aufschlag pro kWh (z.B. 0.05 für 5 Cent) (optional)
    tax: 0.99 # Zusätzlicher prozentualer Aufschlag (z.B. 0.2 für 20%) (optional) 
```
Output `evcc tariff` mit 0.99 ct/%
```
From                   To                     Price/Cost
2024-06-13 00:00:00    2024-06-13 01:00:00    0.099
2024-06-13 01:00:00    2024-06-13 02:00:00    0.093
2024-06-13 02:00:00    2024-06-13 03:00:00    0.084
2024-06-13 03:00:00    2024-06-13 04:00:00    0.085
2024-06-13 04:00:00    2024-06-13 05:00:00    0.084
2024-06-13 05:00:00    2024-06-13 06:00:00    0.099
2024-06-13 06:00:00    2024-06-13 07:00:00    0.122
2024-06-13 07:00:00    2024-06-13 08:00:00    0.131
2024-06-13 08:00:00    2024-06-13 09:00:00    0.115
2024-06-13 09:00:00    2024-06-13 10:00:00    0.084
2024-06-13 10:00:00    2024-06-13 11:00:00    0.069
2024-06-13 11:00:00    2024-06-13 12:00:00    0.062
2024-06-13 12:00:00    2024-06-13 13:00:00    0.053
2024-06-13 13:00:00    2024-06-13 14:00:00    0.047
2024-06-13 14:00:00    2024-06-13 15:00:00    0.043
2024-06-13 15:00:00    2024-06-13 16:00:00    0.054
2024-06-13 16:00:00    2024-06-13 17:00:00    0.077
2024-06-13 17:00:00    2024-06-13 18:00:00    0.080
2024-06-13 18:00:00    2024-06-13 19:00:00    0.107
2024-06-13 19:00:00    2024-06-13 20:00:00    0.140
2024-06-13 20:00:00    2024-06-13 21:00:00    0.188
2024-06-13 21:00:00    2024-06-13 22:00:00    0.164
2024-06-13 22:00:00    2024-06-13 23:00:00    0.108
2024-06-13 23:00:00    2024-06-14 00:00:00    0.085
```

VolkerK62, 2024-06-15:
ich habe es mal mit `awattar` verglichen.
Dort wird `tax` berücksichtigt, `costs` aber auch nicht.

In dem Zusammenhang ist mir aufgefallen, dass ein negativer Grundpreis durch `tax` noch negativer wird.
Ist zwar mathemtsich richtig, aber ich denke, steuerlich eher nicht.

andig, 2024-06-15:
Tax einfach positiver Betrag?

andig, 2024-06-15:
Gute Frage. Kann das jemand kommentieren?

closed 2024-06-15 by andig