evcc-io/evcc #14340: Tesla wird nicht über EVCC geregelt

Neophyte85, 2024-06-13:
### Describe the bug

Seit heute ist keinerlei Steuerung über EVCC möglich.

Das einzige was neu ist ist meine Software Version vom Tesla 2024.20.1.
Ist eine die recht frisch ist und noch nicht so häufig verteilt.

Neugestartet habe ich die Wallbox und EVCC. EVCC eben auch noch testweise auf 0.127.0 gebracht. Vorher gleiches Prob bei der 126.6.

Wallbox ist die TWC3. Token habe ich vor die upload hier rausgenommen.

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: dfda0d2e12df53dc1ae4bb7a419fa478ed6499952fe200bb0313fce9c6936074

tariffs:
  currency: EUR # (default EUR)
  grid:
    # static grid price
    type: fixed
    price: 0.34 # [currency]/kWh

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.078 # [currency]/kWh

# trial token, valid until 2024-03-20
sponsortoken: 

interval: 6s # control cycle interval

meters:
- type: template
  template: sungrow-hybrid 
  id: 1  
  host: 192.168.178.244  
  port: 502  
  usage: grid  
  modbus: tcpip  
  name: grid1
- type: template
  template: sungrow-hybrid 
  id: 1  
  host: 192.168.178.244  
  port: 502  
  usage: pv  
  modbus: tcpip  
  name: pv2
#- type: template
#  template: sungrow-hybrid 
[... 1040 lines cut ...]
[site ] DEBUG 2024/06/13 17:25:08 ----
[lp-1 ] DEBUG 2024/06/13 17:25:08 charge power: 3634W
[lp-1 ] DEBUG 2024/06/13 17:25:08 charge currents: [5 5.1 5.1]A
[site ] DEBUG 2024/06/13 17:25:08 pv power: 3505W
[site ] DEBUG 2024/06/13 17:25:08 grid meter: 307W
[site ] DEBUG 2024/06/13 17:25:08 site power: 307W
[lp-1 ] DEBUG 2024/06/13 17:25:08 charge voltages: [239 239 240]V
[lp-1 ] DEBUG 2024/06/13 17:25:08 detected connected phases: 3p
[lp-1 ] DEBUG 2024/06/13 17:25:08 detected active phases: 3p
[lp-1 ] DEBUG 2024/06/13 17:25:08 !! session: chargeRater.chargedEnergy=1.5 - chargedAtStartup=0.8
[lp-1 ] DEBUG 2024/06/13 17:25:08 charger status: C
[lp-1 ] DEBUG 2024/06/13 17:25:08 soc estimated: 47.87% (vehicle: 47.00%)
[lp-1 ] DEBUG 2024/06/13 17:25:08 vehicle soc: 48%
[lp-1 ] DEBUG 2024/06/13 17:25:08 vehicle soc limit: 80%
[lp-1 ] DEBUG 2024/06/13 17:25:08 vehicle range: 232km
[lp-1 ] DEBUG 2024/06/13 17:25:09 pv charge current: 4.56A = 5A + -0.445A (307W @ 3p)
```

### What type of operating system are you running?

Linux

### Version

0.127.0

StefanSchoof, 2024-06-13:
Take a look at #14226

Neophyte85, 2024-06-13:
Sorry wollte keinen doppelten Threat aufmachen. Hielt den anderen threat für einen anderen Fehler weil ich nicht als Meldung "too many conections" bekomme.  

Mein Problem trat auch sofort beim ersten anschließen auf. Nicht erst nach 30 min.

StefanSchoof, 2024-06-13:
Da haben auch einige den Fehler mit dem `Client.Timeout exceeded while awaiting headers`, z.B. https://github.com/evcc-io/evcc/issues/14226#issuecomment-2156002855

closed 2024-06-13 by andig